arendst/Tasmota #9720: Wrong voltage monitoring / not possible to calibrate [Gosund SP112 | BlitzWolf BW-SHPS]

Z3NOX, 2020-11-02:
### PROBLEM DESCRIPTION
In a self-built image for usage with Gosund SP112 (aka BlitzWolf BW-SHPS, using CSE7766) the wrong voltage is displayed (125V) and I am not able to calibrate it via the given [calibration instructions](https://tasmota.github.io/docs/Power-Monitoring-Calibration/)

```
00:14:39 CMD: VoltageCal
00:14:39 MQT: stat/tasmota_socket5/RESULT = {"Command":"Error"}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund SP112
- [x] Tasmota binary firmware version number used: Tasmota 9.0.0.3 (same with 8.5.1)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: platformio
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
00:16:22 CMD: Backlog Template; Module; GPIO 255
00:16:22 MQT: stat/tasmota_socket5/RESULT = {"NAME":"Gosund SP112 USB","GPIO":[321,3072,320,3104,1,225,0,0,1,1,224,1,32,0],"FLAG":0,"BASE":18}
00:16:22 MQT: stat/tasmota_socket5/RESULT = {"Module":{"0":"Gosund SP112 USB"}}
00:16:22 MQT: stat/tasmota_socket5/RESULT = {"GPIO0":{"321":"Led_i2"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"225":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
00:17:09 CMD: status 0
00:17:09 MQT: stat/tasmota_socket5/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota_socket5","FriendlyName":["tasmota_socket5","tasmota_socket52"],"Topic":"tasmota_socket5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:17:09 MQT: stat/tasmota_socket5/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:05:01","StartupUTC":"2020-11-02T23:12:08","Sleep":50,"CfgHolder":102,"BootCount":2,"BCResetTime":"2020-11-02T23:56:57","SaveCount":14,"SaveAddress":"F6000"}}
00:17:09 MQT: stat/tasmota_socket5/STATUS2 = {"StatusFWR":{"Version":"9.0.0.3(tasmota)","BuildDateTime":"2020-11-02T23:52:53","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"490/699"}}
00:17:09 MQT: stat/tasmota_socket5/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID1","SSID2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C8000100068000005A00000000000000","00000000","00006000","00000000"]}}
00:17:09 MQT: stat/tasmota_socket5/STATUS4 = {"StatusMEM":{"ProgramSize":477,"Free":524,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F800583","04040001","0000004F","010013C0","40004880","00004000","00000000"],"Drivers":"1,2,3,4,8,9,10,29","Sensors":"1,2,3,5"}}
00:17:09 MQT: stat/tasmota_socket5/STATUS5 = {"StatusNET":{"Hostname":"tasmota_socket5-2486","IPAddress":"192.168.0.89","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"D8:F1:XX:XX:XX:XX","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:17:09 MQT: stat/tasmota_socket5/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.121","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BB29B6","MqttUser":"tasmota_socket5","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:17:09 MQT: stat/tasmota_socket5/STATUS7 = {"StatusTIM":{"UTC":"2020-11-02T23:17:09","Local":"2020-11-03T00:17:09","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:41","Sunset":"17:25"}}
00:17:09 MQT: stat/tasmota_socket5/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:17:09 MQT: stat/tasmota_socket5/STATUS10 = {"StatusSNS":{"Time":"2020-11-03T00:17:09","ENERGY":{"TotalStartTime":"2020-11-02T23:56:57","Total":0.002,"Yesterday":0.000,"Today":0.002,"Power":1,"ApparentPower":4,"ReactivePower":4,"Factor":0.23,"Voltage":125,"Current":0.033}}}
00:17:09 MQT: stat/tasmota_socket5/STATUS11 = {"StatusSTS":{"Time":"2020-11-03T00:17:09","Uptime":"0T00:05:01","UptimeSec":301,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"SSID2","BSSId":"44:4E:YY:YY:YY:YY","Channel":13,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:02"}}}
00:17:17 MQT: tele/tasmota_socket5/STATE = {"Time":"2020-11-03T00:17:17","Uptime":"0T00:05:09","UptimeSec":309,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"SSID2","BSSId":"44:4E:YY:YY:YY:YY","Channel":13,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:02"}}
00:17:17 MQT: tele/tasmota_socket5/SENSOR = {"Time":"2020-11-03T00:17:17","ENERGY":{"TotalStartTime":"2020-11-02T23:56:57","Total":0.002,"Yesterday":0.000,"Today":0.002,"Period":1,"Power":1,"ApparentPower":4,"ReactivePower":4,"Factor":0.23,"Voltage":125,"Current":0.034}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
00:19:51 CMD: VoltageCal
00:19:51 SRC: WebConsole from 192.168.0.130
00:19:51 CMD: Group 0, Index 1, Command "VOLTAGECAL", Data ""
00:19:51 MQT: stat/tasmota_socket5/RESULT = {"Command":"Error"}
00:19:51 DMP: 55 5A 02 C3 D0 00 05 A5 00 3B C4 07 14 47 4A 53 58 43 EB D0 61 02 DC D2
```

### TO REPRODUCE
Flash, Start, Setting Template according to the [reference](https://templates.blakadder.com/gosund_SP112.html)

### EXPECTED BEHAVIOUR
The Voltage should either be around ~230V or I should be able to callibrate it via `VoltageCal`.

### SCREENSHOTS
-
[... 157 lines cut ...]

#ifdef USE_EXS_DIMMER                           // Add support for ES-Store WiFi Dimmer (+1k5 code)
#undef USE_EXS_DIMMER                           // Add support for ES-Store WiFi Dimmer (+1k5 code)
#endif

#ifdef USE_PWM_DIMMER                           // Add support for MJ-SD01/acenx/NTONPOWER PWM dimmers (+2k2 code, DGR=0k4)
#undef USE_PWM_DIMMER                           // Add support for MJ-SD01/acenx/NTONPOWER PWM dimmers (+2k2 code, DGR=0k4)
#endif

#ifdef USE_SONOFF_D1                            // Add support for Sonoff D1 Dimmer (+0k7 code)
#undef USE_SONOFF_D1                            // Add support for Sonoff D1 Dimmer (+0k7 code)
#endif

#ifdef USE_WS2812
#undef USE_WS2812
#endif

#ifdef USE_IR_REMOTE
#undef USE_IR_REMOTE
#endif

#endif  // _USER_CONFIG_OVERRIDE_H_
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-03:
Probably the GPIO assignments for the power monitoring IC is not correct. The issue you have is typical.

Z3NOX, 2020-11-03:
According to the implementation of  [CSE7766](https://github.com/arendst/Tasmota/blob/24280bcdeac2d977f078dc16ae6a8a2a70d55a26/tasmota/xnrg_02_cse7766.ino) [to the extend I understand it…] the GPIO pin for TX is not used at all. So If I see some numbers from the power monitoring IC, which I do, the RX pin should be correct.

Am I wrong or is there no calibration implemented for this component? If yes, do you have any idea on why the read voltage value is wrong?

Jason2866, 2020-11-04:
This is not good:
```
00:19:51 CMD: VoltageCal
00:19:51 SRC: WebConsole from 192.168.0.130
00:19:51 CMD: Group 0, Index 1, Command "VOLTAGECAL", Data ""
00:19:51 MQT: stat/tasmota_socket5/RESULT = {"Command":"Error"}
```
Try ``VoltageSet 230`

Z3NOX, 2020-11-04:
Same with `VoltageSet 230`
```
22:39:37 CMD: VoltageCal 230
22:39:37 MQT: stat/tasmota_socket5/RESULT = {"Command":"Error"}
```

closed 2020-11-04 by Z3NOX