arendst/Tasmota #9724: TasmotaClient Home Assistant autodetect

CIURI86, 2020-11-03:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

I've started to use the Tasmota client option to control a relay board with Arduino. I'm using the Arduino to control the relay board for now but in the future I want to expand this to switches and sensors .The Arduino flash is quite small and I would prefer if there where an option to use Home Assistant autodetect features from the esp8266 to publish the GPIO's from the Arduino.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
If it's possible I would like to have an option to automatically add the Arduino GPIO's to Home Assistant and use them like the esp8266 GPIO's.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
 A second option would be to manually add the MQTT publish syntaxes to the Arduino's code but as the controller's flash is limited it would use a lot of flash space .

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-03:
Since it is a special use case it will not be implemented. The resources of the ESP82XX are limited too. You have the possibilty to configure in Home Assistant.

sfromis, 2020-11-03:
Or you can yourself from Tasmota make and publish a JSON payload suitable for Home Assistant to autodetect.

closed 2020-11-04 by ascillato2