arendst/Tasmota #9736: 9.0.0.3 prevents send radio  / Sonoff RF Bridge

Elix-g, 2020-11-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After flashing the latest, self compiled development version, no radio signals are being sent anymore but no errors thrown. Going back to self compiled 9.0.0.2 (same user_config_override) resolves this issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [y ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [y] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [y] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [y] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____ Sonoff RF Bridge
- [ ] Tasmota binary firmware version number used: _____ 9.0.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: _____ VSC 1.51
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

privnet/tasmorfgw/stat/TEMPLATE = {"NAME":"Sonoff Ir Bridge","GPIO":[32,3200,1,3232,1,1,0,0,1,320,1,0,0,0],"FLAG":0,"BASE":25}
privnet/tasmorfgw/stat/MODULE = {"Module":{"25":"Sonoff Bridge"}}
privnet/tasmorfgw/stat/GPIO = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"224":"Relay1"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"225":"Relay2"},"GPIO5":{"226":"Relay3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"227":"Relay4"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

privnet/tasmorfgw/stat/RULE = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":507,"Free":4,"Rules":"ON tele-power1 DO IF (%value%=ON and %var1%=OFF) var1 ON;RfKey2 ELSEIF (%value%=OFF and %var1%=ON) var1 OFF;RfKey1 ENDIF ENDON ON tele-power2 DO IF (%value%=ON and %var2%=OFF) var2 ON;RfKey4 ELSEIF (%value%=OFF and %var2%=ON) var2 OFF;RfKey3 ENDIF ENDON ON tele-power3 DO IF (%value%=ON and %var3%=OFF) var3 ON;RfKey6 ELSEIF (%value%=OFF and %var3%=ON) var3 OFF;RfKey5 ENDIF ENDON ON tele-power4 DO IF (%value%=ON and %var4%=OFF) var4 ON;RfKey8 ELSEIF (%value%=OFF and %var4%=ON) var4 OFF;RfKey7 ENDIF ENDON"}
privnet/tasmorfgw/stat/RULE = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":502,"Free":9,"Rules":"ON RfReceived#RfKey==10 DO IF (%var1%=ON) Power1 0 ELSEIF (%var1%=OFF) Power1 1 ENDIF ENDON ON RfReceived#RfKey==9 DO IF (%var2%=ON) Power2 0 ELSEIF (%var2%=OFF) Power2 1 ENDIF ENDON ON RfReceived#RfKey==12 DO IF (%var3%=ON) Power3 0 ELSEIF (%var3%=OFF) Power3 1 ENDIF ENDON ON RfReceived#RfKey==11 DO IF (%var4%=ON) Power4 0 ELSEIF (%var4%=OFF) Power4 1 ENDIF ENDON ON var5#state DO Publish privnet/tasmo6103/cmnd/POWER %value% ENDON ON var6#state DO Publish privnet/tasmo6103/cmnd/POWER %value% ENDON"}
privnet/tasmorfgw/stat/RULE = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":437,"Free":74,"Rules":"ON RfReceived#RfKey==14 DO IF (%var5%=ON) var5 OFF ELSEIF (%var5%=OFF) var5 ON ENDIF ENDON ON RfReceived#RfKey==13 DO IF (%var6%=ON) var6 OFF ELSEIF (%var6%=OFF) var6 ON ENDIF ENDON ON time#minute|3 DO Power0 ENDON ON event#T1 DO IF (%value%=%var5%) var5 ELSE var5 %value% ENDIF ENDON ON event#T2 DO var6 %value% ENDON ON mqtt#connected DO Backlog Subscribe T1,privnet/tasmo6101/tele/POWER;Subscribe T2,privnet/tasmo6102/tele/POWER ENDON"}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

privnet/tasmorfgw/stat/STATUS = {"Status":{"Module":25,"DeviceName":"TasmoRfGW","FriendlyName":["TasmoRfGW","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmorfgw","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
privnet/tasmorfgw/stat/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://syslog.privnet.xmas/tasmota","RestartReason":"Software/System restart","Uptime":"0T00:09:39","StartupUTC":"2020-11-04T14:43:50","Sleep":50,"CfgHolder":4890,"BootCount":4,"BCResetTime":"2020-04-06T15:13:57","SaveCount":141,"SaveAddress":"F7000"}}
privnet/tasmorfgw/stat/STATUS2 = {"StatusFWR":{"Version":"9.0.0.2(tasmota)","BuildDateTime":"2020.10.12 20:31:28","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"685/699"}}
privnet/tasmorfgw/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"syslog.privnet.xmas","LogPort":514,"SSId":["fingerwech",""],"TelePeriod":180,"Resolution":"558180C0","SetOption":["20008019","2805C80001004B0000005A00000000000000","000002C8","00206080","00000000"]}}
privnet/tasmorfgw/stat/STATUS4 = {"StatusMEM":{"ProgramSize":605,"Free":396,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8F8A04C7","04218001","000014CF","010393C0","04807905","00024004","00000000"],"Drivers":"1,2,3,4,5,6,8,9,10,16,17,22,24,27,29,32,37,41","Sensors":"1,2,3,4,5,6,9,10,37,52,62"}}
privnet/tasmorfgw/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmo-rfgw.privnet.xmas","IPAddress":"192.168.179.245","Gateway":"192.168.179.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.179.23","Mac":"xxx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
privnet/tasmorfgw/stat/STATUS6 = {"StatusMQT":{"MqttHost":"openhab.privnet.xmas","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CC3FE7","MqttUser":"tasmorfgw","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
privnet/tasmorfgw/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-04T14:53:29","Local":"2020-11-04T15:53:29","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:31","Sunset":"16:58"}}
privnet/tasmorfgw/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-04T15:53:29"}}
privnet/tasmorfgw/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-04T15:53:29","Uptime":"0T00:09:39","UptimeSec":579,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"fingerwech","BSSId":"44:4E:6D:40:CB:AD","Channel":6,"RSSI":100,"Signal":-28,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Flash 9.0.0.3 and try to switch lights with web buttons

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Check commits between 9.0.0.2 and 9.0.0.3. Related to Setoption114 perhaps?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Elix-g, 2020-11-04:
Tested with pre-compiled 9.0.0.3 now and have identical results.

Thx,
Elix

arendst, 2020-11-04:
What if you use the default Sonoff Bridge module (25) instead of the tweaked template?

arendst, 2020-11-04:
Mine still works with latest 9.0.0.3

Elix-g, 2020-11-04:
You hit the point! With module 25, it's back to normal. Of course the rules do not make sense anymore and have to be reworked completely in order to get back the desired functionality when I cannot define fake relays anymore.

Thanks a lot!

closed 2020-11-04 by arendst