arendst/Tasmota #9737: TuyaMCU 63,X (5 speed buttons for Fan) not working properly or configuration missing

vitoralvimb, 2020-11-04:
### PROBLEM DESCRIPTION
_I am using the AVATTO Fan Light FLS-02 with the tasmota version 8.5.1. 

I have the Device integrated with Home Assistant and is working very well.  The problem I´ve found is with Tasmota GUI Buttons._

I'd configured a 5 Speeds Fan Controller (FnId 63) to the dpId 3 using "Backlog TuyaMCU 63,3;".
So, I can see the 5 speed buttons in the Tasmota Gui (until now, it´s all OK).

But the problem is:
When we press, on the GUI, one of the 5 speed buttons (0 to 4), Tasmota send a TuyaSend message to the dpId (3 in my case).

**Example pressing SPEED 2**
```
23:56:20 SRC: WebGui
23:56:20 CMD: Group 0, Index 2, Command "TUYASEND", Data "3,2"
23:56:21 TYA: Send "55aa00060008030200040000000218"
23:56:21 MQT: stat/switchwall/fanlight1/RESULT = {"TuyaSend":"Done"}
```

The button is sending TuyaSend2 (Index 2) but the Fan is ENUM and it is waiting for TuyaSend4 (Index 4).
So, the 5 speed buttons are not working. Nothing happen with TuyaSend Index 2. 

If I manually send **TuyaSend4 3,2** by Console, for example, it works pretty well.
**Result**
```
14:18:07 CMD: Group 0, Index 4, Command "TUYASEND", Data "3,2"
14:18:07 TYA: Send "55aa00060005030400010214"
14:18:07 MQT: stat/switchwall/fanlight1/RESULT = {"TuyaSend":"Done"}
14:18:07 {"TuyaReceived":{"Data":"55AA03070005030400010218","Cmnd":7,"CmndData":"0304000102","DpType4Id3":2,"3":{"DpId":3,"DpIdType":4,"DpIdData":"02"}}}

```
Is it possible to configure the Speed Button to send TuyaSend4 (Index 4)? Or its a problem in Tasmota solution ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Avatto Fan Light FLS-02
- [X] Tasmota binary firmware version number used: 8.5.1
  - [X] Pre-compiled
  - [] Self-compiled
    - [] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer 1.2 for windows
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```
14:16:12 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
14:16:13 SRC: Backlog
14:16:13 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
14:16:13 MQT: stat/switchwall/fanlight1/RESULT = {"NAME":"AVATTO Smart Wifi Fan Light","GPIO":[0,107,0,108,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":54}
14:16:13 SRC: Backlog
14:16:13 CMD: Group 0, Index 1, Command "MODULE", Data ""
14:16:13 MQT: stat/switchwall/fanlight1/RESULT = {"Module":{"0":"AVATTO Smart Wifi Fan Light"}}
14:16:13 SRC: Backlog
14:16:13 CMD: Group 0, Index 1, Command "GPIO", Data "255"
14:16:13 MQT: stat/switchwall/fanlight1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
I am using Rules but there is nothing to do with the problem reported. With or without rules the problem is the same.
**Rule1** = Events received from Home Assistant to set Speed and Turn On the Fan
**Rule2** = To send the information to Home Assistant when the Speed is changed in the Phisical Switch
```
14:29:57 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2"
14:29:57 SRC: Backlog
14:29:57 CMD: Group 0, Index 1, Command "RULE", Data ""
[... 29 lines cut ...]
_Very simple. Just need to execute the command:
TuyaMCU 63,x (replacing x for any dpId you have) _

You will have 5 speed buttons on the Tasmota GUI

Press one of them on the GUI (0 to 4). Tasmota will send a TuyaSend message to the chosen dpId (3 in my case).

**Example pressing SPEED 2**
```
23:56:20 SRC: WebGui
23:56:20 CMD: Group 0, Index 2, Command "TUYASEND", Data "3,2"
23:56:21 TYA: Send "55aa00060008030200040000000218"
23:56:21 MQT: stat/switchwall/fanlight1/RESULT = {"TuyaSend":"Done"}
```

### EXPECTED BEHAVIOUR
_I expect that the Fan Speed Button could send TuyaSend4 command (Index 4), once the fan is waiting ENUM comand_

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-11-04:
Thanks for reporting, I already aware of the bug.

The fan controller will be removed from the configuration as soon as I'll have time to do it and will be replaced by up to 3 different enum type4 (range 0 - 10). The webgui buttons will be probably removed because are just a waste of space. Perhaps I'll add some sliders with snap points, didn't decided yet. 

As important note you should consider Tuya has almost dropped completely ESP devices and is replacing them with ARM microcontrollers not supported by Tasmota. For that reason I'm not considering the Tuya platform as important as it was.

Closing for now as there is no solution at the moment but fear not, I'll mention this issue in my update when it will be ready so you will have a feedback.

closed 2020-11-04 by effelle

rt400, 2020-11-29:
@effelle @vitoralvimb i have also the AVATTO Fan Light FLS-02 and the fan work fine with rules.
my problem is that if i turn on the light from the switch the homeassistant wont update . i try to fix it with rules but with no success . you have the same issue ?

vitoralvimb, 2020-11-29:
> @effelle @vitoralvimb i have also the AVATTO Fan Light FLS-02 and the fan work fine with rules.
> my problem is that if i turn on the light from the switch the homeassistant wont update . i try to fix it with rules but with no success . you have the same issue ?

There is no need to use rules for light. Check your MQTT Topic configuration on TASMOTA and check on HASS. Probably the problem is there.
I am using FLS-02 bought on November/2020.

I have no problems and the light works perfectly:

Tasmota Template:
{“NAME”:“AVATTO Smart Wifi Fan Light”,“GPIO”:[0,107,0,108,0,0,0,0,0,0,0,0,0],“FLAG”:0,“BASE”:54}

TOPIC (I use it but you can choose yours):
switchwall/fanlight01

Tasmota CONSOLE:
BackLog TuyaMCU 11,1; TuyaMCU 63,3; TuyaMCU 12,9; WebButton1 Fan; WebButton2 Light; 

Rules:
Rule1 ON EVENT#3,0 DO Backlog Publish cmnd/%topic%/TuyaSend4 3,0; POWER1 ON endon  ON EVENT#3,2 DO Backlog Publish cmnd/%topic%/TuyaSend4 3,2; POWER1 ON endon  ON EVENT#3,4 DO Backlog Publish cmnd/%topic%/TuyaSend4 3,4; POWER1 ON endon;
Rule2 ON TuyaReceived#Data=55AA03070005030400010016 DO Publish stat/%topic%/SPEED 3,0 endon  ON TuyaReceived#Data=55AA03070005030400010117 DO Publish stat/%topic%/SPEED 3,2 endon  ON TuyaReceived#Data=55AA03070005030400010218 DO Publish stat/%topic%/SPEED 3,2 endon  ON TuyaReceived#Data=55AA03070005030400010319 DO Publish stat/%topic%/SPEED 3,2 endon  ON TuyaReceived#Data=55AA0307000503040001041A DO Publish stat/%topic%/SPEED 3,4 endon;
Rule1 1;
Rule2 1;

On HASS:
fan:
  - platform: mqtt
    name: "MQTT Switch Wall Fan 01"
    qos: 0
    state_topic: "stat/switchwall/fanlight01/POWER1"
    command_topic: "cmnd/switchwall/fanlight01/POWER1"
    speed_state_topic: "stat/switchwall/fanlight01/SPEED"
    speed_command_topic: "cmnd/switchwall/fanlight01/EVENT"
    payload_on: 'ON'
    payload_off: 'OFF'
    payload_low_speed: '3,0'
    payload_medium_speed: '3,2'
    payload_high_speed: '3,4'
    availability_topic: tele/switchwall/fanlight01/LWT
    payload_available: Online
    payload_not_available: Offline
    speeds:
      - low
      - medium
      - high

light:
  - platform: mqtt
    name: "MQTT Switch Wall Fan Light 01"
    qos: 1
    state_topic: "stat/switchwall/fanlight01/POWER2"
    command_topic: "cmnd/switchwall/fanlight01/POWER2"
    payload_on: "ON"
    payload_off: "OFF"
    availability_topic: "tele/switchwall/fanlight01/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"

On the LOVELACE UI I am using a custom component for FAN. Very intersting.
I am using the custom FrontEnd called fan-control-entity-row. You can find it on HACS

ON LOVELACE:
type: entities
title: Fans
show_header_toggle: false
entities:
  entity: fan.mqtt_switch_wall_fan_1
  type: ‘custom:fan-control-entity-row’
  name: Bedroom
  customTheme: false

If you press any speed, it will turn on the fan at the right speed.

rt400, 2020-11-30:
yes i did like this. but when i turn on the light from switch in the Homeassistant the state not change.. only if i turn the light from the Homeassistant UI. so i never know if someone turn on the light from the switch. the fan control work grate and update the state in HA.

vitoralvimb, 2020-12-05:
> yes i did like this. but when i turn on the light from switch in the Homeassistant the state not change.. only if i turn the light from the Homeassistant UI. so i never know if someone turn on the light from the switch. the fan control work grate and update the state in HA.

Using the Tasmota Device Console and the MQTT Explorer you can debug it.
Check the state and command topic when your turn on the phisical switch. Check if there is something wrong in the topic or other problem. If you can, write here your Console LOG and the Topic when you do this action.

rt400, 2020-12-06:
do you have the same issue ? cause we have the same module...

rt400, 2020-12-06:
> Using the Tasmota Device Console and the MQTT Explorer you can debug it.
> Check the state and command topic when your turn on the phisical switch. Check if there is something wrong in the topic or [...]

do you have the same issue ? cause we have the same module...

vitoralvimb, 2020-12-06:
No, for me its working perfectly! Using exactly the configuration I have
wrote above.

rt400, 2020-12-08:
Well it's really weird, do not know what I did but suddenly it updates the status. 
Really weird ... could it be that a firmware update fixed the problem?