arendst/Tasmota #9739: NTPServer Fallback fails

kugelkopf123, 2020-11-04:
### PROBLEM DESCRIPTION
If NTPSERVER1 does not exist/is unreachable, tasmota does not use NTPServer2.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WemosD1Mini
- [x] Tasmota binary firmware version number used: latest Dev 2020.11.04 23:34:38
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PIO
- [x] Flashing tools used: PIO->esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
00:18:04 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
00:18:04 SRC: Backlog
00:18:04 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
00:18:04 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:18:04 SRC: Backlog
00:18:04 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
00:18:04 RSL: RESULT = {"Module":{"18":"Generic"}}
00:18:04 SRC: Backlog
00:18:04 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
00:18:04 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
00:18:08 RSL: STATE = {"Time":"1970-01-01T00:18:08","Uptime":"0T00:18:10","UptimeSec":1090,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"WasEinDrama","BSSId":"38:10:D5:50:B1:27","Channel":6,"RSSI":62,"Signal":-69,"LinkCount":3,"Downtime":"0T00:00:09"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
00:19:00 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
00:19:00 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_7E5A4D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:19:00 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:19:02","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"1970-01-01T00:00:00","SaveCount":16,"SaveAddress":"3F3000"}}
00:19:00 RSL: STATUS2 = {"StatusFWR":{"Version":"9.0.0.3(tasmota)","BuildDateTime":"2020.11.04 23:34:38","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"358/699"}}
00:19:00 RSL: STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WasEinDrama",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000008","00006000","00000000"]}}
00:19:00 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":505,"Free":1540,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000407","80800586","04080001","00000087","000000C0","00000008","00000000","00000000"],"Drivers":"1,2,3,4,8,10","Sensors":"1,2,3,6,53"}}
00:19:00 RSL: STATUS5 = {"StatusNET":{"Hostname":"esp8266Test","IPAddress":"192.168.0.24","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:7E:5A:4D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:19:00 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:19:00","Local":"1970-01-01T00:19:00","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":99}}
00:19:00 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:19:00"}}
00:19:00 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:19:00","Uptime":"0T00:19:02","UptimeSec":1142,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"WasEinDrama","BSSId":"38:10:D5:50:B1:27","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":3,"Downtime":"0T00:00:09"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Just use a ntpserver1 that does not exist. on NTPserver2 must be a Server that exist.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
If NTPServer1 does not exist or work. Switch to NTPserver2 address.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-11-06:
It seems to be a problem in core/SDK.

I have rewritten the NTP client based on work from EspEasy and it works as expected in my local dev version. 

Not yet released as I need to do some more tweaking regarding correct sync.

arendst, 2020-11-06:
Give it a try and let me know if something goes wrong.

kugelkopf123, 2020-11-06:
Ok! Will give it a try tonight.

kugelkopf123, 2020-11-06:
Yes! It works! NTPserver3, too! Thanks!!

closed 2020-11-06 by kugelkopf123