arendst/Tasmota #9748: Feature Request: Support for RGB controllers with an integrated microphone

lullerboi, 2020-11-06:
**Have you looked for this feature in other issues and in the docs?**  

- Yes, found this, but it goes far beyond my expectations: https://github.com/arendst/Tasmota/issues/4350

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

- Not really. My devices lack a feature that was possible with their stock firmware.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

- Feature Request: Support for RGB controllers with an integrated [microphone](https://smile.amazon.de/gp/product/B08CMMWSPR)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

- I don't know ... should I use something like McLighting instead? That would hurt me :(

**Additional context**  
_Add any other context or screenshots about the feature request here._

- If any Dev is willing to look at this I'm willing to gift one of these controllers to him.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-06:
I have some mid-term plans for something similar. I plan to get the sound from a microphone like MAX4466. Then use the multicast device group feature to remotely control lights. The fft module is ready so we could even think of changing colors depending on frequencies. 

But you may need to wait a few weeks for that.

ascillato, 2020-11-06:
> I plan to get the sound from a microphone like MAX4466.

That "sounds" great!!!! :+1:

lullerboi, 2020-11-06:
> But you may need to wait a few weeks for that.

Will do. Patiently.
If you need one of the devices I linked feel free to contact me. They are easy to flash (OTA) using tuya-convert.

Big THANK YOU in advance!

s-hadinger, 2020-11-07:
I have also ordered the same device. I didn't know there existed led drivers with microphones

s-hadinger, 2020-11-15:
I bought the Gosund LED controller and was abler to tuya-convert it (with the latest version). That's good progress.

lullerboi, 2020-11-15:
Great news and way faster than expected. Thank you!

You should already get a reading from the sensor using analog input. And I *hope* it's also wired to the digital pins so you can f.e. read values for "high/middle/bass". I have no clue if it's possible to find out without opening the controller.

I have three tasmota lights in my living room (including this strip) so my dream would be controlling all of them with websend commands, driven by the mic sensor. If that would be possible I think this will be an AWESOME feature.

s-hadinger, 2020-12-11:
Still on the todo list

laenan8466, 2020-12-25:
So @s-hadinger you were able to bring tasmota to the device? Was this a big issue? Searching for a WS2812/SK6812 controller that can run tasmota without a hassle to flash it.

s-hadinger, 2020-12-26:
Installing Tasmota on the device is easy with Tuya-convert. The Microphone part is not coded yet.

closed 2020-12-27 by ascillato2