arendst/Tasmota #9750: filter out scanned/connected BT sensors

martinlindjarv, 2020-11-06:
**Have you looked for this feature in other issues and in the docs?**  
Yes, the future i'd like is partly like NRFIgnore (it would be one half) and the second part is limiting BT sensors per tasmota device.
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No it's not related to any problem.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I'd like to have some kind on string or parameter (which should be driveable trough mqtt) where i can specify mac addresses to scan / monitor. 
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._
The problem arises when there's small house but big enough for one collector to lose connectivity. I'd like to have two collectors - one for each floor. At the moment bot collectors connect to both floors sensors and drain battery due to bad connection.
**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-11-06:
Yes, that makes sense.

I have already thought about it, but never had the perfect idea so far.

It would probably be possible to extend the -ignore and -use command to work with MAC addresses, like that:
Nrfignore x - like before, ignore a type
Nrfignore aabbccddeeff or aa:bb:cc:dd:ee:dd:ff - ignore device with MAC

Do you mean something like that?

martinlindjarv, 2020-11-06:
At first i was thinking that some sensors (like nrfingnore now) might be excluded. But idea about excluding mac addresses would be even better. 
Maybe it can be json of different mac addresses?

Staars, 2020-11-07:
Before I begin with the „wrong“ driver: Which device do you use?
(I assume it would make sense to implement this for all 3 drivers at the end)

martinlindjarv, 2020-11-07:
At the moment i'm having two different devices - MiFlora and LYWSD03MMC.
And i'd actually would like to exclude mifloga devices and use only LYWSD03MMC.
But it would be excellent if i could use some mifloras on one "collector" and others on other.

Staars, 2020-11-07:
😁 No, I mean your bluetooth device (HM-1x, ESP32 or NRF24)?

martinlindjarv, 2020-11-07:
oh :) esp32 :)

Staars, 2020-11-14:
I finally have started with it.
So your suggestion is a JSON like this(?):
`MI32Ignore {"0": ["AA:BB:CC:DD:EE:FF", "FF:EE:DD:CC:BB:AA"]}`

At least https://jsonlint.com wants a leading string before the array for a valid JSON. 

The simple approach would be (similar to other commands) like I wrote above:
`MI32Ignore AA:BB:CC:DD:EE:FF`

martinlindjarv, 2020-11-14:
Can simple approach have multiple mac addresses or just one ?
In my opinion one address at a time would be sufficient but only one wouldn't work - that's why i suggested josn.
If multiple mac addresses can be added without json - one at a time - i think would be ok.

And afterward's it would be excellent to get the list of ignored mac's on specific device as well...

Staars, 2020-11-14:
You can chain as many commands in a rule with BACKLOG as there is space in (a specific part of the) memory, which is constrained for a JSON too.
And yes, there will be a list output to the console, like that:
```
17:44:13 CMD: mi32ignore
17:44:13 11:22:33:44:55:66
17:44:13 11:22:33:11:22:33
```

martinlindjarv, 2020-11-14:
But will the list be available over mqtt as well? So one could use home automation to drive what collector collects what ... and don't need to log into collector directly?

Staars, 2020-11-14:
Yes, this is the plan.

closed 2020-12-15 by github-actions[bot]