arendst/Tasmota #9752: Is setoption114 for 1 switch/relay possible (shelly2.5)

kroon040, 2020-11-06:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
setoption 114 1 detach both switches to the relays (on shelly2.5)
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
It should ben nice to have the option enabled per switch/relay so switch 1 is attached to relay1 and swtich2 is detached to relay2
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
other firmware espurna
**Additional context**  
_Add any other context or screenshots about the feature request here._
it should be nice to have
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-06:
Decoupling a specific switch can be done by a rule. A rule example is in the documentation.

Elix-g, 2020-11-06:
May I ask where in the documentation? Can't find an example neither in the documentation of SetOption114, nor in the rules documentation nor in the Shelly 2.5 description.

ascillato2, 2020-11-06:
https://tasmota.github.io/docs/Rules/#button-single-press-double-press-and-hold

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-06 by ascillato2