arendst/Tasmota #9756: Sonoff POW R1 revert to Sonoff Basic after 9.1.0 Update

tobox, 2020-11-06:
### PROBLEM DESCRIPTION

I updated 25 Sonoffs yesterday from 8.2-8.5 to 9.1.0. The Updates ran through rather smoothly. I compiled the firmware myself, but only with some settings changes. All Sonoffs came back properly, including all POW R2 models. However, all POW (non-R2) reverted back to Sonoff Basic configuration.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: 9.1.0
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: PlatformIO
- [ ] Flashing tools used: OTA Update
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

20:04:15 CMD: Backlog Template; Module; GPIO 255
20:04:15 MQT: tasmotas/sonoff-power-carport/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,4704],"FLAG":0,"BASE":18}
20:04:15 MQT: tasmotas/sonoff-power-carport/stat/RESULT = {"Module":{"1":"Sonoff Basic"}}
20:04:15 MQT: tasmotas/sonoff-power-carport/tele/HASS_STATE = {"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-05T18:05:28","Module or Template":"Sonoff Basic","RestartReason":"Software/System restart","Uptime":"0T00:01:02","Hostname":"Sonoff-Power-Carport","IPAddress":"192.168.207.161","RSSI":"94","Signal (dBm)":"-53","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:07","MqttCount":1,"LoadAvg":20}
20:04:15 MQT: tasmotas/sonoff-power-carport/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
No rules are used.

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

20:04:47 CMD: Status 0
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS = {"Status":{"Module":1,"DeviceName":"Sonoff-Power-Carport","FriendlyName":["Sonoff-Power-Carport"],"Topic":"sonoff-power-carport","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://192.168.207.40/html/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:34","StartupUTC":"2020-11-06T19:03:13","Sleep":50,"CfgHolder":4617,"BootCount":68,"BCResetTime":"2020-03-30T07:05:41","SaveCount":1449,"SaveAddress":"F6000"}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-05T18:05:28","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"427/699"}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.207.40","LogPort":514,"SSId":["ToBoX",""],"TelePeriod":15,"Resolution":"559380C0","SetOption":["000A8009","2805C8000100068000005AFF000000000000","000002C8","00006000","00000000"]}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS4 = {"StatusMEM":{"ProgramSize":604,"Free":396,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FCBE797","04368001","000404CF","01001BC0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,9,15,34"}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-Power-Carport","IPAddress":"192.168.207.161","Gateway":"192.168.207.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.207.1","Mac":"68:C6:3A:D5:C2:12","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.207.240","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D5C212","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-06T19:04:47","Local":"2020-11-06T20:04:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:46","Sunset":"17:21"}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-06T20:04:47"}}
20:04:47 MQT: tasmotas/sonoff-power-carport/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-06T20:04:47","Uptime":"0T00:01:34","UptimeSec":94,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"ToBoX","BSSId":"7C:FF:4D:FF:70:C9","Channel":11,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Error occurs directly after reboot after setting device to POW (6)

00:00:00 CFG: Loaded from flash at F5, Count 1450
00:00:00 FRC: Some settings have been reset (6)
00:00:00 Project tasmota Sonoff-Power-Carport Version 9.1.0(tasmota)-2_7_4_5
00:00:04 WIF: Connecting to AP1 ToBoX Channel 11 BSSId 7C:FF:4D:FF:70:C9 in mode 11N as Sonoff-Power-Carport...
00:00:05 WIF: Connected
20:05:52 HTP: Web server active on Sonoff-Power-Carport with IP address 192.168.207.161
20:05:52 QPC: Reset
20:05:53 MQT: Attempting connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Update POW from 8.5 to 9.1.0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Should have remained a POW, not revert to BASIC.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-06:
> Some settings have been reset (6)

That means that the bootloop protection has been triggered. That protection clears the settings of a bricked or stuck device in order to allow you regaining access to your device.

It is not a bug, it is a feature. Something was scrambled in your stored settings of your device (tasmota settings, rules or wifi SDK settings) and a factory settings' reset was applied.

If you want to know more about that, please check https://tasmota.github.io/docs/Commands/#setoption36

closed 2020-11-06 by ascillato2

tobox, 2020-11-07:
Thanks for your answers. I was able to confirm that the error must have been in my compiled firmware. I downgraded to the official 8.5.1 files and upgraded to the official 9.1.0 and the error did not occur. I am currently recompiling the development branch to see if I can reproduce the error.

tobox, 2020-11-07:
I was able to track down the issue to this compile option:

#define USE_WS2812_DMA                         // DMA supports only GPIO03 (= Serial RXD) (+1k mem). When USE_WS2812_DMA is enabled expect Exceptions on Pow 

And guess what, the documentation of the flag explains everything. Sorry for the bug report, my bad...