arendst/Tasmota #9761: Shelly 1PM restart after toggle with exception EPC

kopfkraft, 2020-11-07:
### PROBLEM DESCRIPTION
Using tasmota 9.1.0 (as well as 8.5.1) shelly 1PM devices tend to restart after a button toggle. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 1PM
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
cmnd/tasmota_19BC5B/backlog template; Module; GPIO 255
stat/tasmota_19BC5B/LOGGING 13:35:05 MQT: Data Size 26, "cmnd/tasmota_19BC5B/backlog template; Module; GPIO 255"
stat/tasmota_19BC5B/LOGGING 13:35:05 SRC: MQTT
stat/tasmota_19BC5B/LOGGING 13:35:05 CMD: Group 0, Index 1, Command "BACKLOG", Data "template; Module; GPIO 255"
stat/tasmota_19BC5B/LOGGING 13:35:05 SRC: Backlog
stat/tasmota_19BC5B/LOGGING 13:35:05 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
stat/tasmota_19BC5B/RESULT {"NAME":"Shelly 1PM","GPIO":[320,0,0,0,192,2720,0,0,0,0,0,224,0,4736],"FLAG":0,"BASE":18}
stat/tasmota_19BC5B/LOGGING 13:35:05 MQT: stat/tasmota_19BC5B/RESULT = {"NAME":"Shelly 1PM","GPIO":[320,0,0,0,192,2720,0,0,0,0,0,224,0,4736],"FLAG":0,"BASE":18}
stat/tasmota_19BC5B/LOGGING 13:35:05 SRC: Backlog
stat/tasmota_19BC5B/LOGGING 13:35:05 CMD: Group 0, Index 1, Command "MODULE", Data ""
stat/tasmota_19BC5B/RESULT {"Module":{"0":"Shelly 1PM"}}
stat/tasmota_19BC5B/LOGGING 13:35:05 MQT: stat/tasmota_19BC5B/RESULT = {"Module":{"0":"Shelly 1PM"}}
stat/tasmota_19BC5B/LOGGING 13:35:06 SRC: Backlog
stat/tasmota_19BC5B/LOGGING 13:35:06 CMD: Group 0, Index 1, Command "GPIO", Data "255"
stat/tasmota_19BC5B/RESULT {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"192":"Switch_n1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}
stat/tasmota_19BC5B/LOGGING 13:35:06 MQT: stat/tasmota_19BC5B/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"192":"Switch_n1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
stat/tasmota_19BC5B/STATUS {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_19BC5B","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
stat/tasmota_19BC5B/STATUS1 {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:22:33","StartupUTC":"2020-11-07T12:14:36","Sleep":50,"CfgHolder":4617,"BootCount":42,"BCResetTime":"2020-11-05T22:49:41","SaveCount":253,"SaveAddress":"FA000"}}
stat/tasmota_19BC5B/STATUS2 {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-05T10:23:59","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"386/699"}}
stat/tasmota_19BC5B/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":1,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WLAN-IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
stat/tasmota_19BC5B/STATUS4 {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
stat/tasmota_19BC5B/STATUS5 {"StatusNET":{"Hostname":"tasmota_19BC5B-7259","IPAddress":"192.168.4.169","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"D8:BF:C0:19:BC:5B","Webserver":0,"WifiConfig":4,"WifiPower":17.0}}
stat/tasmota_19BC5B/STATUS6 {"StatusMQT":{"MqttHost":"192.168.4.205","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_19BC5B","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
stat/tasmota_19BC5B/STATUS7 {"StatusTIM":{"UTC":"2020-11-07T12:37:09","Local":"2020-11-07T13:37:09","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:47","Sunset":"17:19"}}
stat/tasmota_19BC5B/STATUS9 {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
stat/tasmota_19BC5B/STATUS10 {"StatusSNS":{"Time":"2020-11-07T13:37:09","Switch1":"ON","ANALOG":{"Temperature":33.8},"ENERGY":{"TotalStartTime":"2020-11-05T22:49:41","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0},"TempUnit":"C"}}
stat/tasmota_19BC5B/STATUS11 {"StatusSTS":{"Time":"2020-11-07T13:37:09","Uptime":"0T00:22:33","UptimeSec":1353,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WLAN-IOT","BSSId":"FE:EC:DA:84:48:E5","Channel":1,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:05"}}}
stat/tasmota_19BC5B/STATUS12 {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000e169","00000000","00000000"],"EXCVADDR":"00000001","DEPC":"00000000","CallChain":["4024f8fc","402595e0","40105dee","40105dd0","402595e0","4026f893","402595e0","40254dba","40101d62","4024ee9a","4028a03a","4028a03a","4028a03f","4024f084","4024f084","4024742b","40255e14","401013a8","402385ec","4023864a","40238721","40243dd8","402441ec","40216639","40244231","40201ab8","40208bd3","4021fc36","4021fba4","40201ab8","40207f6a"]}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

simply switch on and off several times (sometimes it happens immediately, sometimes it takes longer ... could be aligned to some activity like the "discovery" or "telemetry" ...)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

well, dont reboot

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-07:
Have you a second one? It seems some Shellys have a weak power supply.
Exception 29 is typical for problems with power supply.

arendst, 2020-11-07:
Just tried with my 1PM and using a Sonoff 4CH as switch. With ``blinktime 4`` (400 mSec) and ``blinkcount 20`` the lamp flashes nicely and the Shelly does not exception.

What happens without a load? Is it an inductive load?

kopfkraft, 2020-11-07:
My tests next to the computer are performed with a 6.3 W LED bulb (E27 socket, regular crashes). unscrewing it, i was not yet able to "crash" the shelly 1pm. Another shelly 1pm is installed to switch a halogen lamp - it is even more problematic and crashes way more... 

I am not much of an electronics expert (yet :)), however, the bulb or the shelly (not 100% sure) has the ability to turn off my monitor (a samsung computer monitor). On switching, it turns black and the source is wrong (no idea how this could happen ...). but it is reproduceable (they are on the same socket multplier ... 

Is there anything I could do to make them work? Maybe removing some routines, features (telemetry, discovery) ... i just ordered more items (i thought it might be the 1pm ...), and might rething that action. I am bit stuck here, as i ordered the Athom product line and found myself with items i could not program (due to the 2MB firmware not upgradeable and the missing serial connection ports). So i went on and found the really cool shelly devices (small, programmable, serial port without soldering ...). But it seems I cannot switch ... rather an "essential" function :) ... What would you recommend as "usable" products one can use to experiment with tasmota (which i like!)?

sfromis, 2020-11-07:
If your computer monitor is affected, it it sounds like some electrical issue. I'd triple-check appropriate wiring, and make sure to understand exactly what and why to connect where.

kopfkraft, 2020-11-07:
... it feels different (EM?) .. however, once i remove the led bulb, the monitor "behaves" and shally works fine too ... 

the wiring is all ok (all certfied parts) ... but your hint was good -> i replaced the 6.5 W LED with an oldfashioned 75 W "wolfram" (real filament) lamp and it appears to work just fine.... is it possible, that a just bought (IKEA) LED bulb causes such interference (it seems to)?

kopfkraft, 2020-11-07:
ok, thanks a lot for the comments! I tested without a bulb (works), with an osram classic 75 W (wolfram) lamp (works), a philips 7 W LED bulb (works) and a "Müller-Licht" 6.3 W bulb bought at IKEA (does not work). The later regulary forces shelly to restart and occasionally manages to render my monitor black (reproducable) ... the only connection i see (monitor - bulb) is that they take the power from the same source (multiplier jack). ... ever had that problem, suggestions? (not really related to tasmota anymore i guess). 

Nevertheless, if you have recommendations for devices -> me thankful!

Jason2866, 2020-11-07:
I was in contact with Athom CEO regarding the 2M builds. ALL issues with there modified Tasmota software is solved. The deliver since end of October a full compatible 1M Tasmota version.
**Tasmota is in no way related to Athom!!**  Tasmota has NO commercial benefits

kopfkraft, 2020-11-07:
hey, no worries! I bought those devices because I wanted to get to know the platform and you list them as native on your extremely long devices list (that is like a "amazon recommends" sticker :) ). I read that those devices are now available with OTA updates ... but it wont make mine work as i already bought them - and I know you are not married (probably the name would change to MOULECOOL ínstead of ATHOM ;) ). 

So I went on and bought something that i could pin down and stick my needles in as in "serial port" and so i found the shelly devices. The dimension or size is just great, as they fit in my installation boxes. They do switching and provide feedback (temperature and power consumption) that i find extremely important to build "feedback loops" ... but no worries, i recently ordered some more rudimentory devices "d1 mini" and "esp32" boards and just got my first webserver to run (and tasmota as well!!!). I do play around with computers/code for a couple of years (33 actually :)) and got interessted in these little devices - congrats, really cool stuff you do!!! My experience in electronics is not as "rich" ... learning! I had to do with the server side of IoT though ... was a platform similar to conrad connect or IFTTT ... cloud, scaling, oauth ... 

Thanks a lot for the tip, already helped a lot!

Jason2866, 2020-11-07:
You will find other `addicted` in our Discord
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

kopfkraft, 2020-11-07:
will do!
thanks!

closed 2020-11-07 by kopfkraft

schrej-zz, 2020-11-11:
Hi,
I just want to mention, that I habe the same issues with a shelly 2.5.
After some toggles the device resets. (8.5.1 & 9.1.0)

schrej-zz, 2020-11-11:
Some more evaluation.
I did a reset 1 and set the Shelly up from scratch.
Same behavior: After a random number of toggles on POWER1 the device crashes.
I don't see a Status12. Reset reason is: External system
I changed the load from input POWER1 to POWER2 und vice versa to see, if the load is the reason.
But still POWER1 crashes. No issues with POWER2.
Shall I open a separate issue?

arendst, 2020-11-11:
No need. Just read this issue from the top down and perform the tasks needed to fix your issue.

majuss, 2021-01-13:
The docs are listing this template: `{"NAME":"Shelly 1PM","GPIO":[56,0,0,0,82,134,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}` which is not working for me... but the one posted from op is: `{"NAME":"Shelly 1PM","GPIO":[320,0,0,0,192,2720,0,0,0,0,0,224,0,4736],"FLAG":0,"BASE":18}`

sfromis, 2021-01-13:
Those two templates are the same. The component numbering system was changed with Tasmota version 9, assigning new numbers for the same functions. Tasmota should convert old template to new if used, or when upgrading.

majuss, 2021-01-13:
Thank you for the info. The problem was something else :)