arendst/Tasmota #9765: ESP32 wroom not boot. it boot only with power on USB

megamarco833, 2020-11-07:
### PROBLEM DESCRIPTION
 i have three ESP32 esp-wroom32:
two with yellow pins and one with black pin, but all like this:
https://www.aliexpress.com/item/32845423155.html?spm=a2g0s.9042311.0.0.27424c4dYMzqRH
i upload succesfull the bin file of last release for ESP32 beta version

i find an issue when i power up the ESP:

if i use the USB cable connected to my PC, it boot up normally, and i have no problem on using it.

if i use the 5v and GND pins connected to a power suplly the ESP continue to reboot and red led continuos blinking (if yellow pin ESP) and led continue to light on (if black pin ESP)

the sampe source of power supply 5v is powering on some nodemcu esp8266 that are correctly working.

is it a "known issue" ?
is it related in someway to firmware?

thanks

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  ESP32 wroom
- [x] Tasmota binary firmware version number used: 9.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool.py with: **esptool.py --chip esp32 --port COM28 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32.bin**
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
17:35:26 CMD: Backlog Template; Module; GPIO 255
17:35:26 MQT: stat/tasmota_FD311C/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
17:35:26 MQT: stat/tasmota_FD311C/RESULT = {"Module":{"1":"ESP32-DevKit"}}
17:35:26 MQT: stat/tasmota_FD311C/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34 ...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  none

```
- [ ] Provide the output of this command: `Status 0`:
```
17:36:09 CMD: Status 0
17:36:09 MQT: stat/tasmota_FD311C/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_FD311C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Vbat power on reset","Uptime":"0T00:01:00","StartupUTC":"2020-11-07T16:35:09","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-11-07T15:36:00","SaveCount":14}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T12:32:13","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"372/699"}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS4 = {"StatusMEM":{"ProgramSize":1052,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29","Sensors":"1,2,3,5,6"}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS5 = {"StatusNET":{"Hostname":"tasmota_FD311C-4380","IPAddress":"192.168.0.56","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"3C:71:BF:FD:31:1C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.105","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FD311C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS7 = {"StatusTIM":{"UTC":"2020-11-07T16:36:09","Local":"2020-11-07T17:36:09","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:47","Sunset":"17:19"}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS10 = {"StatusSNS":{"Time":"2020-11-07T17:36:09"}}
17:36:09 MQT: stat/tasmota_FD311C/STATUS11 = {"StatusSTS":{"Time":"2020-11-07T17:36:09","Uptime":"0T00:01:00","UptimeSec":60,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"NETGEAR","BSSId":"E0:3F:49:A8:C5:70","Channel":11,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
none

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-11-07:
Could be due to an insufficient amperage available, I've seen such with recurring brownout booting. Probably not a Tasmota issue in any way.

Jason2866, 2020-11-07:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

megamarco833, 2020-11-07:
> 
> 
> Could be due to insufficient amperage available, I've seen such with recurring brownout booting. Probably not a Tasmota issue in any way.

many thanks for your comment!
....what kind of power supply i should use?
actually i'm on this:
https://it.aliexpress.com/item/33037781855.html?spm=a2g0s.9042311.0.0.27424c4dzGof2O
5v 10A
i think that 10A should be more than enough....i'm wrong?
what kind of power supply i should use to power on 6 ESP8266 and ESP32 ?
thanks!!!
regards

Jason2866, 2020-11-07:
Not only Amperes count. Some caps for filtering. Google for `esp32 hardware design guidelines`

closed 2020-11-07 by ascillato2