arendst/Tasmota #9768: Would it be possible to provide a release with expressions and conditions enabled?

victorclaessen, 2020-11-07:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
The problem is that there is a feature (expressions/conditions) that is not exposed to the general public. Yes, it is possible for people to compile it for themselves. But isn't that just a whole lot of wasted time when you consider the total time requirement integrated over many people?

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Could you please include in the next release a binary that includes expressions and the if statement? (#define USE_RULES, #define USE_EXPRESSION and #define SUPPORT_IF_STATEMENT)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
The alternative is that many people have to spend the time to set up a build environment. Of course they will learn something, but they may not be willing to invest the time. And why should they? Compiling Tasmota may not be the thing they really really want to achieve, when they search deep down inside themselves.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-11-07:
Specialized variants built daily - https://github.com/tasmota/binaries/tree/circleci-build#rules-with-all-the-extras-enabled

victorclaessen, 2020-11-07:
Ah, cool. That's probably already enough for me, but perhaps for other people it would be good to have such a build included in the stable release program?

Jason2866, 2020-11-07:
The variants are stable (builded from Tasmota master branch) just not official build variants.

lalo-uy, 2020-11-07:
Will be nice to have a link to this in the docs.

meingraham, 2020-11-07:
@lalo-uy 

See the tip on the builds page - https://tasmota.github.io/docs/Builds/
>Tip
>
>You might find some of the features you need included in one of our unofficial experimental builds over at https://github.com/tasmota/binaries.

victorclaessen, 2020-11-07:
> 
> 
> @lalo-uy
> 
> See the tip on the builds page - https://tasmota.github.io/docs/Builds/
> 
> > Tip
> > You might find some of the features you need included in one of our unofficial experimental builds over at https://github.com/tasmota/binaries.

Yes, you are right. I totally missed that. Only now that you've pointed me to it I actually see what it says. Sorry about that. Will close this now. Thanks!

closed 2020-11-07 by victorclaessen