arendst/Tasmota #9773: Feature request support for LED transitions

dave-parsons, 2020-11-08:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No.  Transitions are supported on MagicHome controllers using ESPHome and Native firmware.   Would like the same capability in Tasmota.

**Describe the solution you'd like**  
support transitions - particularly during on / off move to desired brightness.

**Describe alternatives you've considered**  
ESPHome, Native

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-08:
Did you try `Fade` command? It looks to do what you are requesting

dave-parsons, 2020-11-09:
Ah yes - fade with MQTT a good option.

closed 2020-11-09 by dave-parsons