arendst/Tasmota #9780: SwitchMode15 for Multi-Press Functions

MisterBJ, 2020-11-08:
**Have you looked for this feature in other issues and in the docs?**  
I searched the documentation and issues but found no match.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I have installed a button from Gira (015100) with a Shelly i3. Tasmota 9.1.0 is installed on the Shelly. The button is configured as Button1IN in the template. Setoption73 is active. The Shelly recognizes whether the button has been pressed once, twice, three times or long and sends its MQTT message accordingly. But a single signal is always sent in a certain cycle.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
A function like SwitchMode15 that only sends an MQTT message when the status changes. A delay when the internal counter for keystrokes should be automatically set to 0 (without mqtt message) would perhaps also make sense.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
The only alternative would be the original Shelly firmware ... but who will;)

**Additional context**  
_Add any other context or screenshots about the feature request here._
I've seen the problem well enough with my own English.

**(Please, remember to close the issue when the problem has been addressed)**  
I will do that.

effelle, 2020-11-16:
> But a single signal is always sent in a certain cycle

I'm not understanding what you want to archive, what should be this signal? 
The equivalent of `SetOption73` for switches exists an is [`SetOption114`](https://tasmota.github.io/docs/Buttons-and-Switches/#setoption114).

MisterBJ, 2020-11-16:
hi effele,

here is a printout from the console:
18:12:17 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"DOUBLE"}}
18:12:18 MQT: stat/ShellyI34/RESULT = {"Button3":{"Action":"SINGLE"}}
18:12:20 MQT: stat/ShellyI34/RESULT = {"Button3":{"Action":"SINGLE"}}
18:12:20 MQT: stat/ShellyI34/RESULT = {"Button2":{"Action":"DOUBLE"}}
18:12:22 MQT: stat/ShellyI34/RESULT = {"Button2":{"Action":"SINGLE"}}
18:12:26 MQT: stat/ShellyI34/RESULT = {"Button3":{"Action":"SINGLE"}}
18:12:29 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}

explanation:
18:12:17 i pressed the button twice
18:12:22 new state "single" without doing anything. the problem is e.g. openHAB receives the new status and then reacts to it incorrectly. 
So I want a command like SwitchMode 15 only for buttons.
I hope this is a good explanation of the problem.

Short information1: So that we do not misunderstand a technical drawing of the wall switch I use.
[Link to drawing](https://encrypted-tbn0.gstatic.com/images?q=tbn%3AANd9GcTE2UZ3lA17H5n_U_appT6n0wL6lbXt0nuwTA&usqp=CAU)

Short information2: I don't know why Button2 is also "Double", because Button2 and Button3 have no switches or buttons.

Best regards,
Björn

effelle, 2020-11-16:
Please issue and paste here the result of `Backlog Template; Module; GPIO 255; Status 0`

effelle, 2020-11-16:
I'm a bit confused, 
you said nothing is hooked on buttons 2 and 3 but you have them configured:
```py
{"129":"Button_in2"},"GPIO13":{"130":"Button_in3"}
```
If they reacts  with no reasons you have some kind of RF interference or poor power signal.
Also, in your gpio list there is no `button1` but instead a Switch is configured. How comes you have a button1 reporting?
My suggestion is to remove the two buttons if not in use and see how the device responds.

MisterBJ, 2020-11-16:
> you said nothing is hooked on buttons 2 and 3 but you have them configured

Since it's a Shelly i3, I set up all inputs but didn't connect a real button / switch.
I took your advice and disabled the two entrances. After the change there are no responds.

> Also, in your gpio list there is no button1 but instead a Switch is configured. How comes you have a button1 reporting?

Sorry, this was just a test and I forgot to change it. I changed it again. See log:

`22:02:41 CMD: savedata 1
22:02:41 MQT: stat/ShellyI34/RESULT = {"SaveData":"ON"}
22:02:47 CMD: setoption 73 1
22:02:47 MQT: stat/ShellyI34/RESULT = {"Command":"Error"}
22:02:48 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:02:55 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:02:55 CMD: setoption73 1
22:02:55 MQT: stat/ShellyI34/RESULT = {"SetOption73":"ON"}
22:03:04 CMD: savedata 0
22:03:04 MQT: stat/ShellyI34/RESULT = {"SaveData":"OFF"}
22:03:10 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:19 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:27 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:30 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"TRIPLE"}}
22:03:31 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:32 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:35 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:40 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:44 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:52 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}`

The current config:
`22:09:17 CMD: Backlog Template; Module; GPIO 255; Status 0
22:09:17 MQT: stat/ShellyI34/RESULT = {"NAME":"Shelly i3","GPIO":[0,0,0,0,0,0,0,0,0,0,128,0,0,0],"FLAG":0,"BASE":18}
22:09:17 MQT: stat/ShellyI34/RESULT = {"Module":{"0":"Shelly i3"}}
22:09:17 MQT: stat/ShellyI34/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"128":"Button_in1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
22:09:17 MQT: stat/ShellyI34/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["ShellyI34"],"Topic":"ShellyI34","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[15,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:09:17 MQT: stat/ShellyI34/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:55","StartupUTC":"2020-11-16T21:02:22","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-11-16T17:58:21","SaveCount":225,"SaveAddress":"FB000"}}
22:09:17 MQT: stat/ShellyI34/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020.11.05 10:38:46","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"381/699"}}
22:09:17 MQT: stat/ShellyI34/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BJ24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00800000","00006000","00000001"]}}
22:09:17 MQT: stat/ShellyI34/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":400,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
22:09:17 MQT: stat/ShellyI34/STATUS5 = {"StatusNET":{"Hostname":"ShellyI34-5902","IPAddress":"192.168.1.81","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"10:52:1C:F1:D7:0E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:09:17 MQT: stat/ShellyI34/STATUS6 = {"StatusMQT":{"MqttHost":"smartpi","MqttPort":8883,"MqttClientMask":"ShellyI34","MqttClient":"ShellyI34","MqttUser":"openhab","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:09:17 MQT: stat/ShellyI34/STATUS7 = {"StatusTIM":{"UTC":"2020-11-16T21:09:17","Local":"2020-11-16T22:09:17","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:01","Sunset":"17:08"}}
22:09:17 MQT: stat/ShellyI34/STATUS10 = {"StatusSNS":{"Time":"2020-11-16T22:09:17"}}
22:09:17 MQT: stat/ShellyI34/STATUS11 = {"StatusSTS":{"Time":"2020-11-16T22:09:17","Uptime":"0T00:06:55","UptimeSec":415,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BJ24","BSSId":"98:9B:CB:58:B1:12","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}
22:09:20 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}`

effelle, 2020-11-16:
```py
22:03:10 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:19 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:27 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:30 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"TRIPLE"}}
22:03:31 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:32 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:35 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:40 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:44 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
22:03:52 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
```
If those log are from the device triggering alone then you have a interference issue.
How long are the cable connected to the button?
Try to play a bit with `ButtonDebounce`, or if you need just a single and double press use a switch with switchmode5 and enable setoption114. Switch code is less sensible to interference.

MisterBJ, 2020-11-17:
> How long are the cable connected to the button?

In my test setup, the cables are short, see picture.
[Test Case](https://abload.de/img/unadjustednonraw_thumedkl7.jpg)

> Try to play a bit with ButtonDebounce

Even at 1000 milliseconds button1 triggers even though I don't use the button:
17:22:12 MQT: stat/ShellyI34/RESULT = {"ButtonDebounce":1000}
17:22:29 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
17:22:59 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}

> or if you need just a single and double press use a switch with switchmode5 and enable setoption114

I would prefer multipress, but if nothing works, I'll have to try it.

effelle, 2020-11-17:
I still think you have some  interference issues, but since I'm not familiar with the Shelly better ask to someone more used to them. 
What do you think @arendst ?

MisterBJ, 2020-11-17:
`Button`
18:28:55 QPC: Reset
18:28:56 MQT: tele/ShellyI34/STATE = {"Time":"2020-11-17T18:28:56","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BJ24","BSSId":"98:9B:CB:58:B1:12","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}
=======
No reaction when a button is pressed.

`Button_i`
18:17:34 QPC: Reset
18:17:34 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}
=======
No reaction when a button is pressed. But the Shelly resets itself completely and I had to re-flash it.

effelle, 2020-11-17:
If with `button_i` the Shelly resets means you need an inverted button. Tasmota thinks you are holding down the button for 40 seconds and then will reset to factory. When it happens you should see the device in access point mode ready to be configured, no need to flash it again.

MisterBJ, 2020-11-18:
> When it happens you should see the device in access point mode ready to be configured

Good to know, I'll try it next time.

> Do a test with button_n

With `button_n` he only recognizes SINGLE:
`16:03:25 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:03:27 QPC: Reset
16:03:28 MQT: tele/ShellyI34/STATE = {"Time":"2020-11-18T16:03:28","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BJ24","BSSId":"98:9B:CB:58:B1:12","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}
16:03:30 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}
16:03:44 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:03:45 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:03:48 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:03:50 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:03:54 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}
16:04:03 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:04:06 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:04:07 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:04:10 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:04:14 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}
16:04:14 CMD: ButtonDebounce 1000
16:04:14 MQT: stat/ShellyI34/RESULT = {"ButtonDebounce":1000}
16:04:51 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}
16:05:07 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}
16:05:22 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}
16:05:23 CMD: ButtonDebounce 50
16:05:23 MQT: stat/ShellyI34/RESULT = {"ButtonDebounce":50}
16:05:28 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"SINGLE"}}
16:05:33 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}
16:05:48 MQT: stat/ShellyI34/RESULT = {"Button1":{"Action":"HOLD"}}`

closed 2020-12-19 by github-actions[bot]

MisterBJ, 2021-02-01:
Hello everybody,

on Saturday I tried a new Shelly i3 with the latest firmware (original) 1.9.0 and mqtt and so the status is only transmitted if the switch was actually operated. So the way it should be. In order to rule out a hardware defect in the old Shelly I3, I deleted its flash and flashed the original firmware on it and the device works perfectly.
It seems that the problem is with Tasmota or the settings, right?

Best regards,
Bjorn

MisterBJ, 2021-02-26:
@effelle 

Hi Guys,

I recognized Tasmota's problem.
The Shelly i3 sends its status every 60 seconds is normal.
But the original firmware always sends the following JSON container:
{"event": "SSS", "event_cnt": 18}
- That means the original FW always sends the last event, Tasmota always sends single instead of the last status (e.g. double) after pressing the buttons.
- In order to be able to react to a change, the original FW sends a counter (not Tasmota, at least I didn't find anything) that only answers when the event changes, e.g. from SSS to LS.
Maybe you can take this with you for the future;)

Best regards,
Bjorn

PsychoRS, 2022-09-12:
Has this issue been solved? Faced exactly the same problem

barbudor, 2022-09-12:
Which problem? There have been so many things discussed here....
I suggest that you open a new Discussion providing exact details of your configuration and problem

PsychoRS, 2022-09-13:
> Which problem? There have been so many things discussed here.... I suggest that you open a new Discussion providing exact details of your configuration and problem

The fact that the Shelly i3 is constantly sending ghost single presses when idle with Tasmota. Tried with the same debounce time that I use with ESPHome, with higher one, but it keeps sending single presses no matter what I do. It's working perfectly (no ghost presses) using ESPHome with a little debounce time (70ms).

It makes the Shelly i3 unusable with Tasmota if you want button inputs for multiclick actions.