arendst/Tasmota #9784: 8N2 serial configuration available for Smart Meter Interface

tiagofreire-pt, 2020-11-08:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Some newer firmware versions on `EDP Distribuição` smart meters (Landis+Gyr zcxe110cr), using MODBUS interface, are requiring a serial configuration of 8N**2**, instead the 8N1 currently available on Tasmota. This was detected by a trial-and-error approach, using an alternative rs485-to-wifi converter, capable of such serial configuration (8N2).

**Describe the solution you'd like**  
Adding support for Smart Meter Interface, on Tasmota, to serial communication with 2 stop bits (e.g., 8N2).

**Describe alternatives you've considered**  
No such alternatives available on SMI.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

tiagofreire-pt, 2020-11-08:
The issue is about the lines [2189](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_53_sml.ino#L2189) and [2197](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_53_sml.ino#L2197) of the `xsns_53_sml.ino` file.

I propose a new flag for the 2 stop bits:

```
- m2 = MODBus binary coding with serial mode 8N2
- M2 = MODBus binary coding with serial mode 8E2
```

In: https://tasmota.github.io/docs/Smart-Meter-Interface/

s-hadinger, 2020-11-08:
If it's only about sending 8N2 instead of 8N1, maybe we can just remove 8N1 and put 8N2 instead. 8N2 is backwards compatible with 8N1 when sending, just slightly slower. 

When receiving Tasmota will silently ignore the additional stop bit.

kugelkopf123, 2020-11-09:
@Gemu2015

gemu2015, 2020-11-09:
@s-hadinger 
i agree this should work.

@tiagofreire-pt 

to be sure if this works with your meter edit the code below

if (meter_ss[meters]->hardwareSerial()) {
          if (meter_desc_p[meters].type=='M') {
            Serial.begin(meter_desc_p[meters].params, SERIAL_8E1);
          }
          ClaimSerial();
          //Serial.setRxBufferSize(512);
        }

to

if (meter_ss[meters]->hardwareSerial()) {
          if (meter_desc_p[meters].type=='M') {
            Serial.begin(meter_desc_p[meters].params, SERIAL_8E2);
          } else {
            Serial.begin(meter_desc_p[meters].params, SERIAL_8N2);
          }
          ClaimSerial();
          //Serial.setRxBufferSize(512);
        }

if will do a pr witch defaults to 2 stopp bits soon

gemu2015, 2020-11-09:
tested with my MODBus device. SERIAL_8N2 does NOT work. so need a switch.

@tiagofreire-pt
please report if SERIAL_8N2 works with your meter.

ill then do  a pr with your proposed  switch   m2 or M2

nikito7, 2020-11-15:
m working
m2 working
Sagemcom

Can we delay first tele? Too many zeros

17:22:38 MQT: tele/edpbox1/SENSOR = {"Time":"2020-11-15T17:22:38","EB":{"Clock_H":0,"Clock_M":0,"Clock_S":0,"Voltage_L1":236.8,"Voltage_L2":235.8,"Voltage_L3":236.7,"Current_Sum":2.4,"Current_L1":1.1,"Current_L2":0.0,"Current_L3":1.3,"Power":459,"Power_L1":175,"Power_L2":3,"Power_L3":279,"PFactor":0.805,"PFactor_L1":0.670,"PFactor_L2":1.000,"PFactor_L3":0.888,"Frequency":49.9,"Energy_R1":0.0,"Energy_R2":0.0,"Energy_R3":0.0,"Energy_Sum":0.0,"Energy_Export":0.000,"Energy_Q1":0.00,"Energy_Q4":0.000}}

closed 2020-12-16 by github-actions[bot]