arendst/Tasmota #9788: Sonoff Zigbee bridge with Tasmota - include device info in the message

mgx0, 2020-11-09:
**Have you looked for this feature in other issues and in the docs?**  
yes but maybe I'm blind. Or is there any option to set to get this?

**Is your feature request related to a problem? Please describe.**  
no

**Describe the solution you'd like**  
get device manufacturer and model included in MQTT message like zigbee2mqtt does

**Describe alternatives you've considered**  
get device information by asking the bridge, but that's not that fast

**Additional context**  

this is what I get from zigbee2mqtt, as an example
```
battery: 51
device: object
applicationVersion: 2
friendlyName: "sensorname"
ieeeAddr: "0x00158xxxxxxxx"
manufacturerID: 4151
manufacturerName: "LUMI"
model: "WSDCGQ01LM"
networkAddress: 15462
powerSource: "Battery"
type: "EndDevice"
elapsed: 25
humidity: 57.92
last_seen: "2020-11-09T13:49:54+01:00"
linkquality: 7
temperature: 14.32
voltage: 2915
```

**(Please, remember to close the issue when the problem has been addressed)**  
okay

s-hadinger, 2020-11-09:
Hi, part of the information can be retrieved with `ZbStatus2`. There will be a new command to also dump the last known information (the ones displayed in the web UI). Stay tuned.

kroon040, 2020-11-10:
also the last seen? that should be nice for the smoke detector, with also the notation in the web (so last seen 40m ago)

mgx0, 2020-11-10:
yeah that would be also nice. It is displayed on the Tasmota web UI so it does exist
would be handy also for the water leak detector

mgx0, 2020-11-10:
but looking at the data I'm getting, the message contents vary. Sometimes there is BatteryPercentage and sometimes not. This applies as well for temperature/humidity data and other sensor data. Looks like the sensors send multiple data which zigbee2mqtt combined into one.

an example from Xiaomi water leakage sensor

[image]
[image]

s-hadinger, 2020-11-11:
#9810 introduces a new `ZbInfo` command to display all known information of a device, including the last_seen in both seconds from now and absolute epoch time.

mgx0, 2020-11-11:
that's great. I believe everyone will welcome if all of this information is supplied in every message coming from device, so we don't have to ask for it when we get updates.

I think it's a useful feature

s-hadinger, 2020-11-12:
@mgx0 I'm reluctant to include all the information with every messages for two reasons: 1/ it will create a massive amout of traffic with essentially redundant information, 2/ messages received also include the source endpoint which may be important with regards to which actuator was used. `ZbInfo` does not publish the endpoint information on purpose (although the information is internally kept) for the sake of simplicity. If I added the endpoints to ZbInfo, it would quickly become unreadable.

MQTT gateways have much more horsepower than the tiny ESP8266, data persistence should definitely be done at this level.

mgx0, 2020-11-12:
I understand. But what about an option to include it? a checkbox where everyone accepts the risk?
or is it really a no-go?

s-hadinger, 2020-11-12:
I would prefer a Rule: whenever you receive a message, trigger a `ZbInfo` right after. Which makes me think that `ZbInfo` should maybe have a different prefix to avoid infinite loop.

mgx0, 2020-11-12:
yeah for me that is a workaround from the start, but it is acceptable. It's true that I ran zigbee2mqtt on Pi4 so that power cannot be compared. should I close this issue then?

s-hadinger, 2020-11-13:
You can use a simple rule as this one: `Rule3 on zbreceived#? do zbinfo %ZBDEVICE% endon`

I understand it's not perfect though because you may miss some information that was not kept into memory (Tasmota doesn't remember all attributes)

mgx0, 2020-11-13:
good, thanks for that, I'll close this issue. but I have a question, last one.

when you look at the screenshot attached, you see two messages (look only at the originalPayload object, the others are calculated to keep up the structure I have built for zigbee2mqtt). Both arrived at the same second. Is the second one zbInfo? 

I have noticed that the second message is sent more frequently than the first one (that contains the sensor data). Is it some heartbeat message from the sensor itself or is it Tasmota sending this?

thanks a lot, I'll close this issue after you answer. or you can close it if you don't mind

[image]

s-hadinger, 2020-11-14:
Yes, that's part of the Xiaomi folklore. They sent sometimes messages compliant to ZCL, sometimes with their own format, including resending the `ModelId`.
Tasmota groups both messages into a single when they arrive within 350 ms.

mgx0, 2020-11-14:
Perfect, thanks for perfect response, I appreciate it. Closing now

closed 2020-11-14 by mgx0