arendst/Tasmota #9794: BMP/BME280 no readings

zloty777, 2020-11-09:
### PROBLEM DESCRIPTION
Hi
After upgrading from 8.5.1 to 9.1.0.1 (9.1.0 initially tried) I have no readings from these sensors. When rolling back to 8.5.1 all works perfectly fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-09:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

zloty777, 2020-11-09:
Hi
I have sent bug teport for the very first time here and do not really understand how to complete this form. Are you able to help me what details do I need to put in here? I am using Wemos D1 mini with both bme and bmp 280 sensors. When I rolled back to tasmota 8.5.1 all worked perfectly fine. When upgraded few times, just to check if I was doing something wrong, nothing appears on the website gui nor in Home Assistant. The only reading which seems to work, as the value changes from time to time, is wifi signal, status.

Many thanks for your helo
Regards
Darek

zloty777, 2020-11-09:
Hi
After upgrading from 8.5.1 to 9.1.0.1 (9.1.0 initially tried) I have no readings from these sensors. When rolling back to 8.5.1 all works perfectly fine.

zloty777, 2020-11-09:
Sorted. tasmota-sensors.bin must be used to get it working. Up to version 8.5.1 I figured to working it with tasmota.bin.

closed 2020-11-09 by zloty777