arendst/Tasmota #9795: DS18x20 powered by Power1 not detected

peterpepo, 2020-11-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
DS18s20 sensor powered from D8/GPIO15 is not detected after reboot/wake up from deepsleep. D8/GPIO15 is set as Relay1 controlled by Power1 1 / Power1 0.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini v2.0
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
MQT: stat/tasmota_XYZ123/RESULT = {"Module":{"18":"Generic"}}
MQT: stat/tasmota_XYZ123/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"4096":"DeepSleep"},"GPIO5":{"1312":"DS18x20"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"4864":"ADC Range"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
MQT: stat/tasmota_XYZ123/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":146,"Free":365,"Rules":"ON Dimmer#Boot DO RuleTimer1 30 ENDON ON Rules#Timer=1 DO DeepSleepTime 15 ENDON ON System#Boot DO Power1 1 ENDON ON System#Save DO Power1 0 ENDON"}
MQT: stat/tasmota_XYZ123/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
MQT: stat/tasmota_XYZ123/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
MQT: stat/tasmota_0EB047/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_XYZ123","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
MQT: stat/tasmota_0EB047/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:09:48","StartupUTC":"2020-11-09T18:19:46","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2020-11-09T18:09:56","SaveCount":71,"SaveAddress":"F5000"}}
MQT: stat/tasmota_0EB047/STATUS2 = {"StatusFWR":{"Version":"9.1.0(sensors)","BuildDateTime":"2020-11-07T11:59:31","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"442/699"}}
MQT: stat/tasmota_0EB047/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MY-AP","MY-AP-TWO"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
MQT: stat/tasmota_0EB047/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
MQT: stat/tasmota_0EB047/STATUS5 = {"StatusNET":{"Hostname":"tasmota_XYZ123-4167","IPAddress":"192.168.2.173","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"CE:BC:BC:FA:B0:48","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
MQT: stat/tasmota_0EB047/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.204","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_XYZ123","MqttUser":"mosq_sensorusr","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
MQT: stat/tasmota_0EB047/STATUS7 = {"StatusTIM":{"UTC":"2020-11-09T18:29:34","Local":"2020-11-09T19:29:34","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T04:00:00","Timezone":"+01:00","Sunrise":"03:58","Sunset":"17:16"}}
MQT: stat/tasmota_0EB047/STATUS10 = {"StatusSNS":{"Time":"2020-11-09T19:29:34","ANALOG":{"Range":71}}}
MQT: stat/tasmota_0EB047/STATUS11 = {"StatusSTS":{"Time":"2020-11-09T19:29:34","Uptime":"0T00:09:48","UptimeSec":588,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"MY-AP","BSSId":"AA:BB:CC:DD:EE:FF","Channel":1,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect DS18s20 DATApin to GPIO5, GNDpin to GND, PWRpin to GPIO15, pullup 4k7 resistor to PWRpin and DATApin. Turn the Relay1 off on System#Save. Turn the Relay1 on on System#Boot.

When Relay1 is Toggled off (in WEBui) and device reset (Reset1 or plug-off-and-on), DS18s20 is not detected. The Relay is powered on though (I can see in WEBui and I have also connected green led, so I know when the pin is HIGH/LOW).

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
DS18s20 can be powered off/on for power saving purposes. DS18s20 is correctly initialized when powered on System#Boot by Relay1.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-11-09:
Yes, sensors are supposed to be detected at boot time. With a DS18B20 sensor, I see this shortly after boot along with detection of other devices: `DSB: Sensors found 1`. Without it being up, the number is obviously 0, and that's the end of it.

peterpepo, 2020-11-09:
How to turn it off/on as described here: https://tasmota.github.io/docs/DeepSleep/#esp8266-deep-sleep-side-effects

>  I use one GPIO to trigger a BC337 transistor to switch OFF all connected devices during deep sleep.

sfromis, 2020-11-09:
Well, that may be fine for devices+drivers supporting being powered down/up, which does not include DS18x20.

peterpepo, 2020-11-10:
I've managed to achieve desired behaviour by moving ds18x20 power pin to D8, which is LOW during deepsleep. To prevent accidentaly turning the relay off via MQTT/WEBgui, I changed PowerOnstate1 to 4 (on on boot and disable further manipulation). Since manual manipulation is not needed anymore, I removed it from the rules.

Please can you verify, this is correct/expected behaviour and I can use it like this. (In another words - it won't be removed in next release. Or the fact it works is just a coincidence?).
If this is correct, please can you add to documentation of DeepSleep (https://tasmota.github.io/docs/DeepSleep/#esp8266-deep-sleep-side-effects)?

This setup seems to work for me:

1. Configure D8 / GPIO15 (magicpin) as Power1.
2. Move DS18x20 PWR to D8 / GPIO15.
3. Change PowerOnState for Power1 to 4
> 4 = turn relay(s) ON and disable further relay control 

> PowerOnState1 4

4. Remove power control from rules, thus following remains

> Rule1
ON Dimmer#Boot DO RuleTimer1 30 ENDON
ON Rules#Timer=1 DO DeepSleepTime 15 ENDON

It has been running for last 12 hours and seems to work. My indicator led as well as DS18S20 turn on after Esp8266 wakes from sleep, send the temperature and falls asleep again. Toggling Relay (in webgui) nor in homeassistant doesn't affect the functionality - control is in fact disabled in tasmota.

ascillato2, 2020-11-10:
> I've managed to achieve desired behaviour by moving ds18x20 power pin to D8, which is LOW during deepsleep.

Great!!! Thanks for sharing your solution.

> Please can you verify, this is correct/expected behaviour and I can use it like this.

You are correct.

> If this is correct, please can you add to documentation of DeepSleep

As you are enjoying Tamosta, please, help us and edit that article adding this information. (just click on the pencil in the top right of that page, then edit that page and click on propose pull request in order to be reviewed) Thanks!!!!!!

closed 2020-11-10 by ascillato2