arendst/Tasmota #9800: Shutter target position incorrect

Mikk36, 2020-11-10:
### PROBLEM DESCRIPTION
Shutter target position is incorrectly set for some ranges and is reported back incorrectly as well.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff 4CH R2
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer 1.2
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
10:58:38 MQT: hall_heating_controller/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:58:39 MQT: hall_heating_controller/stat/RESULT = {"Module":{"7":"Sonoff 4CH"}}
10:58:39 MQT: hall_heating_controller/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"35":"Button4"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
10:59:11 MQT: hall_heating_controller/stat/STATUS = {"Status":{"Module":7,"DeviceName":"Tasmota Hall Heating Controller","FriendlyName":["Hall Heating Radiator Valve","Hall Heating Controller Close","Hall Heating Fan Power","Hall Heating Controller 4"],"Topic":"hall_heating_controller","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
10:59:11 MQT: hall_heating_controller/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:19:03","StartupUTC":"2020-11-10T09:40:08","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2020-11-09T09:53:32","SaveCount":569,"SaveAddress":"FB000"}}
10:59:11 MQT: hall_heating_controller/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"489/699"}}
10:59:11 MQT: hall_heating_controller/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LakritoLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AC029","2805C8000100060000005A00000000000000","40000200","00006000","00000000"]}}
10:59:11 MQT: hall_heating_controller/stat/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:59:11 MQT: hall_heating_controller/stat/STATUS5 = {"StatusNET":{"Hostname":"hall_heating_controller-7061","IPAddress":"10.0.7.12","Gateway":"10.0.0.1","Subnetmask":"255.0.0.0","DNSServer":"10.0.0.2","Mac":"24:62:AB:4B:BB:95","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:59:11 MQT: hall_heating_controller/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.7.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4BBB95","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:59:11 MQT: hall_heating_controller/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-10T09:59:11","Local":"2020-11-10T10:59:11","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:52","Sunset":"17:15"}}
10:59:11 MQT: hall_heating_controller/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-10T10:59:11","Shutter1":{"Position":0,"Direction":0,"Target":0}}}
10:59:11 MQT: hall_heating_controller/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-10T10:59:11","Uptime":"0T00:19:03","UptimeSec":1143,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"LakritoLAN","BSSId":"18:E8:29:E7:4E:BE","Channel":1,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
10:59:11 MQT: hall_heating_controller/stat/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":1440,"Close":1440,"50perc":50,"Delay":0,"Opt":"0100","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
11:00:47 CMD: ShutterPosition1 45
11:00:47 SRC: WebConsole from 10.0.1.5
11:00:47 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "45"
11:00:47 SHT: Pos. in: payload 45 (2), payload 45, idx 1, src 9
11:00:47 SHT: lastsource 9:, real 0, target 866503, payload 45
11:00:47 SHT: Delay Start. var0 <99>=<>, max10s?
11:00:47 SHT: Delay Start. Done
11:00:47 SRC: Shutter
11:00:47 SHT: Switched relay: 1 by Shutter
11:00:47 SHT: Shutter 1: source: Shutter, powerstate_local 1, ShutterGlobal.RelayCurrentMask 1, manual change 0
11:00:47 MQT: hall_heating_controller/tele/STATE = {"Time":"2020-11-10T11:00:47","Uptime":"0T00:20:39","UptimeSec":1239,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"LakritoLAN","BSSId":"18:E8:29:E7:4E:BE","Channel":1,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:00:47 MQT: hall_heating_controller/stat/RESULT = {"POWER1":"ON"}
11:00:47 MQT: hall_heating_controller/stat/POWER1 = ON (retained)
11:00:47 MQT: hall_heating_controller/stat/RESULT = {"ShutterPosition1":45}
11:00:48 SHT: time: 24, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 23000, next_stop 0, target: 866503,  max_vel_change 0, dir: 1
11:00:48 SHT: time: 27, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 26000, next_stop 0, target: 866503,  max_vel_change 0, dir: 1
11:00:48 SHT: Shutter 1 Real 26000, Start 0, Stop 866503, Dir 1, Delay 0, Rtc 1.35 [s], Freq 0, PWM 0
11:00:48 MQT: hall_heating_controller/stat/RESULT = {"Shutter1":{"Position":1,"Direction":1,"Target":30}}
11:00:48 SHT: time: 28, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 27000, next_stop 0, target: 866503,  max_vel_change 0, dir: 1
11:00:48 SHT: time: 29, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 28000, next_stop 0, target: 866503,  max_vel_change 0, dir: 1
[... 354 lines cut ...]

```

### TO REPRODUCE
Configure module -> Sonoff 4CH (7)
Console:
Interlock 1,2
Interlock 1
SetOption80 1
ShutterMode1 1
ShutterRelay1 1
ShutterOpenDuration1 144
ShutterCloseDuration1 144
ShutterPosition1 45

### EXPECTED BEHAVIOUR
I expect shutter to move to position 45, when commanded so. It should also run for roughly 65 seconds out of 144 (ShutterOpenDuration1 144). Instead, it runs for 43 seconds, which stands for the reported target position 30.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Mikk36, 2020-11-10:
From testing I got these results:

command | target
-- | --
0 | 0
1 | 1
2 | 2
3 | 3
4 | 4
5 | 5
6 | 6
7 | 7
8 | 8
9 | 9
10 | 10
11 | 11
12 | 12
13 | 13
14 | 14
15 | 15
16 | 16
17 | 17
18 | 18
19 | 19
20 | 20
21 | 21
22 | 22
23 | 23
24 | 24
25 | 0
26 | 0
27 | 0
28 | 0
29 | 0
30 | 30
31 | 31
32 | 32
33 | 33
34 | 34
35 | 35
36 | 36
37 | 37
38 | 31
39 | 32
40 | 33
41 | 34
42 | 35
43 | 36
44 | 37
45 | 30
46 | 31
47 | 32
48 | 33
49 | 34
50 | 50
51 | 51
52 | 52
53 | 53
54 | 54
55 | 55
56 | 56
57 | 57
58 | 51
59 | 52
60 | 53
61 | 54
62 | 55
63 | 56
64 | 57
65 | 50
66 | 51
67 | 52
68 | 53
69 | 54
70 | 70
71 | 71
72 | 72
73 | 73
74 | 74
75 | 75
76 | 76
77 | 77
78 | 71
79 | 72
80 | 73
81 | 74
82 | 75
83 | 76
84 | 77
85 | 70
86 | 71
87 | 72
88 | 73
89 | 74
90 | 90
91 | 91
92 | 92
93 | 93
94 | 94
95 | 95
96 | 96
97 | 97
98 | 98
99 | 99
100 | 100

Mikk36, 2020-11-17:
In addition to that, you can see that it reports the position to be 0 between 24 and 30.

ascillato2, 2020-12-02:
Thread continues at https://github.com/arendst/Tasmota/issues/9906

closed 2020-12-02 by ascillato2

stefanbode, 2020-12-15:
I already fixed something in this area. It is a problem with the long run time of 144 Seconds. This causes the overflow. I will check if and how I can make some changes.

stefanbode, 2020-12-15:
I can reproduce the bug with the setting and will work on a fix. A Undo of the change is not possible because of new feature requirements for speed control

Mikk36, 2020-12-16:
Thank you, will wait for a new release.

Mikk36, 2020-12-18:
This seems to have been a partial fix, it still doesn't move to the correct position (target is incorrect).
`10:25:03 MQT: hall_heating_controller/stat/RESULT = {"ShutterPosition1":67}
10:25:03 MQT: hall_heating_controller/stat/RESULT = {"Shutter1":{"Position":50,"Direction":1,"Target":52}}`

Mikk36, 2021-01-08:
@stefanbode , any thoughts on this?

PR #10145 (2020-12-16): Fixed overflow on runtime > 100 sek #9800
## Description:
With 144 seconds runtime the calibration run into an overflow of the percentage calculation. Fixed with uint64 expand and reorder of multipliers.
**Related issue (if applicable):** fixes #9800
## Checklist:
  - [ x] The pull request is done against the latest dev branch
  - [x ] Only relevant files were touched
  - [x ] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ x] The code change is tested and works on Tasmota c [... cut]