arendst/Tasmota #9804: Tasmota fade resetting or not working with HA integration

RangerFX4, 2020-11-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hi,

I tried Tasmota 9.1.0 with setOption19 0 to integrate with HA but the problem is fade will always reset to 0 when I turned it off. furthermore, when turning the light on, the light will not open or not fade up when using relay to power the light then turn on the led. I made a script to have a delay turning the relay then the led. with so19 1 its working perfectly. 

TIA!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Basic
- [X] Tasmota binary firmware version number used: 9.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer 1.2
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
22:04:06 MQT: 505_cabinet_led_2/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"1376":"WS2812"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

22:15:23 MQT: 505_cabinet_led_2/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
22:15:23 MQT: 505_cabinet_led_2/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
22:15:23 MQT: 505_cabinet_led_2/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS = {"Status":{"Module":1,"DeviceName":"505 - Cabinet 2","FriendlyName":["505 - Cabinet 2 Relay","505 - Cabinet 2 LED"],"Topic":"505_cabinet_led_2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:43:21","StartupUTC":"2020-11-10T13:24:26","Sleep":50,"CfgHolder":4617,"BootCount":411,"BCResetTime":"2020-06-30T11:13:53","SaveCount":2241,"SaveAddress":"F6000"}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"414/699"}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["80028009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS5 = {"StatusNET":{"Hostname":"505_cabinet_led_2-4209","IPAddress":"10.0.40.150","Gateway":"10.0.40.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.40.1","Mac":"84:F3:EB:AF:70:71","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF7071","MqttUser":"hass","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-10T14:07:47","Local":"2020-11-10T22:07:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+08:00","Sunrise":"14:52","Sunset":"00:15"}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-10T22:07:47"}}
22:07:47 MQT: 505_cabinet_led_2/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-10T22:07:47","Uptime":"0T00:43:21","UptimeSec":2601,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Dimmer":75,"Color":"191,191,191","HSBColor":"0,0,75","Channel":[75,75,75],"Scheme":0,"Width":1,"Fade":"OFF","Speed":6,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IoT","BSSId":"C6:FB:E4:41:04:95","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. so19 0 and Fade 1 in Tasmota console
2.Using this in HA as turning on the relay and light

```

light:
  - platform: template
    lights:
      505_cabinet_led_2_template:
        value_template: "{{ is_state('light.505_cabinet_2_led', 'on') }}"
        turn_on:
          - service: switch.turn_on
            data:
              entity_id: switch.505_cabinet_2_relay
          - delay: 
              milliseconds: 300
          - service: light.turn_on
            data:
              entity_id: light.505_cabinet_2_led  
        turn_off:        
          - service: light.turn_off
            data:
              entity_id: light.505_cabinet_2_led
          - delay: 
              milliseconds: 3500
          - service: switch.turn_off
            data:
              entity_id: switch.505_cabinet_2_relay
```
3. turning on the switch 
4. turning off the switch
5. fade reset

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Fade will reset turning on off the switch

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

N/A

**(Please, remember to close the issue when the problem has been addressed)**

pcdiem, 2020-11-10:
Is SetOption85 (device groups) on?

RangerFX4, 2020-11-10:
> Is SetOption85 (device groups) on?

I just checked and it is off. Relay and LED (GPIO14) is on one Sonoff basic attached.

22:48:10 MQT: 505_cabinet_led_2/stat/RESULT = {"SetOption85":"OFF"}

emontnemery, 2020-11-10:
Home Assistant will set fade when the light is turned on or off.
To set a default fade, modify `light_profiles.csv`, and add a default row for your light:

For example, if your light's entity_id is `light.living_room`, to make it default to brihtness=100, and a 1s transition time;
```
id,x,y,brightness, transition
living_room.default 0,0,100,1
```

Note that the first row is a header row and will be ignored.

effelle, 2020-11-10:
Closing as in not a Tasmota issue since is about the native integration under Home Assistant. 
Please use [HaTasmota](https://github.com/emontnemery/hatasmota) or [Home Assistant](https://github.com/home-assistant/core) github if you have issue with the integration, thanks.

closed 2020-11-10 by effelle

RangerFX4, 2020-11-11:
Thanks for the tip. adding "transition: 3" for 3seconds to the script solved the problem.
```
  - platform: template
    lights:
      505_cabinet_led_2_template:
        value_template: "{{ is_state('light.505_cabinet_2_led', 'on') }}"
        turn_on:
          - service: switch.turn_on
            data:
              entity_id: switch.505_cabinet_2_relay
          - delay: 
              milliseconds: 300
          - service: light.turn_on
            data:
              entity_id: light.505_cabinet_2_led  
              transition: 3
        turn_off:        
          - service: light.turn_off
            data:
              entity_id: light.505_cabinet_2_led
              transition: 3
          - delay: 
              milliseconds: 3500
          - service: switch.turn_off
            data:
              entity_id: switch.505_cabinet_2_relay
```

emontnemery, 2021-01-29:
The Tasmota integration has been updated to not set speed or fade unless `transition` is actively set in the light turn on/off command from HA: https://github.com/home-assistant/core/issues/45620