arendst/Tasmota #9807: Zigbee Light MQTT request/improvement

Bibinsa, 2020-11-10:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Zigbee Light MQTT improvement

In order to have Home Assistant sync'd with Tasmota-Zigbee, it could be useful to have the ability to
Publish ZbLight JSON message on a per device topic like` tele/Zigbee/5ADF/LIGHTSTATE` or something else.

Home Assistant could subscribe to this topic for state/brightness/color as every keys are present (Power, Dimmer, X, Y, ColorMode, Hue)

If not, does anyone have any idea on how to get Power/Dimmer/X/Y at every update for MQTT subscriber ?
Thanks !

s-hadinger, 2020-11-11:
This is what `SetOption89` is supposed to do. Are you asking for something different?

Bibinsa, 2020-11-11:
> This is what `SetOption89` is supposed to do. Are you asking for something different?

Yes, `SetOption89` is doing well for tele response but not for `ZbLight` response.

```
12:39:53 CMD: ZbSend { "device":"0x34A7", "send":{"Power":"true"} }
12:39:53 MQT: tasmota/zbbridge-1/stat/RESULT = {"ZbSend":"Done"}
12:39:53 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbResponse":{"Device":"0x34A7","Name":"tradrfri_bulb_1","Command":"0006!01","Status":0,"StatusMessage":"SUCCESS","Endpoint":1,"LinkQuality":89}}
12:39:54 MQT: tasmota/zbbridge-1/tele/34A7/SENSOR = {"tradrfri_bulb_1":{"Device":"0x34A7","Name":"tradrfri_bulb_1","Power":1,"Endpoint":1,"LinkQuality":89}}
12:40:10 CMD: zblight 0x34A7
12:40:10 MQT: tasmota/zbbridge-1/stat/RESULT = {"ZbLight":{"tradrfri_bulb_1":{"Device":"0x34A7","Name":"tradrfri_bulb_1","Reachable":true,"Power":1,"Dimmer":254,"X":30138,"Y":26909,"CT":370,"ColorMode":2,"Hue":360}}}
12:40:10 MQT: tasmota/zbbridge-1/stat/RESULT = {"ZbLight":"Done"}
```

For color status we neeed `ZbLight` response because `ZbResponse` does not include all items (Dimmer, Power, X, Y CT, ...).
Unless I missed something in HomeAssistant Integration.

s-hadinger, 2020-11-11:
Ok, got it. I will unify the different places where the JSON is published.

PR #9835 (2020-11-13): Zigbee more consistent messages for ZbInfo and ZbLight
## Description:

Small refactoring about how messages are sent out, for more consistency between `ZbReceived`, `ZbInfo` and `ZbLight`. Now all three follow the same scheme with regards to SetOptions.

** **BREAKING CHANGE** **
`ZbInfo` now reports messages as `{"ZbInfo":{...}}` instead of `{"ZbRecevied":{...}}`, this avoids infinite loops in Rules, see #9788 

Also fixed a small bug of `"Hue"` being sent in `ZbLight` when it shouldn't.

**Related issue (if applicable):** fixes #9807 #9788 

## Checklist:
  - [x] The pull request is done ag [... cut]
closed 2020-11-13 by s-hadinger