arendst/Tasmota #9808: rule command output controll topic via setoption or own topic verbose level

riker65, 2020-11-11:
**Have you looked for this feature in other issues and in the docs?**  

yes nothing similar found 

**Is your feature request related to a problem? Please describe.**  
mqtt command output from rules are flushing the broker.
setoption4 only creates distingtion  
**Describe the solution you'd like**  

set verbose level for rule commands output 0 (supress) to 5 (debug) 
set topic for rules command output

**Describe alternatives you've considered**  
rule command output via own topic additional to stat, cmnd, tele

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-11:
Please, try command `weblog`

Thanks

riker65, 2020-11-11:
Hi I did set all logging to 0 no change

sfromis, 2020-11-11:
What was asked on Discord was to get rid of the /RESULT messages posted to MQTT for commands like Var.

Jason2866, 2020-11-11:
Why? The broker can easily handle this.

riker65, 2020-11-11:
> What was asked on Discord was to get rid of the /RESULT messages posted to MQTT for commands like Var.

But setoption4 ist only changing structure of message no amount of verbose level
No verbose level at all

riker65, 2020-11-11:
> Why? The broker can easily handle this.

Actually I am not sure.
Having a lot of disconnects and try to find the root cause.

sfromis, 2020-11-11:
The verbose level is simply not at all involved in this, except for confusion about what you mean. And it is not likely that MQTT traffic is the cause of disconnects.

ascillato, 2020-11-12:
> Why? The broker can easily handle this.
> 
> Actually I am not sure.
> Having a lot of disconnects and try to find the root cause.

How many devices do you have?

I have 50 Tasmota devices and most of them are in weblog 4 for testing, plus TV, pc, phones etc.

And the broker usage is veeeeery low. The raspberry where the mqtt broker is, is using about 1% of the CPU. So, I don't think your disconnections are due to MQTT messages. Check your wifi signal, your amount of devices supported by your wifi router, your wifi router quality, the wifi channel used, the wifi channels used close to your home, the type of wifi connection (20mb / 40mb, etc), your security protocol, etc. The firmware version of your router and update if there is any new version. Please, check in the Tasmota docs for wifi troubleshoting.

Your request to be able to set the loglevel of rule output is Ok, but as explained in an older issue, it requires a huge modification of Tasmota, because the MQTT report and the console output are the same. They use the same buffer. So, sorry, but your request is not likely to be implemented easily and not very soon.

riker65, 2020-11-12:
> > Why? The broker can easily handle this.
> > Actually I am not sure.
> > Having a lot of disconnects and try to find the root cause.
> 
> How many devices do you have?
> 
> I have 50 Tasmota devices and most of them are in weblog 4 for testing, plus TV, pc, phones etc.
> 
> And the broker usage is veeeeery low. The raspberry where the mqtt broker is, use about 1% of the CPU. So, I don't think your disconnections are due to MQTT messages. Check your wifi signal, your amount of devices supported by your wifi router, your wifi router quality, the wifi channel used, the wifi channels used close to your home, the type of wifi connection (20mb / 40mb, etc), your security protocol, etc. The firmware version of your router and update if there is any new version. Please, check in the Tasmota docs for wifi troubleshoting.
> 
> Your request to be able to set the loglevel of rule output is Ok, but as explained in #7953, it requires a huge modification of Tasmota, because the MQTT report routines and the console output are the same. They use the same buffer. So, sorry, but your request is not likely to be implemented easily and not very soon.

Hi 
my setup is using MQTT within FHEM [not](https://fhem.de/commandref.html#MQTT2_SERVER) Mosquitto
so all messages are routed through fhem .
having permormance issues with fhem broker , I assume caused by too many messages floading the logs.
probably will check with mosquitto only

another topic which I assume is related:#9809
[https://github.com/arendst/Tasmota/issues/9809](url)

sfromis, 2020-11-12:
Then you have problems with FHEM, not the broker part. If FHEM is to bad at ignoring messages, it may help to not subscribe to the unwanted messages, hence SetOption4. Did you try that before assuming that it would not be useful?

riker65, 2020-11-12:
> Then you have problems with FHEM, not the broker part. If FHEM is too bad at ignoring messages, it may help to not subscribe to the unwanted messages, hence SetOption4. Did you try that before assuming that it would not be useful?

Thanks just got the setoption4 info from you yesterday. filtering them now in FHEM
but because #9809 still is causing too much traffic

riker65, 2020-11-12:
maybe you can confirm #9809 as an issue? or what am I missing ?

Jason2866, 2020-11-12:
As @ascillato explained it is a hughe effort to change. Since it is not a problem hitting many setups it will not be done. Sorry.

closed 2020-11-13 by ascillato2