arendst/Tasmota #9809: rule  triggers too often using tele- prefix for Temp Sensor DS18B20

riker65, 2020-11-11:
### PROBLEM DESCRIPTION
In rules, trigger Tele- 
is triggering contant and not as described in intervall defined with  teleperiod

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- x[ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ESP generic ESP01
- [x] Tasmota binary firmware version number used: 

Program Version | 9.1.0(fullrules)
-- | --
2020-11-06T12:14:19
2_7_4_5/2.2.2-dev(38a443e)

  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:03:47 MQT: stat/TA_WP_ESP0143/TEMPLATE = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:03:48 MQT: stat/TA_WP_ESP0143/MODULE = {"Module":{"0":"Generic"}}
12:03:48 MQT: stat/TA_WP_ESP0143/GPIO = {"GPIO0":{"0":"None"},"GPIO1":{"1312":"DS18x20"},"GPIO2":{"257":"Relay_i2"},"GPIO3":{"224":"Relay1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
12:05:47 CMD: Backlog Rule1; Rule2; Rule3
12:05:47 MQT: stat/TA_WP_ESP0143/RULE = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":743,"Free":136,"Rules":"on System#Boot do \tBacklog var1 B1.6;var16 Tr0;var9 57;var14 A;VAR10 INIT-%timestamp%;teleperiod 15; publish tele/WP/%topic%/A RB %timestamp%;rule0 1;power1 on;var16 Tr0   endon  on Power1#State=0 do \t backlog publish tele/WP/%topic%/P1_0 P1_0-Stop; endon on Power1#State=0 do \t backlog     publish tele/WP/%topic%/S400 %timestamp%;publish cmnd/TA_WP_H_P1_P2_Lolin1/ruletimer2 2;  endon  on Power1#State=1 do \t backlog     publish cmnd/TA_WP_H_P1_P2_Lolin1/power2 1;  endon on event#SOLL_SSRHSichT do \tbacklog ME..."}
12:05:48 MQT: stat/TA_WP_ESP0143/RULE = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":487,"Free":24,"Rules":"on system#boot do var2 B2.58 endon  on Tele-DS18B20#Temperature do \tbacklog var11 %value%; var13 Tele-%timestamp% endon  on var11#state>%MEM9% do \tbacklog teleperiod 300;var16 Tr0;var15 0;publish tele/WP/%topic%/WPT_Sich_Hot I%VAR1%S_M%MEM9% ; var4 HOT; RuleTimer1 6;publish cmnd/TA_WP_H_P1_P2_Lolin1/RT3 2 ;publish cmnd/TA_WP_H_P1_P2_Lolin1/RT2 3; endon  on var11#state<%MEM8% do  \tbacklog var16 Tr1;var15 1;IF (%VAR14% = A) power1 1 ENDIF ;teleperiod 100;var6 R_V11-%timestamp%; endon "}
12:05:48 RUL: Rule3 on system#boot do var3 B3.41 endon  on Time#Minute|5 do 	backlog publish tele/WP/%topic%/THSim5 HSich-Max%MEM9%C-%var1%%var2%%var3%-%timestamp%;var,mem endon on Time#Minute|10 do 	backlog publish tele/WP/%topic%/THSim10 HSich-Max%MEM9%C-%var1%%var2%%var3%-%timestamp%;var,mem endon on Time#Minute|60 do 	backlog publish tele/WP/%topic%/THSim60 HSich-Max%MEM9%C-%var1%%var2%%var3%-%timestamp%;var,mem endon on Time#Minute|5 do backlog VAR16 Tr0; Var15 0; publish tele/%topic%/WP/h_var16
12:05:48 RUL: Rule3 +v16-%var16%-%var1%%var2%%var3% endon   on event#TR=ON  do publish cmnd/TA_WP_H_P1_P2_Lolin1/power2 1 endon on event#TR=R3ON  do publish cmnd/TA_WP_H_P1_P2_Lolin1/power3 1 endon
12:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:07:07 CMD: status 0
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS = {"Status":{"Module":0,"DeviceName":"TA_WP_ESP0143","FriendlyName":["TA_WP_ESP0143","Tasmota2"],"Topic":"TA_WP_ESP0143","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":4,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"1T14:36:36","StartupUTC":"2020-11-09T19:30:31","Sleep":50,"CfgHolder":4617,"BootCount":519,"BCResetTime":"2020-04-12T09:38:08","SaveCount":2297,"SaveAddress":"F4000"}}
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS2 = {"StatusFWR":{"Version":"9.1.0(fullrules)","BuildDateTime":"2020-11-06T12:14:19","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"493/699"}}
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.0.22","LogPort":514,"SSId":["TR_ESP","TR7272"],"TelePeriod":100,"Resolution":"558180C0","SetOption":["00008019","2805C8000100060000005A00000000000000","008000C8","00006000","00000000"]}}
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS4 = {"StatusMEM":{"ProgramSize":612,"Free":388,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004084","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,38","Sensors":"1,2,3,4,5,6"}}
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS5 = {"StatusNET":{"Hostname":"TA_WP_ESP0143-2479","IPAddress":"192.168.6.116","Gateway":"192.168.0.31","Subnetmask":"255.255.240.0","DNSServer":"192.168.0.31","Mac":"EC:FA:BC:9B:E9:AF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.9","MqttPort":1893,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BE9AF","MqttUser":"DVES_USER","MqttCount":12,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS7 = {"StatusTIM":{"UTC":"2020-11-11T10:07:07","Local":"2020-11-11T12:07:07","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":"+02:00","Sunrise":"08:53","Sunset":"18:14"}}
12:07:08 MQT: stat/TA_WP_ESP0143/STATUS10 = {"StatusSNS":{"Time":"2020-11-11T12:07:07","DS18B20":{"Id":"021313A9DEAA","Temperature":17.4},"TempUnit":"C"}}
12:07:08 MQT: stat/TA_WP_ESP0143/STATUS11 = {"StatusSTS":{"Time":"2020-11-11T12:07:08","Uptime":"1T14:36:37","UptimeSec":138997,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":67,"MqttCount":12,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":2,"SSId":"TR7272","BSSId":"46:A4:3C:DD:17:E1","Channel":10,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:09"}}}
12:07:08 MQT: tele/TA_WP_ESP0143/STATE = {"Time":"2020-11-11T12:07:08","Uptime":"1T14:36:37","UptimeSec":138997,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":101,"MqttCount":12,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":2,"SSId":"TR7272","BSSId":"46:A4:3C:DD:17:E1","Channel":10,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:09"}}
12:07:08 MQT: tele/TA_WP_ESP0143/SENSOR = {"Time":"2020-11-11T12:07:08","DS18B20":{"Id":"021313A9DEAA","Temperature":17.4},"TempUnit":"C"}
12:07:08 RUL: TELE-DS18B20#TEMPERATURE performs "backlog var11 17.4; var13 Tele-2020-11-11T12:07:08"

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
I used setoption4 to have more clear info on output
12:07:07 CMD: status 0
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS = {"Status":{"Module":0,"DeviceName":"TA_WP_ESP0143","FriendlyName":["TA_WP_ESP0143","Tasmota2"],"Topic":"TA_WP_ESP0143","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":4,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:07:07 MQT: stat/TA_WP_ESP0143/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"1T14:36:36","StartupUTC":"2020-11 [... cut ...]
2020-11-11_09:35:46 TA_WP_ESP0143 Var13: Tele-2020-11-11T10:35:44
2020-11-11_09:35:46 TA_WP_ESP0143 Var13: Tele-2020-11-11T10:35:46
2020-11-11_09:35:48 TA_WP_ESP0143 Var13: Tele-2020-11-11T10:35:47
2020-11-11_09:35:49 TA_WP_ESP0143 Var13: Tele-2020-11-11T10:35:49
2020-11-11_09:35:51 TA_WP_ESP0143 Var13: Tele-2020-11-11T10:35:51
2020-11-11_09:35:52 TA_WP_ESP0143 Var13: Tele-2020-11-11T10:35:53
2020-11-11_09:35:54 TA_WP_ESP0143 Var13: Tele-2020-11-11T10:35:54

**(Please, remember to close the issue when the problem has been addressed)**

riker65, 2020-11-11:
even a simple rule as 
`on var1 do var12 RV1-%timestamp% endon`

was triggered multiple times without changing var1.

`
zwh100@ub9:/opt/fhem/log$  cat TA_WP_ESP0143-2020-11-11.log  | grep Var12:
2020-11-11_11:20:04 TA_WP_ESP0143 Var12: RV1-2020-11-11T12:20:01
2020-11-11_11:20:04 TA_WP_ESP0143 Var12: RV1-2020-11-11T12:20:02
2020-11-11_11:20:30 TA_WP_ESP0143 Var12: RV1-2020-11-11T12:20:28
2020-11-11_11:20:30 TA_WP_ESP0143 Var12: RV1-2020-11-11T12:20:29
2020-11-11_11:25:04 TA_WP_ESP0143 Var12: RV1-2020-11-11T12:25:01
2020-11-11_11:30:02 TA_WP_ESP0143 Var12: RV1-2020-11-11T12:30:01
2020-11-11_11:35:02 TA_WP_ESP0143 Var12: RV1-2020-11-11T12:35:01
zwh100@ub9:/opt/fhem/log$ cat TA_WP_ESP0143-2020-11-11.log  | grep Var1:
zwh100@ub9:/opt/fhem/log$ cat TA_WP_ESP0143-2020-11-11.log  | grep Var2:
2020-11-11_11:20:30 TA_WP_ESP0143 Var2: B2.59
zwh100@ub9:/opt/fhem/log$
`

Jason2866, 2020-11-12:
`\tBacklog` is not a valid command

riker65, 2020-11-12:
Hi I thought in fullrules version if support is enabled

riker65, 2020-11-12:
Will check for /tbacklog and update

riker65, 2020-11-12:
> `\tBacklog` is not a valid command
> There is no `if` support in any precompiled binary!
> Dont expect a predictable result with so many errors in the rules

22:29:54 RSL: stat/TA_WP_ESP0143/IF = {"If":"Done"} so IF support is enabld in  precompiled fullrule version

riker65, 2020-11-12:
> > `\tBacklog` is not a valid command
> > There is no `if` support in any precompiled binary!
> > Dont expect a predictable result with so many errors in the rules
> 
> 22:29:54 RSL: stat/TA_WP_ESP0143/IF = {"If":"Done"} so IF support is enabld in precompiled fullrule version

\t is ignored I think, rules are working ....in general, ut Tele-DSxxx is triggering permanently

Jason2866, 2020-11-12:
Please provide a simple correct rule which generates the issue.
No one can help without.

riker65, 2020-11-12:
`on Tele-DS18B20#Temperature do
backlog var11 %value%; var13 Tele-%timestamp% endon `

this simple rule is constant triggering and not on Teleperiod intervall only.

so I think Tele- is ignored.

does this help?

thanks for checking

sfromis, 2020-11-12:
That exact rule (copy/paste) works as expected on a board of mine, running 9.1.0.1. Looks like an issue local to your setup. I'd suggest doing a `Reset 6`, reconfig what's needed and test again.

riker65, 2020-11-13:
> That exact rule (copy/paste) works as expected on a board of mine, running 9.1.0.1. Looks like an issue local to your setup. I'd suggest doing a `Reset 6`, reconfig what's needed and test again.

ok strange . will follow your advise and update

ascillato2, 2020-11-17:
Hi, any news on this? Were you able to solve your issue?  Thanks.

closed 2020-11-17 by ascillato2

riker65, 2020-11-18:
Sorry was busy could not test so far will update later