arendst/Tasmota #9811: 32bit data for KNX Tx/Rx

BBBits, 2020-11-11:
**Have you looked for this feature in other issues and in the docs?**  
--
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
no.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like to send an RF code Via KNX
I tried it but the number format is wrong
I.E.
00:10:07 RUL: RFRECEIVED#DATA performs "knxtx_val1 0X7CD7EC"
00:10:07 KNX: KNX TX 1 = 0.00 sent to 7.1.0
00:10:07 MQT: stat/tasmota_378864/RESULT = {"KnxTx_Val1":"0.00"}
The reason is that I have a Number of 8 button RF remotes and would run out of rule space if I tried to create a rule for each one.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I wrote my own prog to do it but KNX 32bit receive data isn't supported by tasmota.
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-11-12:
Knxtx_val is for sending numeric values. It can't send text or hex values. For text or hex, a new command/feature should be added. Adding the feature request label for that.

Anyway, if you have 8 buttons as you said, why you run out of rule space? Now Tasmota make a compression of rules so, a lot of rules can fit in memory. Have you tried all your codes? Are you using latest Tasmota?

BBBits, 2020-11-13:
Hi,
there are 8 buttons on each remote.
currently have 4 remotes , more on the way.
also have pir's .
when i use setoption28 1 , it then gives a decimal number for rf received .

14:23:18 RUL: RFRECEIVED#DATA performs "knxtx_val1 1706469"
14:23:18 KNX: KNX TX 1 = 1706469.00 sent to 7.1.0
14:23:18 RSL: RESULT = {"KnxTx_Val1":"1706469.00"}

then on another device
14:23:18 KNX: Received from 7.1.0 Command Write: -629.32 to KNX RX 1
14:23:18 KNX: Received from 7.1.0 Command Write: -629.32 to KNX RX 1
14:23:18 KNX: Received from 7.1.0 Command Write: -629.32 to KNX RX 1

BBBits, 2020-11-13:
BTW , 
I just upgraded a device from 8.4 to 9 and there is no longer an entry in the 
"Data to Send to Group Addresses" dropdown for buttons or switches that are configured.
I had 4 swiches setup on a joystick and it has stopped working.

arendst, 2020-11-13:
> I just upgraded a device from 8.4 to 9 and there is no longer an entry in the
> "Data to Send to Group Addresses" dropdown for buttons or switches that are configured.
> I had 4 swiches setup on a joystick and it has stopped working.

I'll try to fix this now.

BBBits, 2020-11-14:
Thanks Theo,
9.1.0.1  and joystick knxing again.

ascillato, 2020-11-16:
The KNX DataPointTypes (DPT) are listed [here](https://support.knx.org/hc/en-us/articles/115001133744-Datapoint-Type)

Tasmota is using DPT9 (16bit Float) for `KnxTx_Valn` commands and sensors.

Let's make it simple and just change for sensors and for `KnxTx_Valn` commands, the KNX encoding from DPT9 to DPT14. This increment in resolution makes no increase for RAM/Flash usage due to Tasmota was already using internally 32 Bits Float Variables. In fact, the KNX encoding for DPT14 don't need the use of POW or LOG, reducing the flash usage.

This is going to be a breaking change, but I think, we should have used 32 bits from the start. Any user sending any sensor value with Tasmota KNX, or using `KnxTx_Valn` commands, should upgrade all their interconnected Tasmota devices and also upgrade this DPT in its KNX Network.

I'm now working on this.

ascillato2, 2020-11-16:
Fixed by #9888 

Please test it. Thanks.

closed 2020-11-16 by ascillato2

BBBits, 2020-11-17:
Thanks Adrian !
It works .  Ye Ha
I can't get the Tx happening on an esp32.
It says it is in the log 
17/11/20 19:51:50.586 19:51:48 RUL: RFRECEIVED#DATA performs "knxtx_val1 1706472"
17/11/20 19:51:50.598 19:51:48 KNX: KNX TX 1 = 1706472.00 sent to 7.1.0
but it isn't heard on network.
I will have a tinker in the driver in the next few days & see If I can get it going

BBBits, 2020-11-17:
Found problem
lib\libesp32\ESP32-to-ESP8266-compat\src\ESP8266WiFi.h
missing code in 
        void beginPacketMulticast(IPAddress multicast, uint16_t port, IPAddress interfaceAddr)
        {
        WiFiUDP::beginMulticastPacket(); // ******  this was not there *******
        }

BBBits, 2020-11-17:
@arendst 
WiFiUDP::beginMulticastPacket()  not updated in commit

arendst, 2020-11-17:
I know. I expect there need to be variables within the function so I need to investigate what variables.

BBBits, 2020-11-17:
it works as long as the multicast is initiated first 
and it is in the knx lib
beginMulticast(WiFi.localIP(),  MULTICAST_IP, MULTICAST_PORT);

BBBits, 2020-11-17:
perhaps something like this
            if(WiFiUDP::beginMulticastPacket()==0){
               WiFiUDP::beginMulticast(multicast, port);
               WiFiUDP::beginMulticastPacket();
            }

BBBits, 2020-11-17:
thanks theo,
not sure why , but have tried doing an ota with the .gz file a number of times and it has failed each time with
"Upload buffer miscompare"
worked with .bin file so tried the qz again and it failed .
has there been a change with esp32 gz or do i have some odd build problem

BBBits, 2020-12-08:
New remotes arrived and first one doesn't work in rules
Received 14829752.00  & cmd Data is 1482975  
20:42:16 KNX: Received from 7.1.0 Command Write: 14829752.00 to KNX RX 1
20:42:16 SRC: Knx
20:42:16 CMD: Group 0, Index 1, Command "EVENT", Data "KNXRX_VAL1=1482975"
20:42:16 MQT: stat/tasmota_C4365F/RESULT = {"Event":"Done"}
20:42:16 KNX: Received from 7.1.0 Command Write: 14829752.00 to KNX RX 1
20:42:16 KNX: Received from 7.1.0 Command Write: 14829752.00 to KNX RX 1

BBBits, 2020-12-08:
there are at least 2 declarations of 
char command[25];  that I increased to char command[35]; & it solved the problem
first is at line # 593.
I don't fully understand all the code so not gonna do a PR

ascillato, 2020-12-08:
https://github.com/arendst/Tasmota/pull/10096