arendst/Tasmota #9818: LOLIN (WEMOS) D1 mini I2C Dual Motor Driver TB6612FNG

Brackbouille, 2020-11-12:
**Have you looked for this feature in other issues and in the docs?**

Hello, I checked the docs, this is not available, but a request was opened
https://github.com/arendst/Tasmota/issues/7719

This was for an old version, A new and actual version is available

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

I would love to have i2c support in tasmota for WeMos D1 Mini motor shield, this is a missed in the i2c list

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Product link 

https://www.aliexpress.com/i/32700182142.html

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-12:
Hi,

A motor shiels for wemos is already supported. Please, check if the driver added in https://github.com/arendst/Tasmota/pull/7764 works for you. Thanks.

closed 2020-11-12 by ascillato2

dcast78, 2021-01-29:
The code of the old version V1(https://github.com/wemos/WEMOS_Motor_Shield_Arduino_Library) and the new version (https://github.com/wemos/LOLIN_I2C_MOTOR_Library) are very different.

The driver added in #7764 is not working with new version.

Please add a support for new version.

Thanks

ascillato, 2021-01-29:
Please, use the libraries already included in Tasmota that are tuned to use less RAM and FLASH. Thanks.

Brackbouille, 2021-03-29:
Hi Ascillato

I bought the lollin V2 motor drivers but i can't make it work with tasmota.

I took the correct bin (sensors), configure SDA and SCL.
I can detect the motor drivers at 0x30 with I2Cscan but how could i use it ?

I really apreciate if you could help me to make it work.

regards

ascillato, 2021-03-29:
Hi,

Inside the driver you have a brief usage explanation:

https://github.com/arendst/Tasmota/blob/d7d750c928224f52289f72576611f34c085f2cca/tasmota/xdrv_34_wemos_motor_v1.ino#L22-L45

Also, there is a full explanation in the **show and tell** section of discussions at:

https://github.com/arendst/Tasmota/discussions/categories/show-and-tell?discussions_q=Motor+category%3A%22Show+and+tell%22

If you want to chat with the people that are already using it, please visit the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a very dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)