arendst/Tasmota #9823: Full support of the Xiaomi Aqara Opple switch 3 bands? (ZbBridge)

pingel20, 2020-11-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Double click and long click (some keys) are not supported on the Aqara Opple switch 3 bands
ModelId: lumi.remote.b686opcn01

Double clicking the keys behaves as a single click.
Key 1 and 2 (upper row) do not respond to long click, they show as a single click.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: 9.1.0_____
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:05:26 MQT: stat/tasmota_D7F2A1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:05:27 MQT: stat/tasmota_D7F2A1/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
15:05:27 MQT: stat/tasmota_D7F2A1/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Double click a key, long click key 1 or 2

using Sonoff ZbBridge with Tasmota 9.1.0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A correct MQTT message when double clicking a key or long clicking key 1 or 2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-12:
Please report `weblog 3` logs when you do a simple and a double click.

pingel20, 2020-11-12:
Button 1 single click:

```
20:42:56 ZIG: {"ZbEZSPReceived":"450000040106000101000100007EC0CC595EFFFF03118C0002"}
20:42:56 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":92,"securityuse":0,"seqnumber":126,"fc":"0x11","manuf":"0x0000","transact":140,"cmdid":"0x00","payload":""}}
20:42:56 ZIG: ZbZCLRawReceived: {"0x5E59":{"0006!00":"","Power":0,"Endpoint":1,"LinkQuality":92}}
20:42:56 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":92}}}
```

Button 1 double click:

```
20:43:10 ZIG: {"ZbEZSPReceived":"450000040106000101000100007FBCCB595EFFFF03118D0002"}
20:43:10 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":89,"securityuse":0,"seqnumber":127,"fc":"0x11","manuf":"0x0000","transact":141,"cmdid":"0x00","payload":""}}
20:43:10 ZIG: ZbZCLRawReceived: {"0x5E59":{"0006!00":"","Power":0,"Endpoint":1,"LinkQuality":89}}
20:43:10 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":89}}}
```

---------------------------------------------------

Button 6 single click:

```
20:48:20 ZIG: {"ZbEZSPReceived":"4500000401000301010001000080F8DA595EFFFF0E118E4C034500070000000000000002"}
20:48:20 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":768,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":128,"securityuse":0,"seqnumber":128,"fc":"0x11","manuf":"0x0000","transact":142,"cmdid":"0x4C","payload":"0345000700000000000000"}}
20:48:20 ZIG: ZbZCLRawReceived: {"0x5E59":{"0300!4C":"0345000700000000000000","ColorTempStepDown":69,"Endpoint":1,"LinkQuality":128}}
20:48:20 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0300!4C":"0345000700000000000000","ColorTempStepDown":69,"Endpoint":1,"LinkQuality":128}}}

```

Button 6 double click:

```
20:49:32 ZIG: {"ZbEZSPReceived":"4500000401000301010001000081FFDD595EFFFF0E118F4C034500070000000000000002"}
20:49:32 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":768,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":136,"securityuse":0,"seqnumber":129,"fc":"0x11","manuf":"0x0000","transact":143,"cmdid":"0x4C","payload":"0345000700000000000000"}}
20:49:32 ZIG: ZbZCLRawReceived: {"0x5E59":{"0300!4C":"0345000700000000000000","ColorTempStepDown":69,"Endpoint":1,"LinkQuality":136}}
20:49:32 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0300!4C":"0345000700000000000000","ColorTempStepDown":69,"Endpoint":1,"LinkQuality":136}}}

```

(button 1: upper left, button 6: lower right)

MattWestb, 2020-11-13:
@s-hadinger The opple switches have 2 working modes. One is light steering with On/Off, Dimm Upp/Down and scenes control. The other mode is one HA mode with click click click. . . .

pingel20, 2020-11-14:
It would be nice to catch all button actions (single, double, long click), then you can use them for whatever you like.

For completeness I also have the logging of button 1 and 2 (top row), single and long click:

Button 1 single click:

```
15:23:11 ZIG: {"ZbEZSPReceived":"4500000401060001010001000098FCDB595EFFFF0311B50002"}
15:23:11 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":131,"securityuse":0,"seqnumber":152,"fc":"0x11","manuf":"0x0000","transact":181,"cmdid":"0x00","payload":""}}
15:23:11 ZIG: ZbZCLRawReceived: {"0x5E59":{"0006!00":"","Power":0,"Endpoint":1,"LinkQuality":131}}
15:23:11 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":131}}}

```
Button 1 long click:

```
15:24:07 ZIG: {"ZbEZSPReceived":"4500000401060001010001000099ECD7595EFFFF0311B60002"}
15:24:07 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":120,"securityuse":0,"seqnumber":153,"fc":"0x11","manuf":"0x0000","transact":182,"cmdid":"0x00","payload":""}}
15:24:07 ZIG: ZbZCLRawReceived: {"0x5E59":{"0006!00":"","Power":0,"Endpoint":1,"LinkQuality":120}}
15:24:07 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":120}}}
```

------------------------------------------

Button 2 single click:

```
15:26:27 ZIG: {"ZbEZSPReceived":"450000040106000101000100009AD8D2595EFFFF0311B70102"}
15:26:27 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":107,"securityuse":0,"seqnumber":154,"fc":"0x11","manuf":"0x0000","transact":183,"cmdid":"0x01","payload":""}}
15:26:27 ZIG: ZbZCLRawReceived: {"0x5E59":{"0006!01":"","Power":1,"Endpoint":1,"LinkQuality":107}}
15:26:27 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":107}}}

```

Button 2 long click:

```
15:27:18 ZIG: {"ZbEZSPReceived":"450000040106000101000100009BD4D1595EFFFF0311B80102"}
15:27:18 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":105,"securityuse":0,"seqnumber":155,"fc":"0x11","manuf":"0x0000","transact":184,"cmdid":"0x01","payload":""}}
15:27:18 ZIG: ZbZCLRawReceived: {"0x5E59":{"0006!01":"","Power":1,"Endpoint":1,"LinkQuality":105}}
15:27:18 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":105}}}

```

s-hadinger, 2020-12-10:
Hold on, it's still on my todo list.

s-hadinger, 2020-12-15:
As MattWestb mentioned it, the switch has 2 operations mode. The default is the one you have and makes no difference between single/double/long presses.

left click: turn off
right click: turn on
left double click: light dim down (by steps of 33%)
right double click: light dim up (by steps of 33%)
long left click: warm white
long right click: cold white

To read of change the operation mode, you need to use the following commands:

Read:
```
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Cluster":"0xFCC0","Read":9}
```

Write:
```
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"FCC0/0009%10":1}}
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"FCC0/0009%10":0}}
```

Can you please confirm? If it works, I can add the `OppleMode` attribute to make it simpler

s-hadinger, 2020-12-15:
In the coming changes, it will be:

Read:
```
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Read":{"OppleMode":true}}
```

Write:
```
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":0}}
```

MattWestb, 2020-12-15:
NICE Stephan !!

I think I can investing in some OPPLE switches now then deCONZ was making them HA only and not possible shushing / changing operating modes of them so it was a braking deal for my :-))))

The Santa Claus is bringing good things this year in both soft (ware) and hard (ware) packages !

s-hadinger, 2020-12-15:
Yes clearing the backlog for Christmas

closed 2020-12-15 by s-hadinger

MattWestb, 2020-12-15:
Guten Rutsch :-))

pingel20, 2020-12-15:
Running dev version 9.1.0.2 of today:

Initially I checked the mode with
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Read":{"OppleMode":true}}
this gave OppleMode:0

then tried to change to mode 1 with
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}

this gives:
```
15:20:21 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}
15:20:21 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
15:20:29 ZIG: {"ZbEZSPReceived":"800042595E"}
15:20:29 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
15:20:38 ZIG: {"ZbEZSPReceived":"800042595E"}
15:20:38 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
15:20:47 ZIG: {"ZbEZSPReceived":"800042595E"}
15:20:47 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
```
Reading the mode with
ZbSend {"Device":"0x5E59","Manuf":"0x115F","Read":{"OppleMode":true}}
now also gives the same MAC_INDIRECT_TIMEOUT messages, so I can no longer read the operation mode, but I think it didn't change because button 2 single and double click both give a "Power":1 message.

If you need any further info please ask.

s-hadinger, 2020-12-15:
`MAC_INDIRECT_TIMEOUT` means that the device is not reachable (proably in sleep mode). Please press a button just before sending the command.

pingel20, 2020-12-15:
I already did that. Next result:

Two button presses and then a ZbSend Write 

```
20:01:22 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0300!4C":"0145000700000000000000","ColorTempStepUp":69,"Endpoint":1,"LinkQuality":120}}}
20:01:34 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0300!4C":"0345000700000000000000","ColorTempStepDown":69,"Endpoint":1,"LinkQuality":113}}}
20:01:42 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}
20:01:42 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
20:01:50 ZIG: {"ZbEZSPReceived":"800042595E"}
20:01:50 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
20:01:59 ZIG: {"ZbEZSPReceived":"800042595E"}
20:01:59 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
20:02:09 ZIG: {"ZbEZSPReceived":"800042595E"}
20:02:09 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
```

s-hadinger, 2020-12-15:
I have no idea why the coordinator can't find a route. Can you copy the Zigbee map diagram (needs the latest dev version)

pingel20, 2020-12-16:
I gave the device a friendly name. 0x5E59 is the Xiaomi switch

Map:
[image]

The Map function is very nice!!

pingel20, 2020-12-17:
Here is some web log level 3 information when sending the write instruction for the OppleMode.
Perhaps this gives a clue.

```
14:08:23 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}
14:08:23 SRC: WebConsole from 192.168.178.10
14:08:23 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}"
14:08:23 ZIG: guessing endpoint 1
14:08:23 ZIG: ZbEZSPSend 340000595E0401C0FC010140010000010109145F11010209001001
14:08:23 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
14:08:23 ZIG: {"ZbEZSPReceived":"34000024"}
14:08:31 ZIG: {"ZbEZSPReceived":"800042595E"}
14:08:31 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
14:08:40 ZIG: {"ZbEZSPReceived":"800042595E"}
14:08:40 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
14:08:49 ZIG: {"ZbEZSPReceived":"800042595E"}
14:08:49 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbRouteError":{"ShortAddr":"0x5E59","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
14:08:51 ZIG: {"ZbEZSPReceived":"3F0000595E0401C0FC01014001000024016600"}
```

qlwz, 2020-12-19:
12:45:10.190 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","OppleMode":0,"Endpoint":1,"LinkQuality":119}}}
12:45:10.793 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":109}}}
12:45:16.592 CMD: ZbSend {"Device":"0x9C27","Manuf":"0x115F","Write":{"FCC0/0009%10":1}}
12:45:16.601 RSL: stat/tasmota_21D76C/RESULT = {"ZbSend":"Done"}
12:45:17.746 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":127}}}
12:45:18.795 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":132}}}
12:45:20.290 RSL: tele/tasmota_21D76C/RESULT = {"ZbState":{"Status":20,"Message":"Disable Pairing mode"}}
12:45:21.194 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0003!01":"","IdentifyQuery":true,"Endpoint":1,"LinkQuality":179}}}
12:45:21.291 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0003!01":"","IdentifyQuery":true,"Endpoint":1,"LinkQuality":179}}}
12:45:26.103 CMD: ZbSend {"Device":"0x9C27","Manuf":"0x115F","Cluster":"0xFCC0","Read":9}
12:45:26.111 RSL: stat/tasmota_21D76C/RESULT = {"ZbSend":"Done"}
12:45:27.055 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","OppleMode":0,"Endpoint":1,"LinkQuality":163}}}
12:45:27.458 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":156}}}
12:45:28.707 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":153}}}
12:45:30.955 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":158}}}
12:45:32.702 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":161}}}
12:45:34.499 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":163}}}
12:45:54.870 ZIG: sending `ZbMap 0x0000`
12:45:54.982 RSL: tele/tasmota_21D76C/RESULT = {"ZbMap":{"Device":"0x0000","Status":0,"StatusMessage":"SUCCESS","Total":9,"Start":1,"Map":[{"Device":"0xFC72","DeviceType":"Device","RxOnWhenIdle":false,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":170},{"Device":"0x8DBD","DeviceType":"Router","RxOnWhenIdle":true,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":0},{"Device":"0x5DEA","DeviceType":"Device","RxOnWhenIdle":false,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":170}]}}
12:45:55.130 RSL: tele/tasmota_21D76C/RESULT = {"ZbMap":{"Device":"0x0000","Status":0,"StatusMessage":"SUCCESS","Total":9,"Start":4,"Map":[{"Device":"0x1E60","DeviceType":"Device","RxOnWhenIdle":false,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":170},{"Device":"0xD15F","DeviceType":"Device","RxOnWhenIdle":false,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":170},{"Device":"0xFDEC","DeviceType":"Device","RxOnWhenIdle":false,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":170}]}}
12:45:55.280 RSL: tele/tasmota_21D76C/RESULT = {"ZbMap":{"Device":"0x0000","Status":0,"StatusMessage":"SUCCESS","Total":9,"Start":7,"Map":[{"Device":"0x101D","DeviceType":"Device","RxOnWhenIdle":false,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":170},{"Device":"0x440A","DeviceType":"Device","RxOnWhenIdle":false,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":170},{"Device":"0x9C27","DeviceType":"Device","RxOnWhenIdle":false,"Relationship":"Child","PermitJoin":null,"Depth":1,"LinkQuality":160}]}}
12:45:56.815 ZIG: sending `ZbMap 0x9C27`
12:46:02.800 ZIG: ZbMap done
12:46:28.249 CMD: ZbSend {"Device":"0x9C27","Manuf":"0x115F","Write":{"OppleMode":1}}
12:46:28.260 RSL: stat/tasmota_21D76C/RESULT = {"ZbSend":"Done"}
12:46:30.203 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":161}}}
12:46:33.100 RSL: tele/tasmota_21D76C/SENSOR = {"ZbReceived":{"0x9C27":{"Device":"0x9C27","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":150}}}
12:46:34.894 RSL: tele/tasmota_21D76C/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}
12:46:40.248 CMD: ZbSend {"Device":"0x9C27","Manuf":"0x115F","Write":{"OppleMode":1}}
12:46:40.259 RSL: stat/tasmota_21D76C/RESULT = {"ZbSend":"Done"}
12:46:41.904 RSL: tele/tasmota_21D76C/STATE = {"Time":"2020-12-19T12:46:41","Uptime":"0T00:05:11","UptimeSec":311,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"qlwz","BSSId":"E2:63:DA:E7:93:0F","Channel":6,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:04"}}
12:46:46.895 RSL: tele/tasmota_21D76C/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}
12:46:49.900 CMD: ZbSend {"Device":"0x9C27","Manuf":"0x115F","Cluster":"0xFCC0","Read":9}
12:46:49.909 RSL: stat/tasmota_21D76C/RESULT = {"ZbSend":"Done"}
12:46:56.985 RSL: tele/tasmota_21D76C/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}

setting OppleMode error

s-hadinger, 2020-12-23:
@qlwz 

You need to press a button on the device to wake it up right before sending `ZbSend {"Device":"0x9C27","Manuf":"0x115F","Write":{"OppleMode":1}}``

Also please try to read again the attribute at the end: `ZbSend {"Device":"0x9C27","Manuf":"0x115F","Cluster":"0xFCC0","Read":9}`

pingel20, 2020-12-23:
@s-hadinger 

Immediately after the pairing process I can send an OppleMode write command without errors.
But reading the attribute tells me the mode did not change.

Log:

```
14:06:51 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
14:06:52 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","Manufacturer":"LUMI","ModelId":"lumi.remote.b686opcn01","Endpoint":1,"LinkQuality":123}}}
14:06:53 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbState":{"Status":33,"Device":"0x5E59","Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0105","DeviceVersion":1,"InClusters":["0x0000","0x0003","0x0001"],"OutClusters":["0x0003","0x0006","0x0008","0x0300"]}}
14:06:55 ZIG: auto-bind `ZbBind {"Device":"0x5E59","Endpoint":1,"Cluster":"0x0001"}`
14:06:55 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbBind":{"Device":"0x5E59","Status":0,"StatusMessage":"SUCCESS"}}
14:06:57 ZIG: auto-bind `ZbBind {"Device":"0x5E59","Endpoint":1,"Cluster":"0x0006"}`
14:06:57 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbBind":{"Device":"0x5E59","Status":0,"StatusMessage":"SUCCESS"}}
14:06:59 ZIG: auto-bind `ZbBind {"Device":"0x5E59","Endpoint":1,"Cluster":"0x0008"}`
14:06:59 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbBind":{"Device":"0x5E59","Status":0,"StatusMessage":"SUCCESS"}}
14:07:01 ZIG: auto-bind `ZbBind {"Device":"0x5E59","Endpoint":1,"Cluster":"0x0300"}`
14:07:02 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbBind":{"Device":"0x5E59","Status":0,"StatusMessage":"SUCCESS"}}
14:07:03 ZIG: auto-bind `ZbSend {"Device":"0x5E59","Config":{"BatteryVoltage":{"MinInterval":3600,"MaxInterval":14400,"ReportableChange":0.1},"BatteryPercentage":{"MinInterval":3600,"MaxInterval":14400,"ReportableChange":1}}}`
14:07:04 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","ConfigResponse":{"BatteryPercentage":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":126}}}
14:07:17 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0008!02":"015507000000","DimmerStepDown":85,"Endpoint":1,"LinkQuality":97}}}
14:07:21 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}
14:07:21 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
14:07:52 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Cluster":"0xFCC0","Read":9}
14:07:52 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
14:07:53 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","OppleMode":0,"Endpoint":1,"LinkQuality":123}}}
14:09:17 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0008!02":"005507000000","DimmerStepUp":85,"Endpoint":1,"LinkQuality":126}}}
```

qlwz, 2020-12-23:
Me too, can write successfully, but cannot read successfully

s-hadinger, 2020-12-23:
Thanks but I need `weblog 3` logs when you write the attribute. I need to know the response from the device

pingel20, 2020-12-23:
Weblog 3, no problem
Same actions: pairing, a button press ('DimmerStepUp'), write OppleMode and then read back

```
15:36:00 ZIG: ZbZCLRawReceived: {"0x5E59":{"ConfigResponse":{"BatteryPercentage":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":126}}
15:36:00 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","ConfigResponse":{"BatteryPercentage":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":126}}}
15:36:14 ZIG: {"ZbEZSPReceived":"4500000401080001010001000017FCDB595EFFFF0911050200550700000002"}
15:36:14 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":8,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":131,"securityuse":0,"seqnumber":23,"fc":"0x11","manuf":"0x0000","transact":5,"cmdid":"0x02","payload":"005507000000"}}
15:36:14 ZIG: ZbZCLRawReceived: {"0x5E59":{"0008!02":"005507000000","DimmerStepUp":85,"Endpoint":1,"LinkQuality":131}}
15:36:14 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","0008!02":"005507000000","DimmerStepUp":85,"Endpoint":1,"LinkQuality":131}}}
15:36:32 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}
15:36:32 SRC: WebConsole from 192.168.178.10
15:36:32 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}"
15:36:32 ZIG: guessing endpoint 1
15:36:32 ZIG: ZbEZSPSend 340000595E0401C0FC0101400100001E0109145F111E0209001001
15:36:32 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
15:36:32 ZIG: {"ZbEZSPReceived":"340000AA"}
15:36:33 ZIG: {"ZbEZSPReceived":"3F0000595E0401C0FC010140010000AA010000"}
15:36:33 ZIG: {"ZbEZSPReceived":"4500000401C0FC01010001000018F4D9595EFFFF081C5F111E048D090002"}
15:36:33 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64704,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":126,"securityuse":0,"seqnumber":24,"fc":"0x1C","manuf":"0x115F","transact":30,"cmdid":"0x04","payload":"8D0900"}}
15:36:33 ZIG: ZbZCLRawReceived: {"0x5E59":{"Endpoint":1,"LinkQuality":126}}
15:36:33 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbState":{"Status":20,"Message":"Pairing mode disabled"}}
15:37:07 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Cluster":"0xFCC0","Read":9}
15:37:07 SRC: WebConsole from 192.168.178.10
15:37:07 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x5E59","Manuf":"0x115F","Cluster":"0xFCC0","Read":9}"
15:37:07 ZIG: guessing endpoint 1
15:37:07 ZIG: ZbEZSPSend 340000595E0401C0FC0101400100001F0107145F111F000900
15:37:07 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
15:37:07 ZIG: {"ZbEZSPReceived":"340000AB"}
15:37:08 ZIG: {"ZbEZSPReceived":"3F0000595E0401C0FC010140010000AB010000"}
15:37:08 ZIG: {"ZbEZSPReceived":"4500000401C0FC01010001000019F8DA595EFFFF0A1C5F111F01090000200002"}
15:37:08 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64704,"srcaddr":"0x5E59","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":128,"securityuse":0,"seqnumber":25,"fc":"0x1C","manuf":"0x115F","transact":31,"cmdid":"0x01","payload":"0900002000"}}
15:37:08 ZIG: ZbZCLRawReceived: {"0x5E59":{"FCC0/0009":0,"Endpoint":1,"LinkQuality":128}}
15:37:08 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","OppleMode":0,"Endpoint":1,"LinkQuality":128}}}
```

s-hadinger, 2020-12-23:
Aha, thanks. I had the ZCL type wrong. It was `uint8` and no `bool`.

Please try with #10246 (wait for a few hours until the binaries are avaialble)

qlwz, 2020-12-23:
successfully

pingel20, 2020-12-24:
@s-hadinger 

This looks really nice!

Setting the OppleMode succeeds and a confirmation is sent:
```
12:06:45.263 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Write":{"OppleMode":1}}
12:06:45.273 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
12:06:45.877 MQT: tele/tasmota_D7F2A1/RESULT = {"ZbResponse":{"Device":"0x5E59","Command":"FCC0_04","Status":0,"StatusMessage":"SUCCESS","Endpoint":1,"LinkQuality":128}}
12:07:23.299 CMD: ZbSend {"Device":"0x5E59","Manuf":"0x115F","Cluster":"0xFCC0","Read":9}
12:07:23.306 MQT: stat/tasmota_D7F2A1/RESULT = {"ZbSend":"Done"}
12:07:24.285 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","OppleMode":1,"Endpoint":1,"LinkQuality":131}}}

```
Button clicks:

```
13:52:35.631 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","MultiInValue":1,"click":"single","Endpoint":3,"LinkQuality":81}}}
13:52:45.786 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","MultiInValue":2,"click":"double","Endpoint":3,"LinkQuality":107}}}
13:52:50.166 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","MultiInValue":3,"click":3,"Endpoint":3,"LinkQuality":107}}}

```
A triple click is presented as "click":3 - should this be "click":"triple" to be more consistent with the single and double clicks?

```
13:53:09.811 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","MultiInValue":1,"click":"single","Endpoint":3,"LinkQuality":105}}}
13:53:13.109 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","MultiInValue":2,"click":"double","Endpoint":3,"LinkQuality":105}}}
13:53:21.418 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","MultiInValue":0,"action":"hold","Endpoint":3,"LinkQuality":107}}}
13:53:23.659 MQT: tele/tasmota_D7F2A1/5E59/SENSOR = {"ZbReceived":{"0x5E59":{"Device":"0x5E59","MultiInValue":255,"action":"release","Endpoint":3,"LinkQuality":107}}}

```
The "hold" and "release" actions have the "action" key. For further handling of the messages and creating rules it would be nice if they also have the "click" key. So "click":"hold" and "click":"release"

Mr. Hadinger, thank you for your work. I'm very happy with the result.
Vielen Dank!

pingel20, 2020-12-24:
How can I make a rule like:

on ZbReceived#0x5E59#click=single#Endpoint=3 do publish xxxxx/yyyyy endon ?

s-hadinger, 2020-12-24:
You are right, this is inconsistent. I will also rename the attribute `"Click"` instead of `"click"` (capitalization).

You cannot have a rule to match both click and endpoint. But in your case you can use `SetOption101 1`. This will append the endpoint number to the attribute if endpoint>1.

So in your case the rule should be:
on ZbReceived#0x5E59#click3=single do publish xxxxx/yyyyy endon ?

pingel20, 2020-12-25:
Thank you. I'll have a play with this.

pingel20, 2021-01-07:
**Just to summarize and for documentation:**

**Setting the Opple mode** should be done right after pairing the device for best results.

An operational device might give several MAC_INDIRECT_TIMEOUT error messages and mode setting will fail.

So:

1. (Re)Pair your Opple device
2. Press a button to check the device is working
3. Set Opple mode:  `ZbSend {"Device":"0x1234","Manuf":"0x115F","Write":{"OppleMode":1}}`
and check console output for SUCCESS message
4. Read back actual setting: `ZbSend {"Device":"0x1234","Manuf":"0x115F","Read":{"OppleMode":true}}`

Done.

pingel20, 2021-01-10:
Thanks for latest dev version.

Do you want to maintain the 'Action' key?
A payload now has to be searched for both 'Click' and 'Action' to see what kind of button press a user performed.

My suggestion:
"Click":"hold"  and
"Click":"release"

s-hadinger, 2021-01-10:
You are right, I'm moving all actions to `Click`. This also makes the code smaller.

pingel20, 2021-01-11:
Works a treat!
```
 {"ZbReceived":{"0x5E59":{"Device":"0x5E59","Name":"Xiaomi_switch_3_bands","MultiInValue":2,"Click":"double","click":"double","Endpoint":4,"LinkQuality":105}}}
 {"ZbReceived":{"0x5E59":{"Device":"0x5E59","Name":"Xiaomi_switch_3_bands","MultiInValue":0,"Click":"hold","action":"hold","Endpoint":5,"LinkQuality":105}}}
 {"ZbReceived":{"0x5E59":{"Device":"0x5E59","Name":"Xiaomi_switch_3_bands","MultiInValue":255,"Click":"release","action":"release","Endpoint":5,"LinkQuality":110}}}
```
Thank you.

qntris, 2024-07-12:
@pingel20 , do these instructions work for the two button version of Opple too. On my side, I've tried them multiple times without success (although I might get a "Success" message", the operating mode of the Opple Buttons remain as lamp switch - power, dimming etc.). Here's the output post "weblog 3" logging level:
`19:29:44 ZIG: rebooting CC2530 device
19:29:45 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":1,"Message":"CC2530 booted","RestartReason":"Watchdog","MajorRel":2,"MinorRel":6}}
19:29:45 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":50,"MajorRel":2,"MinorRel":6,"MaintRel":3,"Revision":20190608}}
19:29:45 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
19:29:46 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":40,"NewState":9,"Message":"Started as coordinator"}}
19:29:46 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":51,"IEEEAddr":"0x00124B000947E106","ShortAddr":"0x0000","DeviceType":7,"DeviceState":9,"NumAssocDevices":3,"AssocDevicesList":["0xBF02","0x9005","0x0680"]}}
19:29:47 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
19:29:47 ZIG: Zigbee started
19:29:47 ZIG: Zigbee device information in Flash (86 bytes)
19:32:33 CMD: weblog 3
19:32:33 MQT: stat/tasmota_273D2D/RESULT = {"WebLog":3}
19:32:34 CFG: Saved to flash at FB, Count 1641, Bytes 4096
19:32:37 HTP: Main Menu
19:32:39 SRC: WebGui from 192.168.0.100
19:32:39 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data ""
19:32:39 ZIG: ZbZNPSent 25360FFCFF3C00
19:32:39 MQT: stat/tasmota_273D2D/RESULT = {"ZbPermitJoin":"Done"}
19:32:39 ZIG: {"ZbZNPReceived":"653600"}
19:32:39 ZIG: {"ZbZNPReceived":"45CB3C"}
19:32:39 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":21,"Message":"Enable Pairing mode for 60 seconds"}}
19:32:39 ZIG: {"ZbZNPReceived":"45B6000000"}
19:32:40 HTP: Console
19:32:50 ZIG: {"ZbZNPReceived":"45CAA03D5255793CDF8CCF040000"}
19:32:50 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x04CF8CDF3C795552","ShortAddr":"0x3DA0","ParentNetwork":"0x0000"}}
19:32:51 ZIG: {"ZbZNPReceived":"45C1A03DA03D5255793CDF8CCF0484"}
19:32:51 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x04CF8CDF3C795552","ShortAddr":"0x3DA0","PowerSource":true,"ReceiveWhenIdle":false,"Security":false}}
19:32:51 ZIG: ZbZNPSent 2505A03DA03D
19:32:51 ZIG: {"ZbZNPReceived":"650500"}
19:32:51 ZIG: {"ZbZNPReceived":"4585A03D00A03D06010203040506"}
19:32:51 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01","0x02","0x03","0x04","0x05","0x06"]}}
19:32:51 ZIG: ZbZNPSent 240202A03D00000000000001000001000001301E070000010004000500
19:32:51 ZIG: {"ZbZNPReceived":"640200"}
19:32:52 ZIG: {"ZbZNPReceived":"4480F00101"}
19:32:52 MQT: tele/tasmota_273D2D/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}
19:32:52 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 113 bytes)
19:32:53 ZIG: ZbZNPSent 2504A03DA03D01
19:32:53 ZIG: {"ZbZNPReceived":"650400"}
19:32:53 ZIG: {"ZbZNPReceived":"4584A03D00A03D1601040105010103000003000100040300060008000003"}
19:32:53 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":33,"Device":"0x3DA0","Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0105","DeviceVersion":1,"InClusters":["0x0000","0x0003","0x0001"],"OutClusters":["0x0003","0x0006","0x0008","0x0300"]}}
19:32:54 ZIG: ZbZNPSent 2504A03DA03D02
19:32:54 ZIG: {"ZbZNPReceived":"650400"}
19:32:55 ZIG: {"ZbZNPReceived":"4584A03D00A03D0E0204010301010103000206000300"}
19:32:55 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":33,"Device":"0x3DA0","Endpoint":"0x02","ProfileId":"0x0104","DeviceId":"0x0103","DeviceVersion":1,"InClusters":["0x0003"],"OutClusters":["0x0006","0x0003"]}}
19:32:56 ZIG: ZbZNPSent 2504A03DA03D03
19:32:56 ZIG: {"ZbZNPReceived":"650400"}
19:32:57 ZIG: ZbZNPSent 2504A03DA03D04
19:32:57 ZIG: {"ZbZNPReceived":"650400"}
19:32:58 ZIG: {"ZbZNPReceived":"45C0A03D83"}
19:32:58 ZIG: {"ZbZNPReceived":"45C0A03D83"}
19:32:59 ZIG: auto-bind `ZbBind {"Device":"0x3DA0","Endpoint":1,"Cluster":"0x0001"}`
19:32:59 ZIG: ZbZNPSent 2521A03D5255793CDF8CCF040101000306E14709004B120001
19:32:59 ZIG: {"ZbZNPReceived":"652100"}
19:33:00 ZIG: {"ZbZNPReceived":"45A1A03D00"}
19:33:00 MQT: tele/tasmota_273D2D/RESULT = {"ZbBind":{"Device":"0x3DA0","Status":0,"StatusMessage":"SUCCESS"}}
19:33:01 ZIG: {"ZbZNPReceived":"448100000100A03D010100630070680900000718000A2000201EA03D1D"}
19:33:01 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1,"srcaddr":"0x3DA0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":99,"securityuse":0,"seqnumber":0,"fc":"0x18","manuf":"0x0000","transact":0,"cmdid":"0x0A","payload":"2000201E"}}
19:33:01 ZIG: ZbZCLRawReceived: {"0x3DA0":{"0001/0020":30,"Endpoint":1,"LinkQuality":99}}
19:33:01 ZIG: auto-bind `ZbBind {"Device":"0x3DA0","Endpoint":1,"Cluster":"0x0006"}`
19:33:01 ZIG: ZbZNPSent 2521A03D5255793CDF8CCF040106000306E14709004B120001
19:33:01 ZIG: {"ZbZNPReceived":"652100"}
19:33:01 MQT: tele/tasmota_273D2D/3DA0/SENSOR = {"ZbReceived":{"0x3DA0":{"Device":"0x3DA0","BatteryVoltage":3,"BatteryPercentage":100,"Endpoint":1,"LinkQuality":99}}}
19:33:02 ZIG: {"ZbZNPReceived":"45A1A03D00"}
19:33:02 MQT: tele/tasmota_273D2D/RESULT = {"ZbBind":{"Device":"0x3DA0","Status":0,"StatusMessage":"SUCCESS"}}
19:33:03 ZIG: auto-bind `ZbBind {"Device":"0x3DA0","Endpoint":1,"Cluster":"0x0008"}`
[... cut ...]
19:35:18 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x3DA0","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":126,"securityuse":0,"seqnumber":0,"fc":"0x11","manuf":"0x0000","transact":20,"cmdid":"0x01","payload":""}}
19:35:18 ZIG: ZbZCLRawReceived: {"0x3DA0":{"0006!01":"","Power":1,"Endpoint":1,"LinkQuality":126}}
19:35:18 MQT: tele/tasmota_273D2D/3DA0/SENSOR = {"ZbReceived":{"0x3DA0":{"Device":"0x3DA0","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":126}}}
19:35:51 ZIG: {"ZbZNPReceived":"45CB00"}
19:35:51 MQT: tele/tasmota_273D2D/RESULT = {"ZbState":{"Status":20,"Message":"Disable Pairing mode"}}`

pingel20, 2024-07-15:
Your Manufacturer code is wrong: needs to be 0x115F

So try:
ZbSend {"Device":"0x3DA0","Manuf":"0x115F","Write":{"OppleMode":1}}

PR #10142 (2020-12-15): Zigbee support for Opple operation mode #9823
## Description:

Changing mode of the switch requires the following commands:

Read mode:
```
ZbSend {"Device":"<device>","Manuf":"0x115F","Read":{"OppleMode":true}}
```

Write mode (0 or 1):
```
ZbSend {"Device":"<device>","Manuf":"0x115F","Write":{"OppleMode":1}}
ZbSend {"Device":"<device>","Manuf":"0x115F","Write":{"OppleMode":0}}
```

**Related issue (if applicable):** fixes #9823

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR a [... cut]
PR #10472 (2021-01-09): Better support for triple and quadruple clicks
## Description:

Support for `triple` and `quadruple` value for `Click` instead of 3-4.

**Warning** Breaking change: the attributes reported for Aqara remotes are now `Action` and `Click` (capitalized) instead of `action` and `click`.
For the next 3 months, `action` and `click` are kept as deprecated and unchanged. They will be dropped in 3 months.

**Related issue (if applicable):** fixes #9823

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/f [... cut]
PR #10494 (2021-01-10): Zigbee Opple Aqara remover Action to Click
## Description:

This supersedes #10472 and uses only `Click` attribute name for values `hold`, `release` and `shake`.

**Related issue (if applicable):** fixes #9823

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc4
  - [x] I accep [... cut]