arendst/Tasmota #9825: ZBBridge tasmota 9.1.0 doesn't save MQTT user and pass.

zandadoum, 2020-11-12:
Hello,

### PROBLEM DESCRIPTION
recently got my first sonoff zbbridge. tasmotized it following digiblurs tutorials.

latest release version (9.1.0) and latest OTA file

almost everything works through ZHA but i wanted to configure it to use MQTT like many of my other tasmotized devices.

i go to config other and enable MQTT. that works. device reboots. i check the setting and it is active
then i go into config MQTT i set the server, port and all the other info. i accept. device reboots...

but when i go into config MQTT again, the user and password were not saved and have been reverted back to default values.
and obviously, in home assistant, nothing new appears in MQTT

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): SONOFF ZBBRIDGE
- [ ] Tasmota binary firmware version number used: 9.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: TASMOTIZER
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

18:14:54 HTP: Console
18:14:57 CMD: Backlog Template; Module; GPIO 255
18:14:57 SRC: WebConsole from 192.168.69.69
18:14:57 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
18:14:57 SRC: Backlog
18:14:57 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
18:14:57 RSL: RESULT = {"NAME":"Sonoff ZHABridge","GPIO":[320,5088,0,5120,323,322,0,0,0,0,0,0,32,0],"FLAG":0,"BASE":18}
18:14:57 SRC: Backlog
18:14:57 CMD: Group 0, Index 1, Command "MODULE", Data ""
18:14:57 RSL: RESULT = {"Module":{"0":"Sonoff ZHABridge"}}
18:14:57 SRC: Backlog
18:14:57 CMD: Group 0, Index 1, Command "GPIO", Data "255"
18:14:57 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"5088":"TCP Tx"},"GPIO2":{"0":"None"},"GPIO3":{"5120":"TCP Rx"},"GPIO4":{"323":"Led_i4"},"GPIO5":{"322":"Led_i3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

18:15:43 CMD: status 0
18:15:43 SRC: WebConsole from 192.168.69.69
18:15:43 CMD: Group 0, Index 1, Command "STATUS", Data "0"
18:15:43 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Sonoff_ZBBRidge_01","FriendlyName":["Sonoff_ZBBRidge_01"],"Topic":"tasmota_E32671","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:15:43 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:16:42","StartupUTC":"2020-11-12T16:59:01","Sleep":50,"CfgHolder":4617,"BootCount":35,"BCResetTime":"2020-11-08T10:39:53","SaveCount":64,"SaveAddress":"F6000"}}
18:15:43 RSL: STATUS2 = {"StatusFWR":{"Version":"9.1.0(zbbridge)","BuildDateTime":"2020-11-07T12:01:19","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"447/699"}}
18:15:43 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Tiphares2.4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00088001","2805C8000100060000005A00000000000000","00008000","00006000","00000000"]}}
18:15:43 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
18:15:43 RSL: STATUS5 = {"StatusNET":{"Hostname":"Sonoff_ZBBridge_01","IPAddress":"192.168.69.252","Gateway":"192.168.69.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.69.99","Mac":"98:F4:AB:E3:26:71","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:15:43 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-11-12T17:15:43","Local":"2020-11-12T18:15:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:55","Sunset":"17:12"}}
18:15:43 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-11-12T18:15:43"}}
18:15:43 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-11-12T18:15:43","Uptime":"0T00:16:42","UptimeSec":1002,"Vcc":3.427,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Tiphares2.4","BSSId":"EC:08:6B:B4:11:B2","Channel":11,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

tasmotize a sonoff zbbridge. enable mqtt, try to input mqtt user and pass

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

expected to keep those settings saved and start showing in the MQTT integration in HA

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-11-12:
Please, provide the console log when your device restarts after you save your mqtt settings. Thanks.

zandadoum, 2020-11-12:
here it is:

`
19:15:25 CFG: CR 447/699, Busy 0
19:15:25 MQT: MqttHost 192.168.69.99, MqttPort 1883, MqttClient DVES_%06X, Topic tasmota_%06X, FullTopic %prefix%/%topic%/

00:00:00 CFG: Loaded from flash at F4, Count 66
00:00:00 SNS: Hardware Serial
00:00:00 SRC: Restart
00:00:00 Project tasmota Sonoff_ZBBRidge_01 Version 9.1.0(zbbridge)-2_7_4_5
00:00:00 TCP: from MCU: FE
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 **PRIVATE WIFI NAME** Channel 11 BSSId **PRIVATE MAC** in mode 11N as Sonoff_ZBBridge_01...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Connected
19:15:33 NTP: UTC 2020-11-12T18:15:32, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
19:15:33 HTP: Web server active on Sonoff_ZBBridge_01 with IP address 192.168.69.252
19:15:34 MQT: Attempting connection...
19:15:34 MQT: Connected
19:15:34 MQT: tele/tasmota_E32671/LWT = Online (retained)
19:15:34 MQT: cmnd/tasmota_E32671/POWER = 
19:15:34 MQT: Subscribe to cmnd/tasmota_E32671/#
19:15:34 MQT: Subscribe to cmnd/tasmotas/#
19:15:34 MQT: Subscribe to cmnd/DVES_E32671_fb/#
19:15:34 MQT: tele/tasmota_E32671/INFO1 = {"Module":"Sonoff ZHABridge","Version":"9.1.0(zbbridge)","FallbackTopic":"cmnd/DVES_E32671_fb/","GroupTopic":"cmnd/tasmotas/"}
19:15:34 MQT: tele/tasmota_E32671/INFO2 = {"WebServerMode":"Admin","Hostname":"Sonoff_ZBBridge_01","IPAddress":"192.168.69.252"}
19:15:34 MQT: tele/tasmota_E32671/INFO3 = {"RestartReason":"Software/System restart"}
19:15:34 RUL: SYSTEM#BOOT performs "TCPStart 8888"
19:15:34 SRC: Rule
19:15:34 CMD: Group 0, Index 1, Command "TCPSTART", Data "8888"
19:15:34 TCP: Starting TCP server on port 8888
19:15:34 MQT: stat/tasmota_E32671/RESULT = {"TCPStart":"Done"}
19:15:35 HTP: Main Menu
19:15:36 APP: Boot Count 37
19:15:36 CFG: Saved to flash at FB, Count 67, Bytes 4096
19:15:38 MQT: tele/tasmota_E32671/STATE = {"Time":"2020-11-12T19:15:38","Uptime":"0T00:00:12","UptimeSec":12,"Vcc":3.427,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"**PRIVATE WIFI NAME**","BSSId":"**PRIVATE MAC**","Channel":11,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:06"}}

`

sfromis, 2020-11-12:
I can confirm that updating MQTT user from the web UI also did not work for me, but updating via the command `MqttUser` worked as expected. Nothing interesting in the log when saving from the page, and it does not appear to be something resetting values during the restart triggered by updating.

Doing the same on another device, I saw MqttUser included in the console message appearing when saving from the web UI, in contrast to this conspicuously missing on the ZbBridge. Looks like a real difference between the binary for ZbBridge and in this case the regular tasmota.bin.gz.

Both devices on 9.1.0

closed 2020-11-15 by ascillato2