arendst/Tasmota #9829: Power to relay only lasts a few seconds before switching off when using Thermostat

blaggersdev, 2020-11-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

When switching on the Thermostat with ThermostatModeSet 1 (having set TempTargetSet to a level well above current temperature) the relay powers on for a few seconds, then off, and continues cycling.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ *] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [* ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [* ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [* ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [* ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ *] Device used (e.g., Sonoff Basic): _TH10____
- [* ] Tasmota binary firmware version number used: __9.1.0.1___
  - [ ] Pre-compiled
  - [* ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [* ] Flashing tools used: ____OTA_
- [ *] Provide the output of command: `Backlog Template; Module; GPIO 255`:
19:07:58 MQT: sonoff-woodpecker-heating/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:07:58 MQT: sonoff-woodpecker-heating/stat/RESULT = {"Module":{"4":"Sonoff TH"}}
19:07:58 MQT: sonoff-woodpecker-heating/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"1248":"SI7021"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS = {"Status":{"Module":4,"DeviceName":"Woodpecker Heating","FriendlyName":["Woodpecker Heating"],"Topic":"sonoff-woodpecker-heating","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[3,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:36:36","StartupUTC":"2020-11-12T18:34:01","Sleep":50,"CfgHolder":4617,"BootCount":57,"BCResetTime":"2020-11-02T19:55:43","SaveCount":3864,"SaveAddress":"F9000"}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0.1(tasmota)","BuildDateTime":"2020-11-12T18:27:18","Boot":7,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"401/699"}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Systems",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["040A8008","2805C8000100060000005AFF000000000000","00000200","00006000","00000000"]}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS4 = {"StatusMEM":{"ProgramSize":539,"Free":464,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1420C2","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FC04782","04360001","00B400C3","00000000","C004F981","00000404","00000000"],"Drivers":"1,2,4,6,8,9,10,12,18,19,20,21,22,24,26,27,29,30,35,37,39","Sensors":"1,2,4,5,6,15,17,18,20,56"}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-woodpecker-heating-4444","IPAddress":"192.168.11.94","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"CC:50:E3:54:B1:5C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.11.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_54B15C","MqttUser":"mqttadmin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-12T19:10:37","Local":"2020-11-12T19:10:37","StartDST":"2020-03-29T01:00:00","EndDST":"2020-10-25T02:00:00","Timezone":99,"Sunrise":"06:55","Sunset":"16:12"}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-12T19:10:37","SI7021":{"Temperature":13.2,"Humidity":84.2,"DewPoint":10.6},"TempUnit":"C"}}
19:10:37 MQT: sonoff-woodpecker-heating/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-12T19:10:37","Uptime":"0T00:36:36","UptimeSec":2196,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [* ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
18:36:52 MQT: sonoff-woodpecker-heating/stat/RESULT = {"ThermostatModeSet":1}
18:36:54 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-11-12T18:36:54","Uptime":"0T00:02:53","UptimeSec":173,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:36:54 MQT: sonoff-woodpecker-heating/stat/RESULT = {"POWER1":"OFF"}
18:36:54 MQT: sonoff-woodpecker-heating/stat/POWER1 = OFF
18:37:01 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-11-12T18:37:01","Uptime":"0T00:03:00","UptimeSec":180,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:37:01 MQT: sonoff-woodpecker-heating/stat/RESULT = {"POWER1":"ON"}
18:37:01 MQT: sonoff-woodpecker-heating/stat/POWER1 = ON
18:37:03 MQT: sonoff-woodpecker-heating/tele/HASS_STATE = {"Version":"9.1.0.1(tasmota)","BuildDateTime":"2020-11-12T18:27:18","Module or Template":"Sonoff TH","RestartReason":"Software/System restart","Uptime":"0T00:03:02","Hostname":"sonoff-woodpecker-heating-4444","IPAddress":"192.168.11.94","RSSI":"56","Signal (dBm)":"-72","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:03","MqttCount":1,"LoadAvg":19}
18:37:10 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-11-12T18:37:10","Uptime":"0T00:03:09","UptimeSec":189,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:37:10 MQT: sonoff-woodpecker-heating/tele/SENSOR = {"Time":"2020-11-12T18:37:10","SI7021":{"Temperature":13.2,"Humidity":83.7,"DewPoint":10.5},"TempUnit":"C"}
18:37:24 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-11-12T18:37:24","Uptime":"0T00:03:23","UptimeSec":203,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:37:24 MQT: sonoff-woodpecker-heating/stat/RESULT = {"POWER1":"OFF"}
18:37:24 MQT: sonoff-woodpecker-heating/stat/POWER1 = OFF

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Keep Thermostat settings as defaults (for heating), enter setpoint and ThermostatModeSet 1.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
According to TimeMinActionSet = 4 (the default) the relay should be powered for a minimum of 4 minutes.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-07:
@arijav 

Can you take a look on this? Thanks.

arijav, 2020-12-08:
I take the point and will have a look

arijav, 2020-12-08:
@blaggersdev : What was your TempTargetSet value and the temperature reported by the sensor?. Please compile with "#define DEBUG_THERMOSTAT" in your "user_config_override.h" file and post the serial output. With the information provided I am not able to identify if there is an issue.

blaggersdev, 2020-12-09:
> @blaggersdev : What was your TempTargetSet value and the temperature reported by the sensor?. Please compile with "#define DEBUG_THERMOSTAT" in your "user_config_override.h" file and post the serial output. With the information provided I am not able to identify if there is an issue.

blaggersdev, 2020-12-09:
@arijav I'll compile a build later this evening as you suggest and hope to get back to you tomorrow.

blaggersdev, 2020-12-09:
@arjav Just compiled and tested with 9.1.0.2. All settings are on default; user_config_override.h has been altered for a SI7021 sensor; setpoint was 18 and sensor reading was 11.8.
Print from console is below. Is this what you need?

20:52:11 CMD: THERMOSTATMODESET 1
20:52:11 MQT: sonoff-woodpecker-heating/stat/RESULT = {"ThermostatModeSet":1}
20:52:13 MQT: domoticz/in = {"idx":792,"nvalue":1,"svalue":""}
20:52:13 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-12-09T20:52:13","Uptime":"0T00:13:59","UptimeSec":839,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"ON","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:52:13 MQT: sonoff-woodpecker-heating/stat/RESULT = {"POWER1":"ON"}
20:52:13 MQT: sonoff-woodpecker-heating/stat/POWER1 = ON
20:52:13 MQT: domoticz/in = {"idx":791,"nvalue":1,"svalue":""}
20:52:23 MQT: sonoff-woodpecker-heating/tele/HASS_STATE = {"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-09T20:33:39","Module or Template":"Sonoff TH","RestartReason":"Software/System restart","Uptime":"0T00:14:09","Hostname":"sonoff-woodpecker-heating-4444","IPAddress":"192.168.11.94","RSSI":"60","Signal (dBm)":"-70","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:04","MqttCount":2,"LoadAvg":19}
20:52:24 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-12-09T20:52:24","Uptime":"0T00:14:10","UptimeSec":850,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"ON","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:52:24 MQT: sonoff-woodpecker-heating/tele/SENSOR = {"Time":"2020-12-09T20:52:24","SI7021":{"Temperature":11.8,"Humidity":73.8,"DewPoint":7.3},"TempUnit":"C"}
20:52:28 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-12-09T20:52:28","Uptime":"0T00:14:14","UptimeSec":854,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:52:28 MQT: sonoff-woodpecker-heating/stat/RESULT = {"POWER1":"OFF"}
20:52:28 MQT: sonoff-woodpecker-heating/stat/POWER1 = OFF
20:52:38 MQT: sonoff-woodpecker-heating/tele/HASS_STATE = {"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-09T20:33:39","Module or Template":"Sonoff TH","RestartReason":"Software/System restart","Uptime":"0T00:14:24","Hostname":"sonoff-woodpecker-heating-4444","IPAddress":"192.168.11.94","RSSI":"60","Signal (dBm)":"-70","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:04","MqttCount":2,"LoadAvg":19}
20:52:39 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-12-09T20:52:39","Uptime":"0T00:14:25","UptimeSec":865,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:52:39 MQT: sonoff-woodpecker-heating/tele/SENSOR = {"Time":"2020-12-09T20:52:39","SI7021":{"Temperature":11.7,"Humidity":73.8,"DewPoint":7.2},"TempUnit":"C"}

blaggersdev, 2020-12-09:
@arijav Made an error compiling and reran. setpoint 15  and external SI702, all else at defaults:

Given the difference between the setpoint and actual temp I would expect the boiler to stay on in 'ramp-up' mode - yet this switches off:

22:21:25 MQT: homeassistant/binary_sensor/54B15C_SW_6/config =  (retained)
22:21:25 MQT: homeassistant/device_automation/54B15C_SW_7_TOGGLE/config =  (retained)
22:21:25 MQT: homeassistant/device_automation/54B15C_SW_7_HOLD/config =  (retained)
22:21:25 MQT: homeassistant/binary_sensor/54B15C_SW_7/config =  (retained)
22:21:25 MQT: homeassistant/device_automation/54B15C_SW_8_TOGGLE/config =  (retained)
22:21:25 MQT: homeassistant/device_automation/54B15C_SW_8_HOLD/config =  (retained)
22:21:25 MQT: homeassistant/binary_sensor/54B15C_SW_8/config =  (retained)
22:21:25 MQT: homeassistant/sensor/54B15C_SI7021_Temperature/config = {"name":"Woodpecker Heating SI7021 Temperature","stat_t":"sonoff-woodpecker-heating/tele/SENSOR","avty_t":"sonoff-woodpecker-heating/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"54B15C_SI7021_Temperature","dev":{"ids":["54B15C"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['SI7021']['Temperature']}}"} (retained)
22:21:26 RSL: homeassistant/sensor/54B15C_SI7021_Humidity/config = {"name":"Woodpecker Heating SI7021 Humidity","stat_t":"sonoff-woodpecker-heating/tele/SENSOR","avty_t":"sonoff-woodpecker-heating/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"54B15C_SI7021_Humidity","dev":{"ids":["54B15C"]},"unit_of_meas":"%","dev_cla":"humidity","frc_upd":true,"val_tpl":"{{value_json['SI7021']['Humidity']}}"}
22:21:26 RSL: homeassistant/sensor/54B15C_SI7021_DewPoint/config = {"name":"Woodpecker Heating SI7021 DewPoint","stat_t":"sonoff-woodpecker-heating/tele/SENSOR","avty_t":"sonoff-woodpecker-heating/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"54B15C_SI7021_DewPoint","dev":{"ids":["54B15C"]},"unit_of_meas":"°C","ic":"mdi:weather-rainy","frc_upd":true,"val_tpl":"{{value_json['SI7021']['DewPoint']}}"}
22:21:26 RSL: homeassistant/cover/54B15C_SHT_1/config = 
22:21:26 RSL: homeassistant/cover/54B15C_SHT_2/config = 
22:21:27 RSL: homeassistant/cover/54B15C_SHT_3/config = 
22:21:27 RSL: homeassistant/cover/54B15C_SHT_4/config = 
22:21:27 RSL: homeassistant/light/54B15C_LI_1/config = 
22:21:27 RSL: homeassistant/switch/54B15C_RL_1/config = {"name":"Woodpecker Heating","stat_t":"sonoff-woodpecker-heating/tele/STATE","avty_t":"sonoff-woodpecker-heating/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"sonoff-woodpecker-heating/cmnd/POWER1","val_tpl":"{{value_json.POWER1}}","pl_off":"OFF","pl_on":"ON","uniq_id":"54B15C_RL_1","dev":{"ids":["54B15C"]}}
22:21:28 RSL: homeassistant/light/54B15C_LI_2/config = 
22:21:28 RSL: homeassistant/switch/54B15C_RL_2/config = 
22:21:28 RSL: homeassistant/light/54B15C_LI_3/config = 
22:21:28 RSL: homeassistant/switch/54B15C_RL_3/config = 
22:21:29 RSL: homeassistant/light/54B15C_LI_4/config = 
22:21:29 RSL: homeassistant/switch/54B15C_RL_4/config = 
22:21:29 RSL: homeassistant/light/54B15C_LI_5/config = 
22:21:29 RSL: homeassistant/switch/54B15C_RL_5/config = 
22:21:30 RSL: homeassistant/light/54B15C_LI_6/config = 
22:21:30 RSL: homeassistant/switch/54B15C_RL_6/config = 
22:21:30 RSL: homeassistant/light/54B15C_LI_7/config = 
22:21:30 RSL: homeassistant/switch/54B15C_RL_7/config = 
22:21:31 RSL: homeassistant/light/54B15C_LI_8/config = 
22:21:31 RSL: homeassistant/switch/54B15C_RL_8/config = 
22:21:31 RSL: homeassistant/sensor/54B15C_status/config = {"name":"Woodpecker Heating status","stat_t":"sonoff-woodpecker-heating/tele/HASS_STATE","avty_t":"sonoff-woodpecker-heating/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"sonoff-woodpecker-heating/tele/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"54B15C_status","dev":{"ids":["54B15C"],"name":"Woodpecker Heating","mdl":"Sonoff TH","sw":"9.1.0.2(tasmota)","mf":"Tasmota"}}
22:21:31 MQT: tasmota/discovery/CC50E354B15C/config =  (retained)
22:21:31 HTP: Console
22:21:37 MQT: sonoff-woodpecker-heating/tele/HASS_STATE = {"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-09T22:18:11","Module or Template":"Sonoff TH","RestartReason":"Software/System restart","Uptime":"0T00:01:00","Hostname":"sonoff-woodpecker-heating-4444","IPAddress":"192.168.11.94","RSSI":"54","Signal (dBm)":"-73","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:05","MqttCount":1,"LoadAvg":20}
22:21:38 MQT: Attempting connection...
22:21:38 MQT: Connected
22:21:38 MQT: sonoff-woodpecker-heating/tele/LWT = Online (retained)
22:21:38 MQT: sonoff-woodpecker-heating/cmnd/POWER = 
22:21:38 MQT: Subscribe to sonoff-woodpecker-heating/cmnd/#
22:21:38 MQT: Subscribe to sonoffs/cmnd/#
22:21:38 MQT: Subscribe to cmnd/DVES_54B15C_fb/#
22:21:38 MQT: Subscribe to homeassistant/status
22:21:40 MQT: sonoff-woodpecker-heating/tele/STATE = {"Time":"2020-12-09T22:21:40","Uptime":"0T00:01:03","UptimeSec":63,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Systems","BSSId":"02:9F:C2:A7:83:9C","Channel":1,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:21:40 MQT: sonoff-woodpecker-heating/tele/SENSOR = {"Time":"2020-12-09T22:21:40","SI7021":{"Temperature":12.3,"Humidity":74.2,"DewPoint":7.8},"TempUnit":"C"}
22:21:49 WIF: Checking connection...
[... cut ...]
22:28:42 Thermostat[ctr_output].time_integral_pi: 0
22:28:42 Thermostat[ctr_output].time_total_pi: 1200
22:28:42 Thermostat[ctr_output].temp_measured_gradient: 24000
22:28:42 Thermostat[ctr_output].time_rampup_deadtime: 0
22:28:42 Thermostat[ctr_output].temp_rampup_meas_gradient: 0
22:28:42 Thermostat[ctr_output].time_ctr_changepoint: 1380
22:28:42 Thermostat[ctr_output].temp_rampup_output_off: 0
22:28:42 Thermostat[ctr_output].time_ctr_checkpoint: 1980
22:28:42 uptime: 480
22:28:42 power: 1
22:28:42 ------ Thermostat End ------
22:28:42

arijav, 2020-12-10:
I might be mistaken but at first sight what I see is that something outside the driver is switching off the output. Example:
Before the driver is called:
22:28:28 MQT: sonoff-woodpecker-heating/stat/POWER1 = OFF

Driver is called at the timepoint 22:28:42 as seen in the log activity and it turns again ON the output:
22:28:42: MQT: sonoff-woodpecker-heating/stat/POWER1 = ON

So it seems for me that the driver tries to keep the output ON in every call (every minute) but something else is turning it off. Could you check if you have some rule or something that is switching your output off?

arendst, 2020-12-10:
arijav is probably correct as you can see from the logging. At 22:28:228 it says ``SRC: MQTT``. This means an external signal is received from MQTT to power off.

You'll have to find out what sends this power off command.

closed 2020-12-10 by ascillato2