arendst/Tasmota #9831: Telegram send issue if "chatid" contains a minus at the beginning

FerFemNemBem, 2020-11-12:
[diff.txt](https://github.com/arendst/Tasmota/files/5532889/diff.txt)

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sending a Telegram message fails if the chatid starts with a minus sign.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: 9.1.0
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VSCode + PIO
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
Console output here:
18:37:27 CMD: tmchatid -123456789
18:37:27 MQT: keller/heizung/warmwassersensor/RESULT = {"TmChatId":"-123456789"}
18:37:49 CMD: tmsend test
18:37:49 SRC: WebConsole from 192.168.178.62
18:37:49 CMD: Group 0, Index 1, Command "TMSEND", Data "test"
18:37:49 TGM: sendMessage
18:37:49 TGM: Cmnd bot54354356:FGHRWHTHJ-GTERZER_DFGFHGHGFHFGHJHJ/sendMessage?chat_id=123456789&text=test
18:37:50 TGM: Connected in 983 ms, max ThunkStack used 4484
18:37:51 TGM: Response {"ok":false,"error_code":400,"description":"Bad Request: chat not found"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Telegram uses for group chats IDs with a minus at the beginning "tmchatid -123456789"
If you doublecheck at the console using "chatid" you will see the correct ID.
If you want to send a message to this chat it will fail.
The reason seems to be the following commit: https://github.com/arendst/Tasmota/commit/40ad4f7ed01411d9b38e7ff230840885cc27ce4b#diff-50b7d293806b164cb42a4bc8d06a84fbcacaf1803a6d30e42701f62367bdbdc4

Inside the commit the "chat_id" was changed from String to "uint_32_t".
Before this commit it was working also with chatIDs using a minus sign at the beginning.
Changing the "chat_id" from "uint_32_t" to "int_32_t" would fix this issue.
I have appended a diff.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Message should be sent.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-11-13:
Thx!  your solution will fix it. I'll update soon.

FerFemNemBem, 2020-11-13:
Thank you for fixing that fast!

closed 2020-11-13 by FerFemNemBem