arendst/Tasmota #9846: tasmota-display is missing "Display" commands

Ruebezahl11, 2020-11-14:
### PROBLEM DESCRIPTION
_a binary compiled for tasmota-display environment does not include "Display" group of commands

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): nodemcu
- [ ] Tasmota binary firmware version number used: 9.1.0
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: platformio
- [ x] Flashing tools used: esptool
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
07:09:15 CMD: Backlog Template; Module; GPIO 255`:
07:09:15 RSL: stat/tasmota_302151/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:09:15 RSL: stat/tasmota_302151/RESULT = {"Module":{"18":"Generic"}}
07:09:16 RSL: stat/tasmota_302151/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
07:12:13 RSL: stat/tasmota_302151/STATUS = {"Status":{"Module":18,"DeviceName":"Wecker","FriendlyName":["Tasmota"],"Topic":"tasmota_302151","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:12:13 RSL: stat/tasmota_302151/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T09:17:56","StartupUTC":"2020-11-13T20:54:17","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2020-11-13T21:55:26","SaveCount":45,"SaveAddress":"F9000"}}
07:12:13 RSL: stat/tasmota_302151/STATUS2 = {"StatusFWR":{"Version":"9.1.0(display)","BuildDateTime":"2020-11-13T21:53:23","Boot":6,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"415/699"}}
07:12:13 RSL: stat/tasmota_302151/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mrt150n","mrt150n_EXT"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
07:12:13 RSL: stat/tasmota_302151/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":28,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAAA59E","04049FA1","000000C3","00000000","00004080","00000020","00000000"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
07:12:13 RSL: stat/tasmota_302151/STATUS5 = {"StatusNET":{"Hostname":"tasmota_302151-0337","IPAddress":"192.168.3.57","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"(IP unset)","Mac":"C8:2B:96:30:21:51","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:12:13 RSL: stat/tasmota_302151/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_302151","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:12:13 RSL: stat/tasmota_302151/STATUS7 = {"StatusTIM":{"UTC":"2020-11-14T06:12:13","Local":"2020-11-14T07:12:13","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:58","Sunset":"17:10"}}
07:12:13 RSL: stat/tasmota_302151/STATUS10 = {"StatusSNS":{"Time":"2020-11-14T07:12:13"}}
07:12:13 RSL: stat/tasmota_302151/STATUS11 = {"StatusSTS":{"Time":"2020-11-14T07:12:13","Uptime":"0T09:17:56","UptimeSec":33476,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"mrt150n","BSSId":"80:1F:02:EA:7A:8E","Channel":11,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
compiled and uploaded with
platformio run --target upload

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Display should return display setting, DisplayModel should be available to choose type of display

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-14:
Iirc the command (s) are only available if a display is configured (i2c or spi) and recognized
from Tasmota

ascillato2, 2020-11-15:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-15 by ascillato2

Ruebezahl11, 2020-11-15:
Hi, my impression was that "DisplayModel" command serves to tell tasmota what kind of display I have (and which i2c addresses it uses). I understand your response that tasmota auto-detects display first and then allows to change its findings?

meingraham, 2020-11-15:
According to the information you provided you have not configured the GPIO that you are using for I2C.