arendst/Tasmota #9853: PZEM004T on Wemos D1 Mini, does not work after upgrade to version 9.1.0

raulmorales, 2020-11-14:
### PROBLEM DESCRIPTION
does not work after upgrade to version 9.1.0. It does not show any value : voltage, power, etc on the web or on the console

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x ] Tasmota binary firmware version number used: 9.1.0.1(tasmota)
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"18":"Generic"}}
 {"GPIO0":{"0":"None"},"GPIO1":{"1504":"PZEM004 Rx"},"GPIO2":{"0":"None"},"GPIO3":{"1472":"PZEM0XX Tx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:38:08 MQT: sonoff-pow-general/stat/STATUS = {"Status":{"Module":18,"DeviceName":"Sonoff-pow-general","FriendlyName":["Sonoff-pow-general"],"Topic":"sonoff-pow-general","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T02:10:29","StartupUTC":"2020-11-14T08:27:39","Sleep":50,"CfgHolder":4617,"BootCount":37,"BCResetTime":"2020-11-13T18:39:46","SaveCount":85,"SaveAddress":"F9000"}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0.1(tasmota)","BuildDateTime":"2020-11-13T20:04:11","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"403/699"}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dd-2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS4 = {"StatusMEM":{"ProgramSize":598,"Free":404,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-pow-general-1775","IPAddress":"192.168.5.129","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.5.100","Mac":"60:01:94:0C:26:EF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.100","MqttPort":32797,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0C26EF","MqttUser":"raul","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-14T10:38:08","Local":"2020-11-14T11:38:08","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:58","Sunset":"17:10"}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:38:08 MQT: sonoff-pow-general/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-14T11:38:08","ENERGY":{"TotalStartTime":"2020-11-13T18:39:46","Total":1.516,"Yesterday":1.516,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
11:38:10 MQT: sonoff-pow-general/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-14T11:38:08","Uptime":"0T02:10:29","UptimeSec":7829,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"dd-2G","BSSId":"88:C3:97:C2:C1:42","Channel":5,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upgrade from version 8.5.1 to 9.1.0
I have restored the configuration, applied template and the error persists

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-11-14:
Cannot reproduce. Works just fine.

[image]

Your logging shows a baudrate of 115200 which is wrong for the PZEM004T. It should be 9600. You might also want to consider switching to hardware serial by swapping both Tx/Rx like:

[image]

raulmorales, 2020-11-14:
Thank you very much, i changed the baudrate and switching to hardware serial by swapping both Tx/Rx

closed 2020-11-14 by raulmorales