arendst/Tasmota #9859: Unknown DevGroupName of tasmotas

jmorris644, 2020-11-14:
### PROBLEM DESCRIPTION
I was using DevGroupNames with two switches. I needed to redeploy the switches into single use switches. I ran the DevGroupName1 0 in the console to delete the DevGroup that I had created. When I issued the console command DevGroupNames it showed all 4 as being blank. But the switches were still in sync. I then ran the DevGroupStatus command and a group called 'tasmotas' existed. I had never directly created this group. The only to remove this group that I found was to so85 to 0. When I so85 back to 1 the group was gone. For the switches final state I did so85 back to 0 as I am not needing groups on the switches. In my searching I did find info about a 'tasmotas' in GroupTopic but did not see a relevance to GroupDev.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Tuya MCU
- [x ] Tasmota binary firmware version number used: 9.1.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: OTA
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

THIS IS POST FIX AS DESCRIBED ABOVE

19:35:32 MQT: lights/stat/dimMasterBathMirrors/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:35:32 MQT: lights/stat/dimMasterBathMirrors/RESULT = {"Module":{"54":"Tuya MCU"}}
19:35:32 MQT: lights/stat/dimMasterBathMirrors/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS = {"Status":{"Module":54,"DeviceName":"dimMasterBathMirrors","FriendlyName":["dimMasterBathMirrors"],"Topic":"dimMasterBathMirrors","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:00:36","StartupUTC":"2020-11-14T18:38:56","Sleep":50,"CfgHolder":4617,"BootCount":375,"BCResetTime":"2020-06-05T19:23:13","SaveCount":1517,"SaveAddress":"F6000"}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"449/699"}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MorrisNoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS5 = {"StatusNET":{"Hostname":"dimMasterBathMirrors","IPAddress":"192.168.30.10","Gateway":"192.168.30.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.30.1","Mac":"98:F4:AB:BF:D0:FE","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BFD0FE","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS7 = {"StatusTIM":{"UTC":"2020-11-14T18:39:32","Local":"2020-11-14T19:39:32","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:58","Sunset":"17:10"}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS10 = {"StatusSNS":{"Time":"2020-11-14T19:39:32"}}
19:39:32 MQT: lights/stat/dimMasterBathMirrors/STATUS11 = {"StatusSTS":{"Time":"2020-11-14T19:39:32","Uptime":"0T00:00:36","UptimeSec":36,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":15,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"MorrisNoT","BSSId":"E2:63:DA:1E:7D:42","Channel":11,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

UNKNOWN
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

That there be no 'tasmotas' group

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pcdiem, 2020-11-14:
From https://tasmota.github.io/docs/Device-Groups:

The device group name is set with the DevGroupName command (GroupTopic prior to v8.2.0.3). If a device group name is not set for a group, the MQTT group topic is used (with the device group number appended for device group numbers > 1)

jmorris644, 2020-11-14:
> From https://tasmota.github.io/docs/Device-Groups:
> 
> The device group name is set with the DevGroupName command (GroupTopic prior to v8.2.0.3). If a device group name is not set for a group, the MQTT group topic is used (with the device group number appended for device group numbers > 1)

Understood. As I have never used GroupTopics prior I am thinking that the default 'tasmotas' MQTT Group was leftover through the upgrades. The reason I am saying this is that after I set the DevGroupName to 0 and  prior to setting so85 option to 0, a DevGroupName showed all 4 DevGroups as "". But a DevGroupStatus showed the 'tasmotas' group. Once I set the so85 to 0 the DevGroup commands returned an error. As maybe expected. I then turned so85 back to 1 for experimentation purposes the 'tasmotas' DevGroup reappears and the two switches are once again linked. I see no where in the UI to set or change the MQTT group setting.

pcdiem, 2020-11-14:
That's all exactly as designed. With SO85 on, if a DevGroupName is not defined, it uses the MQTT GroupTopic as documented. There is no way in the UI to set the group topic. It's set with the GroupTopic<x> command.

jmorris644, 2020-11-15:
OK, thanks. I will close the issue.

closed 2020-11-15 by jmorris644