arendst/Tasmota #9860: Zigbee2Tasmota - Support Request Tuya 4 Gang wall switch (TS0601, TZE200_aqnazj70)

utech-git, 2020-11-14:
### PROBLEM DESCRIPTION
I'd like support on zigbee2tasmota for this [4 gang in wall zigbee switch](https://www.aliexpress.com/item/4000973462125.html)
Below is the zbstatus3 response (I masked a few data that I don't know if i'm supposed to mask or not)
{"ZbStatus3":[{"Device":"0xAAAA","Name":"switch","IEEEAddr":"0xAAAAAAAAAAAAAAAA","ModelId":"TS0601","Manufacturer":"_TZE200_aqnazj70","Endpoints":[1],"Config":[]}]}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [X] Tasmota binary firmware version number used: 9.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
20:46:56 CMD: Backlog Template; Module; GPIO 255
20:46:56 SRC: WebConsole from 192.168.0.3
20:46:56 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
20:46:56 SRC: Backlog
20:46:56 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:46:56 MQT: stat/zigbeebridge/RESULT = {"NAME":"ZHA ZBBridge","GPIO":[320,5088,0,5120,323,322,0,0,0,0,0,0,32,0],"FLAG":0,"BASE":18}
20:46:56 SRC: Backlog
20:46:56 CMD: Group 0, Index 1, Command "MODULE", Data ""
20:46:56 MQT: stat/zigbeebridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
20:46:56 SRC: Backlog
20:46:56 CMD: Group 0, Index 1, Command "GPIO", Data "255"
20:46:56 MQT: stat/zigbeebridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
2
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
No Rules used
```
- [X] Provide the output of this command: `Status 0`:
```
20:47:48 MQT: stat/zigbeebridge/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["0xXXXX,XXXX"],"Topic":"zigbeebridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:47:48 MQT: stat/zigbeebridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software Watchdog","Uptime":"0T00:14:36","StartupUTC":"2020-11-14T19:33:12","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2020-09-04T15:27:18","SaveCount":158,"SaveAddress":"F4000"}}
20:47:48 MQT: stat/zigbeebridge/STATUS2 = {"StatusFWR":{"Version":"9.1.0(zbbridge)","BuildDateTime":"2020-11-07T12:01:19","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"428/699"}}
20:47:48 MQT: stat/zigbeebridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":4,"LogHost":"XXX.XXX.XXX.XXX","LogPort":514,"SSId":["XXXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006082","00000000"]}}
20:47:48 MQT: stat/zigbeebridge/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
20:47:48 MQT: stat/zigbeebridge/STATUS5 = {"StatusNET":{"Hostname":"zigbeebridge-4595","IPAddress":"192.168.0.166","Gateway":"XXX.XXX.XXX.XXX","Subnetmask":"255.255.255.0","DNSServer":"XXX.XXX.XXX.XXX","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:47:48 MQT: stat/zigbeebridge/STATUS6 = {"StatusMQT":{"MqttHost":"XXX.XXX.XXX.XXX","MqttPort":XXX,"MqttClientMask":"XXXXX","MqttClient":"XXXXX","MqttUser":"XXXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:47:48 MQT: stat/zigbeebridge/STATUS7 = {"StatusTIM":{"UTC":"2020-11-14T19:47:48","Local":"2020-11-14T20:47:48","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:58","Sunset":"17:10"}}
20:47:48 MQT: stat/zigbeebridge/STATUS10 = {"StatusSNS":{"Time":"2020-11-14T20:47:48"}}
20:47:48 MQT: stat/zigbeebridge/STATUS11 = {"StatusSTS":{"Time":"2020-11-14T20:47:48","Uptime":"0T00:14:36","UptimeSec":876,"Vcc":3.472,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"Wifi":{"AP":1,"SSId":"XXXXXXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":9,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:03"}}}
20:47:48 MQT: stat/zigbeebridge/STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["40214dc5","00000000","4010194b"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4021d9f2","4023ed74","402242d9","4021ddc9","40243060","4022522f","40239e00","402256ec","40240000","4027a239","402757e0","4027a1dc","40275764","402757a4","4020b252","4027574c","40230ca9","40215875","40227ea1","4023fea6","4021d891","4023dcae","40106671","4023e234","4021d9f2","40100f37","40000000","40100f37","401011bb","40243328","40227fac"]}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
Gang 1 button turn on:
20:51:04 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":139,"securityuse":0,"seqnumber":232,"fc":"0x09","manuf":"0x0000","transact":15,"cmdid":"0x02","payload":"00550101000101"}}
20:51:04 ZIG: ZbZCLRawReceived: {"0xCC8E":{"EF00/0101":1,"Endpoint":1,"LinkQuality":139}}
20:51:04 MQT: tele/zigbeebridge/CC8E/SENSOR = {"ZbReceived":{"switch":{"Device":"0xCC8E","Name":"switch","EF00/0101":1,"Endpoint":1,"LinkQuality":139}}}

Gang 1 button turn off:
20:52:37 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":107,"securityuse":0,"seqnumber":26,"fc":"0x09","manuf":"0x0000","transact":17,"cmdid":"0x02","payload":"00570101000100"}}
20:52:37 ZIG: ZbZCLRawReceived: {"0xCC8E":{"EF00/0101":0,"Endpoint":1,"LinkQuality":107}}
20:52:37 MQT: tele/zigbeebridge/CC8E/SENSOR = {"ZbReceived":{"switch":{"Device":"0xCC8E","Name":"switch","EF00/0101":0,"Endpoint":1,"LinkQuality":107}}}

Gang 2 turn on
20:51:57 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":110,"securityuse":0,"seqnumber":1,"fc":"0x09","manuf":"0x0000","transact":16,"cmdid":"0x02","payload":"00560201000101"}}
20:51:57 ZIG: ZbZCLRawReceived: {"0xCC8E":{"EF00/0102":1,"Endpoint":1,"LinkQuality":110}}
[... 11 lines cut ...]
And the power status is the number after the "Cluster/Command": 1 for ON and 0 for OFF. ie:
Gang1 ON: "EF00/0101":1
Gang1 OFF: "EF00/0101":0
Gang4 ON: "EF00/0104":1
Gang4 OFF: "EF00/0104":0

But since I didn't find any information on the "tuya cluster", how do I discover what zigbee command should I use to turn on/off each gang?

I know that the device ModelID "TS0601" have been used for other devices that is not a wall switch, like [here](https://github.com/arendst/Tasmota/issues/9510) (in which case it was a thermostat), so I added the Manufacturer: _TZE200_aqnazj70

Apparently this device is already supported on Zigbee2MQTT, if that helps: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c638c6af586f862802ef629fb05677d438cf5bf7/devices.js#L1402 and https://github.com/Koenkk/zigbee2mqtt/issues/4792

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-15:
Thanks for this perfectly documented issue. It should as simple as adding aliases to EF00/0101-0104. That's easy.

What name should it be?  To be consistent with other switches, it could be `Power`, `Power2`, `Power3`, `Power4`. There's a small risk that another number suffix is appended in the case of multiple endpoints and `SetOption101` but there will be no issue with this device that has only 1 endpoint.

utech-git, 2020-11-16:
Good to know this kind of documention helps you help me

I think Power, Power2, Power3 and Power4 should be good enough and will avoid confusion/misunderstanding.

Just so it helps me in the future: with the current release is there any raw command I could use to turn on/off each gang?

I tried many combinations of ``ZbSend {"Device":"switch","Send":"EF00_01/0101"}`` and nothing happed. 
I Tried changing the _command_ "01" after the "\_" from "00" to "0A".
Changed the "\_" for "!" and also nothing.

ascillato2, 2020-11-16:
Gggg

closed 2020-11-16 by ascillato2

s-hadinger, 2020-11-16:
@utech-git Tuya uses a specific encoding as described here: https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d

The decoding part is implemented in Tasmota but not the encoding. You will unfortunately need to encode manually. I'll get you an example later.

s-hadinger, 2020-11-16:
Can you please try the following:
Turn on switch 1: `ZbSend {"Device":"switch","Send":"EF00_00/00550101000101"}`
Turn off switch 1: `ZbSend {"Device":"switch","Send":"EF00_00/00560101000100"}`
Turn on switch 2: `ZbSend {"Device":"switch","Send":"EF00_00/00570201000101"}`
Turn off switch 2: `ZbSend {"Device":"switch","Send":"EF00_00/00580201000100"}`

The second byte in the payload (55-58) is just a sequence number. I don't know how sensitive the device is but it may discard some messages if you send twice the same sequence in a row.

utech-git, 2020-11-17:
Well, I tried the commands and nothing happened. I even tried to change the sequence number to match the next sequence number expected but no success. Here's what I did:

Flashed the device with the [dev branch version 9.1.0.2](http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin). Downloaded now.

First I just pressed the buttons.
Turn on switch 1 (I don't know if that's important, but I pressed the button just once and got 15 of this message. The only thing that changed between them was the "seqnumber" parameter, that is increased on each message. But even with the previous firmware that was already happening)
``15:04:06 ZIG: ZbZCLRawReceived: {"0xCC8E":{"EF00/0101":1,"Endpoint":1,"LinkQuality":126}}
15:04:06 MQT: tele/zigbeebridge/CC8E/SENSOR = {"ZbReceived":{"switch":{"Device":"0xCC8E","Name":"switch","Power":1,"Endpoint":1,"LinkQuality":126}}}
15:04:06 ZIG: {"ZbEZSPReceived2":"51209001450000040100EF0101000100003EF8DA8ECCFFFF0A090D02000D0101000101"}
15:04:06 ZIG: ZbEZSPSentRaw 86
15:04:06 ZIG: {"ZbEZSPReceived":"450000040100EF0101000100003EF8DA8ECCFFFF0A090D02000D0101000101"}
15:04:06 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":128,"securityuse":0,"seqnumber":62,"fc":"0x09","manuf":"0x0000","transact":13,"cmdid":"0x02","payload":"000D0101000101"}}
15:04:06 ZIG: ZbZCLRawReceived: {"0xCC8E":{"EF00/0101":1,"Endpoint":1,"LinkQuality":128}}
15:04:06 MQT: tele/zigbeebridge/CC8E/SENSOR = {"ZbReceived":{"switch":{"Device":"0xCC8E","Name":"switch","Power":1,"Endpoint":1,"LinkQuality":128}}}
15:04:06 ZIG: {"ZbEZSPReceived2":"61209001450000040100EF0101000100003FF8DA8ECCFFFF0A090D02000D0101000101"}``

So, based on this, I can see that something changed: now the "EF00/0101" is already recognized as "Power".
The same goes for the other switches (EF00/0102 is recognized as Power2 and so on).

On the other hand, when I send the commands you asked, nothing happens on the switches.
First I tried the exact command ``ZbSend {"Device":"switch","Send":"EF00_00/00550101000101"}``
Here's the result
``
15:12:05 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"switch","Send":"EF00_00/00550101000101"}"
15:12:05 ZIG: guessing endpoint 1
15:12:05 ZigbeeZCLSend device: 0xCC8E, group: 0x0000, endpoint:1, cluster:0xEF00, cmd:0x00, send:"00550101000101"
15:12:05 ZbSend: shortaddr 0xCC8E, groupaddr 0x0000, cluster 0xEF00, endpoint 0x01, cmd 0x00, data 
15:12:05 ZIG: ZbEZSPSend 3400008ECC040100EF01014001000005010A00050000550101000101
15:12:05 ZIG: adding packet to_send, to_ack:1, to_send:1, to_end:1
15:12:05 MQT: stat/zigbeebridge/RESULT = {"ZbSend":"Done"}
15:12:05 ZIG: Something to_send, to_ack:1, to_send:1, to_end:2
15:12:05 ZIG: ZbEZSPSentRaw 162100013400008ECC040100EF01014001000005010A00050000550101000101
15:12:05 ZIG: {"ZbEZSPReceived2":"62218001340000E5"}
15:12:05 ZIG: new ack/data received, was 1 now 2
15:12:05 ZIG: freeing packet 1 from memory
15:12:05 ZIG: ZbEZSPSentRaw 87
15:12:05 ZIG: {"ZbEZSPReceived":"340000E5"}
15:12:05 ZIG: {"ZbEZSPReceived2":"72219001450000040100EF01010001000089E4D58ECCFFFF0C180501005586010186000186"}
15:12:05 ZIG: ZbEZSPSentRaw 80
15:12:05 ZIG: {"ZbEZSPReceived":"450000040100EF01010001000089E4D58ECCFFFF0C180501005586010186000186"}
15:12:05 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":115,"securityuse":0,"seqnumber":137,"fc":"0x18","manuf":"0x0000","transact":5,"cmdid":"0x01","payload":"005586010186000186"}}
15:12:05 ZIG: ZbZCLRawReceived: {"0xCC8E":{"Endpoint":1,"LinkQuality":115}}
15:12:05 ZIG: {"ZbEZSPReceived2":"022190013F00008ECC040100EF010140010000E5010000"}
15:12:05 ZIG: ZbEZSPSentRaw 81
15:12:05 ZIG: {"ZbEZSPReceived":"3F00008ECC040100EF010140010000E5010000"}
``
And nothing happened.

Then I tried to match the sequence number for the one expected by the device. First I turned on and off the switch1, to get the last sequence number. Heres the output when I turned off switch1 pressing the button:
``
15:14:24 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":136,"securityuse":0,"seqnumber":187,"fc":"0x09","manuf":"0x0000","transact":18,"cmdid":"0x02","payload":"00120101000100"}}
15:14:24 ZIG: ZbZCLRawReceived: {"0xCC8E":{"EF00/0101":0,"Endpoint":1,"LinkQuality":136}}
15:14:24 MQT: tele/zigbeebridge/CC8E/SENSOR = {"ZbReceived":{"switch":{"Device":"0xCC8E","Name":"switch","Power":0,"Endpoint":1,"LinkQuality":136}}}``

As we can see, the last sequence number was "0012" (second byte of the payload).
So I tried ´´ZbSend {"Device":"switch","Send":"EF00_00/00130101000101"}´´
And still no luck:
``
15:15:35 CMD: ZbSend {"Device":"switch","Send":"EF00_00/00130101000101"}
15:15:35 SRC: WebConsole from 192.168.0.3
15:15:35 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"switch","Send":"EF00_00/00130101000101"}"
15:15:35 ZIG: guessing endpoint 1
15:15:35 ZigbeeZCLSend device: 0xCC8E, group: 0x0000, endpoint:1, cluster:0xEF00, cmd:0x00, send:"00130101000101"
15:15:35 ZbSend: shortaddr 0xCC8E, groupaddr 0x0000, cluster 0xEF00, endpoint 0x01, cmd 0x00, data 
15:15:35 ZIG: ZbEZSPSend 3400008ECC040100EF01014001000006010A00060000130101000101
15:15:35 ZIG: adding packet to_send, to_ack:2, to_send:2, to_end:2
15:15:35 MQT: stat/zigbeebridge/RESULT = {"ZbSend":"Done"}
15:15:35 ZIG: Something to_send, to_ack:2, to_send:2, to_end:3
[... cut ...]
15:17:02 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":131,"securityuse":0,"seqnumber":213,"fc":"0x09","manuf":"0x0000","transact":19,"cmdid":"0x02","payload":"00130101000101"}}
15:17:02 ZIG: ZbZCLRawReceived: {"0xCC8E":{"EF00/0101":1,"Endpoint":1,"LinkQuality":131}}
15:17:02 MQT: tele/zigbeebridge/CC8E/SENSOR = {"ZbReceived":{"switch":{"Device":"0xCC8E","Name":"switch","Power":1,"Endpoint":1,"LinkQuality":131}}}``

utech-git, 2020-11-17:
Indeed, it didn't change anything... lol

Heres the output for ``ZbSend {"Device":"switch","Send":"EF00_00/005501010001000101"}``

``21:40:41 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":100,"securityuse":0,"seqnumber":106,"fc":"0x18","manuf":"0x0000","transact":2,"cmdid":"0x01","payload":"005586010186000186000186"}}
21:40:41 ZIG: ZbZCLRawReceived: {"0xCC8E":{"Endpoint":1,"LinkQuality":100}}``

If it helps in any way: one interesting thing is that the device didn't respond with the last byte 01 (possibly followed by 86). Seems like the device ignored it.

It seems that the number of bytes sent in the command payload (the bytes after the "/") must be even.
Since "005501010001000101" has 17 bytes, it ignores the last byte and just respond with the first 16 (each followed by 86).

If I add one more byte on the payload, for example: ``ZbSend {"Device":"switch","Send":"EF00_00/005501010001000101AA"}``
I responds the 18 bytes, followed by 86:

``21:55:52 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":102,"securityuse":0,"seqnumber":189,"fc":"0x18","manuf":"0x0000","transact":10,"cmdid":"0x01","payload":"00558601018600018600018601AA86"}}
21:55:52 ZIG: ZbZCLRawReceived: {"0xCC8E":{"Endpoint":1,"LinkQuality":102}}``

I got a CC2531 zigbee stick here. Would it help if I install zigbee2mqtt and get their log when pressing the switches?

utech-git, 2020-11-17:
Fiddling with it a little bit more on the first command you told me, I tried to change the "command" on the raw bytes sent (byte before the "/") from 00 to 01:
``ZbSend {"Device":"switch","Send":"EF00_01/00550101000101"}``

I get a different output:
``22:34:29 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xCC8E","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":102,"securityuse":0,"seqnumber":198,"fc":"0x08","manuf":"0x0000","transact":19,"cmdid":"0x0B","payload":"0100"}}
22:34:29 MQT: tele/zigbeebridge/RESULT = {"ZbResponse":{"Device":"0xCC8E","Name":"switch","Command":"EF00!01","Status":0,"StatusMessage":"SUCCESS","Endpoint":1,"LinkQuality":102}}``

It seems that the command is now recognized. Altought the nothing happens on the switch 

(I dont know if this tells you anything)

utech-git, 2020-11-17:
A few months ago (when I used z2m) I tried to sniff zigbee traffic using wireshark and the CC2531 but I just couldn't get it to work so I gave up.

I tried it now and it finally worked. So, based on the [link](https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d) you posted before, here's some updates with the wireshark sniffing:

As expected, when I turn on switch 1 the sniffed traffic on the data part of the packet is:
Data: 004B0101000101

It matchs correctly what you said before:
004B is the sequence
0101 is the gang/switch identifier
00 is always 00
01 is the data size
01 is the command data

But when I use the command ``ZbSend {"Device":"switch","Send":"EF00_00/004C0101000100"}`` tasmota console I immediately get a packet sent from the coordinator (0x0000) to the switch (0xcc8e) with the "info" field on Wireshark as "ZCL: Read Attributes, Seq: 5 [Malformed Packet]
[image]

A few packets below I get the "Read Attribute Response" from the device to the coordinator.
This packet says that the attributes 0x4c00, 0x0101 and 0x0100 are unsupported (thus, the 0x86)
[image]

I'll keep digging. But apparently the command is not OK. Because I'm not trying to "read" any atribute, right? I'm trying to SEND a command...

utech-git, 2020-11-18:
Ok. Flashed one CC2531 with the zigbee2mqtt firmware. Flashed other CC2531 I had here with the sniffer firmware. And installed zigbee2mqtt.

Paired the switch with z2m and started sniffing the traffic.

I could controle the switches on z2m with the topics: zigbee2mqtt/0xaaaaaaaaaaaaaa/lx/set, where x is the switch number.
It worked.

But when I sent the mqtt message  ´´zigbee2mqtt/0xaaaaaaaaaaaaaa/lx/set´´ with payload "on" I couldn't sniff any packet going FROM the coordinator TO the device. The packets from the commands are below 824 to 828
[image]

Packet 826 is FROM device TO coordinator. And it has the information we already know: 
0033 = sequence
0101 = switch
00 = always 00
01 = data lenght
01 = data

Am I doing something wrong? When I sent the mqtt messagen, shouldn't I be able to sniff a package from the coordinator to the device?

s-hadinger, 2020-11-18:
Oh my! Sorry!!! I rarely use the raw commands, and it was the wrong option.

It should be:
Turn on switch 1: `ZbSend {"Device":"switch","Send":"EF00!00/00550101000101"}`
Turn off switch 1: `ZbSend {"Device":"switch","Send":"EF00!00/00560101000100"}`
Turn on switch 2: `ZbSend {"Device":"switch","Send":"EF00!00/00570201000101"}`
Turn off switch 2: `ZbSend {"Device":"switch","Send":"EF00!00/00580201000100"}`

The`!` instead of `_` indicates it's a cluster specific command. Command 0x00 in the general command frames is "Read Attributes" - which is definitely wrong.

utech-git, 2020-11-18:
Perfect! Thanks a lot!!!! Now each gang is controlled!! And really helped me figuring out how it works so I can do it by myself with other devices in the future.

Just for the record, here's the output for switch4 on (``ZbSend {"Device":"switch","Send":"EF00!00/00550401000101"}``)

``14:46:22 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xAC7C","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":136,"securityuse":0,"seqnumber":127,"fc":"0x09","manuf":"0x0000","transact":17,"cmdid":"0x02","payload":"00110401000101"}}
14:46:22 ZIG: ZbZCLRawReceived: {"0xAC7C":{"EF00/0104":1,"Endpoint":1,"LinkQuality":136``

And switch4 off (``ZbSend {"Device":"switch","Send":"EF00!00/00550401000100"}``)

``14:47:38 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xAC7C","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":131,"securityuse":0,"seqnumber":153,"fc":"0x09","manuf":"0x0000","transact":18,"cmdid":"0x02","payload":"00120401000100"}}
14:47:38 ZIG: ZbZCLRawReceived: {"0xAC7C":{"EF00/0104":0,"Endpoint":1,"LinkQuality":131}}``

One last question. I know I dont need to change the firmware itself (flashing zigbee devices with tasmota) to control zigbee devices.
But for example on this device: let's say I want to disable switch4 from controlling the relay directly and I want switch4 to just start a scene on Home Assistant (for example, when I press switch4 I send a command to HA and it turns off all the lights). For this I would have to use a custom firmware, right?

Is tasmota planning to "launch" a firmware for this? Like in normal tasmota I can configure each pin to control relays/switch and stuff. It would be nice to do this with zigbee devices. And also with a custom tasmota firmware I think it could be possible to make each gang "understand" commands like: single/double/triple clicks and long press (the tuya custom firmware on this devices doesn't allow this. At least I think it doesn't. I dont have a tuya zigbee Hub to try configuring it like this)

This way when I press switch 4 relay4 wouldn't be turned on/off, and I could use single click to start scene 1, double click to start scene 2, and so on.

And thanks again!

s-hadinger, 2020-11-18:
Good to know it works now. I would be grateful if you edited the Zigbee repository template with all the findings.

I have the impression that this device is not listed https://zigbee.blakadder.com/switches.html

About customizing the behavior of the device, this is unfortunately out of our reach. It depends on the device. Some switches can be remotely configured to decouple the button and the relay. But this one doesn't seem to be able to do so.

s-hadinger, 2020-12-06:
@utech-git Can you please try with #10074

You will need to re-pair the device or use `ZbRestore` to mark the device as Tuya protocol.

Then you can use:
Turn on switch 1: `ZbSend {"Device":"switch","Write":{"Power":1}}`
Turn off switch 1: `ZbSend {"Device":"switch","Write":{"Power":0}}`
Turn on switch 2: `ZbSend {"Device":"switch","Write":{"Power2":1}}`
Turn off switch 2: `ZbSend {"Device":"switch","Write":{"Power2":0}}`

Can you please confirm it's working?
Thanks

utech-git, 2020-12-06:
Well... it kinda worked. But not for all switches/gang.

"Power" controls switch 1
But all other "Power<x>" controls switch 2.

For example 
`ZbSend {"Device":"switch","Write":{"Power2":1}}` turned on switch 2
`ZbSend {"Device":"switch","Write":{"Power3":0}}` turned off switch 2
`ZbSend {"Device":"switch","Write":{"Power3":1}}` also turned on switch 2

The same goes for Power4 also controlling switch 2.

One thing that I noted is that the value for "Power<x>" is: 0 turns it off. Any other number turns it on.
It would be nice to have 2 for toggle, like "regular" tasmota, if possible.
Like `ZbSend {"Device":"switch","Write":{"Power3":2}}` for toggle

Let me know if I can help with anything else.

s-hadinger, 2020-12-07:
I don't see why it does not work for Power3 and Power4. In my own tests here is what I get (with `WebLog 3`):

```
ZbSend {"Device":"0x6FBD","Write":{"Power3":1}}
ZbEZSPSend 340000BD6F040100EF01014001000005010A11050000040103000101

ZbSend {"Device":"0x6FBD","Write":{"Power4":1}}
ZbEZSPSend 340000BD6F040100EF01014001000007010A11070000060104000101
```

Can you please try again with `WebLog 3` and copy what you see in `ZbEZSPSend` lines?

Also, for Power toggle, it's unfortunately not possible with the Tuya protocol since you can only write a boolean to the Power attribute. Unlesss I missed something, you can't just toggle.

utech-git, 2020-12-07:
Ok, I think I messed up the tests yesterday.
I guess somewhere mid test I turned the whole switch upside down and I made a confusion with the switches. (I didn't installed it on my wall yet. I just power it on and do the tests).

What really is happening is: every Power<x> actually controls Switch 1.

Here are the outputs:

Command: **ZbSend {"Device":"switch","Write":{"Power":1}}**
Console output: ``ZbEZSPSend 3400007CAC040100EF01014001000041010A11410000400101000101``
Outcome: switch 1 turns on
"feedback": 
``19:50:46 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xXXXX","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":126,"securityuse":0,"seqnumber":184,"fc":"0x09","manuf":"0x0000","transact":10,"cmdid":"0x02","payload":"000A0101000101"}}
19:50:46 ZIG: ZbZCLRawReceived: {"0xXXXX":{"EF00/0101":1,"Endpoint":1,"LinkQuality":126}}
19:50:46 MQT: tele/XXXXXXXXXXXXX/SENSOR = {"ZbReceived":{"switch":{"Device":"0xXXXX","Name":"switch","Power":1,"Endpoint":1,"LinkQuality":126}}} (retained)``

Command: **ZbSend {"Device":"switch","Write":{"Power":1}}**
Console output: ``ZbEZSPSend 3400007CAC040100EF01014001000043010A11430000420101000100`` 
Outcome: switch 1 turns off
"Feedback":
``19:52:59 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xXXXX","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":123,"securityuse":0,"seqnumber":212,"fc":"0x09","manuf":"0x0000","transact":11,"cmdid":"0x02","payload":"000B0101000100"}}
19:52:59 ZIG: ZbZCLRawReceived: {"0xXXXX":{"EF00/0101":0,"Endpoint":1,"LinkQuality":123}}
19:52:59 MQT: tele/XXXXXXXXXXXXX/SENSOR = {"ZbReceived":{"switch":{"Device":"0xXXXX","Name":"switch","Power":0,"Endpoint":1,"LinkQuality":123}}} (retained)``

Command: **ZbSend {"Device":"switch","Write":{"Power2":1}}**
Console output: ``ZbEZSPSend 3400007CAC040100EF01014001000045010A11450000440102000101`` 
Outcome: switch 1 turns on
"Feedback":
``19:54:31 ZIG: {"ZbEZSPReceived":"450000040100EF010100010000EDF8DA7CACFFFF0A090C02000C010100010102"}
19:54:31 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xXXXX","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":128,"securityuse":0,"seqnumber":237,"fc":"0x09","manuf":"0x0000","transact":12,"cmdid":"0x02","payload":"000C0101000101"}}
19:54:31 ZIG: ZbZCLRawReceived: {"0xXXXX":{"EF00/0101":1,"Endpoint":1,"LinkQuality":128}}
19:54:31 MQT: tele/XXXXXXXXXXXXX/SENSOR = {"ZbReceived":{"switch":{"Device":"0xXXXX","Name":"switch","Power":1,"Endpoint":1,"LinkQuality":128}}} (retained)``

Command: **ZbSend {"Device":"switch","Write":{"Power4":0}}**
Console output: ``ZbEZSPSend 3400007CAC040100EF01014001000047010A11470000460104000100`` 
Outcome: switch 1 turns off
"Feedback":
``19:55:48 ZIG: {"ZbZCLReceived": {"groupid":0,"clusterid":61184,"srcaddr":"0xXXXX","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":126,"securityuse":0,"seqnumber":6,"fc":"0x09","manuf":"0x0000","transact":13,"cmdid":"0x02","payload":"000D0101000100"}}
19:55:49 ZIG: ZbZCLRawReceived: {"0xXXXX":{"EF00/0101":0,"Endpoint":1,"LinkQuality":126}}
19:55:49 MQT: tele/XXXXXXXXXXXXX/SENSOR = {"ZbReceived":{"switch":{"Device":"0xXXXX","Name":"switch","Power":0,"Endpoint":1,"LinkQuality":126}}} (retained)``

I also did the test with the "old" command, without the "Power", in case it helps

Command: **ZbSend {"Device":"switch","Send":"EF00!00/00550401000100"}**
Console output: ``ZbEZSPSend 3400007CAC040100EF0101400100004A010A014A0000550401000100`` 
Outcome: switch 4 turns on
"Feedback":
``20:06:57 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":61184,"srcaddr":"0xXXXX","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":128,"securityuse":0,"seqnumber":81,"fc":"0x09","manuf":"0x0000","transact":16,"cmdid":"0x02","payload":"00100401000100"}}
20:06:57 ZIG: ZbZCLRawReceived: {"0xXXXX":{"EF00/0104":0,"Endpoint":1,"LinkQuality":128}}
20:06:57 MQT: tele/XXXXXXXXXXXXX/SENSOR = {"ZbReceived":{"switch":{"Device":"0xXXXX","Name":"switch","Power4":0,"Endpoint":1,"LinkQuality":128}}} (retained)``

What most caught my attention between the last two ZbEZSPSend is the byte highlighted here:
The command that doesn't work ("Write":{"Power4":0})
3400007CAC040100EF01014001000047010A1147000046``0104``000100

The command that works ("Send":"EF00!00/00550401000100")
3400007CAC040100EF0101400100004A010A014A000055``0401``000100

Maybe a miscalculation of the byte number when converting the switch id to big endian?

I hope it helps and let me know whenever I can test again

PR #9882 (2020-11-16): Zigbee support for Tuya 4 Gang wall switch TS0601
## Description:

**Related issue (if applicable):** fixes #9860

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.7
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreem [... cut]