arendst/Tasmota #9861: ESP32 TTGO Camera Mini CAM: Init failed to get the frame on time

victorclaessen, 2020-11-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
My camera doesn't work. I flashed tasmota32-webcam.bin onto a TTGO Camera Mini ([link to image of the device](https://www.tinytronics.nl/shop/image/cache/data/product-2670/H382_default-1000x1000.jpg)) and configured it, but I get the error `CAM: Init failed to get the frame on time`.

The pinout configuration for the camera can be found here (black pcb version):
https://github.com/Xinyuan-LilyGO/TTGO_Camera_Mini/blob/master/TTGO_Camera_Mini.ino#L42

This is my configuration in tasmota:
[image]
My assumption was that the CAM_DATA pins 1-8 correspond to the Y2-Y9 pins. Without the CAM_DATA pins, the device is not recognised at all. `[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
21:49:43 CAM: Init failed with error 0x20004
`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): TTGO Camera mini
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:49:49 CMD: Backlog Template; Module; GPIO 255
21:49:49 RSL: stat/tasmota_82EB90/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
21:49:49 RSL: stat/tasmota_82EB90/RESULT = {"Module":{"1":"ESP32-DevKit"}}
21:49:49 RSL: stat/tasmota_82EB90/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"5090":"CAM_DATA3"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"5056":"CAM_SIOC"},"GPIO13":{"5024":"CAM_SIOD"},"GPIO14":{"5089":"CAM_DATA2"},"GPIO15":{"5091":"CAM_DATA4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5092":"CAM_DATA5"},"GPIO19":{"5184":"CAM_PCLK"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA"},"GPIO22":{"608":"I2C SCL"},"GPIO23":{"5093":"CAM_DATA6"},"GPIO24":{"0":"None"},"GPIO25":{"5152":"CAM_HREF"},"GPIO26":{"0":"None"},"GPIO27":{"5120":"CAM_VSYNC"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"}," ...

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
NO RULES

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:51:07 CMD: status 0
21:51:07 RSL: stat/tasmota_82EB90/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_82EB90","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Vbat power on reset","Uptime":"0T00:01:35","StartupUTC":"2020-11-14T20:49:32","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-11-14T21:19:04","SaveCount":13}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS2 = {"StatusFWR":{"Version":"9.1.0(webcam)","BuildDateTime":"2020-11-07T12:35:02","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":240,"Hardware":"ESP32","CR":"373/699"}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Zeezicht_M2M",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS4 = {"StatusMEM":{"ProgramSize":1185,"Free":1984,"Heap":111,"PsrMax":4095,"PsrFree":4095,"ProgramFlashSize":16384,"FlashSize":16384,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","80004000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,81","Sensors":"1,2,3,5,6"}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS5 = {"StatusNET":{"Hostname":"Lilycam","IPAddress":"10.29.7.65","Gateway":"10.29.9.4","Subnetmask":"255.255.240.0","DNSServer":"10.29.0.1","Mac":"8C:AA:B5:82:EB:90","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_82EB90","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS7 = {"StatusTIM":{"UTC":"2020-11-14T20:51:07","Local":"2020-11-14T21:51:07","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:58","Sunset":"17:10"}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS10 = {"StatusSNS":{"Time":"2020-11-14T21:51:07"}}
21:51:07 RSL: stat/tasmota_82EB90/STATUS11 = {"StatusSTS":{"Time":"2020-11-14T21:51:07","Uptime":"0T00:01:35","UptimeSec":95,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Redacted_M2M","BSSId":"1E:EC:BA:8A:63:86","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
21:31:15 RSL: tele/tasmota_82EB90/INFO1 = {"Module":"ESP32-DevKit","Version":"9.1.0(webcam)","FallbackTopic":"cmnd/DVES_82EB90_fb/","GroupTopic":"cmnd/tasmotas/"}
21:31:15 RSL: tele/tasmota_82EB90/INFO2 = {"WebServerMode":"Admin","Hostname":"Lilycam","IPAddress":"10.29.7.65"}
21:31:15 RSL: tele/tasmota_82EB90/INFO3 = {"RestartReason":"Software reset CPU"}
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
21:31:22 CAM: Init failed to get the frame on time
21:31:22 QPC: Reset

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install tasmota 9.1.0 on TTGO Camera Mini, configure as above.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I was expecting the camera to work.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Included screenshot of configuration above

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-15:
This is the default setting
```
// CAMERA_MODEL_AI_THINKER default template pins
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22
```
according to your link there are missing some GPIO defines in your setup
```
//! New version is black PCB, please use these pins
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    32
#define SIOD_GPIO_NUM    13
#define SIOC_GPIO_NUM    12

#define Y9_GPIO_NUM      39
#define Y8_GPIO_NUM      36
#define Y7_GPIO_NUM      38
#define Y6_GPIO_NUM      37
#define Y5_GPIO_NUM      15
#define Y4_GPIO_NUM      4
#define Y3_GPIO_NUM      14
#define Y2_GPIO_NUM      5

#define VSYNC_GPIO_NUM   27
#define HREF_GPIO_NUM    25
#define PCLK_GPIO_NUM    19
```

Jason2866, 2020-11-15:
Edit sorry. Setup looks correct so far.

victorclaessen, 2020-11-15:
I was going back and forth between the 9.1.0 release and the development binary from 2020-11-15. At some point (while I was on the 9.1.0 release) I realised that I made a mistake in CAM_DATA_6 that I linked to pin23 above, but should have been pin 38. I then saw that pin38 was not exposed in this template, so I made a new template. 

{"NAME":"TTGO Camera Mini","GPIO":[544,0,0,0,5090,5088,1,1,5056,5024,5089,5091,0,0,0,5184,0,640,608,0,0,5152,0,5120,0,0,0,0,4992,32,0,0,5094,5092,5093,5095],"FLAG":0,"BASE":1}

And after applying this template, I got a video stream!

I did have trouble moving to the development firmware, because the error returned. In the end, what worked was flashing the original firmware, then erasing flash, then flashing the development release. It may have something to do with the power management feature that it mentioned on [this page](https://github.com/Xinyuan-LilyGO/TTGO_Camera_Mini), something about turning off the current limit of the AXP192. Not sure.

Jason2866, 2020-11-16:
If the camera has PSRAM Tasmota does not support. So you have to reduce to XGA 1024×768
See https://randomnerdtutorials.com/esp32-cam-troubleshooting-guide/

ascillato2, 2020-11-16:
Please when you have time, tell us if this solves your issue. Thanks.

closed 2020-11-16 by ascillato2

victorclaessen, 2020-11-16:
I tried WcResolution 0 through 10, makes no difference. In the original firmware as received from AliExpress, which I backed up but do not have a link to the source from, the streaming works at any resolution (albeit at lower framerates with increasing resolution).

victorclaessen, 2020-11-17:
I found the code that produces the firmware it was shipped with originally [here](https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series) and specifically the [readme for this model](https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/master/docs/T_CameraMini.md). It works at the highest resolution.