arendst/Tasmota #9863: DevGroupName: doesn't work with dashes in the name. Fix or document

hipitihop, 2020-11-15:
### PROBLEM DESCRIPTION
When using group name which include characters like dash `-` devices don't join the device group.
Verified by issuing `DevGroupStatus` which would show `"Members"` as empty array `[ ]`
Example: `patio-leds` does not work however `pled` works fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ElectroDragon ESP LED Strip Board
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: ESPTool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
01:09:49 MQT: dragon-led-3/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
01:09:49 MQT: dragon-led-3/stat/RESULT = {"Module":{"18":"Generic"}}
01:09:49 MQT: dragon-led-3/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"418":"PWM3"},"GPIO13":{"417":"PWM2"},"GPIO14":{"419":"PWM4"},"GPIO15":{"416":"PWM1"},"GPIO16":{"288":"Led1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Set `DevGroupName1 patio-led`

### EXPECTED BEHAVIOUR
Expect devices in this the said group to join and show as "Members" in `DevGroupStatus`

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pcdiem, 2020-11-15:
I don't see this behavior. Do you see the correct group name (patio-leds) for DevGroupName1 on all devices in the group when you enter the DevGroupName command?

18:53:53 MQT: stat/TestLight1/RESULT = {"DevGroupStatus":{"Index":0,"GroupName":"patio-leds","MessageSeq":4,"MemberCount":1,"Members":[{"IPAddress":"192.168.0.251","ResendCount":0,"LastRcvdSeq":2,"LastAckedSeq":4}]}}

ascillato2, 2020-11-16:
When you have time, please, let us know if this solves your issue. Thanks.

hipitihop, 2020-11-19:
I have retested by assigning both devices to `patio-leds` and after they restart they indeed join the group and show as members. I'm puzzled with what I saw but indeed it now works as expected so my apologies for wasting peoples time.