arendst/Tasmota #9864: Dimmer turns on immeidately after being turned off

jonootto, 2020-11-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Using a tuya dimmer, when i use a version higher than 8.5.1, the light will not turn off.
When it's on, then i turn it off, after about a second the lights turn back on automatically.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): BrilliantSmart Jupiter Dimmer
- [ ] Tasmota binary firmware version number used: 8.5.1 (ok) 9.1.0(issue) 9.1.0.1(issue)
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:11:41 CMD: Backlog Template; Module; GPIO 255
08:11:41 SRC: WebConsole from 192.168.1.68
08:11:41 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
08:11:41 SRC: Backlog
08:11:41 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
08:11:41 MQT: stat/bedroom1/RESULT = {"NAME":"Generic","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
08:11:41 SRC: Backlog
08:11:41 CMD: Group 0, Index 1, Command "MODULE", Data ""
08:11:41 MQT: stat/bedroom1/RESULT = {"Module":{"54":"Tuya MCU"}}
08:11:41 SRC: Backlog
08:11:41 CMD: Group 0, Index 1, Command "GPIO", Data "255"
08:11:41 MQT: stat/bedroom1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
08:11:51 TYA: Send "55aa00000000ff"
08:11:51 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:12:18 CMD: Group 0, Index 1, Command "STATUS", Data "0"
08:12:18 MQT: stat/bedroom1/STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"bedroom1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:12:18 MQT: stat/bedroom1/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:25","StartupUTC":"2020-11-15T07:04:53","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-11-14T07:43:01","SaveCount":110,"SaveAddress":"F8000"}}
08:12:18 MQT: stat/bedroom1/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"360/699"}}
08:12:18 MQT: stat/bedroom1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RG-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00DA8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
08:12:18 MQT: stat/bedroom1/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:12:18 MQT: stat/bedroom1/STATUS5 = {"StatusNET":{"Hostname":"bedroom1-1782","IPAddress":"192.168.30.32","Gateway":"192.168.30.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.30.1","Mac":"80:7D:3A:73:E6:F6","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:12:18 MQT: stat/bedroom1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.254","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_73E6F6","MqttUser":"mqttadmin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:12:18 MQT: stat/bedroom1/STATUS7 = {"StatusTIM":{"UTC":"2020-11-15T07:12:18","Local":"2020-11-15T08:12:18","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:00","Sunset":"17:09"}}
08:12:18 MQT: stat/bedroom1/STATUS10 = {"StatusSNS":{"Time":"2020-11-15T08:12:18"}}
08:12:18 MQT: stat/bedroom1/STATUS11 = {"StatusSTS":{"Time":"2020-11-15T08:12:18","Uptime":"0T00:07:25","UptimeSec":445,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":18,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Turn off light
Light turns back on

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Turn off light
Light stays off

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-15:
Please, set weblog to 4, then try to turn your light off and copy and paste the output logs of the console. Thanks.

effelle, 2020-11-15:
If when you light is off the MCU send a dimmer level the light will turn automatically on. If that the case play a bit with `setoption54` and `setoption20`.

jonootto, 2020-11-16:
> Please, set weblog to 4, then try to turn your light off and copy and paste the output logs of the console. Thanks.

Done

jonootto, 2020-11-16:
> `setoption54`

Have tried all four combinations, no change to this behaviour.

jonootto, 2020-11-16:
> If when you light is off the MCU send a dimmer level the light will turn automatically on. If that the case play a bit with `setoption54` and `setoption20`.

Have tried all four combinations, no change to this behaviour.

ascillato, 2020-11-16:
> Here it is behaving normally on 8.5.1:

Great. Thanks. Can you post also the one from latest Tasmota version?

effelle, 2020-11-16:
```py
07:52:36 TYA: RX Relay-1 --> MCU State: Off Current State:Off
07:52:37 TYA: Send "55aa0006000802020004000002bcd3"
07:52:37 {"TuyaReceived":{"Data":"55AA0307000802020004000002BCD7","Cmnd":7,"CmndData":"02020004000002BC","DpType2Id2":700,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000002BC"}}}
07:52:37 TYA: fnId=21 is set for dpId=2
07:52:37 TYA: RX value 700 from dpId 2 
```
After the relay was turned off Tasmota received a message from the MCU with a dimmer level then turned on the light again.
We have no control on what the MCU send out, Tasmota is simply listening. Enabling both setoption54 and setoption 20  should fix this issue, because are made to ignore a dimmer level when a light is off. 
To see more clearly the issue please enable the stat topic with `tuyasend9` and wait for the event. Also add your TuyaMCU configuration.

effelle, 2020-11-16:
See also #9696

ascillato2, 2020-11-16:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

jonootto, 2020-11-17:
Hi, i think this was closed prematurely.
Options 20 and 54 have no effect and this behaviour is still present with both turned on.

> > Here it is behaving normally on 8.5.1:
> 
> Great. Thanks. Can you post also the one from latest Tasmota version?

Yes, thats in the main issue body

jonootto, 2020-11-17:
I think this was closed prematurely.

Options 20 and 54 have make no change to this when both set to 1.

Here's the weblog after tuyasend9

```
04:06:59 SRC: WebGui from 192.168.1.68
04:06:59 TYA: Send "55aa0006000501010001000d"
04:06:59 MQT: tele/bedroom1/STATE = {"Time":"2020-11-17T04:06:59","Uptime":"0T00:04:40","UptimeSec":280,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":43,"MqttCount":1,"POWER":"OFF","Dimmer":39,"Fade":"ON","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}
04:06:59 MQT: stat/bedroom1/RESULT = {"POWER":"OFF"}
04:06:59 MQT: stat/bedroom1/POWER = OFF
04:06:59 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
04:06:59 MQT: stat/bedroom1/DPTYPE1ID1 = 0
04:06:59 TYA: fnId=11 is set for dpId=1
04:06:59 TYA: RX Relay-1 --> MCU State: Off Current State:Off
04:06:59 {"TuyaReceived":{"Data":"55AA03070008020200040000000019","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
04:06:59 MQT: stat/bedroom1/DPTYPE2ID2 = 0
04:06:59 TYA: fnId=21 is set for dpId=2
04:06:59 TYA: RX value 0 from dpId 2 
04:06:59 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
04:06:59 MQT: stat/bedroom1/DPTYPE2ID105 = 0
04:06:59 TYA: fnId=0 is set for dpId=105
04:06:59 TYA: RX value 0 from dpId 105 
04:06:59 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
04:06:59 MQT: stat/bedroom1/DPTYPE1ID1 = 0
04:06:59 TYA: fnId=11 is set for dpId=1
04:06:59 TYA: RX Relay-1 --> MCU State: Off Current State:Off
04:07:00 TYA: Send "55aa0006000802020004000001869c"
04:07:00 {"TuyaReceived":{"Data":"55AA030700080202000400000186A0","Cmnd":7,"CmndData":"0202000400000186","DpType2Id2":390,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000186"}}}
04:07:00 MQT: stat/bedroom1/DPTYPE2ID2 = 390
04:07:00 TYA: fnId=21 is set for dpId=2
04:07:00 TYA: RX value 390 from dpId 2 
04:07:02 WIF: Checking connection...
04:07:05 TYA: Send "55aa00000000ff"
04:07:05 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
04:07:05 TYA: Heartbeat
04:07:16 TYA: Send "55aa00000000ff"
04:07:16 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
04:07:16 TYA: Heartbeat
```

jonootto, 2020-11-17:
I have the below workaround running 
```
04:27:52 MQT: stat/bedroom1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":35,"Free":476,"Rules":"ON Power1#State=0 DO Dimmer 0 ENDON"}
```

effelle, 2020-11-17:
The problem is I can't replicate the issue there (dev 9.1.0.1).
Tested on a device right now:
```py
12:57:07 CMD: backlog setoption20 1; setoption54 1; power1 1; dimmer 90
12:57:07 MQT: stat/tasmota_49A3BC/RESULT = {"SetOption20":"ON"}
12:57:07 MQT: stat/tasmota_49A3BC/RESULT = {"SetOption54":"ON"}
12:57:07 MQT: stat/tasmota_49A3BC/RESULT = {"POWER":"ON"}
12:57:07 MQT: stat/tasmota_49A3BC/POWER = ON
12:57:07 MQT: stat/tasmota_49A3BC/DPTYPE1ID1 = 1
12:57:07 MQT: stat/tasmota_49A3BC/RESULT = {"POWER":"ON","Dimmer":90}
```
then from device I sent a power off and immediately after a dimmer level:
```py
12:57:09 MQT: stat/tasmota_49A3BC/DPTYPE1ID1 = 0
12:57:09 MQT: tele/tasmota_49A3BC/STATE = {"Time":"2020-11-
12:57:09 MQT: stat/tasmota_49A3BC/RESULT = {"POWER":"OFF"}
12:57:09 MQT: stat/tasmota_49A3BC/POWER = OFF
12:57:09 MQT: stat/tasmota_49A3BC/DPTYPE2ID2 = 77
12:57:09 MQT: stat/tasmota_49A3BC/RESULT = {"POWER":"OFF","Dimmer":77}
```
The new dimmer value was received as expected but it didn't turn on the light with so20 and so54 enabled.
Glad you found a workaround anyway. When you send manually a dimmer level Tasmota will ignore the packets for the next 250ms. I'll see if I can add an ignore time for packet also when the power is turned off, just in case. No ETA.

TheReiner, 2020-11-17:
Same problem for me ...
What is the solution  ? Wait 9.1.0.1 ?
Thanks.

h4nc, 2020-11-20:
I had the same issue. So I enabled `setoption54` and `setoption20` like mentioned above and now it's again possible to turn off the light.

However this opened a new issue for me. When I set a brightness value in Home Assistant the light turns on. But turning on the light using the switch in home assistant doesn't work any more. 
Does anyone know how to fix this?

Side note: One nice thing that I noticed is that the light is dimming on now when using the on button in the Tasmota UI. It already did this before for turning off the light but not for turning it on. So before it just turned on without transition. However it still turns on without transition when I use the hardware button on the device.

effelle, 2020-11-20:
@jonootto, @TheReiner and @h4nc please issue in console the command `backlog TuyaMCU; TuyaSend8; so20; so54; so77` and paste here the result. Thanks.

TheReiner, 2020-11-20:
@effelle Thanks ;)

`15:19:17 CMD: backlog TuyaMCU; TuyaSend8; so20; so54; so77
15:19:17 MQT: stat/tasmota_D9E991/RESULT = {"TuyaMCU":[{"fnId":11,"dpId":1},{"fnId":21,"dpId":2}]}
15:19:17 MQT: stat/tasmota_D9E991/RESULT = {"TuyaSend":"Done"}
15:19:17 TYA: MCU Product ID: bSXSSFArVKtc4DyC1.0.0
15:19:17 MQT: stat/tasmota_D9E991/RESULT = {"SetOption20":"OFF"}
15:19:18 MQT: stat/tasmota_D9E991/RESULT = {"SetOption54":"OFF"}
15:19:18 MQT: stat/tasmota_D9E991/RESULT = {"SetOption77":"OFF"}
15:19:23 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF"}
15:19:23 MQT: stat/tasmota_D9E991/POWER = OFF
15:19:25 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON"}
15:19:25 MQT: stat/tasmota_D9E991/POWER = ON`

TheReiner, 2020-11-20:
@effelle With setOption 54 and 20 On
New Problem...
When I turn off the dimmer it puts it at 16% and Turn Off ... and when I turn it on again it puts it at 100%. Before (8.5.1) he put it in the last state. i use HomeBridge... and the MQTT is different with 9.1.

> 15:25:28 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF","Dimmer":100}
> 15:25:28 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF","Dimmer":16}
> 15:25:30 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON"}
> 15:25:30 MQT: stat/tasmota_D9E991/POWER = ON
> 15:25:30 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON","Dimmer":100}

effelle, 2020-11-20:
I need to see them with `weblog 4` enabled...

TheReiner, 2020-11-20:
My HomeBridge Config

> {
>             "type": "lightbulb",
>             "name": "Spot Chambre Cive",
>             "topics": {
>                 "getOn": {
>                     "topic": "stat/tasmota_D9E991/RESULT",
>                     "apply": "return JSON.parse(message).POWER;"
>                 },
>                 "setOn": "cmnd/tasmota_D9E991/power",
>                 "getBrightness": {
>                     "topic": "stat/tasmota_D9E991/RESULT",
>                     "apply": "return JSON.parse(message).Channel1;"
>                 },
>                 "setBrightness": "cmnd/tasmota_D9E991/dimmer"
>             },
>             "onValue": "ON",
>             "offValue": "OFF",
>             "accessory": "mqttthing"
>         },

And with Weblog 4 Enable

> 15:31:53 CMD: Weblog 4 on
> 15:31:53 MQT: stat/tasmota_EE3343/RESULT = {"WebLog":4}
> 15:31:54 CFG: Saved to flash at F7, Count 1582, Bytes 4096
> 15:31:59 TYA: Send "55aa00000000ff"
> 15:31:59 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:31:59 TYA: Heartbeat
> 15:32:10 WIF: Checking connection...
> 15:32:10 TYA: Send "55aa00000000ff"
> 15:32:10 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:32:10 TYA: Heartbeat
> 15:32:21 TYA: Send "55aa00000000ff"
> 15:32:21 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:32:21 TYA: Heartbeat
> 15:32:30 WIF: Checking connection...
> 15:32:32 TYA: Send "55aa00000000ff"
> 15:32:32 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:32:32 TYA: Heartbeat
> 15:32:35 MQT: tele/tasmota_EE3343/STATE = {"Time":"2020-11-20T15:32:35","Uptime":"3T04:10:48","UptimeSec":274248,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Dimmer":16,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"8Filles","BSSId":"A0:55:4F:66:C3:38","Channel":9,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:05"}}
> 15:32:43 TYA: Send "55aa00000000ff"
> 15:32:43 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:32:43 TYA: Heartbeat
> 15:32:50 WIF: Checking connection...

effelle, 2020-11-20:
Now that you have enabled weblog 4 paste the result of `backlog TuyaMCU; TuyaSend8`.

TheReiner, 2020-11-20:
i really don't understand the dimmer back to 16%

there is the log...

> 15:49:15 MQT: stat/tasmota_D9E991/RESULT = {"WebLog":4}
> 15:49:16 CFG: Saved to flash at F7, Count 1376, Bytes 4096
> 15:49:18 CMD: backlog TuyaMCU; TuyaSend8.
> 15:49:18 SRC: WebConsole from 10.15.67.46
> 15:49:18 CMD: Group 0, Index 1, Command "BACKLOG", Data "TuyaMCU; TuyaSend8."
> 15:49:18 SRC: Backlog
> 15:49:18 CMD: Group 0, Index 1, Command "TUYAMCU", Data ""
> 15:49:18 MQT: stat/tasmota_D9E991/RESULT = {"TuyaMCU":[{"fnId":11,"dpId":1},{"fnId":21,"dpId":2}]}
> 15:49:18 SRC: Backlog
> 15:49:18 CMD: Group 0, Index 8, Command "TUYASEND", Data "."
> 15:49:18 TYA: Read MCU state
> 15:49:18 TYA: Send "55aa0001000000"
> 15:49:18 MQT: stat/tasmota_D9E991/RESULT = {"TuyaSend":"Done"}
> 15:49:18 {"TuyaReceived":{"Data":"55AA000100156253585353464172564B746334447943312E302E305A","Cmnd":1,"CmndData":"6253585353464172564B746334447943312E302E30"}}
> 15:49:18 TYA: MCU Product ID: bSXSSFArVKtc4DyC1.0.0
> 15:49:18 TYA: Send "55aa0002000001"
> 15:49:18 {"TuyaReceived":{"Data":"55AA000200020E0011","Cmnd":2,"CmndData":"0E00"}}
> 15:49:18 TYA: RX MCU configuration Mode=2
> 15:49:18 TYA: Read MCU state
> 15:49:18 TYA: Send "55aa0008000007"
> 15:49:18 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
> 15:49:18 TYA: fnId=11 is set for dpId=1
> 15:49:18 TYA: RX Relay-1 --> MCU State: Off Current State:Off
> 15:49:18 {"TuyaReceived":{"Data":"55AA00070008020200040000001026","Cmnd":7,"CmndData":"0202000400000010","DpType2Id2":16,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000010"}}}
> 15:49:18 TYA: fnId=21 is set for dpId=2
> 15:49:18 TYA: RX value 16 from dpId 2 
> 15:49:19 TYA: Send "55aa00000000ff"
> 15:49:19 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:49:19 TYA: Heartbeat
> 15:49:23 MQT: Data Size 2, "cmnd/tasmota_D9E991/power ON"
> 15:49:23 SRC: MQTT
> 15:49:23 CMD: Group 0, Index 1, Command "POWER", Data "ON"
> 15:49:23 TYA: Send "55aa0006000501010001010e"
> 15:49:23 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON"}
> 15:49:23 MQT: stat/tasmota_D9E991/POWER = ON
> 15:49:23 CFG: Saved to flash at F6, Count 1377, Bytes 4096
> 15:49:23 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
> 15:49:23 TYA: fnId=11 is set for dpId=1
> 15:49:23 TYA: RX Relay-1 --> MCU State: On Current State:On
> 15:49:23 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
> 15:49:23 TYA: fnId=11 is set for dpId=1
> 15:49:23 TYA: RX Relay-1 --> MCU State: On Current State:On
> 15:49:23 {"TuyaReceived":{"Data":"55AA00070008020200040000001026","Cmnd":7,"CmndData":"0202000400000010","DpType2Id2":16,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000010"}}}
> 15:49:23 TYA: fnId=21 is set for dpId=2
> 15:49:23 TYA: RX value 16 from dpId 2 
> 15:49:23 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
> 15:49:23 TYA: fnId=11 is set for dpId=1
> 15:49:23 TYA: RX Relay-1 --> MCU State: On Current State:On
> 15:49:23 {"TuyaReceived":{"Data":"55AA00070008020200040000006D83","Cmnd":7,"CmndData":"020200040000006D","DpType2Id2":109,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000006D"}}}
> 15:49:23 TYA: fnId=21 is set for dpId=2
> 15:49:23 TYA: RX value 109 from dpId 2 
> 15:49:23 SRC: Switch
> 15:49:23 CMD: Group 0, Index 3, Command "DIMMER", Data "100"
> 15:49:23 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON","Dimmer":100}
> 15:49:24 CFG: Saved to flash at F5, Count 1378, Bytes 4096
> 15:49:25 WIF: Checking connection...
> 15:49:27 MQT: Data Size 3, "cmnd/tasmota_D9E991/power OFF"
> 15:49:27 SRC: MQTT
> 15:49:27 CMD: Group 0, Index 1, Command "POWER", Data "OFF"
> 15:49:27 TYA: Send "55aa0006000501010001000d"
> 15:49:27 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF"}
> 15:49:27 MQT: stat/tasmota_D9E991/POWER = OFF
> 15:49:27 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
> 15:49:27 TYA: fnId=11 is set for dpId=1
> 15:49:27 TYA: RX Relay-1 --> MCU State: Off Current State:Off
> 15:49:27 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
[... 3 lines cut ...]
> 15:49:27 TYA: fnId=21 is set for dpId=2
> 15:49:27 TYA: RX value 109 from dpId 2 
> 15:49:27 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
> 15:49:27 TYA: fnId=11 is set for dpId=1
> 15:49:27 TYA: RX Relay-1 --> MCU State: Off Current State:Off
> 15:49:27 {"TuyaReceived":{"Data":"55AA00070008020200040000001026","Cmnd":7,"CmndData":"0202000400000010","DpType2Id2":16,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000010"}}}
> 15:49:27 TYA: fnId=21 is set for dpId=2
> 15:49:27 TYA: RX value 16 from dpId 2 
> 15:49:27 SRC: Switch
> 15:49:27 CMD: Group 0, Index 3, Command "DIMMER", Data "16"
> 15:49:27 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF","Dimmer":16}
> 15:49:28 CFG: Saved to flash at F4, Count 1379, Bytes 4096
> 15:49:29 TYA: Send "55aa00000000ff"
> 15:49:30 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:49:30 TYA: Heartbeat
> 15:49:41 TYA: Send "55aa00000000ff"
> 15:49:41 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:49:41 TYA: Heartbeat
> 15:49:45 WIF: Checking connection...
> 15:49:52 TYA: Send "55aa00000000ff"
> 15:49:52 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:49:52 TYA: Heartbeat
> 15:50:02 TYA: Send "55aa00000000ff"
> 15:50:02 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
> 15:50:02 TYA: Heartbeat

effelle, 2020-11-20:
> i really don't understand the dimmer back to 16%

Is what the device is transmitting to Tasmota.  
Please issue `backlog dimmerrange; ledtable` and paste here the result.

TheReiner, 2020-11-20:
When Turn on "30%"

[image]

```py
16:11:06 MQT: Data Size 2, "cmnd/tasmota_D9E991/power ON"
16:11:06 SRC: MQTT
16:11:06 CMD: Group 0, Index 1, Command "POWER", Data "ON"
16:11:06 TYA: Send "55aa0006000501010001010e"
16:11:06 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON"}
16:11:06 MQT: stat/tasmota_D9E991/POWER = ON
16:11:06 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
16:11:06 TYA: fnId=11 is set for dpId=1
16:11:06 TYA: RX Relay-1 --> MCU State: On Current State:On
16:11:06 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
16:11:06 TYA: fnId=11 is set for dpId=1
16:11:06 TYA: RX Relay-1 --> MCU State: On Current State:On
16:11:06 {"TuyaReceived":{"Data":"55AA00070008020200040000001026","Cmnd":7,"CmndData":"0202000400000010","DpType2Id2":16,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000010"}}}
16:11:06 TYA: fnId=21 is set for dpId=2
16:11:06 TYA: RX value 16 from dpId 2 
16:11:06 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
16:11:06 TYA: fnId=11 is set for dpId=1
16:11:06 TYA: RX Relay-1 --> MCU State: On Current State:On
16:11:06 {"TuyaReceived":{"Data":"55AA00070008020200040000006D83","Cmnd":7,"CmndData":"020200040000006D","DpType2Id2":109,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000006D"}}}
16:11:06 TYA: fnId=21 is set for dpId=2
16:11:06 TYA: RX value 109 from dpId 2 
16:11:06 SRC: Switch
16:11:06 CMD: Group 0, Index 3, Command "DIMMER", Data "100"
16:11:06 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON","Dimmer":100}
16:11:07 CFG: Saved to flash at FB, Count 1380, Bytes 4096
16:11:07 TYA: Send "55aa00000000ff"
16:11:07 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
16:11:07 TYA: Heartbeat
16:11:18 TYA: Send "55aa00000000ff"
16:11:18 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
16:11:18 TYA: Heartbeat
16:11:25 WIF: Checking connection...
16:11:29 TYA: Send "55aa00000000ff"
16:11:29 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
16:11:29 TYA: Heartbeat
16:11:40 TYA: Send "55aa00000000ff"
16:11:40 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
16:11:40 TYA: Heartbeat`

![Screenshot 2020-11-20 at 16 14 29](https://user-images.githubusercontent.com/18024918/99816109-8642d300-2b4b-11eb-9ce3-ed4dc709e8b7.png)

```py
16:13:19 TYA: Heartbeat
16:13:21 MQT: Data Size 3, "cmnd/tasmota_D9E991/power OFF"
16:13:21 SRC: MQTT
16:13:21 CMD: Group 0, Index 1, Command "POWER", Data "OFF"
16:13:21 TYA: Send "55aa0006000501010001000d"
16:13:21 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF"}
16:13:21 MQT: stat/tasmota_D9E991/POWER = OFF
16:13:21 CFG: Saved to flash at FA, Count 1381, Bytes 4096
16:13:21 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:13:21 TYA: fnId=11 is set for dpId=1
16:13:21 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:13:21 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:13:21 TYA: fnId=11 is set for dpId=1
16:13:21 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:13:21 {"TuyaReceived":{"Data":"55AA00070008020200040000006D83","Cmnd":7,"CmndData":"020200040000006D","DpType2Id2":109,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000006D"}}}
16:13:21 TYA: fnId=21 is set for dpId=2
16:13:21 TYA: RX value 109 from dpId 2 
16:13:21 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:13:21 TYA: fnId=11 is set for dpId=1
16:13:21 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:13:21 {"TuyaReceived":{"Data":"55AA00070008020200040000001026","Cmnd":7,"CmndData":"0202000400000010","DpType2Id2":16,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000010"}}}
16:13:21 TYA: fnId=21 is set for dpId=2
16:13:21 TYA: RX value 16 from dpId 2 
16:13:21 SRC: Switch
16:13:21 CMD: Group 0, Index 3, Command "DIMMER", Data "16"
16:13:21 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF","Dimmer":16}
16:13:22 CFG: Saved to flash at F9, Count 1382, Bytes 4096
16:13:25 WIF: Checking connection...
16:13:31 TYA: Send "55aa00000000ff"
16:13:31 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
16:13:31 TYA: Heartbeat`

effelle, 2020-11-20:
Can we focus on Tasmota please? 
I don't know nor I support HomeBridge, all the test must be done inside the console.  
Also:
> Please issue backlog dimmerrange; ledtable and paste here the result.

TheReiner, 2020-11-20:
OK Always back on 16% inside Tasmota...  But FYI, i only update Tasmota 9.1, not HomeBridge. And all my problem become since 9.1. i Think i will come back to 8.5.1
Thanks and Good Luck.

The log... (Without come bridge command, only inside Tasmota)

```py
16:29:00 TYA: Send "55aa0006000501010001000d"
16:29:00 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF"}
16:29:00 MQT: stat/tasmota_D9E991/POWER = OFF
16:29:00 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:29:00 TYA: fnId=11 is set for dpId=1
16:29:00 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:29:00 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:29:00 TYA: fnId=11 is set for dpId=1
16:29:00 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:29:00 {"TuyaReceived":{"Data":"55AA00070008020200040000007288","Cmnd":7,"CmndData":"0202000400000072","DpType2Id2":114,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000072"}}}
16:29:00 TYA: fnId=21 is set for dpId=2
16:29:00 TYA: RX value 114 from dpId 2 
16:29:00 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:29:00 TYA: fnId=11 is set for dpId=1
16:29:00 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:29:00 {"TuyaReceived":{"Data":"55AA00070008020200040000001026","Cmnd":7,"CmndData":"0202000400000010","DpType2Id2":16,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000010"}}}
16:29:00 TYA: fnId=21 is set for dpId=2
16:29:00 TYA: RX value 16 from dpId 2 
16:29:00 SRC: Switch
16:29:00 CMD: Group 0, Index 3, Command "DIMMER", Data "16"
16:29:00 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF","Dimmer":16}
16:29:01 CFG: Saved to flash at FA, Count 1389, Bytes 4096
16:29:02 SRC: WebGui from 10.15.67.46
16:29:02 CMD: Group 0, Index 1, Command "DIMMER", Data "55"
16:29:02 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"OFF","Dimmer":55}
16:29:03 TYA: Send dim skipped value 55 for dpid 2
16:29:03 CFG: Saved to flash at F9, Count 1390, Bytes 4096
16:29:03 SRC: WebGui from 10.15.67.46
16:29:03 TYA: Send "55aa0006000501010001010e"
16:29:03 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON"}
16:29:03 MQT: stat/tasmota_D9E991/POWER = ON
16:29:03 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
16:29:03 TYA: fnId=11 is set for dpId=1
16:29:03 TYA: RX Relay-1 --> MCU State: On Current State:On
16:29:03 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
16:29:03 TYA: fnId=11 is set for dpId=1
16:29:03 TYA: RX Relay-1 --> MCU State: On Current State:On
16:29:03 {"TuyaReceived":{"Data":"55AA00070008020200040000001026","Cmnd":7,"CmndData":"0202000400000010","DpType2Id2":16,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000010"}}}
16:29:03 TYA: fnId=21 is set for dpId=2
16:29:03 TYA: RX value 16 from dpId 2 
16:29:03 SRC: Switch
16:29:03 CMD: Group 0, Index 3, Command "DIMMER", Data "16"
16:29:03 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON","Dimmer":16}
16:29:03 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
16:29:03 TYA: fnId=11 is set for dpId=1
16:29:03 TYA: RX Relay-1 --> MCU State: On Current State:On
16:29:03 {"TuyaReceived":{"Data":"55AA00070008020200040000007288","Cmnd":7,"CmndData":"0202000400000072","DpType2Id2":114,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000072"}}}
16:29:03 TYA: fnId=21 is set for dpId=2
16:29:03 TYA: RX value 114 from dpId 2 
16:29:03 SRC: Switch
16:29:03 CMD: Group 0, Index 3, Command "DIMMER", Data "100"
16:29:03 MQT: stat/tasmota_D9E991/RESULT = {"POWER":"ON","Dimmer":100}
16:29:04 CFG: Saved to flash at F8, Count 1391, Bytes 4096
16:29:05 WIF: Checking connection...
16:29:05 TYA: Send "55aa00000000ff"
16:29:05 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
16:29:05 TYA: Heartbeat
16:29:07 HTP: Console
```

jonootto, 2020-11-20:
> @jonootto, @TheReiner and @h4nc please issue in console the command `backlog TuyaMCU; TuyaSend8; so20; so54; so77; dimmerange; ledtable` with `Weblog 4` as debug level and paste here the result. Thanks.

```
20:22:12 CMD: Group 0, Index 1, Command "BACKLOG", Data "TuyaMCU; TuyaSend8; so20; so54; so77; dimmerange; ledtable"
20:22:12 SRC: Backlog
20:22:12 CMD: Group 0, Index 1, Command "TUYAMCU", Data ""
20:22:12 MQT: stat/bedroom1/RESULT = {"TuyaMCU":[{"fnId":11,"dpId":1},{"fnId":21,"dpId":2}]}
20:22:12 SRC: Backlog
20:22:12 CMD: Group 0, Index 8, Command "TUYASEND", Data ""
20:22:12 TYA: Read MCU state
20:22:12 TYA: Send "55aa0001000000"
20:22:12 MQT: stat/bedroom1/RESULT = {"TuyaSend":"Done"}
20:22:12 {"TuyaReceived":{"Data":"55AA0301002A7B2270223A226773726D6D736669683165766F796A66222C2276223A22312E302E30222C226D223A307D83","Cmnd":1,"CmndData":"7B2270223A226773726D6D736669683165766F796A66222C2276223A22312E302E30222C226D223A307D"}}
20:22:12 TYA: MCU Product ID: {"p":"gsrmmsfih1evoyjf","v":"1.0.0","m":0}
20:22:12 TYA: Send "55aa0002000001"
20:22:12 {"TuyaReceived":{"Data":"55AA0302000004","Cmnd":2}}
20:22:12 TYA: RX MCU configuration Mode=0
20:22:12 TYA: Read MCU state
20:22:12 TYA: Send "55aa0008000007"
20:22:12 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
20:22:12 TYA: fnId=11 is set for dpId=1
20:22:12 TYA: RX Relay-1 --> MCU State: Off Current State:Off
20:22:13 {"TuyaReceived":{"Data":"55AA0307000802020004000003E602","Cmnd":7,"CmndData":"02020004000003E6","DpType2Id2":998,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000003E6"}}}
20:22:13 TYA: fnId=21 is set for dpId=2
20:22:13 TYA: RX value 998 from dpId 2 
20:22:13 SRC: Switch
20:22:13 CMD: Group 0, Index 3, Command "DIMMER", Data "100"
20:22:13 MQT: tele/bedroom1/STATE = {"Time":"2020-11-20T20:22:13","Uptime":"1T10:19:53","UptimeSec":123593,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:22:13 MQT: stat/bedroom1/RESULT = {"POWER":"OFF","Dimmer":100}
20:22:13 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
20:22:13 TYA: fnId=0 is set for dpId=105
20:22:13 TYA: RX value 0 from dpId 105 
20:22:13 SRC: Backlog
20:22:13 CMD: Group 0, Index 20, Command "SO", Data ""
20:22:13 MQT: stat/bedroom1/RESULT = {"SetOption20":"ON"}
20:22:13 SRC: Backlog
20:22:13 CMD: Group 0, Index 54, Command "SO", Data ""
20:22:13 MQT: stat/bedroom1/RESULT = {"SetOption54":"ON"}
20:22:13 SRC: Backlog
20:22:13 CMD: Group 0, Index 77, Command "SO", Data ""
20:22:13 MQT: stat/bedroom1/RESULT = {"SetOption77":"OFF"}
20:22:13 SRC: Backlog
20:22:13 CMD: Group 0, Index 1, Command "DIMMERANGE", Data ""
20:22:13 MQT: stat/bedroom1/RESULT = {"Command":"Unknown"}
20:22:14 SRC: Backlog
20:22:14 CMD: Group 0, Index 1, Command "LEDTABLE", Data ""
20:22:14 MQT: stat/bedroom1/RESULT = {"LedTable":"ON"}
20:22:15 CFG: Saved to flash at F8, Count 373, Bytes 4096
20:22:15 TYA: Send "55aa00000000ff"
20:22:15 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
20:22:15 TYA: Heartbeat
20:22:21 MQT: tele/bedroom1/HASS_STATE = {"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-11-18T20:29:28","Module or Template":"Tuya MCU","RestartReason":"Software/System restart","Uptime":"1T10:20:01","Hostname":"bedroom1-1782","IPAddress":"192.168.30.32","RSSI":"66","Signal (dBm)":"-67","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:03","MqttCount":1,"LoadAvg":19}
```

jonootto, 2020-11-20:
I'll also add, though i don't know if it should be a separate issue, that sending a dimmer command while the light is off (with the above options enabled) turns the light to 100% before dimming.

```

20:36:43 CMD: dimmer 20
20:36:43 SRC: WebConsole from 192.168.1.68
20:36:43 CMD: Group 0, Index 1, Command "DIMMER", Data "20"
20:36:43 MQT: tele/bedroom1/STATE = {"Time":"2020-11-20T20:36:43","Uptime":"1T10:34:23","UptimeSec":124463,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":20,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:36:43 MQT: stat/bedroom1/RESULT = {"POWER":"OFF","Dimmer":20}
20:36:43 CFG: Saved to flash at F7, Count 406, Bytes 4096
20:36:44 TYA: Send "55aa0006000802020004000000c8dd"
20:36:44 TYA: Send "55aa00000000ff"
20:36:44 {"TuyaReceived":{"Data":"55AA0307000802020004000000C8E1","Cmnd":7,"CmndData":"02020004000000C8","DpType2Id2":200,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C8"}}}
20:36:44 TYA: fnId=21 is set for dpId=2
20:36:44 TYA: RX value 200 from dpId 2 
20:36:44 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
20:36:44 TYA: Heartbeat
20:36:44 WIF: Checking connection...
20:36:55 TYA: Send "55aa00000000ff"
20:36:55 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
20:36:55 TYA: Heartbeat
```

effelle, 2020-11-20:
> I'll also add, though i don't know if it should be a separate issue, that sending a dimmer command while the light is off (with the above options enabled) turns the light to 100% before dimming.

As far as I can see Tasmota get just one message for the dimmer ( `RX value 200 from dpId 2` ) and in between nothing more. If you turn off the light and send a similar dimmer command, let's say 25, it still revert to 100% before applying the correct value?  
I can't reproduce it, but I'll put some logs on light.ino to see if I can see some spikes.  
 
> Also, after this happens and the light is on, the gui and console still shows it as off, so tasmota must be passing the new dimmer value to the MCU, and the MCU is deciding to turn on?

Good question, and that concern me a bit. If you have both options enabled the relay is completely separated by the dimmer, My fear is your device send a set of information (would be not the first time, happens a lot with rgcw lights). If is the case the device will change state, but Tasmota will not know it until the MCU send out the next update. 
I'll prepare a special build as soon as possible for you to test.
I tested some devices (from different persons) and just on one we found a similar issue, solved completely enabling so20 and 54. 
What device and model are you using? I would like to check the tuya dev board to see if I can gather more information. 
Do you have any idea what is linked to dpid 105? It popped more than once on your logs and there is a chance is used by the physical button/slider on the device (if is a wall switch for example).

jonootto, 2020-11-20:
> As far as I can see Tasmota get just one message for the dimmer ( `RX value 200 from dpId 2` ) and in between nothing more. If you turn off the light and send a similar dimmer command, let's say 25, it still revert to 100% before applying the correct value?

I've just tested this a number of times in a row, it seems slightly inconsistent. I start with the power off and dimmer set to 0, then send a dimmer <x> command. Sometimes nothing happens, sometimes it turns the light on to the set dimmer value (but the power in tasmota remains off). Very occasionally it goes to full brightness then dims down - but this seems to be really uncommon now, though it was consistent earlier. It does happen consistently when i trigger it through MQTT by home assistant though. 

Here's what happens when i try control the light through homeassistant MQTT (the first dimmer command got ignored too):
```
23:04:34 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
23:04:34 TYA: fnId=11 is set for dpId=1
23:04:34 TYA: RX Relay-1 --> MCU State: Off Current State:Off
23:04:34 {"TuyaReceived":{"Data":"55AA03070008020200040000000019","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
23:04:34 TYA: fnId=21 is set for dpId=2
23:04:34 TYA: RX value 0 from dpId 2 
23:04:34 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
23:04:34 TYA: fnId=0 is set for dpId=105
23:04:34 TYA: RX value 0 from dpId 105 
23:04:34 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
23:04:34 TYA: fnId=11 is set for dpId=1
23:04:34 TYA: RX Relay-1 --> MCU State: Off Current State:Off
23:04:35 MQT: Data Size 2, "cmnd/bedroom1/POWER ON"
23:04:35 SRC: MQTT
23:04:35 CMD: Group 0, Index 1, Command "POWER", Data "ON"
23:04:35 TYA: Send "55aa0006000501010001010e"
23:04:35 MQT: tele/bedroom1/STATE = {"Time":"2020-11-20T23:04:35","Uptime":"0T00:00:52","UptimeSec":52,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":0,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:04:35 MQT: stat/bedroom1/RESULT = {"POWER":"ON"}
23:04:35 MQT: stat/bedroom1/POWER = ON
23:04:35 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
23:04:35 TYA: fnId=11 is set for dpId=1
23:04:35 TYA: RX Relay-1 --> MCU State: On Current State:On
23:04:35 {"TuyaReceived":{"Data":"55AA0307000802020004000003E804","Cmnd":7,"CmndData":"02020004000003E8","DpType2Id2":1000,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000003E8"}}}
23:04:35 TYA: fnId=21 is set for dpId=2
23:04:35 TYA: RX value 1000 from dpId 2 
23:04:35 SRC: Switch
23:04:35 CMD: Group 0, Index 3, Command "DIMMER", Data "100"
23:04:35 MQT: tele/bedroom1/STATE = {"Time":"2020-11-20T23:04:35","Uptime":"0T00:00:52","UptimeSec":52,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":27,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:04:35 MQT: stat/bedroom1/RESULT = {"POWER":"ON","Dimmer":100}
23:04:35 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
23:04:35 TYA: fnId=0 is set for dpId=105
23:04:35 TYA: RX value 0 from dpId 105 
23:04:35 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
23:04:35 TYA: fnId=11 is set for dpId=1
23:04:35 TYA: RX Relay-1 --> MCU State: On Current State:On
23:04:35 CFG: Saved to flash at F4, Count 537, Bytes 4096
23:04:38 TYA: Send "55aa00000000ff"
23:04:38 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
23:04:38 TYA: Heartbeat
23:04:38 MQT: Data Size 2, "cmnd/bedroom1/Dimmer 38"
23:04:38 SRC: MQTT
23:04:38 CMD: Group 0, Index 1, Command "DIMMER", Data "38"
23:04:38 MQT: tele/bedroom1/STATE = {"Time":"2020-11-20T23:04:38","Uptime":"0T00:00:55","UptimeSec":55,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":47,"MqttCount":1,"POWER":"ON","Dimmer":38,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:04:38 MQT: stat/bedroom1/RESULT = {"POWER":"ON","Dimmer":38}
23:04:38 MQT: Data Size 2, "cmnd/bedroom1/POWER ON"
23:04:38 SRC: MQTT
23:04:38 CMD: Group 0, Index 1, Command "POWER", Data "ON"
23:04:38 TYA: Send "55aa0006000501010001010e"
23:04:38 MQT: tele/bedroom1/STATE = {"Time":"2020-11-20T23:04:38","Uptime":"0T00:00:55","UptimeSec":55,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":50,"MqttCount":1,"POWER":"ON","Dimmer":38,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:04:38 MQT: stat/bedroom1/RESULT = {"POWER":"ON"}
[... cut ...]
>What device and model are you using? I would like to check the tuya dev board to see if I can gather more information.
Do you have any idea what is linked to dpid 105? It popped more than once on your logs and there is a chance is used by the physical button/slider on the device (if is a wall switch for example).

105 is a timer to turn off in seconds

Its a [BrilliantSmart Jupiter Dimmer](https://www.brilliantsmart.com.au/smart-products/electrical/smart-jupiter-2/)
[Template](https://templates.blakadder.com/brilliantsmart_20693.html)

effelle, 2020-11-20:
Puzzling, but I suppose there a first time for everything. Let me think on it a bit. 
Are you using the mqtt discovery or the official integration? If is the former please give a spin to the integration, we have a better flow for dimmable light there. It will not solve your Tuya issue anyway but perhaps HAss will be more efficient.

jonootto, 2020-11-20:
> Puzzling, but I suppose there a first time for everything. Let me think on it a bit.
> Are you using the mqtt discovery or the official integration? If is the former please give a spin to the [...]

Switching over now, the new integration looks great!

jonootto, 2020-11-20:
Spent a bit more time testing the turn on characteristics.

I'm turning it on when dimmer= 0, which is because that's the only state where i can turn it off. It makes sense that turning it on from 0 would set it to 100 i think. The next dimmer value i send through the console is ignored however, and even sending the same value multiple times is ignored. for example:

1. Light is off and dimmer=0

2.

effelle, 2020-11-20:
> Switching over now, the new integration looks great!

Thanks, we are working a lot in it!
Ok, I think I have enough data to try to fix it. I'll drop you a line if some tests will be needed.

effelle, 2020-11-21:
@jonootto  If the device buton controls the relay do a test please.
Disable MQTT in Tasmota (Configuration --> configure Other), remove the dimmer from TuyaMCU configuration and assign the fnId 11 to a dummy dpId ( tuyamcu 11,99 ).
Now set the weblog to 4 and press the button to turn off the relay. Paste here the log.
After that turn on the relay and from Tasmota send a `tuyasend1 1,0` and paste also this relevant log.
Thanks.

MrT76, 2020-11-21:
Hello, I'm new to the forum and don't have a lot of experience. I also did a few attempts with the Tuya mcu dimmer today because I have the problem myself. I don't know if it helps, but my conclusion is that the control via the slider works perfectly. This means that when I move the slider to OFF the light stays off, when I switch the switch to on, it returns to the last light intensity.

MrT76, 2020-11-21:
15:05:25 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "0"
15:05:25 SRC: Light
15:05:25 TYA: Send "55aa0006000501010001000d"
15:05:25 MQT: stat/tasmota/RESULT = {"POWER":"OFF","Dimmer":0}
15:05:25 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:05:25 TYA: fnId=11 is set for dpId=1
15:05:25 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:05:25 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:05:25 TYA: fnId=11 is set for dpId=1
15:05:25 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:05:25 {"TuyaReceived":{"Data":"55AA00070008020200040000008CA2","Cmnd":7,"CmndData":"020200040000008C","DpType2Id2":140,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000008C"}}}
15:05:25 TYA: fnId=21 is set for dpId=2
15:05:25 TYA: RX value 140 from dpId 2 
15:05:25 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:05:25 TYA: fnId=11 is set for dpId=1
15:05:25 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:05:25 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
15:05:25 TYA: fnId=21 is set for dpId=2
15:05:25 TYA: RX value 0 from dpId 2 
15:05:26 CFG: in Flash gespeichert am F4, zählen 5747, Bytes 4096
15:05:34 TYA: Send "55aa00000000ff"
15:05:34 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
15:05:34 TYA: Heartbeat
15:05:35 HTP: Befehl
15:09:38 TYA: fnId=21 is set for dpId=2
15:09:38 TYA: RX value 0 from dpId 2 
15:09:39 CFG: in Flash gespeichert am FA, zählen 5749, Bytes 4096
15:09:40 WIF: Prüfe Verbindung...
15:09:44 HTP: Konsole
15:09:47 TYA: Send "55aa00000000ff"
15:09:47 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
15:09:47 TYA: Heartbeat
15:09:49 HTP: Hauptmenü
15:09:53 SRC: WebGui from 192.168.xxx.xx
15:09:53 TYA: Send "55aa0006000501010001010e"
15:09:53 MQT: stat/tasmota/RESULT = {"POWER":"ON"}
15:09:53 MQT: stat/tasmota/POWER = ON
15:09:53 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
15:09:53 TYA: fnId=11 is set for dpId=1
15:09:53 TYA: RX Relay-1 --> MCU State: On Current State:On
15:09:53 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
15:09:53 TYA: fnId=11 is set for dpId=1
15:09:53 TYA: RX Relay-1 --> MCU State: On Current State:On
15:09:53 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
15:09:53 TYA: fnId=21 is set for dpId=2
15:09:53 TYA: RX value 0 from dpId 2 
15:09:53 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
15:09:53 TYA: fnId=11 is set for dpId=1
15:09:53 TYA: RX Relay-1 --> MCU State: On Current State:On
15:09:53 {"TuyaReceived":{"Data":"55AA00070008020200040000008CA2","Cmnd":7,"CmndData":"020200040000008C","DpType2Id2":140,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000008C"}}}
15:09:53 TYA: fnId=21 is set for dpId=2
15:09:53 TYA: RX value 140 from dpId 2 
15:09:53 SRC: Switch
15:09:53 CMD: Gruppe 0, Index 3, Befehl "DIMMER", Daten "64"
15:09:53 MQT: stat/tasmota/RESULT = {"POWER":"ON","Dimmer":64}
15:09:54 CFG: in Flash gespeichert am F9, zählen 5750, Bytes 4096
15:09:55 HTP: Hauptmenü
15:09:57 SRC: WebGui from 192.168.xxx.xx
15:09:57 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "0"
15:09:57 SRC: Light
15:09:58 TYA: Send "55aa0006000501010001000d"
15:09:58 MQT: stat/tasmota/RESULT = {"POWER":"OFF","Dimmer":0}
15:09:58 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:09:58 TYA: fnId=11 is set for dpId=1
15:09:58 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:09:58 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:09:58 TYA: fnId=11 is set for dpId=1
15:09:58 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:09:58 {"TuyaReceived":{"Data":"55AA00070008020200040000008CA2","Cmnd":7,"CmndData":"020200040000008C","DpType2Id2":140,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000008C"}}}
15:09:58 TYA: fnId=21 is set for dpId=2
15:09:58 TYA: RX value 140 from dpId 2 
15:09:58 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
15:09:58 TYA: fnId=11 is set for dpId=1
15:09:58 TYA: RX Relay-1 --> MCU State: Off Current State:Off
15:09:58 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
15:09:58 TYA: fnId=21 is set for dpId=2
15:09:58 TYA: RX value 0 from dpId 2 
15:09:58 CFG: in Flash gespeichert am F8, zählen 5751, Bytes 4096
15:09:58 TYA: Send "55aa00000000ff"
15:09:58 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
15:09:58 TYA: Heartbeat
15:10:00 WIF: Prüfe Verbindung...
15:10:01 HTP: Hauptmenü
15:10:03 HTP: Konsole

MrT76, 2020-11-21:
I can't explain where the 64% are stored, when I set the slider to OFF it sends dimmer 0 and when I switch it to on, the last value "64 in this example" comes back for the dimmer

MrT76, 2020-11-21:
I noticed something else.
If I switch the light on and off without changing the dimm level via the slider in the web browser, it switches on again automatically. However, if I change the dimm level while it is switched on, I can switch it off using the button on the web without it automatically switching on again.

MrT76, 2020-11-21:
18:58:38 TYA: RX value 70 from dpId 2 
18:58:38 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
18:58:38 TYA: fnId=11 is set for dpId=1
18:58:38 TYA: RX Relay-1 --> MCU State: Off Current State:Off
18:58:38 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
18:58:38 TYA: fnId=21 is set for dpId=2
18:58:38 TYA: RX value 0 from dpId 2 
18:58:39 CFG: in Flash gespeichert am FA, zählen 5821, Bytes 4096
18:58:39 TYA: Send "55aa0006000802020004000000465b"
18:58:39 {"TuyaReceived":{"Data":"55AA0007000802020004000000465C","Cmnd":7,"CmndData":"0202000400000046","DpType2Id2":70,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000046"}}}
18:58:39 TYA: fnId=21 is set for dpId=2
18:58:39 TYA: RX value 70 from dpId 2 
18:58:39 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
18:58:39 TYA: fnId=11 is set for dpId=1
18:58:39 TYA: RX Relay-1 --> MCU State: On Current State:Off
18:58:39 SRC: Switch
18:58:39 MQT: stat/tasmota/RESULT = {"POWER":"ON"}
18:58:39 MQT: stat/tasmota/POWER = ON
18:58:39 {"TuyaReceived":{"Data":"55AA0007000802020004000000465C","Cmnd":7,"CmndData":"0202000400000046","DpType2Id2":70,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000046"}}}
18:58:39 TYA: fnId=21 is set for dpId=2
18:58:39 TYA: RX value 70 from dpId 2 
18:58:40 CFG: in Flash gespeichert am F9, zählen 5822, Bytes 4096
18:58:41 WIF: Prüfe Verbindung...
18:58:44 SRC: WebGui from 192.168.xxx.xx
18:58:44 CMD: Gruppe 0, Index 1, Befehl "DIMMER", Daten "16"
18:58:44 MQT: stat/tasmota/RESULT = {"POWER":"ON","Dimmer":16}
18:58:44 CFG: in Flash gespeichert am F8, zählen 5823, Bytes 4096
18:58:44 TYA: Send "55aa0006000802020004000000465b"
18:58:44 {"TuyaReceived":{"Data":"55AA0007000802020004000000465C","Cmnd":7,"CmndData":"0202000400000046","DpType2Id2":70,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000046"}}}
18:58:44 TYA: fnId=21 is set for dpId=2
18:58:44 TYA: RX value 70 from dpId 2 
18:58:44 SRC: WebGui from 192.168.xxx.xx
18:58:44 TYA: Send "55aa0006000501010001000d"
18:58:44 MQT: stat/tasmota/RESULT = {"POWER":"OFF"}
18:58:44 MQT: stat/tasmota/POWER = OFF
18:58:44 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
18:58:44 TYA: fnId=11 is set for dpId=1
18:58:44 TYA: RX Relay-1 --> MCU State: Off Current State:Off
18:58:44 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
18:58:44 TYA: fnId=11 is set for dpId=1
18:58:44 TYA: RX Relay-1 --> MCU State: Off Current State:Off
18:58:44 {"TuyaReceived":{"Data":"55AA0007000802020004000000465C","Cmnd":7,"CmndData":"0202000400000046","DpType2Id2":70,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000046"}}}
18:58:44 TYA: fnId=21 is set for dpId=2
18:58:44 TYA: RX value 70 from dpId 2 
18:58:44 SRC: Switch
18:58:44 CMD: Gruppe 0, Index 3, Befehl "DIMMER", Daten "32"
18:58:44 MQT: stat/tasmota/RESULT = {"POWER":"OFF","Dimmer":32}
18:58:44 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
18:58:44 TYA: fnId=11 is set for dpId=1
18:58:44 TYA: RX Relay-1 --> MCU State: Off Current State:Off
18:58:44 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
18:58:44 TYA: fnId=21 is set for dpId=2
18:58:44 TYA: RX value 0 from dpId 2 
18:58:45 CFG: in Flash gespeichert am F7, zählen 5824, Bytes 4096
18:58:45 TYA: Send dim skipped value 70 for dpid 2
18:58:47 HTP: Konsole

MrT76, 2020-11-21:
In summary, I would say that off only works properly after changing the dimming level.

MrT76, 2020-11-21:
I changed the button type for GPIO0 to type 2 so that it works fine.

MrT76, 2020-11-21:
Changing the button type from 1 to 2 temporarily resulted in that I could switch several times without errors. After that, switching off only works if I change the dimming level in the switched-on state.

MrT76, 2020-11-22:
I can currently recommend the following configuration,

stat/tasmota/RESULT = {"ButtonTopic":""}
stat/tasmota/RESULT = {"SetOption1":"ON"}
stat/tasmota/RESULT = {"SetOption20":"ON"}
stat/tasmota/RESULT = {"SetOption54":"ON"}

 only the command "power off" doesn´t work fine, which i have temporarily replaced with "dimmer 0"

effelle, 2020-11-22:
>  I changed the button type for GPIO0 to type 2 so that it works better but not perfect

If is a MCU you are just connected to tx/rx. have a button connected to GPIO0 means is controlled by Tasmota directly. Care to expand it a bit?

For all interested here it is a experimental build with a more verbose log to try to catch a possible cause (remembering I can' reproduce the issue here). 
If you are not confortable flashing a device or if you can't reflash it easily trough serial in case something goes south please don't test it. I take no responsibility.
Is meant to be tested with so20/54 on AND off, logs must be pasted here with weblog level set to 4.

[tasmota.bin.gz](https://github.com/arendst/Tasmota/files/5579688/tasmota.bin.gz)

jonootto, 2020-11-23:
> Is meant to be tested with so20/54 on AND off, logs must be pasted here with weblog level set to 4.
Something seems to be fixed? I can now issue power 0 and it turns off.

both 54 and 20 on, all the commands work fine:
```python
16:19:51 CMD: power 0
16:19:51 SRC: WebConsole from 192.168.1.68
16:19:51 CMD: Group 0, Index 1, Command "POWER", Data "0"
16:19:51 TYA: Send "55aa0006000501010001000d"
16:19:51 MQT: tele/bedroom1/STATE = {"Time":"2020-11-23T16:19:51","Uptime":"0T00:00:09","UptimeSec":9,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":20,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:19:51 MQT: stat/bedroom1/RESULT = {"POWER":"OFF"}
16:19:51 MQT: stat/bedroom1/POWER = OFF
16:19:51 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:19:51 TYA: fnId=11 is set for dpId=1
16:19:51 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:19:51 {"TuyaReceived":{"Data":"55AA03070008020200040000000019","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
16:19:51 TYA: fnId=21 is set for dpId=2
16:19:51 TYA: RX value 0 from dpId 2 
16:19:51 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
16:19:51 TYA: fnId=0 is set for dpId=105
16:19:51 TYA: RX value 0 from dpId 105 
16:19:51 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:19:51 TYA: fnId=11 is set for dpId=1
16:19:51 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:19:52 APP: Boot Count 24
16:19:52 MQT: tele/bedroom1/STATE = {"Time":"2020-11-23T16:19:52","Uptime":"0T00:00:10","UptimeSec":10,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER":"OFF","Dimmer":20,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:19:52 CFG: Saved to flash at FA, Count 647, Bytes 4096
16:19:53 TYA: Send "55aa00000000ff"
16:19:53 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
16:19:53 TYA: Heartbeat
16:19:57 CMD: power 1
16:19:57 SRC: WebConsole from 192.168.1.68
16:19:57 CMD: Group 0, Index 1, Command "POWER", Data "1"
16:19:57 TYA: TuyaSetChannels: duty 20 hex 000000000000 index 1
16:19:57 TYA: Send "55aa0006000802020004000000c8dd"
16:19:57 TYA: Send "55aa0006000501010001010e"
16:19:57 MQT: tele/bedroom1/STATE = {"Time":"2020-11-23T16:19:57","Uptime":"0T00:00:15","UptimeSec":15,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":20,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:19:57 MQT: stat/bedroom1/RESULT = {"POWER":"ON"}
16:19:57 MQT: stat/bedroom1/POWER = ON
16:19:57 {"TuyaReceived":{"Data":"55AA0307000802020004000000C8E1","Cmnd":7,"CmndData":"02020004000000C8","DpType2Id2":200,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000C8"}}}
16:19:57 TYA: fnId=21 is set for dpId=2
16:19:57 TYA: RX value 200 from dpId 2 
16:19:57 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
16:19:57 TYA: fnId=11 is set for dpId=1
16:19:57 TYA: RX Relay-1 --> MCU State: On Current State:On
16:19:59 CMD: dimmer 60
16:19:59 SRC: WebConsole from 192.168.1.68
16:19:59 CMD: Group 0, Index 1, Command "DIMMER", Data "60"
16:19:59 MQT: tele/bedroom1/STATE = {"Time":"2020-11-23T16:19:59","Uptime":"0T00:00:17","UptimeSec":17,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":30,"MqttCount":1,"POWER":"ON","Dimmer":60,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:19:59 MQT: stat/bedroom1/RESULT = {"POWER":"ON","Dimmer":60}
16:19:59 TYA: TuyaSetChannels: duty 60 hex 000000000000 index 1
16:19:59 TYA: Send "55aa0006000802020004000002586f"
16:19:59 {"TuyaReceived":{"Data":"55AA03070008020200040000025873","Cmnd":7,"CmndData":"0202000400000258","DpType2Id2":600,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000258"}}}
16:19:59 TYA: fnId=21 is set for dpId=2
16:19:59 TYA: RX value 600 from dpId 2 
16:20:00 CFG: Saved to flash at F9, Count 648, Bytes 4096
16:20:02 CMD: power 0
16:20:02 SRC: WebConsole from 192.168.1.68
16:20:02 CMD: Group 0, Index 1, Command "POWER", Data "0"
16:20:02 TYA: Send "55aa0006000501010001000d"
16:20:02 MQT: tele/bedroom1/STATE = {"Time":"2020-11-23T16:20:02","Uptime":"0T00:00:20","UptimeSec":20,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":47,"MqttCount":1,"POWER":"OFF","Dimmer":60,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"C6:FB:E4:74:42:B7","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:20:02 MQT: stat/bedroom1/RESULT = {"POWER":"OFF"}
16:20:02 MQT: stat/bedroom1/POWER = OFF
[... cut ...]
16:22:03 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
16:22:03 TYA: fnId=11 is set for dpId=1
16:22:03 TYA: RX Relay-1 --> MCU State: Off Current State:Off
16:22:13 TYA: Send "55aa00000000ff"
16:22:13 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
16:22:13 TYA: Heartbeat
16:22:22 WIF: Checking connection...
16:22:24 TYA: Send "55aa00000000ff"
16:22:24 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
16:22:24 TYA: Heartbeat
```

jonootto, 2020-11-23:
This has also fixed my HA dimmer issues, so from my perspective if this fix is integrated, I'd be happy to close it.

effelle, 2020-11-23:
> Something seems to be fixed? I can now issue power 0 and it turns off

Yes, and that confirms my suspects. Since your device send immediately the state of the dimmer / timer after a power off I added a grace period of 250ms that will force Tasmota to ignore the packets on power off state. 
I'm not sure is the best solution but for now is what I found.
TuyaMCU is a very complicated thing, there is not a default behaviour for serial packets transmission and the manufacturer has a lot of freedom. Sometimes too much! 
This forces me to play cat and mouse every now and then.  
The fix will be added on next update as soon as I'll finish the sensors for Tuya. Once merged this issue will be closed automatically.

jonootto, 2020-11-26:
I've observed that enabling fade causes it to go back to bad behaviour

effelle, 2020-11-26:
Fade is not supported for TuyaMCU, is why we have a specific dimmer command for it. 
If the fade was sent by HA integration consider it as a bug, because we'll remove it .

closed 2020-11-28 by arendst

ptaudio, 2021-11-16:
still not fixed in 9.5