arendst/Tasmota #9870: Can we have native auto discovery in Zigbee2Tasmota?

lukejagger, 2020-11-15:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Can we have the native auto discovery feature that exists in the standard Tasmota, in the TasmotaZBBridge / Zigbee2Tasmota so that new zigbee devices can be auto-discovered.
This would allow for instance, the new native discovery feature in Home Assistant to discover these Tasmota/MQTT devices as they are added, rather than requiring YAML to be added for each device.

**Describe alternatives you've considered**  
The ZHA implementation is the alternative, but you can only have a single ZHA ZBBridge in HA, so it's not possible to have multiple outbuildings (in my case) each with its own ZBBridge.

**Additional context**  
N/A

**(Please, remember to close the issue when the problem has been addressed)**

lukejagger, 2020-11-15:
Reference: **https://www.home-assistant.io/integrations/tasmota/**

Jason2866, 2020-11-15:
There is no Autodiscovery planned for Zigbee.

lukejagger, 2020-11-15:
Can I ask why, as this would make this a much more accessible solution for users.

lukejagger, 2020-11-15:
It's this item from the above link that I'm referring to, just to clarify:
'Tasmota devices configured for native discovery (SetOption19 0)'

Jason2866, 2020-11-15:
Because it is highly complicated to do and would use a lot of flash space.
Better to ask HA developer to do a better ZHA integration.
The developer of zigbee2Tasmota has no knowledge of HomeAssistant.
If "someone" find a clever and small way to implement a PR is welcome

lukejagger, 2020-11-15:
A solution using either setoption 19 0  or  setoption 19 1  would be acceptable.  The original Tasmota firmware can do this already, but it needs implementing for the Tasmota side of Z2T as well I think.

Jason2866, 2020-11-15:
I know what SetOption19 does. It is NOT just a little change of this part. Zigbee is **completly** different to implement as i wrote already.
To be clear will not be done from the developers which are currently contributing to Tasmota

lukejagger, 2020-11-15:
Thanks Jason, I wasn't disputing your knowledge of the options available :) just making sure the feature request was complete.  Are you the developer of this component?

Jason2866, 2020-11-15:
I did not take it as offend 😀 s-hadinger is the one who does all! the zigbee for Tasmota

emontnemery, 2020-11-15:
There are multiple existing zigbee solutions for HA: deconz, zha, zigbee2mqtt. If Tasmota already works with zha but there are limitations in zha, it would be better to improve zha IMHO.

effelle, 2020-11-15:
I agree with @Jason2866 and @emontnemery and from my personal point of view it's not logic add Zigbee to the discovery when there are already other integrations related to it under Home Assistant.
Anyway, at least some Zigbee related entities can already be grabbed by the discovery: the sensors.
Just enable `SetOption89` (and `SetOption83` if you want a friendly name instead of a 16bit short address id) to have them discovered. 
Closing this issue as it has been answered.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-15 by effelle