arendst/Tasmota #9871: New Tuya protocol possibly?

sanchosk, 2020-11-15:
### PROBLEM DESCRIPTION
After manually flashing tuya-based electric socket switch, I cannot make the device work properly. I can manipulate the relays, but the buttons and wifi LED seem to be driven by onboard chip connected to the ESP8266 using something resembling serial line and I cannot find out what that is

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): WiFi Smart Power Strip Intelligent EU Plug Electrical USB Sockets Wireless Timer Remote Independent Control by Google Home Alexa, aliexpress
- [X] Tasmota binary firmware version number used: 9.1.0(tasmota)
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:27:51 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
12:27:51 SRC: Backlog
12:27:51 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
12:27:51 RSL: stat/tasmota_6696DE/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:27:52 SRC: Backlog
12:27:52 CMD: Group 0, Index 1, Command "MODULE", Data ""
12:27:52 RSL: stat/tasmota_6696DE/RESULT = {"Module":{"54":"Tuya MCU"}}
12:27:52 SRC: Backlog
12:27:52 CMD: Group 0, Index 1, Command "GPIO", Data "255"
12:27:52 RSL: stat/tasmota_6696DE/RESULT = {"GPIO0":{"2304":"Tuya Rx"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"2272":"Tuya Tx"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:29:28 SRC: WebConsole from 192.168.11.54
12:29:28 CMD: Group 0, Index 1, Command "STATUS", Data "0"
12:29:28 RSL: stat/tasmota_6696DE/STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_6696DE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:11:35","StartupUTC":"2020-11-15T11:17:53","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-11-15T12:15:52","SaveCount":17,"SaveAddress":"FB000"}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"355/699"}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wlan_home2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS5 = {"StatusNET":{"Hostname":"smartStrip","IPAddress":"192.168.21.182","Gateway":"192.168.21.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.21.254","Mac":"50:02:91:66:96:DE","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6696DE","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS7 = {"StatusTIM":{"UTC":"2020-11-15T11:29:28","Local":"2020-11-15T12:29:28","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:00","Sunset":"17:09"}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS10 = {"StatusSNS":{"Time":"2020-11-15T12:29:28"}}
12:29:28 RSL: stat/tasmota_6696DE/STATUS11 = {"StatusSTS":{"Time":"2020-11-15T12:29:28","Uptime":"0T00:11:35","UptimeSec":695,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXX","BSSId":"XXXXXXXXX","Channel":11,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash the socket using USB2Serial adapter, boot, change the template to Tuya and modify the pin layout to GPIO0 and GPIO14.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
After sending the 0x55 0xaa init, there will be some sort of reply.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

It seems from TuyaConvert forum that Tuya modified their serial protocol.
I tried to connect using 2 usb2serial converters and set them to 115200bps and 9600bps baud rate, but I cannot seem to grab any reasonable outputs.
Can anyone suggest how to proceed further?

Jason2866, 2020-11-15:
All we know is in the docs. https://tasmota.github.io/docs/TuyaMCU/
Since you are the only one who has the device we cant help.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

effelle, 2020-11-15:
We don't have access to buttons and leds driven by a tuyamcu. Not withour re-flashing the mcu chip and that is out the scope of Tasmota. 
We support both 9600 and 115200bps TuyaMCU. Use `SetOption97` to switch between the two speeds.

closed 2020-11-15 by ascillato2