arendst/Tasmota #9884: Trinamic motor driver support

lordmundi, 2020-11-16:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
I'd like to use the newer trinamic stepper motor drivers with Tasmota.  Not really a problem with the existing code.

**Describe the solution you'd like**  
I'd like to see support added to drive a stepper motor using the super silent Trinamic motor drivers such as the TMC2226

**Describe alternatives you've considered**  
There is an A4988 stepper motor driver available, but it won't be as quiet as the modes supported by Trinamic

**Additional context**  
Lots of projects are moving to the Trinamic stepper motor drivers because of the drastically reduced noise and increased efficiency (example: https://www.youtube.com/watch?v=MhK4cp_ZRAU&feature=youtu.be&t=301). I think this could open up Tasmota to be used in many other projects where a stepper motor is involved, such as driving a silent pan/tilt mount for a security camera, or a device where you need to not scare or alert a pet, etc.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-16:
Adding missing data for TMC2226.

[Official Datasheet: TMC2226_Datasheet_V106.pdf](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2226_Datasheet_V106.pdf)
[TMC2226_Calculations.xlsx](https://github.com/MarlinFirmware/Marlin/files/4877390/TMC2226_Calculations.xlsx)
[Arduino's driver](http://downloads.arduino.cc/libraries/github.com/teemuatlut/TMCStepper-0.7.1.zip)
[Github code examples of teemuatlut](https://github.com/teemuatlut/TMCStepper)

Adding the Feature Request Label to see if there is someone interested on working on this.

TrDA-hab, 2020-12-11:
@lordmundi 

I think support for the TMC2226 is not required, just because they are super quiet - the TMC2208 is super quiet too.
Instead of the A4988, I'm just using the TMC2208 for now, and they are very quiet. Here is my wiring diagram.

[image]

closed 2020-12-19 by github-actions[bot]