arendst/Tasmota #9890: BACnet option welcome

Langstein, 2020-11-16:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Can you add a connection to support BACnet (ISO 16484-5) on Tasmota devices. BACnet is a powerful iot protocol to connect devices. There is a project to create BACnet devices. Take a look at: http://bacnet.sourceforge.net/ and also https://github.com/dasrue/esp8266-BACnet. I have some experiance in BACnet an can support the project in that part. But I'm not qualified enough to write it by my own. You can use https://sourceforge.net/projects/yetanotherbacnetexplorer/ as a easy to use BACnet client.

So, if you add a minimal BACnet Server connected to a example sensor I can try to add more BACnet Objects an functionalities like alarms, scheduling, trending and control.

Hopefully it works.

Jason2866, 2020-11-16:
Tasmota uses mqtt for communication. Like many many IoT devices.
Tasmota supports already a lot of sensors. What would be the benefit?

Langstein, 2020-11-17:
The benefit is to use these devices not only in home automation systems, you can use it in building automation as well. The BACnet Standard is use from more than 1200 vendors (http://www.bacnet.org/VendorID/BACnet%20Vendor%20IDs.htm). Most of them only support complex control systems. The tasmota-devices can fill a gap of small sensors and actors. BACnet is a ISO BMS Standard since 2003 https://en.wikipedia.org/wiki/BACnet.

The handling to install tasmota devices is user friendly. To do an integration of MQTT devices in huge amounts is a lot of manual job. BACnet supports self explaining mechanisms (Who-is/I'am and so on).  For sure we can use tasmota devices by using an MQTT to BACnet gateway, but this is one more unneeded source of error inside a Building management system. The ability to support BACnet direktly is much more efficient. So that is the reason I'm hoping tasmota will support in future.

closed 2020-12-18 by github-actions[bot]

RaphaelSquair, 2024-05-27:
Hi I recently had a demand to implement this feature, anyone would be interested in joining me in co-develpment? and @Jason2866 could you give me support if I'd Tried to integrate this protocol in the smart metering feature on the scripting section?

barbudor, 2024-05-27:
@RaphaelSquair it's not clear
From the above discussions it looks like BACnet was considered as an alternative to MQTT : the protocole to connect a Tasmotized device to a backend (in addition to MQTT, Tasmota for example also support KNX)

If you are refering to Smart Meter in scripting, do you mean using BACnet on Tasmota to receive data from a BACnet device and send it over MQTT, something similar to existing support for MODBUS ?

Note that for anything regarding Scripting, you'd better talk to @gemu2015 who is Scripting's father

Jason2866, 2024-05-27:
Sorry, probably everyone has more knowledge in scripter code than me ;-)

RaphaelSquair, 2024-05-27:
Hi @barbudor  yes this is exactly what I want to do!! 

You are right I should talk with @gemu2015  about this,

barbudor, 2024-05-27:
@RaphaelSquair In htat case, I suggest openning a new discussion making it clear of a different objective that this thread
Thanks