arendst/Tasmota #9894: DevGroupShare not working as expected

lullerboi, 2020-11-17:
### PROBLEM DESCRIPTION

5 Devices: 3 in Living Room, 1 in Office / Bedroom
2 Groups: "AlleLampen" with all 5 in it, "WoZi" with the 3 Living Room Lights
Intended use:
- All Devices follow speed and fade settings of one specific Device
- The Devices in "WoZi" also follow the scheme settings of that Device

Config on all Devices:

SetOption85 1
devgroupname1 AlleLampen
devgroupshare 4,0

Sample Output:
_16:39:22 CMD: devgroupstatus
16:39:22 RSL: RESULT = {"DevGroupStatus":{"Index":0,"GroupName":"AlleLampen","MessageSeq":9,"MemberCount":4,"Members":[{"IPAddress":"192.168.0.22","ResendCount":8,"LastRcvdSeq":7,"LastAckedSeq":9},{"IPAddress":"192.168.0.24","ResendCount":6,"LastRcvdSeq":6,"LastAckedSeq":9},{"IPAddress":"192.168.0.23","ResendCount":9,"LastRcvdSeq":2,"LastAckedSeq":9},{"IPAddress":"192.168.0.21","ResendCount":4,"LastRcvdSeq":7,"LastAckedSeq":9}]}}
16:44:12 CMD: devgroupshare
16:44:12 RSL: RESULT = {"DevGroupShare":{"In":"4","Out":"0"}}_

Config on "WoZi" Devices:

devgroupname2 WoZi
devgroupshare 12,0 (4+8)

Sample Output:
_16:40:38 CMD: devgroupstatus
16:40:38 RSL: RESULT = {"DevGroupStatus":{"Index":0,"GroupName":"AlleLampen","MessageSeq":7,"MemberCount":4,"Members":[{"IPAddress":"192.168.0.24","ResendCount":4,"LastRcvdSeq":6,"LastAckedSeq":7},{"IPAddress":"192.168.0.25","ResendCount":4,"LastRcvdSeq":9,"LastAckedSeq":7},{"IPAddress":"192.168.0.23","ResendCount":8,"LastRcvdSeq":2,"LastAckedSeq":7},{"IPAddress":"192.168.0.22","ResendCount":4,"LastRcvdSeq":7,"LastAckedSeq":7}]}}
16:40:44 CMD: devgroupstatus2
16:40:44 RSL: RESULT = {"DevGroupStatus":{"Index":1,"GroupName":"WoZi","MessageSeq":7,"MemberCount":2,"Members":[{"IPAddress":"192.168.0.23","ResendCount":8,"LastRcvdSeq":2,"LastAckedSeq":7},{"IPAddress":"192.168.0.22","ResendCount":5,"LastRcvdSeq":6,"LastAckedSeq":7}]}}
16:43:40 CMD: devgroupshare
16:43:40 RSL: RESULT = {"DevGroupShare":{"In":"C","Out":"0"}}_

Config on the "leader" Device:

devgroupshare 0,12
Sample Output:
_16:34:32 CMD: devgroupstatus
16:34:32 RSL: RESULT = {"DevGroupStatus":{"Index":0,"GroupName":"AlleLampen","MessageSeq":2,"MemberCount":4,"Members":[{"IPAddress":"192.168.0.22","ResendCount":0,"LastRcvdSeq":7,"LastAckedSeq":2},{"IPAddress":"192.168.0.21","ResendCount":0,"LastRcvdSeq":7,"LastAckedSeq":2},{"IPAddress":"192.168.0.25","ResendCount":0,"LastRcvdSeq":9,"LastAckedSeq":2},{"IPAddress":"192.168.0.24","ResendCount":0,"LastRcvdSeq":0,"LastAckedSeq":1}]}}
16:34:37 CMD: devgroupstatus2
16:34:37 RSL: RESULT = {"DevGroupStatus":{"Index":1,"GroupName":"WoZi","MessageSeq":2,"MemberCount":2,"Members":[{"IPAddress":"192.168.0.22","ResendCount":0,"LastRcvdSeq":6,"LastAckedSeq":2},{"IPAddress":"192.168.0.21","ResendCount":0,"LastRcvdSeq":7,"LastAckedSeq":2}]}}
16:43:10 CMD: devgroupshare
16:43:10 RSL: RESULT = {"DevGroupShare":{"In":"0","Out":"C"}}_

When changing **scheme** the two Devices in the living room will follow without problems. However when changing **speed** or **fade** settings, none of the other Devices will follow.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): 2 x Gosund RGB LED Strip, 3 x MoKo RGBW LED Bulb
- [x] Tasmota binary firmware version number used: 9.1.0(tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Initially tuya-convert, then OTA to the latest version
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
16:47:57 CMD: Backlog Template; Module; GPIO 255
16:47:57 RSL: RESULT = {"NAME":"LED Strip","GPIO":[0,0,0,0,32,417,0,0,416,418,0,0,0,1],"FLAG":0,"BASE":18}
16:47:57 RSL: RESULT = {"Module":{"0":"LED Strip"}}
16:47:58 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"417":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"418":"PWM3"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4704":"ADC Input"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
[Should not be relevant]
[... 189 lines cut ...]
16:52:47 DGR: Received AlleLampen message from 192.168.0.22: seq=7, flags=64
16:52:49 CMD: speed 5
16:52:49 SRC: WebConsole from 192.168.0.5
16:52:49 CMD: Group 0, Index 1, Command "SPEED", Data "5"
16:52:49 RSL: RESULT = {"Speed":5}
16:52:49 UDP: Packet (119)
16:52:49 UDP: Packet (119)
16:52:49 UDP: Packet (119)
16:52:49 UDP: Packet (119)
16:52:49 UDP: Packet (119)
16:52:49 UDP: Packet (119)
16:52:51 CMD: weblog 2
```

### TO REPRODUCE
Apply my sample configuration. Change fade or speed setting on "leader" device.

### EXPECTED BEHAVIOUR
All devices should follow the setting.

### SCREENSHOTS
N/A, if anything is unclear please ask.

### ADDITIONAL CONTEXT
THANKS for this awesome project and everybody who puts love and time into it!

pcdiem, 2020-11-17:
The local device only sends and receives updates to device group 1 (unless SO88 is enabled). The other device groups are for use by the DevGroupSend command.

lullerboi, 2020-11-17:
Hello, thanks for your answer.

However, are you sure about option 88?

SetOption88 | Make  each relay part of a separate device group. Relay 1 updates are sent  to/received from device group 1, relay 2 updates are sent to/received  from device group 2, etc. For the PWM Dimmer module, make each button be  associated with a different device group.

I don't see any relays involved here.

pcdiem, 2020-11-17:
In your case, there are no relays so SO88 doesn't apply. I just wanted to be clear that with SO88 enabled, the local device does send/receive messages about device groups other than device group 1. In your situation, you'd need to add rules to use DevGroupSend2 to send the fade/speed updates to the devices in the other group.

lullerboi, 2020-11-17:
But it's **DevGroup2** that works properly and **DevGroup1** that doesn't...?

pcdiem, 2020-11-17:
From a code standpoint, I don't see how that could be possible. In your web logs above, I don't see it sending fade or speed changes to either device group. All the DGR messages show only the keep-alive ACKs (flags=64). Could you send a weblog of the following commands on the sender and a receiver: Weblog 4; DevGroupName; DevGroupShare; DevGroupStatus; Fade 0; Fade 1

lullerboi, 2020-11-18:
Weblog 4
backlog DevGroupName; DevGroupShare; DevGroupStatus; Fade 0; Fade 1; scheme 2; scheme 0; DevGroupStatus

"Sender":

18:33:17 RSL: SENSOR = {"Time":"2020-11-18T18:33:17","ANALOG":{"A0":857}}
18:38:17 RSL: STATE = {"Time":"2020-11-18T18:38:17","Uptime":"1T19:15:18","UptimeSec":155718,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":35,"Color":"00005A","HSBColor":"240,100,35","Channel":[0,0,35],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JJSmartWLAN","BSSId":"90:5C:44:A5:B8:B4","Channel":1,"RSSI":100,"Signal":-32,"LinkCount":4,"Downtime":"0T00:05:05"}}
18:38:17 RSL: SENSOR = {"Time":"2020-11-18T18:38:17","ANALOG":{"A0":852}}
18:43:17 RSL: STATE = {"Time":"2020-11-18T18:43:17","Uptime":"1T19:20:18","UptimeSec":156018,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":35,"Color":"00005A","HSBColor":"240,100,35","Channel":[0,0,35],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JJSmartWLAN","BSSId":"90:5C:44:A5:B8:B4","Channel":1,"RSSI":100,"Signal":-32,"LinkCount":4,"Downtime":"0T00:05:05"}}
18:43:17 RSL: SENSOR = {"Time":"2020-11-18T18:43:17","ANALOG":{"A0":853}}
18:48:17 RSL: STATE = {"Time":"2020-11-18T18:48:17","Uptime":"1T19:25:18","UptimeSec":156318,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":35,"Color":"00005A","HSBColor":"240,100,35","Channel":[0,0,35],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JJSmartWLAN","BSSId":"90:5C:44:A5:B8:B4","Channel":1,"RSSI":100,"Signal":-32,"LinkCount":4,"Downtime":"0T00:05:05"}}
18:48:17 RSL: SENSOR = {"Time":"2020-11-18T18:48:17","ANALOG":{"A0":853}}
18:53:17 RSL: STATE = {"Time":"2020-11-18T18:53:17","Uptime":"1T19:30:18","UptimeSec":156618,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":35,"Color":"00005A","HSBColor":"240,100,35","Channel":[0,0,35],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JJSmartWLAN","BSSId":"90:5C:44:A5:B8:B4","Channel":1,"RSSI":100,"Signal":-29,"LinkCount":4,"Downtime":"0T00:05:05"}}
18:53:17 RSL: SENSOR = {"Time":"2020-11-18T18:53:17","ANALOG":{"A0":851}}
18:58:17 RSL: STATE = {"Time":"2020-11-18T18:58:17","Uptime":"1T19:35:18","UptimeSec":156918,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":35,"Color":"00005A","HSBColor":"240,100,35","Channel":[0,0,35],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JJSmartWLAN","BSSId":"90:5C:44:A5:B8:B4","Channel":1,"RSSI":100,"Signal":-32,"LinkCount":4,"Downtime":"0T00:05:05"}}
18:58:17 RSL: SENSOR = {"Time":"2020-11-18T18:58:17","ANALOG":{"A0":850}}
19:03:17 RSL: STATE = {"Time":"2020-11-18T19:03:17","Uptime":"1T19:40:18","UptimeSec":157218,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":35,"Color":"00005A","HSBColor":"240,100,35","Channel":[0,0,35],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JJSmartWLAN","BSSId":"90:5C:44:A5:B8:B4","Channel":1,"RSSI":100,"Signal":-32,"LinkCount":4,"Downtime":"0T00:05:05"}}
19:03:17 RSL: SENSOR = {"Time":"2020-11-18T19:03:17","ANALOG":{"A0":849}}
19:08:17 RSL: STATE = {"Time":"2020-11-18T19:08:17","Uptime":"1T19:45:18","UptimeSec":157518,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":35,"Color":"00005A","HSBColor":"240,100,35","Channel":[0,0,35],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JJSmartWLAN","BSSId":"90:5C:44:A5:B8:B4","Channel":1,"RSSI":100,"Signal":-33,"LinkCount":4,"Downtime":"0T00:05:05"}}
19:08:17 RSL: SENSOR = {"Time":"2020-11-18T19:08:17","ANALOG":{"A0":847}}
19:09:52 CMD: Weblog 4
19:09:52 RSL: RESULT = {"WebLog":4}
19:09:57 DGR: Received AlleLampen message from 192.168.0.21: seq=24, flags=64
19:10:06 CMD:  backlog DevGroupName; DevGroupShare; DevGroupStatus; Fade 0; Fade 1; scheme 2; scheme 0; DevGroupStatus
19:10:06 SRC: WebConsole from 192.168.0.5
19:10:06 CMD: Group 0, Index 1, Command "BACKLOG", Data "DevGroupName; DevGroupShare; DevGroupStatus; Fade 0; Fade 1; scheme 2; scheme 0; DevGroupStatus"
19:10:06 SRC: Backlog
19:10:06 CMD: Group 0, Index 1, Command "DEVGROUPNAME", Data ""
19:10:06 RSL: RESULT = {"DevGroupName1":"AlleLampen","DevGroupName2":"WoZi","DevGroupName3":"","DevGroupName4":""}
19:10:06 SRC: Backlog
19:10:06 CMD: Group 0, Index 1, Command "DEVGROUPSHARE", Data ""
19:10:06 RSL: RESULT = {"DevGroupShare":{"In":"0","Out":"C"}}
19:10:06 SRC: Backlog
19:10:06 CMD: Group 0, Index 1, Command "DEVGROUPSTATUS", Data ""
19:10:06 RSL: RESULT = {"DevGroupStatus":{"Index":0,"GroupName":"AlleLampen","MessageSeq":10,"MemberCount":4,"Members":[{"IPAddress":"192.168.0.22","ResendCount":5,"LastRcvdSeq":29,"LastAckedSeq":10},{"IPAddress":"192.168.0.21","ResendCount":4,"LastRcvdSeq":24,"LastAckedSeq":10},{"IPAddress":"192.168.0.25","ResendCount":3,"LastRcvdSeq":21,"LastAckedSeq":10},{"IPAddress":"192.168.0.24","ResendCount":5,"LastRcvdSeq":18,"LastAckedSeq":10}]}}
19:10:07 SRC: Backlog
19:10:07 CMD: Group 0, Index 1, Command "FADE", Data "0"
19:10:07 RSL: RESULT = {"Fade":"OFF"}
19:10:07 SRC: Backlog
19:10:07 CMD: Group 0, Index 1, Command "FADE", Data "1"
19:10:07 RSL: RESULT = {"Fade":"ON"}
19:10:07 SRC: Backlog
19:10:07 CMD: Group 0, Index 1, Command "SCHEME", Data "2"
19:10:07 RSL: RESULT = {"Scheme":2}
[... cut ...] 5C:44:A5:B8:B4","Channel":1,"RSSI":36,"Signal":-82,"LinkCount":4,"Downtime":"0T00:05:20"}}
19:13:06 WIF: Checking connection...
19:13:10 UDP: Packet (119)
19:13:10 UDP: Packet (119)
19:13:10 UDP: Packet (119)
19:13:10 UDP: Packet (119)
19:13:10 UDP: Packet (119)
19:13:10 DGR: Received AlleLampen message from 192.168.0.21: seq=24, flags=64
19:13:19 CMD: weblog 2
19:13:19 SRC: WebConsole from 192.168.0.5
19:13:19 CMD: Group 0, Index 1, Command "WEBLOG", Data "2"
19:13:19 RSL: RESULT = {"WebLog":2}

pcdiem, 2020-11-18:
It is sending the scheme updates to device group 1 properly. There is a bug in the handling of the DevGroupShare masks that's preventing it from sending the Fade and Speed changes. I fixed the bug and submitted PR https://github.com/arendst/Tasmota/pull/9911 . Thank you for your report and cooperation.

PR #9911 (2020-11-18): Fix DevGroupShare mask handling
## Description:

Add missing variable in DevGroupShare processing.

**Related issue (if applicable):** fixes #9894

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.7
  - [ ] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUT [... cut]
closed 2020-11-18 by arendst