arendst/Tasmota #9896: Shutter: Shutterstop does not work in 9.1.0

schrej-zz, 2020-11-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sending a Shutterstop (via MQTT or console) does not stop the shutters.
Command is visible in console and counting of shutterposition stops, but shutter keeps on moving.
Console also shows "POWER2 ON" and "POWER2 OFF" (after pulsetime period), but relay does definitely not switch.
Entering "POWER2 ON" via console stops the shutter

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Electrodragon (adapted template to switch Relay1 and Relay 2, but this is definitely not the issue)
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: internal file upload
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
18:44:55 MQT: WL_Electrodragon_04/stat/RESULT = {"NAME":"ElectroDragon_i","GPIO":[33,1,32,1,1,1,0,0,224,225,1,1,288,1],"FLAG":0,"BASE":15}
18:44:56 MQT: WL_Electrodragon_04/stat/RESULT = {"Module":{"0":"ElectroDragon_i"}}
18:44:56 MQT: WL_Electrodragon_04/stat/RESULT = {"GPIO0":{"33":"Button2"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"225":"Relay2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"288":"Led1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS = {"Status":{"Module":0,"DeviceName":"WL_Electrodragon_04","FriendlyName":["WL_Electrodragon_04","Tasmota2"],"Topic":"WL_Electrodragon_04","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:19:10","StartupUTC":"2020-11-17T17:26:27","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-11-17T18:11:16","SaveCount":49,"SaveAddress":"FB000"}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"439/699"}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Schreiner",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40002000","00006000","00000000"]}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS5 = {"StatusNET":{"Hostname":"WL_Electrodragon_04-5509","IPAddress":"192.168.2.194","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"84:F3:EB:17:F5:85","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_17F585","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-17T17:45:37","Local":"2020-11-17T18:45:37","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:03","Sunset":"17:06"}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-17T18:45:37","Shutter1":{"Position":30,"Direction":0,"Target":30}}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-17T18:45:37","Uptime":"0T00:19:10","UptimeSec":1150,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:3D:FA:7C","Channel":11,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}}
18:45:37 MQT: WL_Electrodragon_04/stat/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":230,"Close":230,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
18:47:20 SHT: time: 7, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 454000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:20 SHT: time: 8, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 453000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SHT: time: 9, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 452000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SHT: time: 10, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 451000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SHT: time: 11, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 450000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SRC: PulseTimer
18:47:21 SHT: Switched relay: 2 by PulseTime
18:47:21 SHT: Shutter 1: source: PulseTime, powerstate_local 0, ShutterGlobal.RelayCurrentMask 2, manual change 0
18:47:21 MQT: WL_Electrodragon_04/stat/RESULT = {"POWER2":"OFF"}
18:47:21 MQT: WL_Electrodragon_04/stat/POWER2 = OFF
18:47:21 SHT: time: 12, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 449000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SHT: time: 13, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 448000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SHT: time: 14, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 447000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SHT: time: 15, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 446000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SHT: time: 17, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 444000, next_stop 0, target: 0,  max_vel_change 0, dir: -1
18:47:21 SHT: Shutter 1 Real 444000, Start 460000, Stop 0, Dir -1, Delay 0, Rtc 0.85 [s], Freq 0, PWM 0
[... 86 lines cut ...]

### TO REPRODUCE
_Steps to reproduce the behavior:_
SO80
shutterrelay1 1
interlock 1,2
interlock ON
Pulsetime1 5
Pulsetime2 5

Send shutterclose or shutteropen
Send shutterstop after a while

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Shutterstop should really switch relay, to stop shutter

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
It seems, as if the "POWER2 ON and OFF" are shown in console, but relay does not really switch and therefore shutter keeps on moving.

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-11-18:
I do see two possible problems. pulsetime and interlock is normally nothing that should work together. Also I see shuttermode=0. This is automatic detection. Maybe you can set it directly to the correct value, if you don't want to disable the interlock

quattro1987, 2020-11-18:
Hi, yesterday I wrote this text at Discord:
```
Hello everybody,
yesterday I wanted to upgrade my Wemos D1 Mini to the current Tasmota version 9.1.0.bin.
I use the Wemos for shutter control. (Many thanks in advance to Stefan Bode)
Procedure:
1. flash tasmota-DE9.1.0.bin
Generell setings:
Generic (18)
D1 Relay_i 1
D2 Relay_i 2

console commands: 
SetOption80 1
Interlock 1,2
Interlock ON
PulseTime1 2
PulseTime2 2
ShutterOpenDuration 14.5
ShutterCloseDuration 13.9

Status 13:
{"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":145,"Close":139,"50perc":50,"Delay":0,"Opt":"0000","Calib":"300:500:700:900:1000","Mode":"0"}}}

The hardware structure is realized with an optocoupler for bridging up and down buttons. (pulses start and stop the movement)
Riding up and down works without problems, but while calibrating I noticed that the second impulse to stop is not sent.
The console output shows the switching of the respective relay, but the implouse does not arrive at the button.
Even increasing the impulse time to 3, 4 or 8 did not change the behavior.
I have now tested the following versions with exactly the same procedure on the same hardware for comparison.
Works fine:
tasmota-DE8.3.1.bin
tasmota-DE8.4.0.bin
tasmota-DE8.5.0.bin

Does not work:
tasmota-DE8.5.1.bin
tasmota-DE9.1.0.bin
Has anyone already noticed a similar behavior, or is it due to wrong settings?
Greetings
```
I now have the following attempt:
interlock off
shuttermode 1
Pulsetime1 / 2 = 4
Unfortunately the behavior remains the same.
The 2 pulse to stop is shown in the command line, but the optocoupler does not switch or the outputs are not switched on for the second pulse.

schrej-zz, 2020-11-19:
Hi @quattro1987,
thanks for confirming the issue.
Exactly the same behavior.
I enabled Shuttermode1 as @stefanbode requested.
No change in behavior.
As you described: POWER ON/OFF is in the log, but the relay is not switched (I should hear it).
Either the ON/OFF impulse is too fast to switch the relay, or it is only shown in the logs, but is not really switched.

schrej-zz, 2020-11-19:
Hi,
I looked into the code of xdrv_27_shutter.ino, and added delay(500) for testing purposes in  the code starting around line 450:
`case SHT_PULSE:
      uint8_t cur_relay = Settings.shutter_startrelay[i] + (Shutter[i].direction == 1 ? 0 : (uint8_t)(ShutterGlobal.position_mode == SHT_TIME)) ;
      // we have a momentary switch here. Needs additional pulse on same relay after the end
      if ((SRC_PULSETIMER == TasmotaGlobal.last_source || SRC_SHUTTER == TasmotaGlobal.last_source || SRC_WEBGUI == TasmotaGlobal.last_source)) {
        ExecuteCommandPowerShutter(cur_relay, 1, SRC_SHUTTER);
        delay(500);
        // switch off direction relay to make it power less
        if ((1 << (Settings.shutter_startrelay[i])) & TasmotaGlobal.power) {
          ExecuteCommandPowerShutter(Settings.shutter_startrelay[i]+1, 0, SRC_SHUTTER);
          delay(500);
        }
      } else {
        TasmotaGlobal.last_source = SRC_SHUTTER;
      }
`
I compiled this version by my own and everything works again like a charm.
As I assumed, POWER ON/OFF are sent without the pulsetime delay and seem to be too fast for the relays.
With a lelay, the relay switches as expected.

quattro1987, 2020-11-21:
Hi @schrej-zz , for testing I have also connected hardware relays.
Your workaround has the effect that the relays switch.
I noticed a different behavior (the stop impulse is always switched Relay: 2):
```
With "Shutter-Close"
Start Pulse:
"POWER2": "ON" (With PulseTimer)
"POWER2": "OFF"
Stop Pulse:
"POWER2": "ON" (Without PulseTimer)
"POWER2": "OFF"

On "Shutter-Open"
Start Pulse:
"POWER1": "ON" (With PulseTimer)
"POWER1": "OFF"
Stop Pulse:
"POWER2": "ON" (Without PulseTimer)
"POWER2": "OFF"
```

Here is the log output:

```
Shuter-Close:
12:31:27 CMD: Gruppe 0, Index 1, Befehl "SHUTTERPOSITION", Daten "StopClose"
12:31:27 SHT: Pos. in: payload StopClose (9), payload -99, idx 1, src 7
12:31:27 SHT: Pos. in: payload StopClose (0), payload 0, idx 1, src 7
12:31:27 SHT: lastsource 7:, real 290000, target 0, payload 0
12:31:27 SHT: Delay Start. var0 <99>=<>, max10s?
12:31:27 SHT: Delay Start. Done
12:31:27 SRC: Shutter
12:31:27 SHT: Switched relay: 2 by Shutter
12:31:27 SHT: Shutter 1: source: Shutter, powerstate_local 2, ShutterGlobal.RelayCurrentMask 2, manual change 0
12:31:27 RSL: stat/tasmota_D3B56B/RESULT = {"POWER2":"ON"}
12:31:27 RSL: stat/tasmota_D3B56B/POWER2 = ON
12:31:27 RSL: stat/tasmota_D3B56B/RESULT = {"ShutterPosition1":0}
12:31:27 SHT: Shutter 1 Real 288957, Start 290000, Stop 0, Dir -1, Delay 0, Rtc 0.10 [s], Freq 0, PWM 0
12:31:27 RSL: stat/tasmota_D3B56B/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
12:31:27 SHT: time: 2, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 288957, next_stop 0, target: 0,  max_vel_change 0, dir: -1
12:31:28 SHT: time: 4, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 286871, next_stop 0, target: 0,  max_vel_change 0, dir: -1
12:31:28 SHT: time: 5, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 285828, next_stop 0, target: 0,  max_vel_change 0, dir: -1
12:31:28 SRC: PulseTimer
12:31:28 SHT: Switched relay: 2 by PulseTime
12:31:28 SHT: Shutter 1: source: PulseTime, powerstate_local 0, ShutterGlobal.RelayCurrentMask 2, manual change 0
12:31:28 RSL: stat/tasmota_D3B56B/RESULT = {"POWER2":"OFF"}
12:31:28 RSL: stat/tasmota_D3B56B/POWER2 = OFF
12:31:28 SHT: time: 6, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 284785, next_stop 0, target: 0,  max_vel_change 0, dir: -1
12:31:28 SHT: time: 7, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 283742, next_stop 0, target: 0,  max_vel_change 0, dir: -1
...
12:31:41 SHT: time: 279, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 46, next_stop 0, target: 0,  max_vel_change 0, dir: -1
12:31:41 SHT: time: 280, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos -997, next_stop 0, target: 0,  max_vel_change 0, dir: -1
12:31:41 SHT: Stop Shutter 0. Switchmode 1
12:31:42 SRC: Shutter
12:31:42 SHT: Switched relay: 2 by Shutter
12:31:42 SHT: Shutter 1: source: Shutter, powerstate_local 2, ShutterGlobal.RelayCurrentMask 2, manual change 0
12:31:42 RSL: stat/tasmota_D3B56B/RESULT = {"POWER2":"ON"}
12:31:42 RSL: stat/tasmota_D3B56B/POWER2 = ON
12:31:42 SRC: Shutter
12:31:42 SHT: Switched relay: 2 by Shutter
12:31:42 SHT: Shutter 1: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 2, manual change 0
12:31:42 RSL: stat/tasmota_D3B56B/RESULT = {"POWER2":"OFF"}
12:31:42 RSL: stat/tasmota_D3B56B/POWER2 = OFF
12:31:42 SHT: Shutter 1 Real 0, Start 0, Stop 0, Dir 0, Delay 0, Rtc 14.00 [s], Freq 0, PWM 0
12:31:42 RSL: stat/tasmota_D3B56B/SHUTTER1 = 0
12:31:42 RSL: stat/tasmota_D3B56B/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}

Shutter-Open:
12:35:16 CMD: Gruppe 0, Index 1, Befehl "SHUTTERPOSITION", Daten "StopOpen"
12:35:16 SHT: Pos. in: payload StopOpen (8), payload -99, idx 1, src 7
[... 20 lines cut ...]
12:35:17 RSL: stat/tasmota_D3B56B/POWER1 = OFF
12:35:17 SHT: time: 7, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 6000, next_stop 0, target: 290000,  max_vel_change 0, dir: 1
12:35:17 SHT: time: 8, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 7000, next_stop 0, target: 290000,  max_vel_change 0, dir: 1
12:35:17 SHT: time: 9, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 8000, next_stop 0, target: 290000,  max_vel_change 0, dir: 1
12:35:17 SHT: time: 10, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 9000, next_stop 0, target: 290000,  max_vel_change 0, dir: 1
...
12:35:31 SHT: time: 290, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 289000, next_stop 0, target: 290000,  max_vel_change 0, dir: 1
12:35:31 SHT: time: 291, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 290000, next_stop 0, target: 290000,  max_vel_change 0, dir: 1
12:35:31 SHT: Stop Shutter 0. Switchmode 1
12:35:31 SRC: Shutter
12:35:31 SHT: Switched relay: 2 by Shutter
12:35:31 SHT: Shutter 1: source: Shutter, powerstate_local 2, ShutterGlobal.RelayCurrentMask 2, manual change 0
12:35:31 RSL: stat/tasmota_D3B56B/RESULT = {"POWER2":"ON"}
[... cut ...]
12:35:31 SHT: Shutter 1: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 2, manual change 0
12:35:31 RSL: stat/tasmota_D3B56B/RESULT = {"POWER2":"OFF"}
12:35:31 RSL: stat/tasmota_D3B56B/POWER2 = OFF
12:35:31 SHT: Shutter 1 Real 290000, Start 290000, Stop 290000, Dir 0, Delay 0, Rtc 14.55 [s], Freq 0, PWM 0
12:35:31 RSL: stat/tasmota_D3B56B/SHUTTER1 = 100
12:35:31 RSL: stat/tasmota_D3B56B/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100}}
```

Can you observe this behaviour in your own environment?

schrej-zz, 2020-11-24:
Hi @quattro1987,
sorry for replying that late. Was sick.
Yes, I also see the POWER2. I cannot confirm, that it is different without Pulsetime as I have to use pulsetime with my configuration.
But it was the same case with and without the delay and i dought that a delay will change the behavior.
In my environment, this has no negative effect as long as the shutter moves, as it stopps on POWER1 and POWER2.
But it has an impact, when shutters top moving.
I have a buildin calibration with automatically stops the shutters at the at open and close.
If the SHUTTEROPENDURATION is now just a tenth of a second longer than the automatic stop, the shutters are closing again.
If I reduce the SHUTTEROPENDURATION e.g. by a second, the shutters stop too early :-(
We might have two sepaerate issues here:
1. Is the lack of a delay period at SHUTTERSTOP
2. Is the use of the wrong POWER<x> at SHUTTERSTOP

Joerg

schrej-zz, 2020-11-24:
@stefanbode,
as @quattro1987 wrote, the change must have been around version 8.5.1.
I still have a shutter with version 8.2.03. where everything works perfect with absolutely the same configuration.
An upgrade to 8.5.1. or 9.0.1 without any configurational changes leads to the strange behavior.
See the weblog 4 below. It logs the teleperiod at SHUTTERSTOP and it logs the same POWER\<x> at SHUTTERSTOP as used for SHUTTEROPEN or SHUTTERCLOSE.
`0:30:26 SRC: Backlog
10:30:26 CMD: Group 0, Index 1, Command "SHUTTERCLOSE", Data ""
10:30:26 SHT: Pos. in: payload  (0), payload 0, idx 1, src 7
10:30:26 SHT: lastsource 7:, real 32125, target 0, payload 0
10:30:26 SHT: Start in dir 0
10:30:26 SRC: Shutter
10:30:26 SHT: Switched relay: 2 by Shutter
10:30:26 MQT: WL_Electrodragon_06/stat/RESULT = {"POWER2":"ON"}
10:30:26 MQT: WL_Electrodragon_06/stat/POWER2 = ON
10:30:26 MQT: WL_Electrodragon_06/stat/RESULT = {"ShutterClose1":100}
10:30:26 SHT: Shutter1 Real 31555, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 0.30 [s], Freq 0
10:30:26 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":57,"Direction":-1,"Target":100}}
10:30:26 SRC: PulseTimer
10:30:26 SHT: Switched relay: 2 by PulseTime
10:30:26 MQT: WL_Electrodragon_06/stat/RESULT = {"POWER2":"OFF"}
10:30:26 MQT: WL_Electrodragon_06/stat/POWER2 = OFF
10:30:27 SHT: Shutter1 Real 29655, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 1.30 [s], Freq 0
10:30:27 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":65,"Direction":-1,"Target":100}}
10:30:28 SHT: Shutter1 Real 27755, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 2.30 [s], Freq 0
10:30:28 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":74,"Direction":-1,"Target":100}}
10:30:29 SHT: Shutter1 Real 25855, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 3.30 [s], Freq 0
10:30:29 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":82,"Direction":-1,"Target":100}}
10:30:30 SHT: Shutter1 Real 23860, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 4.35 [s], Freq 0
10:30:30 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":92,"Direction":-1,"Target":100}}
10:30:31 SHT: Shutter1 Real 21960, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 5.35 [s], Freq 0
10:30:31 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":95,"Direction":-1,"Target":100}}
10:30:32 SHT: Shutter1 Real 20060, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 6.35 [s], Freq 0
10:30:32 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":96,"Direction":-1,"Target":100}}
10:30:33 SHT: Shutter1 Real 18255, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 7.30 [s], Freq 0
10:30:33 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":96,"Direction":-1,"Target":100}}
10:30:34 SHT: Shutter1 Real 16355, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 8.30 [s], Freq 0
10:30:34 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":96,"Direction":-1,"Target":100}}
10:30:35 SHT: Shutter1 Real 14360, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 9.35 [s], Freq 0
10:30:35 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":97,"Direction":-1,"Target":100}}
10:30:36 SHT: Shutter1 Real 12460, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 10.35 [s], Freq 0
10:30:36 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":97,"Direction":-1,"Target":100}}
10:30:37 SHT: Shutter1 Real 10560, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 11.35 [s], Freq 0
10:30:37 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":98,"Direction":-1,"Target":100}}
10:30:38 SHT: Shutter1 Real 8755, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 12.30 [s], Freq 0
10:30:38 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":98,"Direction":-1,"Target":100}}
10:30:39 SHT: Shutter1 Real 6855, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 13.30 [s], Freq 0
10:30:39 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":99,"Direction":-1,"Target":100}}
10:30:40 SHT: Shutter1 Real 4860, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 14.35 [s], Freq 0
10:30:40 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":99,"Direction":-1,"Target":100}}
10:30:41 SHT: Shutter1 Real 2960, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 15.35 [s], Freq 0
10:30:41 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":99,"Direction":-1,"Target":100}}
10:30:42 SHT: Shutter1 Real 1060, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 16.35 [s], Freq 0
10:30:42 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":100}}
10:30:43 SRC: Shutter
10:30:43 SHT: Switched relay: 2 by Shutter
10:30:43 MQT: WL_Electrodragon_06/stat/RESULT = {"POWER2":"ON"}
10:30:43 MQT: WL_Electrodragon_06/stat/POWER2 = ON
10:30:43 SHT: Shutter1 Real 15, Start 32125, Stop 0, Dir -1, Delay 0, Rtc 16.90 [s], Freq 0
10:30:43 MQT: WL_Electrodragon_06/stat/SHUTTER1 = 100
10:30:43 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100}}
10:30:43 SRC: PulseTimer
10:30:43 SHT: Switched relay: 2 by PulseTime
10:30:43 MQT: WL_Electrodragon_06/stat/RESULT = {"POWER2":"OFF"}
10:30:43 MQT: WL_Electrodragon_06/stat/POWER2 = OFF
10:30:43 CFG: Saved to flash at F7, Count 741, Bytes 4096
10:30:45 WIF: Checking connection...
10:30:49 MQT: Received Topic "WL_Electrodragon_06/cmnd/Backlog", Data Size 12, Data "shutteropen1"
10:30:49 SRC: MQTT
10:30:49 CMD: Group 0, Index 1, Command "BACKLOG", Data "shutteropen1"
10:30:49 SRC: Backlog
[... 83 lines cut ...]
[... cut ...]
10:31:20 MQT: WL_Electrodragon_06/stat/POWER1 = ON
10:31:20 SHT: Shutter1 Real 44000, Start 23115, Stop 44000, Dir 1, Delay 0, Rtc 10.50 [s], Freq 0
10:31:20 MQT: WL_Electrodragon_06/stat/SHUTTER1 = 0
10:31:20 MQT: WL_Electrodragon_06/stat/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}
10:31:20 CFG: Saved to flash at F5, Count 743, Bytes 4096
10:31:20 SRC: PulseTimer
10:31:20 SHT: Switched relay: 1 by PulseTime
10:31:20 MQT: WL_Electrodragon_06/stat/RESULT = {"POWER1":"OFF"}
10:31:20 MQT: WL_Electrodragon_06/stat/POWER1 = OFF`

quattro1987, 2020-11-26:
Hello,
i have tested and think it is due to 2 factors:
1st:
line 433 - 435 always sets the direction to 0, so the variable cur_relay in line 447 cannot be filled correctly,
because the value is always 0.
I cut out line 433 - 435 and put it at position 448.
```
if (Shutter[i].direction !=0) {
    Shutter[i].direction = 0;
	delay(MOTOR_STOP_TIME);
	}
```
This has the effect that when opening both the start and the stop impulse are switched via Relay 1. Furthermore it has the effect that the relay is switched off by the pulsetime.

The 2nd effect is, that the program code in the 2nd If (line 452 - 454)
```
if ((1 << (Settings.shutter_startrelay[i])) & TasmotaGlobal.power) {
          ExecuteCommandPowerShutter(Settings.shutter_startrelay[i]+1, 0, SRC_SHUTTER);
        }
```
is not executed when opening.
Now the situation is as follows:

With "Shutter-Close"
Start Pulse:
"POWER2": "ON" (**With PulseTimer**)
"POWER2": "OFF"
Stop Pulse:
"POWER2": "ON" (**Without PulseTimer**)
"POWER2": "OFF"

On "Shutter-Open"
Start Pulse:
"POWER1": "ON" (**With PulseTimer**)
"POWER1": "OFF"
Stop Pulse:
"POWER2": "ON" (**With PulseTimer**)
"POWER2": "OFF"

It looks to me as if the command from line 450 starts the relay correctly with the pulsetime, but line 453 starts the relay directly again with the command to state 0 = POWER_OFF = Relay Off, and thus cancels the pulsetime.

schrej-zz, 2020-11-30:
Hi @quattro1987,
thanks for digging a little deeper into this issue.
`This has the effect that when opening both the start and the stop impulse are switched via Relay 1. Furthermore it has the effect that the relay is switched off by the pulsetime.
`
I think, start and stop should always be don by the same relay.Maybe that's what you meant.
`On "Shutter-Open"
Start Pulse:
"POWER1": "ON" (With PulseTimer)
"POWER1": "OFF"
Stop Pulse:
"POWER2": "ON" (With PulseTimer)
"POWER2": "OFF"`
Do you really mean POWER2? it should be POWER1

`It looks to me as if the command from line 450 starts the relay correctly with the pulsetime, but line 453 starts the relay directly again with the command to state 0 = POWER_OFF = Relay Off, and thus cancels the pulsetime.`

I don't understand, why POWER_OFF is necessary? The condition is SHT_PULSE, so POWER_OFF should always just come from pulsetime.

Joerg

ascillato2, 2020-12-02:
Thread continues at https://github.com/arendst/Tasmota/issues/9906

schrej-zz, 2020-12-02:
@ascillato2,
I don't see any relation between this issue and the one you are referring to. The only relation I see, is, that the issues started with 8.5.1.
When closing this issue, the discussion is lost.

stefanbode, 2020-12-14:
I will check this one, too. The first analyse with the delay(500) looks reasonable for the problem. I have to check further to see whats happen. With the pulsetime relays the job is more complicated and error can easily happen. I will work on it @ascillato2 please can you reopen?

ascillato2, 2020-12-14:
Reopening as requested.

stefanbode, 2020-12-15:
There are two different shutters possible with pulsetime. One: r1 up and r2 down. Both with pulse. I assume this is your shutter. Then there are also ones with r1: start r2 up/down (without pulse). The power off resets the direction relay to 0 to avoid permanent ON on this relay after stopping the shutter. Anyhow I have submitted a fix that should address the issue finally.

closed 2020-12-15 by ascillato2

schrej-zz, 2020-12-15:
@stefanbode,
yes, you are right. Mine are the first ones.
I now tested current dev branch and the issue really seems to be solved.
Thanks Stefan

PR #10141 (2020-12-15): Shutters: Correct Pulsetime issues.
## Description:

**Related issue (if applicable):** fixes #9896

## Checklist:
  - [ x] The pull request is done against the latest dev branch
  - [x ] Only relevant files were touched
  - [ x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x ] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [ x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla) [... cut]