arendst/Tasmota #9897: Increase space for rules.

Alisand, 2020-11-17:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

I use long rules and run out of space. I changed MAX_RULE_SETS to 8. ESP32 started to overload constantly. How can you expand the space for rules?

meingraham, 2020-11-17:
With text compression used by default for Tasmota rules, each rule set can fit roughly 1K bytes for each rule set or well over 2K  in total for the default three rule sets. The number of rules that can fit in 2K is probably 35-40 individual rules. With 8 rule sets, you are potentially looking at well over 100 rules?

The question becomes, how much complexity does one want to implement with Tasmota rules and in turn the load that places on the device. At some point there has to be a tipping point where it is better to offload some of that complexity onto a home automation hub.

Alisand, 2020-11-17:
I am using tasmotа32sensors. I am interested in collecting RF sensors. Many false codes come from the ether. Each sensor has its own name (full), which is very convenient. For esp 32, it will not be difficult to process 8 or more rules. That would be issue 8266.

meingraham, 2020-11-17:
> not be difficult to process 8 or more rules

A _rule_ consists of a `ON/DO/ENDON` clause. To be clear, a single **_rule set_** can hold multiple _rules_; dozens of _rules_ potentially when one considers the use of text compression included by default with the Tasmota Rules feature. A single **_rule set_** can contain may _rules_. Thus, the Tasmota default of three **_rule sets_** can already hold a substantial number of _rules_. You do not need to increase `MAX_RULE_SETS` in order to use 8 _rules_ and many, many more than 8 _rules_.

closed 2020-11-17 by ascillato2