arendst/Tasmota #9899: Full support of Ikea TRADFRI motion sensor

Bibinsa, 2020-11-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After pairing, Ikea Tradfri Motion sensor does not report Occupancy or even a zigbee message.
Seems to be not paired...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [ ] Tasmota binary firmware version number used: 9.1.0.1 (zigbee)
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

19:44:21 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
19:44:21 zigbee.permit_end_time = 383191159
19:44:21 MQT: tasmota/zbbridge-1/stat/RESULT = {"ZbPermitJoin":"Done"}
19:44:21 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
19:44:43 ZIG: {"ZbEZSPReceived":"230008008300D38E97FEFF570B0004"}
19:44:43 ZIG: {"ZbEZSPReceived":"24008300D38E97FEFF570B000203FFFF"}
19:44:49 ZIG: {"ZbEZSPReceived":"230008018300D38E97FEFF570B0004"}
19:44:49 ZIG: {"ZbEZSPReceived":"24008300D38E97FEFF570B0001000000"}
19:44:49 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x000B57FFFE978ED3","ShortAddr":"0x0083","ParentNetwork":"0x0000","Status":1,"Decision":0}}
19:44:51 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 1337 bytes)
19:44:55 ZIG: {"ZbEZSPReceived":"230008018300D38E97FEFF570B0004"}
19:44:55 ZIG: {"ZbEZSPReceived":"24008300D38E97FEFF570B0001000000"}
19:44:55 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x000B57FFFE978ED3","ShortAddr":"0x0083","ParentNetwork":"0x0000","Status":1,"Decision":0}}
19:44:56 ZIG: {"ZbEZSPReceived":"6200D38E97FEFF570B00"}
19:44:56 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x000B57FFFE978ED3","ShortAddr":"0x0083","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
19:44:57 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
19:44:58 MQT: tasmota/zbbridge-1/tele/0083/SENSOR = {"0x0083":{"Device":"0x0083","Manufacturer":"IKEA of Sweden","ModelId":"TRADFRI motion sensor","Endpoint":1,"LinkQuality":141}}
19:44:59 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbState":{"Status":33,"Endpoint":"0x01","ProfileId":"0xC05E","DeviceId":"0x0850","DeviceVersion":2"InClusters":["0x0000","0x0001","0x0003","0x0009","0x0B05","0x1000"],"OutClusters":["0x0003","0x0004","0x0006","0x0019","0x1000"]}}
19:45:00 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 1374 bytes)
19:45:01 ZIG: auto-bind `ZbBind {"Device":"0x0083","Endpoint":1,"Cluster":"0x0006"}`
19:45:02 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbBind":{"Device":"0x0083","Status":0,"StatusMessage":"SUCCESS"}}
19:45:14 ZIG: Internal ZDO message 0x0006 sent from 0x0083 (broadcast)
19:45:21 MQT: tasmota/zbbridge-1/tele/RESULT = {"ZbState":{"Status":20,"Message":"Pairing mode disabled"}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Pair Ikea Tradfri Motion sensor

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Reporting Occupancy:1 when motion sensor detect presence.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-17:
If I remember well, the IKEA motion sensor does not send Occupancy messages, it only sends Power:on/off messages. They are meant to be bound to a bulb or group and command them directly.

What you can do is create a binding between the sensor and a device, or better between the sensor and a group address. If you want the group messages to show up on the ZBBridge, don't forget to use `ZbListen` to listen to the group address. Add it to a rule to make it automatically triggered when the Zigbee coordinator starts.

ascillato2, 2020-11-17:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-17 by ascillato2

Bibinsa, 2020-11-17:
Thanks, I will try it.

Bibinsa, 2020-11-17:
@s-hadinger The IKEA remote sensor does not send any message since pairing process... 5cm from ZbBridge
Works welll with the Tradfri Gateway but no message with Z2T.

[image]

MattWestb, 2020-11-18:
And there is sending one (Power 6) on with off time (60 Old / 180 New) seconds and no off command ala Zigbee 3 standard for lights  to the bounded group.

Bibinsa, 2020-11-18:
@MattWestb  So I should see on the blog (level 2), a message every 60/180?
The timer on the dashboard UI is 01d today. So I conclude: 0 zigbee message from the pairing.

I will try to forget, pair and bind the sensor with ZbBind.