arendst/Tasmota #9900: Shelly Dimmer 1 - FW 9.1.0 - possible wrong template

mariowitdoek, 2020-11-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I flashed a Shelly Dimmer 1 (one with 5 connectors) OTA with the tool from yaourdt, following the video from digiblur to be sure not to make a mistake and pasting in the template from https://github.com/arendst/Tasmota/commit/b6b35ef86e7e715787a273a9ff0c8c58afb475da
{"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,1,1,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}

Everything for the flashing seems to work but the template is somewhat bizar.
I see temperature (47°C) but nothing related to a dimmer (no on/off and no brightness slider).
The new tasmota (beta) integration in home-assistant also detects a thermometer :S

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): _Shelly Dimmer 1____
- [X ] Tasmota binary firmware version number used: _____
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used: _____https://github.com/yaourdt/mgos-to-tasmota
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:15:19 CMD: Backlog Template; Module; GPIO 255
20:15:19 MQT: domotica/tasmota/shellydimmer1/stat/RESULT = {"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,1,1,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}
20:15:19 MQT: domotica/tasmota/shellydimmer1/stat/RESULT = {"Module":{"0":"Shelly Dimmer 1"}}
20:15:19 MQT: domotica/tasmota/shellydimmer1/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"192":"Switch_n1"},"GPIO13":{"0":"None"},"GPIO14":{"193":"Switch_n2"},"GPIO15":{"288":"Led1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:16:25 CMD: STATUS 0
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS = {"Status":{"Module":0,"DeviceName":"ShellyDimmer1","FriendlyName":["ShellyDimmer1"],"Topic":"shellydimmer1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:40:47","StartupUTC":"2020-11-17T18:35:38","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-11-17T19:13:08","SaveCount":15,"SaveAddress":"FB000"}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"464/699"}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Pretty Fly For A Wifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS5 = {"StatusNET":{"Hostname":"shellydimmer1-2956","IPAddress":"192.168.2.87","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.8","Mac":"CC:50:E3:F3:AB:8C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.13","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F3AB8C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-11-17T19:16:25","Local":"2020-11-17T20:16:25","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:03","Sunset":"17:06"}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS10 = {"StatusSNS":{"Time":"2020-11-17T20:16:25","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":47.9},"TempUnit":"C"}}
20:16:25 MQT: domotica/tasmota/shellydimmer1/stat/STATUS11 = {"StatusSTS":{"Time":"2020-11-17T20:16:25","Uptime":"0T00:40:47","UptimeSec":2447,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Pretty Fly For A Wifi","BSSId":"4C:5E:0C:B3:FF:83","Channel":5,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash OTA, Upgrade and enter template

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
It is like a thermometer but I expect a light

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-11-17:
You are using an old template. See source code for correct template. It's at the top of the file.

arendst, 2020-11-17:
Oh and you're using the wronf tasmota version. You need 9.1.0.2 available from dev download site.

mariowitdoek, 2020-11-17:
Hi, I tried (with the tip of digiblur in discord) the development version but it did not work. Then I followed Jasons procedure and it worked. Indeed I noticed that the template was slightly different between the Jason procedure and the one from yaourdt. I will try it tomorrow and close this thread. 
BTW bedankt voor al uw goed werk. Mocht je in de streek van Brugge zijn, dan trakteer ik u op streekbiertje :p

closed 2020-11-17 by mariowitdoek