arendst/Tasmota #9903: Integration with Home Assistant

xefil, 2020-11-17:
### PROBLEM DESCRIPTION

Hello to all,

I'm trying to integrate a SonOff 4Ch R3 Relay board with HA. The device gets recognized on HA but I cannot change the relay status from HA. Seems something wrong, but cannot understand if it's on HA side or on Tasmota. Strange is that, if I change the relay status from Tasmota, HA gets updated, but if I change the Relay status from HA, nothing happens on the board. I've only noticed that from HA the MQTT publish is not all UPPERCASE like from Tasmota.

Here some details:

Tasmota:

```
Program Version	9.1.0(tasmota)
Build Date & Time	2020-11-07T11:57:45
Core/SDK Version	2_7_4_5/2.2.2-dev(38a443e)
Uptime	0T00:02:55
Flash write Count	67 at 0xF9000
Boot Count	9
Restart Reason	Software/System restart
Friendly Name 1	RELAY02-01
Friendly Name 2	RELAY02-02
Friendly Name 3	RELAY02-03
Friendly Name 4	RELAY02-04
 
MQTT Host	192.168.1.13
MQTT Port	1883
MQTT User	mqtt_user
MQTT Client	DVES_D04DF5
MQTT Topic	GARDEN02
MQTT Group Topic 1	XefilHome/tasmotas/cmnd/
MQTT Full Topic	XefilHome/GARDEN02/cmnd/
MQTT Fallback Topic	cmnd/DVES_D04DF5_fb/
MQTT No Retain	Disabled
```

After boot, if I set `SetOption19 0` the Integration disappear, then setting it back to 1, it comes back correctly in HA:

```
00:01:21 CMD: SetOption19 1
00:01:21 MQT: XefilHome/GARDEN02/RESULT = {"SetOption19":"ON"}
00:01:22 MQT: homeassistant/switch/D04DF5_RL_1/config = {"name":"RELAY02-01","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER1","val_tpl":"{{value_json.POWER1}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_1","dev":{"ids":["D04DF5"]}} (retained)
00:01:22 MQT: homeassistant/switch/D04DF5_RL_2/config = {"name":"RELAY02-02","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER2","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_2","dev":{"ids":["D04DF5"]}} (retained)
00:01:22 MQT: homeassistant/switch/D04DF5_RL_3/config = {"name":"RELAY02-03","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER3","val_tpl":"{{value_json.POWER3}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_3","dev":{"ids":["D04DF5"]}} (retained)
00:01:22 MQT: homeassistant/switch/D04DF5_RL_4/config = {"name":"RELAY02-04","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER4","val_tpl":"{{value_json.POWER4}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_4","dev":{"ids":["D04DF5"]}} (retained)
00:01:22 MQT: homeassistant/sensor/D04DF5_status/config = {"name":"GARDEN02 status","stat_t":"XefilHome/GARDEN02/HASS_STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"XefilHome/GARDEN02/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"D04DF5_status","dev":{"ids":["D04DF5"],"name":"GARDEN02","mdl":"Sonoff 4CH Pro","sw":"9.1.0(tasmota)","mf":"Tasmota"}} (retained)
00:01:29 CMD: SetOption19 0
00:01:29 MQT: XefilHome/GARDEN02/RESULT = {"SetOption19":"OFF"}
00:01:30 LOG: Home Assistant MQTT Discovery disabled.
```
So, then if I turn a relay on/off on Tasmota:

```
00:01:59 MQT: XefilHome/GARDEN02/RESULT = {"POWER1":"ON"}
00:01:59 MQT: XefilHome/GARDEN02/POWER1 = ON
00:02:01 MQT: XefilHome/GARDEN02/STATE = {"Time":"1970-01-01T00:02:01","Uptime":"0T00:02:03","UptimeSec":123,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"XefilHQ","BSSId":"E0:28:6D:7B:CE:48","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}
00:02:01 MQT: XefilHome/GARDEN02/RESULT = {"POWER1":"OFF"}
00:02:01 MQT: XefilHome/GARDEN02/POWER1 = OFF
```

On HA the status gets updated:

```
2020-11-17 22:26:27 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/RESULT: b'{"POWER1":"ON"}'
2020-11-17 22:26:27 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/POWER1: b'ON'
2020-11-17 22:26:29 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/STATE: b'{"Time":"1970-01-01T00:02:01","Uptime":"0T00:02:03","UptimeSec":123,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"XefilHQ","BSSId":"E0:28:6D:7B:CE:48","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}'
2020-11-17 22:26:29 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/RESULT: b'{"POWER1":"OFF"}'
2020-11-17 22:26:29 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/POWER1: b'OFF'
```

But if I do the same operation on HA, nothing happens on the board:

[... 78 lines cut ...]
Written above

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Written above

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

That HA can control the Relay Board.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

---

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Written above

**(Please, remember to close the issue when the problem has been addressed)**

xefil, 2020-11-17:
Same question asked on Home Assistant Community :-)

https://community.home-assistant.io/t/integration-with-tasmota/246306

Hope someone helps me!

Thanks!

Simone

ascillato, 2020-11-18:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

xefil, 2020-11-18:
Ok, I'm not sure on what is that issue related, so, I've posted here. I'll try in chat as well. Thanks. Ah, HA version: 0.117.6

effelle, 2020-11-18:
Cannot replicate it.
With Official integration (so19 0):
```py
2020-11-18 09:00:00 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=switch, service=turn_on, service_data=entity_id=switch.tasmota>
2020-11-18 09:00:00 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=mqtt, service=publish, service_data=topic=cmnd/tasmota_49A3BC/Power1, payload=ON>
2020-11-18 09:00:00 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on cmnd/tasmota_49A3BC/Power1: 'ON', mid: 518

2020-11-18 09:00:01 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=switch, service=turn_on, service_data=entity_id=switch.tasmota2>
2020-11-18 09:00:01 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=mqtt, service=publish, service_data=topic=cmnd/tasmota_49A3BC/Power2, payload=ON>
2020-11-18 09:00:01 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on cmnd/tasmota_49A3BC/Power2: 'ON', mid: 519
```
```py
13:00:01 MQT: Data Size 2, "cmnd/tasmota_49A3BC/Power1 ON"
13:00:01 SRC: MQTT
13:00:01 CMD: Group 0, Index 1, Command "POWER", Data "ON"
13:00:01 MQT: stat/tasmota_49A3BC/RESULT = {"POWER1":"ON"}
13:00:01 MQT: stat/tasmota_49A3BC/POWER1 = ON

13:00:02 MQT: Data Size 2, "cmnd/tasmota_49A3BC/Power2 ON"
13:00:02 SRC: MQTT
13:00:02 CMD: Group 0, Index 2, Command "POWER", Data "ON"
13:00:02 MQT: stat/tasmota_49A3BC/RESULT = {"POWER2":"ON"}
13:00:02 MQT: stat/tasmota_49A3BC/POWER2 = ON
```
With MQTT discovery (so19 1):
```py
2020-11-18 09:05:57 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=switch, service=turn_off, service_data=entity_id=switch.tasmota_2>
2020-11-18 09:05:57 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=mqtt, service=publish, service_data=topic=cmnd/tasmota_49A3BC/POWER1, qos=0, retain=False, payload=OFF>
2020-11-18 09:05:57 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on cmnd/tasmota_49A3BC/POWER1: 'OFF', mid: 538

2020-11-18 09:05:58 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=switch, service=turn_off, service_data=entity_id=switch.tasmota2_2>
2020-11-18 09:05:58 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=mqtt, service=publish, service_data=topic=cmnd/tasmota_49A3BC/POWER2, qos=0, retain=False, payload=OFF>
2020-11-18 09:05:58 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on cmnd/tasmota_49A3BC/POWER2: 'OFF', mid: 539
```
```py
13:05:58 MQT: Data Size 3, "cmnd/tasmota_49A3BC/POWER1 OFF"
13:05:58 SRC: MQTT
13:05:58 CMD: Group 0, Index 1, Command "POWER", Data "OFF"
13:05:58 MQT: stat/tasmota_49A3BC/RESULT = {"POWER1":"OFF"}
13:05:58 MQT: stat/tasmota_49A3BC/POWER1 = OFF

13:05:58 MQT: Data Size 3, "cmnd/tasmota_49A3BC/POWER2 OFF"
13:05:58 SRC: MQTT
13:05:58 CMD: Group 0, Index 2, Command "POWER", Data "OFF"
13:05:59 MQT: stat/tasmota_49A3BC/RESULT = {"POWER2":"OFF"}
13:05:59 MQT: stat/tasmota_49A3BC/POWER2 = OFF
```
As you can see upper/lower case have no impact on topic. Tested changing the state from Tasmota to Ha and from HA to Tasmota. 
Since you checked directly from Mosquitto and still you get no update (but HA correctly received the topics) the problem is elsewhere. Try to cancel the device under HA and then start over with the integration or the mqtt discovery. Reboot HA could help too.

xefil, 2020-11-18:
Thanks @effelle (Federico? CIAO, piacere)
It's possible it has something to do with the case that it's a 4CH relay ? I've a single relayboard same FW but that is working.
I would like to compare all SO options....

effelle, 2020-11-18:
It could be something  with your inveted topic since I donT see the cmnd in your topics.
Try the native integration (so19 0).
Piacere!

xefil, 2020-11-18:
I'm following it into the chat as well. I think it has something to do on that I've the autodiscovery in HA set to personal prefix "discovery_prefix: XefilHome" and the so19 to 1 doesn't then work... I'm not sure...

effelle, 2020-11-18:
Tasmota uses `homeassistant` as prefix. 
If you want to personalize it you must compile your own bin changing the user config.
```py
#define HOME_ASSISTANT_DISCOVERY_PREFIX   "homeassistant"  // Home Assistant discovery prefix
```

xefil, 2020-11-18:
I've got a different approach, to let the discovery work, as suggested by lustyffh in chat.
I've used SO19 1 and the modified the discovery commands and sent it out from cli via mosquitto_sub with correct prefix and that worked 👍 

From:
```
14:53:23 MQT: XefilHome/GARDEN02/RESULT = {"SetOption19":"ON"}
14:53:23 MQT: homeassistant/switch/D04DF5_RL_1/config = {"name":"RELAY02-01","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER1","val_tpl":"{{value_json.POWER1}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_1","dev":{"ids":["D04DF5"]}} (retained)
14:53:23 MQT: homeassistant/switch/D04DF5_RL_2/config = {"name":"RELAY02-02","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER2","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_2","dev":{"ids":["D04DF5"]}} (retained)
14:53:23 MQT: homeassistant/switch/D04DF5_RL_3/config = {"name":"RELAY02-03","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER3","val_tpl":"{{value_json.POWER3}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_3","dev":{"ids":["D04DF5"]}} (retained)
14:53:23 MQT: homeassistant/switch/D04DF5_RL_4/config = {"name":"RELAY02-04","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER4","val_tpl":"{{value_json.POWER4}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_4","dev":{"ids":["D04DF5"]}} (retained)
14:53:23 MQT: homeassistant/sensor/D04DF5_status/config = {"name":"GARDEN02 status","stat_t":"XefilHome/GARDEN02/HASS_STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"XefilHome/GARDEN02/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"D04DF5_status","dev":{"ids":["D04DF5"],"name":"GARDEN02","mdl":"Sonoff 4CH Pro","sw":"9.1.0(tasmota)","mf":"Tasmota"}} (retained)
```

To:

```
mosquitto_pub (HOST/PORT/USER/PWD) -t XefilHome/switch/D04DF5_RL_1/config -r -m '{"name":"RELAY02-01","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER1","val_tpl":"{{value_json.POWER1}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_1","dev":{"ids":["D04DF5"]}}'
mosquitto_pub (HOST/PORT/USER/PWD) -t XefilHome/switch/D04DF5_RL_2/config -r -m '{"name":"RELAY02-02","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER2","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_2","dev":{"ids":["D04DF5"]}}'
mosquitto_pub (HOST/PORT/USER/PWD) -t XefilHome/switch/D04DF5_RL_3/config -r -m '{"name":"RELAY02-03","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER3","val_tpl":"{{value_json.POWER3}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_3","dev":{"ids":["D04DF5"]}}'
mosquitto_pub (HOST/PORT/USER/PWD) -t XefilHome/switch/D04DF5_RL_4/config -r -m '{"name":"RELAY02-04","stat_t":"XefilHome/GARDEN02/STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"XefilHome/GARDEN02/cmnd/POWER4","val_tpl":"{{value_json.POWER4}}","pl_off":"OFF","pl_on":"ON","uniq_id":"D04DF5_RL_4","dev":{"ids":["D04DF5"]}}'
mosquitto_pub (HOST/PORT/USER/PWD) -t XefilHome/sensor/D04DF5_status/config -r -m '{"name":"GARDEN02 status","stat_t":"XefilHome/GARDEN02/HASS_STATE","avty_t":"XefilHome/GARDEN02/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"XefilHome/GARDEN02/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"D04DF5_status","dev":{"ids":["D04DF5"],"name":"GARDEN02","mdl":"Sonoff 4CH Pro","sw":"9.1.0(tasmota)","mf":"Tasmota"}}'

```

BTW some infos are still not updated, not important, like ip address:

`2020-11-18 14:57:34 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/HASS_STATE: b'{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Module or Template":"Sonoff 4CH Pro","RestartReason":"Software/System restart","Uptime":"0T02:51:10","Hostname":"GARDEN02-7606","IPAddress":"192.168.3.29","RSSI":"100","Signal (dBm)":"-39","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:03","MqttCount":1,"LoadAvg":19}'`

it shows only FW version and device. But into 'XefilHome/GARDEN02/HASS_STATE' the informations are received into HA and shown in the payload of the device info.

Ideas?

effelle, 2020-11-18:
No idea, since you are using a  custom solution you are on your own. 
I suggest (again) to use the official integration instead of the MQTT discovery, enabled when setoption19 is `0`. 
Under Home Assistant add it on integration panel if not already there and done.

xefil, 2020-11-18:
ok, back again:

SO19 0 and from HA:

```
2020-11-18 15:16:13 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/RESULT: b'{"SetOption19":"OFF"}'
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on tasmota/discovery/C82B96D04DF5/config: b'{"ip":"192.168.3.30","dn":"GARDEN02","fn":["RELAY02-01","RELAY02-02","RELAY02-03","RELAY02-04",null,null,null,null],"hn":"GARDEN02-3573","mac":"C82B96D04DF5","md":"Sonoff 4CH Pro","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.1.0","t":"GARDEN02","ft":"XefilHome/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,1,1,1,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":1,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0},"lk":1,"lt_st":0,"ver":1}'
2020-11-18 15:16:14 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.tasmota entity: sensor.garden02_ip
2020-11-18 15:16:14 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.tasmota entity: sensor.garden02_signal
2020-11-18 15:16:14 INFO (MainThread) [homeassistant.components.sensor] Not adding entity GARDEN02 Signal because it's disabled
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/STATUS5, mid: 334
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/RESULT, mid: 335
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/RESULT, mid: 336
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/RESULT, mid: 337
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/RESULT, mid: 338
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/LWT, mid: 339
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/STATE, mid: 340
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/LWT (retained): b'Offline'
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/STATE, mid: 341
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/STATE, mid: 342
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/STATE, mid: 343
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/LWT, mid: 344
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/LWT, mid: 345
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/LWT (retained): b'Offline'
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/LWT (retained): b'Offline'
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/LWT, mid: 346
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to XefilHome/GARDEN02/LWT, mid: 347
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/LWT (retained): b'Offline'
2020-11-18 15:16:14 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on XefilHome/GARDEN02/LWT (retained): b'Offline'
```

Why offline?
And the other one (I've two 4Ch boards) is online, but pressing the ON relay the command is not taken from the Board.

In this case bard uodates HA, but HA doesn't act to tasmota :-(

ascillato2, 2020-11-18:
Your fulltopic is custom, not the default. All integrations won't work as you have defined your own fulltopic, you should add manually in HA.

If you want to use the automation integration, you need to use the default fulltopic. You can set the friendly name, the topic, but not the fulltopic.

closed 2020-11-18 by ascillato2

xefil, 2020-11-18:
@ascillato2  sorry, I've reverted the discovery prefix of my whole setup to "homeassistant" (I mean in HA).
Is that what you mean with "fulltopic"? I revert all to default but please help me.
With SO19 to 0 all devices are discovered in HA but only one (on 3) works (Gosund), the others (different hardware sonoff Ch4 Pro R3) not, so I fear it's a Tasmota issue. HA gets updated from Tasmota if on tasmota I change the relay state. But if i change it on HA, nothing happens on those sonoff boards.
Thanks, Simon

ascillato, 2020-11-18:
To revert everything to default, go to the console and type `reset 6`. After it restarts, you can set your topic and friendly name of each device.

xefil, 2020-11-18:
So I've done:

- reset 6
(reboot)
- Config Module to SonOff 4 Ch Pro
(reboot)
- Config Other -> Device Name "GARDEN01" and Friendly Name for 4 relays to RELAY02-0[1-4]
(reboot)
SO19 0

And HA discovered all and it works perfectly.

Thanks @ascillato2 @ascillato *REALLY*

Simon

ascillato, 2020-11-18:
Great to know that now it is working for you!!!!!

ascillato, 2020-11-18:
Checking on the docs at https://tasmota.github.io/docs/Home-Assistant/ I couldn't find any warning about changing the fulltopic. We need to add that. Thanks for pointing this.