arendst/Tasmota #9906: shutter: position percentage not updated and stat SHUTTER1 not published

megamarco833, 2020-11-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
two issue (or possible issue)
1. when i trigger manually sending a power command the status of position percentage of blind is not updated
2. in normal usage, in previous versions was published every time after the moviment a topic in `stat/%topic%/SHUTTER1` with the percentage

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): nodemcu esp8266
- [x] Tasmota binary firmware version number used: 9.1
  - [ ] Pre-compiled
  - [x] Self-compiled (just added support for PCF8576
    - [x] IDE / Compiler used: gitpod
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
12:37:20 CMD: Backlog Template; Module; GPIO 255
12:37:20 MQT: stat/tasmota_blind_new/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:37:20 MQT: stat/tasmota_blind_new/RESULT = {"Module":{"0":"Generic"}}
12:37:21 MQT: stat/tasmota_blind_new/RESULT = {"GPIO0":{"166":"Switch7"},"GPIO1":{"160":"Switch1"},"GPIO2":{"167":"Switch8"},"GPIO3":{"161":"Switch2"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"165":"Switch6"},"GPIO12":{"162":"Switch3"},"GPIO13":{"163":"Switch4"},"GPIO14":{"164":"Switch5"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
12:38:38 CMD: rule1
12:38:38 MQT: stat/tasmota_blind_new/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:38:42 CMD: rule2
12:38:42 MQT: stat/tasmota_blind_new/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":483,"Free":28,"Rules":"on switch1#state=2 do backlog power1 on;delay 5;power1 off; ShutterPosition1 endon on switch1#state=3 do power1 on endon on switch2#state=2 do backlog power2 on;delay 5;power2 off; ShutterPosition1 endon on switch2#state=3 do power2 on endon on switch3#state=2 do backlog power3 on;delay 5;power3 off; ShutterPosition2 endon on switch3#state=3 do power3 on endon on switch4#state=2 do backlog power4 on;delay 5;power4 off; ShutterPosition2 endon on switch4#state=3 do power4 on endon"}
12:38:45 CMD: rule3
12:38:45 MQT: stat/tasmota_blind_new/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
12:39:34 CMD: status 0
12:39:34 MQT: stat/tasmota_blind_new/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota_blind_new","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","Tasmota5","Tasmota6","Tasmota7","Tasmota8"],"Topic":"tasmota_blind_new","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,5,5,5,5,5,5,5],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:37","StartupUTC":"2020-11-18T11:36:57","Sleep":50,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2020-11-17T11:54:32","SaveCount":401,"SaveAddress":"F5000"}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-11-17T11:14:00","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"419/699"}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","1405C8000100060000005A00000000000000","C0000000","00006000","00000000"]}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000FB81","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,28,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS5 = {"StatusNET":{"Hostname":"tasmota_blind_new-8041","IPAddress":"192.168.0.57","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"D8:BF:C0:F9:FF:69","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.105","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F9FF69","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS7 = {"StatusTIM":{"UTC":"2020-11-18T11:39:34","Local":"2020-11-18T12:39:34","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:04","Sunset":"17:05"}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS10 = {"StatusSNS":{"Time":"2020-11-18T12:39:34","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"ON","Shutter1":{"Position":50,"Direction":0,"Target":50},"Shutter2":{"Position":52,"Direction":0,"Target":52}}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS11 = {"StatusSTS":{"Time":"2020-11-18T12:39:34","Uptime":"0T00:02:37","UptimeSec":157,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"NETGEAR","BSSId":"E0:3F:49:A8:C5:70","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}
12:39:34 MQT: stat/tasmota_blind_new/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":150,"Close":150,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}},{"SHT1":{"Relay1":3,"Relay2":4,"Open":150,"Close":150,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
12:40:31 CMD: weblog 4
12:40:31 MQT: stat/tasmota_blind_new/RESULT = {"WebLog":4}
12:40:31 CFG: Saved to flash at F4, Count 402, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

**ISSUE 1.**
  i would like to control my blind configured with:
` relay1 = OPEN
relay2 = CLOSE` 
[... 188 lines cut ...]
12:29:38 MQT: stat/tasmota_blind_new/RESULT = {"POWER2":"ON"}
12:29:38 MQT: stat/tasmota_blind_new/POWER2 = ON
12:29:38 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":49,"Direction":-1,"Target":0}}
12:29:38 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":45,"Direction":-1,"Target":0},"Shutter2":{"Position":72,"Direction":0,"Target":72}}
12:29:39 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":38,"Direction":-1,"Target":0},"Shutter2":{"Position":72,"Direction":0,"Target":72}}
12:29:40 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":32,"Direction":-1,"Target":0},"Shutter2":{"Position":72,"Direction":0,"Target":72}}
12:29:41 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":25,"Direction":-1,"Target":0},"Shutter2":{"Position":72,"Direction":0,"Target":72}}
12:29:42 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":18,"Direction":-1,"Target":0},"Shutter2":{"Position":72,"Direction":0,"Target":72}}
12:29:43 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":12,"Direction":-1,"Target":0},"Shutter2":{"Position":72,"Direction":0,"Target":72}}
[... cut ...]
12:29:46 MQT: stat/tasmota_blind_new/POWER2 = OFF
12:29:46 MQT: stat/tasmota_blind_new/SHUTTER1 = 0
12:29:46 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}`

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

kenschbacher9, 2020-12-01:
I have encountered the same problem.
I have tested - the last working version is 8.5.0 - this bug first occured in 8.5.1 - so I am just downgrading all my rollershutters back to 8.5.0.

megamarco833, 2020-12-01:
> 
> 
> I have encountered the same problem.
> I have tested - the last working version is 8.5.0 - this bug first occured in 8.5.1 - so I am just downgrading all my rollershutters back to 8.5.0.

do you find both issue?

**issue 1) shutter percentage not updated if blink it is used.**
in my example:
shutterposition1 was 55% and after triggering manually `power1 3` the shutterposition1 is still at 55%
so it seams that the percentage is not updated using `power1 3`
it is a bug or i'm missing something? thanks

and 

**issue 2)** the percentage `stat/%topic%/SHUTTER1` in the previous firmware (before 8.5.1) was published every time after the final moviment.
now i see that it is not reported every time after a moviment (from 8.5.1)

thanks!
Marco

kenschbacher9, 2020-12-01:
I was not precise enough, my fault.

Since I do not use blink, I can´t tell anything about issue 1, but I can confirm issue 2 - percentage of shutter is not updated.

Test situation:
Shutters open (SHUTTER1=0), then press down (POWER2), stop after moving downwards a few seconds, SHUTTER1 gets updated, then press up (POWER1). After stopping SHUTTER1 gets updated again.

Console output with Tasmota 8.5.0 - working fine:
`13:37:08 MQT: stat/RolladenEZ/RESULT = {"POWER2":"ON"}
13:37:08 MQT: stat/RolladenEZ/POWER2 = ON
13:37:08 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":0,"Direction":-1,"Target":100}}
13:37:08 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":0,"Direction":-1,"Target":100}}
13:37:09 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":4,"Direction":-1,"Target":100}}
13:37:10 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":12,"Direction":-1,"Target":100}}
13:37:11 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":20,"Direction":-1,"Target":100}}
13:37:12 MQT: stat/RolladenEZ/RESULT = {"POWER2":"OFF"}
13:37:12 MQT: stat/RolladenEZ/POWER2 = OFF
13:37:12 MQT: stat/RolladenEZ/SHUTTER1 = 25
13:37:12 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":25,"Direction":0,"Target":25}}
13:37:18 MQT: stat/RolladenEZ/RESULT = {"POWER1":"ON"}
13:37:18 MQT: stat/RolladenEZ/POWER1 = ON
13:37:18 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":25,"Direction":1,"Target":0}}
13:37:18 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":23,"Direction":1,"Target":0}}
13:37:19 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":16,"Direction":1,"Target":0}}
13:37:20 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":9,"Direction":1,"Target":0}}
13:37:21 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":3,"Direction":1,"Target":0}}
13:37:21 MQT: stat/RolladenEZ/RESULT = {"POWER1":"OFF"}
13:37:21 MQT: stat/RolladenEZ/POWER1 = OFF
13:37:21 MQT: stat/RolladenEZ/SHUTTER1 = 0
13:37:21 MQT: stat/RolladenEZ/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}
13:37:51 MQT: tele/RolladenEZ/STATE = {"Time":"2020-12-01T13:37:51","Uptime":"0T00:05:11","UptimeSec":311,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"UlMxD3_Ss0","BSSId":"34:CE:00:70:53:4D","Channel":6,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:05"}}
13:37:51 MQT: tele/RolladenEZ/SENSOR = {"Time":"2020-12-01T13:37:51","Shutter1":{"Position":0,"Direction":0,"Target":0}}
`

Same test, this time Tasmota 8.5.1:

`13:36:52 MQT: stat/RolladenWZ/RESULT = {"POWER2":"ON"}
13:36:52 MQT: stat/RolladenWZ/POWER2 = ON
13:36:52 MQT: stat/RolladenWZ/RESULT = {"Shutter1":{"Position":0,"Direction":-1,"Target":100}}
13:36:53 MQT: stat/RolladenWZ/RESULT = {"Shutter1":{"Position":4,"Direction":-1,"Target":100}}
13:36:54 MQT: stat/RolladenWZ/RESULT = {"Shutter1":{"Position":9,"Direction":-1,"Target":100}}
13:36:55 MQT: stat/RolladenWZ/RESULT = {"POWER2":"OFF"}
13:36:55 MQT: stat/RolladenWZ/POWER2 = OFF
13:36:57 MQT: stat/RolladenWZ/RESULT = {"POWER1":"ON"}
13:36:57 MQT: stat/RolladenWZ/POWER1 = ON
13:36:57 MQT: stat/RolladenWZ/RESULT = {"Shutter1":{"Position":12,"Direction":1,"Target":0}}
13:36:58 MQT: stat/RolladenWZ/RESULT = {"Shutter1":{"Position":11,"Direction":1,"Target":0}}
13:36:59 MQT: stat/RolladenWZ/RESULT = {"Shutter1":{"Position":6,"Direction":1,"Target":0}}
13:37:00 MQT: stat/RolladenWZ/RESULT = {"Shutter1":{"Position":1,"Direction":1,"Target":0}}
13:37:01 MQT: stat/RolladenWZ/RESULT = {"POWER1":"OFF"}
13:37:01 MQT: stat/RolladenWZ/POWER1 = OFF
13:37:01 MQT: stat/RolladenWZ/SHUTTER1 = 0
13:37:01 MQT: stat/RolladenWZ/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}
13:37:01 MQT: tele/RolladenWZ/STATE = {"Time":"2020-12-01T13:37:01","Uptime":"0T21:21:44","UptimeSec":76904,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":59,"MqttCount":21,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"UlMxD3_Ss0","BSSId":"34:CE:00:70:53:4D","Channel":6,"RSSI":100,"Signal":-50,"LinkCount":2,"Downtime":"0T00:00:17"}}
13:37:01 MQT: tele/RolladenWZ/SENSOR = {"Time":"2020-12-01T13:37:01","Switch1":"OFF","Switch2":"ON","Shutter1":{"Position":0,"Direction":0,"Target":0}}`

What you can see is that SHUTTER1 only gets updated at the end position (0), but not if you stop somewhere between 1 and 99.

btw: After downgrading to 8.5.0 you have to refresh config settings - I have just reloaded an old backup from 8.3.1 which worked fine.

schrej-zz, 2020-12-05:
@stefanbode,
can we still count on you, as I do not see any feedback from you within the last 3 weeks.
Beside this issue, we have 3 further issues that @ascillato2 merged into this one for any reason.
At least two of these issues are not related to this one.
Only coincidence for me is: Everything seems to start with version 8.5.1.
Thanks Joerg

yinonco, 2020-12-07:
@ascillato2 closed my issue #10033 and I don't think it's related to this one

SimoneDomotic, 2020-12-08:
Hello i solved with this solution but obviously it is not the correct way

Enable rule:
rule1 1
rule2 1
`
Rule1
  ON Shutter1#Position DO Backlog Publish stat/%topic%/SHUTTER1 %value%; RuleTimer1 5 ENDON
  ON Rules#Timer=1 DO RuleTimer1 0 ENDON
  ON Rules#Timer=1 DO shutterposition1 ENDON
`

`
Rule2
  ON Shutter2#Position DO Backlog Publish stat/%topic%/SHUTTER2 %value%; RuleTimer2 5 ENDON
  ON Rules#Timer=1 DO RuleTimer2 0 ENDON
  ON Rules#Timer=1 DO shutterposition2 ENDON
`

jtauscher, 2020-12-10:
Hi, do you guys also have the problem that it doesn't have any effect when inverting the relays in the template? I changed the sonoff dual r2 module template (Relay to Relay_i), saved (incl. auto reboot) but it doesn't invert anything. Is this a bug or did I miss something?

megamarco833, 2020-12-10:
hi, i'm not using a sonoff but i'm tested on a esp8266 nodemcu, and i select inverted relay and i works perfectly as inverted
i tested also with esp32 (it's beta version) but also there relay inverted works.
i'm on 9.1.0.2 version

jtauscher, 2020-12-10:
Thanks for your reply, I also use ESP8266 D1 mini, with the latest Tasmota version, but somehow changing relays **in the template** to inverted doesn't take effect. Also when I select "none" or whatever, it doesn't change a thing. Do I understand the way how saving/loading a template works correctly? ->Take a preset, change things u want and save/restart and that's it. Or do I have to load/activate this changed template separately in a way I couldn't find yet? Thanks guys!

megamarco833, 2020-12-10:
but let's wait a moment, i not used template, i manually configure the relay inside configuration -> configuration module
then i select here relay_i <x> as relay inverted
get a try :-)
anyway i think that it's not connected with the two bugs that are here reported:

1. when manually sending a power command, like `power1 3` the status of position percentage of blind is not updated
    
2. in normal usage, in previous versions was published every time after the moviment a topic in stat/%topic%/SHUTTER1 with the percentage

jtauscher, 2020-12-10:
Ah I now I see, when you load the Sonoff Dual R2 preset in the **Config** the relays are hidden, I cannot change the behavior.
I solved it, by changing the relay-GPIOs in the template to "user", save and load the new template in the config and then I can invert the relays in the config. Thanks anyway and sorry I cannot help with your two problems at the moment.

schrej-zz, 2020-12-10:
All,
thanks to @ascillato2 we now have to deal with several (independent?) topics within one issue.
As @stefanbode currently seems to not to be capable to support, I would like to make a suggestion, to identify the version (commit) since these issues arised.
Can the guys with their issues go back to (at least) version 8.3.1 an then go up the versions until your issue arises.
I have a gut feeling, which commit caused the issues.
If we can identify the last working version, I would suggest to revert all commits from Stefan in an upcoming release until someone is able to identify and hopefully can solve the issues.
Joerg

Jason2866, 2020-12-10:
The issue will be left open since it is solved. Undo all the PRs from Stefan will bring back other solved issues. Thats is imho not a good idea.
There is nothing wrong to use a older Tasmota version with the working shutter support
until Stefan or a other one is nailing the issue

megamarco833, 2020-12-11:
hi...Well, for what it's worth i think the same of @Jason2866 
not a good idea reverse back all the efforts done.
better to understand what is changed since version 8.3 only related to shutters.
i can confirm that in version 8.1 was working for my issue opened that is for two bugs:

  1.  when manually sending a power command, like `power1 3` the status of position percentage of blind is not updated

  2.  in normal usage, in previous versions (8.1) was published every time after the movement a topic in `stat/%topic%/SHUTTER<X>` with the percentage

hope that @stefanbode can come back and take a look on both 👍

stefanbode, 2020-12-14:
hi team, sorry for being off some time. Lots of work before Christmas. I will work through and check If I can understand and find the root cause

stefanbode, 2020-12-14:
I can reproduce the error on direct change on relays. It works fine as long as normal open/close or any shuttercomand is used. I will fix this but also point out that switching the relay is not the best way to operate with shutters. There are shutterbuttons and also nice command you can call from backlog

megamarco833, 2020-12-14:
> I can reproduce the error on direct change on relays. It works fine as long as normal open/close or any shuttercomand is used. I will fix this but also point out that switching the relay is not the [...]

hi Stefan and thanks so much for your answer, i appreciate.

you mean instead of use blinktime? 👍 
```
blinktime 5 (5=0,5sec)
blinkcount 1
switchmode1 5 // 5=to manage short/long press
switchmode2 5
power1 3 (3=blink)
```

i use this configuration because in this case i can trigger relay1 (up) or relay2(down) with a shortpress for 0,5seconds, and trigger relay1(up) or relay2(down) with a longpress for whole time shutterOpen or Shutterclose.

i used this rule:
```
rule2 on switch1#state=2 do power1 3 endon on switch1#state=3 do power1 on endon
on switch2#state=2 do power2 3 endon on switch2#state=3 do power2 on endon
rule2 1
```
what's your suggestion? thanks
i can confirm that till version 8.0.1 it was working.
with version 9 or above it is not working.

Related to ISSUE2
if it's possible to reintroduce publishing of every blind position after every kind of moviment  inside `stat/%topic%/SHUTTER<X>`
could be better because every domotic system can listen to that topic and update accordingly the blind<x> position percentage.
This not happened in 9.X versions.
thanks!

ascillato2, 2020-12-14:
Fixed with #10135

megamarco833, 2020-12-14:
@ascillato2 
i tested stefanbode fix and now it works: position is updated inside `stat/%topic%/SHUTTER<X>`
thanks!
Issue2 is now solved.

can you please left open this post for issue1 ?
issue1 is not solved, i try to manual trigger power1 and position is still not updated:
now i ask the percentage and it's 35%:
`stat/tasmota_blind_new/RESULT = {"ShutterPosition1":35}`
```
19:09:48 CMD: shutterposition1
19:09:48 SRC: WebConsole from 192.168.0.4
19:09:48 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data ""
19:09:48 SHT: Pos. in: payload  (0), payload -99, idx 1, src 9
19:09:48 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":35,"Direction":0,"Target":35}}
19:09:48 MQT: stat/tasmota_blind_new/RESULT = {"ShutterPosition1":35}

```

now i trigger maually power1 4 times:

```
19:10:40 CMD: power1 3
19:10:40 SRC: WebConsole from 192.168.0.4
19:10:40 CMD: Group 0, Index 1, Command "POWER", Data "3"
19:10:40 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"Blink ON"}
19:10:40 SHT: Switched relay: 1 by I
19:10:40 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 1
19:10:40 SHT: Shutter 1: Target: 104000, powerstatelocal 0
19:10:40 SHT: Shutter 2: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
19:10:41 SHT: Switched relay: 1 by I
19:10:41 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 1
19:10:41 SHT: Shutter 1: Target: 104000, powerstatelocal 0
19:10:41 SHT: Shutter 2: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
19:10:41 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"Blink OFF"}
19:10:41 SHT: Switched relay: 0 by I
19:10:41 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 0, manual change 0
19:10:41 SHT: Shutter 2: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 0, manual change 0
19:10:41 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"OFF"}
19:10:41 MQT: stat/tasmota_blind_new/POWER1 = OFF
19:10:44 CMD: power1 3
19:10:44 SRC: WebConsole from 192.168.0.4
19:10:44 CMD: Group 0, Index 1, Command "POWER", Data "3"
19:10:44 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"Blink ON"}
19:10:44 SHT: Switched relay: 1 by I
19:10:44 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 1
19:10:44 SHT: Shutter 1: Target: 104000, powerstatelocal 0
19:10:44 SHT: Shutter 2: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
19:10:45 SHT: Switched relay: 1 by I
19:10:45 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 1
19:10:45 SHT: Shutter 1: Target: 104000, powerstatelocal 0
19:10:45 SHT: Shutter 2: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
19:10:45 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"Blink OFF"}
19:10:45 SHT: Switched relay: 0 by I
19:10:45 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 0, manual change 0
19:10:45 SHT: Shutter 2: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 0, manual change 0
19:10:45 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"OFF"}
19:10:45 MQT: stat/tasmota_blind_new/POWER1 = OFF
19:10:48 CMD: power1 3
19:10:48 SRC: WebConsole from 192.168.0.4
19:10:48 CMD: Group 0, Index 1, Command "POWER", Data "3"
19:10:48 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"Blink ON"}
19:10:48 SHT: Switched relay: 1 by I
19:10:48 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 1
19:10:48 SHT: Shutter 1: Target: 104000, powerstatelocal 0
19:10:48 SHT: Shutter 2: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
19:10:49 SHT: Switched relay: 1 by I
19:10:49 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 1
19:10:49 SHT: Shutter 1: Target: 104000, powerstatelocal 0
19:10:49 SHT: Shutter 2: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
[... 31 lines cut ...]
19:10:54 SHT: Shutter 1: Target: 104000, powerstatelocal 0
19:10:54 SHT: Shutter 2: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
19:10:55 SHT: Switched relay: 1 by I
19:10:55 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 1
19:10:55 SHT: Shutter 1: Target: 104000, powerstatelocal 0
19:10:55 SHT: Shutter 2: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
19:10:55 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"Blink OFF"}
19:10:55 SHT: Switched relay: 0 by I
19:10:55 SHT: Shutter 1: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 0, manual change 0
19:10:55 SHT: Shutter 2: source: I, powerstate_local 0, ShutterGlobal.RelayCurrentMask 0, manual change 0
19:10:55 MQT: stat/tasmota_blind_new/RESULT = {"POWER1":"OFF"}
19:10:55 MQT: stat/tasmota_blind_new/POWER1 = OFF
19:11:00 CMD: shutterposition
19:11:00 SRC: WebConsole from 192.168.0.4
19:11:00 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data ""
19:11:00 SHT: Pos. in: payload  (0), payload -99, idx 1, src 9
19:11:00 MQT: stat/tasmota_blind_new/RESULT = {"Shutter1":{"Position":35,"Direction":0,"Target":35}}
19:11:00 MQT: stat/tasmota_blind_new/RESULT = {"ShutterPosition1":35}
```
as you can see the position is still 35% but the shutter was moved, and percentage is not correctly updated.
`stat/tasmota_blind_new/RESULT = {"ShutterPosition1":35}`
so issue1 is still present.

as far as i understand with old version 8.1 this was not happened.
thanks

ascillato2, 2020-12-14:
Reopening as requested

stefanbode, 2020-12-15:
Hi, maybe you can test this without the Blink method. The problem of the Blink is, that this is not handled correctly. If the relay 1 is triggered it should be:
19:10:52 SHT: Shutter 1: Target: 104000, powerstatelocal 1
and not
19:10:52 SHT: Shutter 1: Target: 104000, powerstatelocal 0
it looks like the blink does not trigger the powerstate correctly. If I manually set "power1 on" then it works and the powerlocalstate is ok
in line 598 I calculate the local state with
	power_t powerstate_local = (power >> (Settings.shutter_startrelay[i] -1)) & 3;
Normally in the global "power" the is the bitmatrix of the switched relays. For Blink this seems not to be true. I assume the issue is already fixed. It just do not work with the blink and I try to avoid to make a change to let it also work for blink

Mikk36, 2020-12-15:
@stefanbode, does your fix also fix issue #9800?

megamarco833, 2020-12-15:
> Hi, maybe you can test this without the Blink method. The problem of the Blink is, that this is not handled correctly. If the relay 1 is triggered it should be:
> `19:10:52 SHT: Shutter 1: Target: [...]

hi, i tested with power<x> On / Off and the position now is updated.

ok, so we can not use blink to trigegr relay for some seconds on shortpress.
i now use this workaround and it seams work but has some limits:

`on switch1#state=2 do backlog power1 on;delay 5;power1 off; ShutterPosition1 endon on switch1#state=3 do power1 on endon on switch2#state=2 do backlog power2 on;delay 5;power2 off; ShutterPosition1 endon on switch2#state=3 do power2 on endon `

in this case with shorpress => i trigger relay for 0,5seconds
with longpress => trigegr the relay to complete total moviment up or down

but in this case the delay 5 => 0,5second is not respected, the relay is activated for 1second (i think 1 second is the limit)

there are other rules that can perform it?
or i try to use shutterbutton but i do not find a way to solve.
thanks!

yinonco, 2020-12-15:
Hi, for some reason @ascillato2 closed issue #10033 and I am not using blink at all there so it won't help my situation.
@stefanbode if you can take a look I will appreciate it a lot
thanks

stefanbode, 2020-12-15:
Hi @megamarco833: I would suggest to rewrite the rule a bit to get the proposed solution:
```
on switch1#state=2 do backlog ShutterChange 2 endon on switch1#state=3 do power1 on endon
on switch2#state=2 do backlog ShutterChange -2 endon on switch2#state=3 do power2 on end
```
This would increment or decrement the position with 2%. And yes there is a limit of minimal runtime because TASMOTA is not that quick. I would assume minimum runtime is somewhere of 0.5seconds.

megamarco833, 2020-12-15:
Hi @stefanbode 
there is something strage with ShutterChange

i adapt your suggestion for 4 shutters:
```
rule2 on switch1#state=2 do backlog ShutterStop1; ShutterChange1 2 endon on switch1#state=3 do ShutterOpen1 endon
on switch2#state=2 do backlog ShutterStop1; ShutterChange1 -2 endon on switch2#state=3 do ShutterClose1 endon
on switch3#state=2 do backlog ShutterStop2; ShutterChange2 2 endon on switch3#state=3 do ShutterOpen2 endon
on switch4#state=2 do backlog ShutterStop2; ShutterChange2 -2 endon on switch4#state=3 do ShutterClose2 endon
rule2 1

rule3 on switch5#state=2 do backlog ShutterStop3; ShutterChange3 2 endon on switch5#state=3 do ShutterOpen3 endon
on switch6#state=2 do backlog ShutterStop3; ShutterChange3 -2 endon on switch6#state=3 do ShutterClose3 endon
on switch7#state=2 do backlog ShutterStop4; ShutterChange4 2 endon on switch7#state=3 do ShutterOpen4 endon
on switch8#state=2 do backlog ShutterStop4; ShutterChange4 -2 endon on switch8#state=3 do ShutterClose4 endon
rule3 1
```

i introduce a ShutterStop<x> because in this way if i want to stop a longpress movement i can simply shortpress a button.
but why if i set 2% as shutterchange the percentage changed is not 2%??

issued shutterposition4 and see that it's 59%
the n short press but nothing happened:
```
20:14:06 CMD: shutterposition4
20:14:06 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":59,"Direction":0,"Target":59}}
20:14:06 MQT: stat/tasmota_esp32/RESULT = {"ShutterPosition4":59}
20:14:12 RUL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2"
20:14:12 MQT: stat/tasmota_esp32/RESULT = {"ShutterStop":"Done"}
20:14:12 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":59,"Direction":0,"Target":61}}
20:14:12 MQT: stat/tasmota_esp32/RESULT = {"ShutterChange4":59}
```

now shortpress again and relay move, but not 2% (was59) now it go to 65%
```
20:15:17 RUL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2"
20:15:17 MQT: stat/tasmota_esp32/RESULT = {"ShutterStop":"Done"}
20:15:18 MQT: stat/tasmota_esp32/RESULT = {"POWER7":"ON"}
20:15:18 MQT: stat/tasmota_esp32/POWER7 = ON
20:15:18 MQT: stat/tasmota_esp32/RESULT = {"ShutterChange4":63}
20:15:18 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":60,"Direction":1,"Target":63}}
20:15:19 MQT: stat/tasmota_esp32/RESULT = {"Shutter1":{"Position":62,"Direction":0,"Target":62},"Shutter2":{"Position":25,"Direction":0,"Target":25},"Shutter3":{"Position":55,"Direction":0,"Target":55},"Shutter4":{"Position":65,"Direction":1,"Target":63}}
20:15:19 MQT: stat/tasmota_esp32/RESULT = {"POWER7":"OFF"}
20:15:19 MQT: stat/tasmota_esp32/POWER7 = OFF
20:15:19 MQT: stat/tasmota_esp32/SHUTTER4 = 65
20:15:19 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":65,"Direction":0,"Target":63}}
```

it seams that firmware count also the first command, but phisically the relay was not triggered....it will cause a wrong percentage positioning of the blind...correct?

here i pressed 5 times the switch but relay was triggered only 3 times:
```
20:19:36 RUL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2"
20:19:36 MQT: stat/tasmota_esp32/RESULT = {"ShutterStop":"Done"}
20:19:36 MQT: stat/tasmota_esp32/RESULT = {"POWER7":"ON"}
20:19:36 MQT: stat/tasmota_esp32/POWER7 = ON
20:19:36 MQT: stat/tasmota_esp32/RESULT = {"ShutterChange4":20}
20:19:36 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":18,"Direction":1,"Target":20}}
20:19:37 MQT: stat/tasmota_esp32/RESULT = {"Shutter1":{"Position":24,"Direction":0,"Target":24},"Shutter2":{"Position":25,"Direction":0,"Target":25},"Shutter3":{"Position":55,"Direction":0,"Target":55},"Shutter4":{"Position":19,"Direction":1,"Target":20}}
20:19:37 MQT: stat/tasmota_esp32/RESULT = {"POWER7":"OFF"}
20:19:37 MQT: stat/tasmota_esp32/POWER7 = OFF
20:19:37 MQT: stat/tasmota_esp32/SHUTTER4 = 21
20:19:37 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":21,"Direction":0,"Target":20}}
20:19:39 RUL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2"
20:19:39 MQT: stat/tasmota_esp32/RESULT = {"ShutterStop":"Done"}
20:19:39 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":21,"Direction":0,"Target":22}}
20:19:39 MQT: stat/tasmota_esp32/RESULT = {"ShutterChange4":21}
20:19:41 RUL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2"
20:19:41 MQT: stat/tasmota_esp32/RESULT = {"ShutterStop":"Done"}
20:19:41 MQT: stat/tasmota_esp32/RESULT = {"POWER7":"ON"}
[... 2 lines cut ...]
20:19:42 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":21,"Direction":1,"Target":24}}
20:19:42 MQT: stat/tasmota_esp32/RESULT = {"Shutter1":{"Position":24,"Direction":0,"Target":24},"Shutter2":{"Position":25,"Direction":0,"Target":25},"Shutter3":{"Position":55,"Direction":0,"Target":55},"Shutter4":{"Position":22,"Direction":1,"Target":24}}
20:19:42 MQT: stat/tasmota_esp32/RESULT = {"POWER7":"OFF"}
20:19:42 MQT: stat/tasmota_esp32/POWER7 = OFF
20:19:43 MQT: stat/tasmota_esp32/SHUTTER4 = 25
20:19:43 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":25,"Direction":0,"Target":24}}
20:19:44 RUL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2"
20:19:44 MQT: stat/tasmota_esp32/RESULT = {"ShutterStop":"Done"}
20:19:44 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":25,"Direction":0,"Target":26}}
20:19:44 MQT: stat/tasmota_esp32/RESULT = {"ShutterChange4":25}
20:19:46 RUL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2"
[... cut ...]
20:19:47 MQT: stat/tasmota_esp32/RESULT = {"Shutter1":{"Position":24,"Direction":0,"Target":24},"Shutter2":{"Position":25,"Direction":0,"Target":25},"Shutter3":{"Position":55,"Direction":0,"Target":55},"Shutter4":{"Position":30,"Direction":1,"Target":28}}
20:19:48 MQT: stat/tasmota_esp32/RESULT = {"POWER7":"OFF"}
20:19:48 MQT: stat/tasmota_esp32/POWER7 = OFF
20:19:48 MQT: stat/tasmota_esp32/SHUTTER4 = 31
20:19:48 MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":31,"Direction":0,"Target":28}}
```

why this happen?
thanks!!

stefanbode, 2020-12-16:
The answer in this case is easy. There is a minimum runtime required. 2% seem to low for the minimum time. You can see in the first Shutterchange 2 that the target change from 61->63 but nothing happened. Then you hit again and it changed to 65 (63+2) and this is enough to start a move. As I can see the move takes less than a second. Also quite low. Assumption: You minimum for a change is 4

Second Use case: I can see 4 presses, not 5 (UL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2")

Because of the short runtime after the first press, the position was already 21 and not 20 (small overrun)
MQT: stat/tasmota_esp32/RESULT = {"Shutter4":{"Position":21,"Direction":0,"Target":20}}
Now the next to get skipped because 21->22 is not enough movement.

At the next presses, there was no overrun. Small overrun can happen because of WIFI and other processes running. The Firmware is aware of the overrun but will not go backward.

megamarco833, 2020-12-16:
hi @stefanbode 
ehmm maybe for you the answer is easy but not for me, so sorry if i will make stupid question but i would like to try to understand.

my system is a test now so i not have connected a real  shutter.
i setup to test  `ShutterOpenDuration 15` and `ShutterCloseDuration 8`
so maybe the regulation of  S`hutterChange4 2` with 2% is too low for that reason?
if i will have more seconds to Open or Close duration, the regulation of 2% could be used?

> As I can see the move takes less than a second. Also quite low. Assumption: You minimum for a change is 4

but on "real shutter" where the open duration will be more than 15seconds, i can trigger relay (with a shortpress) for less than a second?

> Second Use case: I can see 4 presses, not 5 (UL: SWITCH7#STATE=2 performs "backlog ShutterStop4; ShutterChange4 2")
>
> Because of the short runtime after the first press, the position was already 21 [...]

ok, but considering for example 5 consecutive shortpress (so regulation of 2%, if it is possible) and considering that somewhere during this 5 consecutive shorpress some overrun can happen, it will cause a wrong / misalignment "real" position percentage of blind vs tasmota software position ?

last question if i have to control relays that are on ESP_B and buttons are on ESP_A
inside ESP_A i should use:

`on switch1#state=2 do backlog publish cmnd/tasmota_blind/ShutterStop1; publish cmnd/tasmota_blind/ShutterChange1 3 endon on switch1#state=3 do publish cmnd/tasmota_blind/ShutterOpen1 on switch1#state=2 do backlog publish cmnd/tasmota_blind/ShutterStop1; publish cmnd/tasmota_blind/ShutterChange1 -3 endon on switch1#state=3 do publish cmnd/tasmota_blind/ShutterClose1`

that's correct?
there is a way to use shutterbutton to perfrom with a shortpress a small regulation percentage of blind and with a longress a total movement of the blind?
if yes, how? i do not find the way....

thanks!

stefanbode, 2020-12-16:
Correct shuttercloseduration drive the minimum time. 10 seconds does mean 2% are 0.2 sec. This are only 4 cycles (50ms TASMOTA hard coded). Very tough. If the runtime is 20seconds then 2% are 8 cycles and this will work. The code requires at least 3 cycles = 0.15sec. Anyhow I assume this short time might create issues. Pulsetime relays will never work here. Just to be aware. 

shutterbutton can be used for other usecase. It will not solve the very short runtime. As mentioned above, independent from the total runtime, the limit is the number of cycles. I tested a bit around and 4 cycles (0.2sec) are the absolute minimum.

```
15:56:26 CMD: Group 0, Index 1, Command "SHUTTERCHANGE", Data "1"
15:56:26 SHT: Pos. in: payload 1 (1), payload 27, idx 1, src 23
15:56:26 SHT: lastsource 23:, real 51000, target 54000, payload 27
15:56:26 SHT: Delay Start. var0 <99>=<>, max10s?
15:56:26 SHT: Delay Start. Done
15:56:26 SRC: Shutter
15:56:26 SHT: Switched relay: 1 by Shutter
15:56:26 SHT: Shutter 1: source: Shutter, powerstate_local 1, ShutterGlobal.RelayCurrentMask 1, manual change 0
15:56:26 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"ON"}
15:56:26 MQT: hm/status/tasmota_A9D47F/POWER1 = ON (retained)
15:56:26 MQT: hm/status/tasmota_A9D47F/RESULT = {"ShutterChange1":27}
15:56:26 SHT: Shutter 1 Real 52000, Start 51000, Stop 54000, Dir 1, Delay 0, Rtc 0.10 [s], Freq 0, PWM 0
15:56:26 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":26,"Direction":1,"Target":27}}
15:56:26 SHT: time: 2, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 52000, next_stop 0, target: 54000,  max_vel_change 0, dir: 1
15:56:26 SHT: time: 3, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 53000, next_stop 0, target: 54000,  max_vel_change 0, dir: 1
15:56:26 SHT: time: 4, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 54000, next_stop 0, target: 54000,  max_vel_change 0, dir: 1
15:56:26 SHT: Stop Shutter 0. Switchmode 0
15:56:26 SRC: Shutter
15:56:26 SHT: Switched relay: 1 by Shutter
15:56:26 SHT: Shutter 1: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
15:56:26 MQT: hm/status/tasmota_A9D47F/RESULT = {"POWER1":"OFF"}
15:56:26 MQT: hm/status/tasmota_A9D47F/POWER1 = OFF (retained)
15:56:26 SHT: Second Relay 2 already OFF at time 5
15:56:26 SHT: Motordelay 0.5sek. Time 5
15:56:26 SHT: Shutter 1 Real 55000, Start 55000, Stop 54000, Dir 0, Delay 0, Rtc 0.25 [s], Freq 0, PWM 0
15:56:26 MQT: hm/status/tasmota_A9D47F/SHUTTER1 = 28 (retained)
15:56:26 MQT: hm/status/tasmota_A9D47F/RESULT = {"Shutter1":{"Position":28,"Direction":0,"Target":27}}
15:56:27 CFG: Saved to flash at 1F8, Count 276, Bytes 4096

```

stefanbode, 2020-12-16:
No, the position is always correct measured. It "knows" that it moved a bit too far and corrected this on the next move.

`ok, but considering for example 5 consecutive shortpress (so regulation of 2%, if it is possible) and considering that somewhere during this 5 consecutive shorpress some overrun can happen, it will cause a wrong / misalignment "real" position percentage of blind vs tasmota software position ?`

megamarco833, 2020-12-17:
> Correct shuttercloseduration drive the minimum time. 10 seconds does mean 2% are 0.2 sec. This are only 4 cycles (50ms TASMOTA hard coded). Very tough. If the runtime is 20seconds then 2% are 8 [...]

[image]

so if i understand correctly i can setup a percentage regulation with a shorpress that must have a numbers of cycles for OPEN and close that could be minimum 4 cycles.
so looking at my reported table witj a 3% regulation i will have 9 cycles for open (and it should be ok) and 4.8cycles for close (that is above the limit of 4 cycles)
so i'm in a safe side, correct?

[image]
with previous setup, i had for close 3.2cycles so it was below 4cycles requested and so that's why some shortpress not cause any movement of shutter, correct?

with new commit of today about milliseconds on tasmota, it will help to drive better the shutter, so also this percentage regulation with low numbers?

stefanbode, 2020-12-18:
All correct, without the last sentence. The millisecond timestamp does not bring any benefit here. There are three different methods of how you can hook in with procedures. e.g. to stop the shutter at the right time:
1.) FUNC_LOOP (whenever there is time and TASMOTA is not busy)
2.) FUNC_EVERY50MS (every 50ms, but only if TASMOTA is not busy. If busy you missed the slot. You do not get the call at all
3.) RTC timer 50ms (hard hook every 50ms as interrupt no matter what is going on.
The shutter is using a combination of 3 and 2. The problem of 3 is, that there are only limited functions that you can do. e.g. no MQTT or any logs. The RTC timer is doing all the calculation of the current position and also speed RAMP-UP/DOWN if you have a speed controlled device. The FUNC_EVERY50MS is responsible for the start/stop and any change that can happen.

With this combination I can assure position calculation is always correct no matter of the load and the WIFI network and do the operations with logging and so on outside the RealTimeClock (RTC) in a normal procedure. Doing RTC at 10ms or less is a very bad idea because we do not have a high-end processor. There is some give-and-take required to assure it runs smooth.

closed 2020-12-18 by ascillato2