arendst/Tasmota #9909: Option to terminate SerialSend with \r instead of \n

Bodypull, 2020-11-18:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I wasn't able to send serial commands to Atlas Scientific's Embedded Systems https://atlas-scientific.com/embedded-solutions/ because thoses devices expect <cr> as terminator

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Option to terminate SerialSend with \r instead of \n

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I just edited tasmota\xdrv_08_serial_bridge.ino line 133, changed \n to \r and recompiled.

**Additional context**  
_Add any other context or screenshots about the feature request here._
It is possible to use SerialSend5 (as hex) and end with 'c0' but it becomes annoying.

**(Please, remember to close the issue when the problem has been addressed)**  

Thanks for Tasmota - love it!

sfromis, 2020-11-18:
Tasmota already supports what you want. `SerialSend3 abc\r`

Bodypull, 2020-11-18:
> Tasmota already supports what you want. `SerialSend3 abc\r`

You are correct. I just compiled the original tasmota and it works with Atlas Scientific's Devices using`SerialSend3 abc\r`.

Thank you very much, I'm closing this.

closed 2020-11-18 by Bodypull