arendst/Tasmota #9913: Buzzer and SSD1306 broken when upgrading from 8.3.1.2 to 9.1.0

pookycade, 2020-11-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Buzzer just makes low clicks instead of sound wave, SSD1306 does not turn on at all. Both work just fine in 8.3.1.2 platformio compile. Piezo is passive, and will work in 9.1 if you set it as PWM. Will not work other than almost inaudible clicks in 9.1 if you set on Buzzer.  SSD1306 when starting up in 8.3.1.2 is off, have to turn it on with DisplayText [O] and it displays just fine. Cannot get same code on same device to turn on SSD1306 in 9.1. Any thoughts about where to look on why this might be broke ? I'm just out of ideas.  Note that i2c is working as VS53L0x distance sensor is working as is a PCF8574 port and both those work in 8.3.1.2 and 9.1. Stock bin in 8.3.1.2 tasmota-display works for buzzer and display. Stock bin 9.1 tasmota-display does not work for either buzzer or display. Stock bin 9.1 tasmota does not work for buzzer (and display commands obviously disabled).  

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____Generic(18)
- [x ] Tasmota binary firmware version number used: _____8.3.1.2 and 9.1.0
  - [x ] Pre-compiled Tasmota.bin and Tasmota-display.bin both firmware version
  - [x ] Self-compiled
    - [ x] IDE / Compiler used: _____Platformio
- [ x] Flashing tools used: _____Tasmotizer
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```04:10:18 RSL: stat/tasmota_7A19E1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
04:10:18 RSL: stat/tasmota_7A19E1/RESULT = {"Module":{"18":"Generic"}}
04:10:18 RSL: stat/tasmota_7A19E1/RESULT = {"GPIO0":{"640":"I2C SDA"},"GPIO1":{"0":"None"},"GPIO2":{"608":"I2C SCL"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"2848":"PN532 Tx"},"GPIO15":{"480":"Buzzer"},"GPIO16":{"2880":"PN532 Rx"},"GPIO17":{"0":"None"}}

  Configuration output here:
V53l0x on 0x20
PCF8574 on 0x29
SSD1306 on 0x3c
Piezo passive buzzer on GPIO 15
RC522 card reader on SPI, hijacked PN532 sensor routines to implement this.

```
- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
03:55:06 CMD: status 0
03:55:06 RSL: stat/tasmota_7A19E1/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","","",""],"Topic":"tasmota_7A19E1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:01:08","StartupUTC":"2020-11-19T02:53:58","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-11-19T03:52:12","SaveCount":12,"SaveAddress":"F8000"}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS2 = {"StatusFWR":{"Version":"9.1.0.1(tasmota)","BuildDateTime":"2020-11-18T14:06:00","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"352/699"}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Zombodor357",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","80000000","00006000","00000000"]}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16400B","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FF02797","04108121","00000007","041000C0","40000A01","00000000","00000000"],"Drivers":"1,2,3,4,7,8,9,10,12,13,16,20,21,24,28,29","Sensors":"1,2,3,40,45"}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS5 = {"StatusNET":{"Hostname":"tasmota_7A19E1-6625","IPAddress":"192.168.0.113","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"50:02:91:7A:19:E1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A19E1","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS7 = {"StatusTIM":{"UTC":"2020-11-19T02:55:06","Local":"2020-11-19T03:55:06","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:06","Sunset":"17:04"}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS10 = {"StatusSNS":{"Time":"2020-11-19T03:55:06","VL53L0X":{"Distance":9999}}}
03:55:06 RSL: stat/tasmota_7A19E1/STATUS11 = {"StatusSTS":{"Time":"2020-11-19T03:55:06","Uptime":"0T00:01:08","UptimeSec":68,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","Wifi":{"AP":1,"SSId":"Zombodor357","BSSId":"F4:F2:6D:BD:D8:09","Channel":11,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
03:54:21 RSL: stat/tasmota_7A19E1/RESULT = {"Buzzer":"Done"}
03:54:33 CMD: DisplayMode 2
03:54:33 RSL: stat/tasmota_7A19E1/RESULT = {"DisplayMode":2}
03:54:41 CMD: DisplayText [O]Me
03:54:41 RSL: stat/tasmota_7A19E1/RESULT = {"DisplayText":"[O]Me"}
03:55:06 CMD: status 0

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Try to either sound the buzzer or display text, it doesn't work

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Should sound buzzer and should show text on OLED.  It doesn't

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-19:
Please, update to Tasmota-sensors.bin. Thanks.

pookycade, 2020-11-19:
Hmm, I am gonna close this topic so we don't waste anyone's time.  I rechecked and with the stock build of 8.3.1 and 9.1.0 the Buzzer and the OLED are broke.  And yet on my custom build they work even though I didn't touch any code to do with the Buzzer. I did modify the library with the OLED since it can be funky on the ESP modules.  But I copied those changes over the my 9.1.0 build and they don't provide the same functionality. This indicates there is some glitch in the design, my code, or something that is causing this lack of functionality except with a custom build that somehow in one build enables it to all work but in most doesn't.  Lots of hours staring at code ahead

closed 2020-11-19 by pookycade