arendst/Tasmota #9914: Arilux Ar-LC04 with Bouffalo Labs BL602

anton-garcias, 2020-11-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I've received a new ordered Arilux AR-LC04 but is the v1.2 and has a new CPU Risc Based BL602 
Chip seems impressive, but I think it's not compatible with present firmware and even with the instruction set and/or the achitecture.
Not sure how to cathegorize the topic.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Study the feassibeness of supporting this particular chip. Maybe it's compatible with the present version but I don't think so.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
none yet

**Additional context**  
_Add any other context or screenshots about the feature request here._
[image]
[image]

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-19:
Sorry, no ESP82xx or ESP32 no Tasmota.

closed 2020-11-19 by ascillato2

anton-garcias, 2020-11-26:
Maybe you can add it to unsupportable devices

ascillato, 2020-11-26:
> Maybe you can add it to unsupportable devices

All devices that don't have ESP8285, ESP8266 or ESP32 are unsupported. There is no need to make a huge list of not supported devices.

It is easier and cleaner for search if we list ONLY the supported devices.

exentio, 2020-12-22:
I don't want to be that guy, but the ESP family **is not here to stay**, unless they switch to RISC-V, and if they do there's very little reason not to support new SoCs that are not made by Espressif. The MagicHome controllers **will** stop using ESP controllers, they all switched to the BL602, and **they won't be the only ones**  

Nobody's asking for a stable and perfectly working release right now, but I definitely suggest you start working on it if you want to keep Tasmota relevant for future products. Let's avoid forks that are essentially the same except for the supported SoC, a unified codebase benefits everyone

Jason2866, 2020-12-22:
No one disagrees that ESP8266 is EoL. Porting Tasmota to a different SoC needs a well working
Arduino port. Without there is no Tasmota possible.
Tasmota is supporting zigbee. Doing this there is (one way) for cloudless home automation solutions.

exentio, 2020-12-22:
> Porting Tasmota to a different SoC needs a well working Arduino port.

Looks like there's one: https://github.com/pine64/ArduinoCore-bouffalo  
Pine64 is very interested in the platform and is working on many tools and docs. I sincerely think it's worth the effort

anton-garcias, 2020-12-22:
Seems that also Expressif is planning to migrare to RiscV with the not-yet-released ESP32-C3 Series SoCs (samples only)
https://www.espressif.com/en/products/socs

Jason2866, 2020-12-22:
@anton-garcias  we hope espressif is doing the needed Arduino port (fast) for the ESP32-C3
There is a high chance Tasmota will be ported too ;-)
The chip is a cheap nice direct (pin compatible) replacement for the ESP8266

normanr, 2021-09-06:
fwiw, feature requests to add support for BL602:
- [PlatformIO](https://github.com/platformio/platformio-core/issues/4046) (required to build Tasmota)
- [ESPHome](https://github.com/esphome/feature-requests/issues/1049) (alternative to Tasmota)

Jason2866, 2021-09-06:
@normanr after this a well done Arduino port on top of the SDK is needed to have the possibility to port Tasmota. Imho the Arduino port will not happen

normanr, 2021-09-06:
acknowledged. The [ArduinoCore-bouffalo](https://github.com/pine64/ArduinoCore-bouffalo) mentioned by exentio looks like it's just a shim from ArduinoCore API to the Bouffalo SDK API, so should be good-enough to plug that gap. Probably the next steps is for someone from the community to create a fork and try, hook this stuff together, and see if it'll even build with everything optional disabled. I may try this in the future if I can get a development environment set up.

I agree that the ESP32-C3 is a good alternative, but it doesn't help those who have already bought pre-made units which have BL602 chips in them and would like to replace the firmware with Tasmota. I found a [reddit thread](https://www.reddit.com/r/RISCV/comments/knsju9/flashing_firmware_to_pinecone_bl602/gorwwu7/) where someone seems to have had some success flashing an SDK sample to a MagicHome unit.

maxgerhardt, 2021-09-13:
If anyone can help test my PlatformIO integration (https://github.com/platformio/platformio-core/issues/4046#issuecomment-918220660), please let me know your results. You should be able to compile + upload to a BL602 based board with the Pine64 Bouffalo Arduino core, such as the DT-BL10 or Pine64 PineCone board.

isaga1234, 2021-09-14:
> If anyone can help test my PlatformIO integration ([platformio/platformio-core#4046 (comment)](https://github.com/platformio/platformio-core/issues/4046#issuecomment-918220660)), please let me know your results. You should be able to compile + upload to a BL602 based board with the Pine64 Bouffalo Arduino core, such as the DT-BL10 or Pine64 PineCone board.

Tx MaxGerhart. I will try (it will not be fast as I'm busy with other topics, but I still have the arilux devices mentioned above)

Minims, 2022-04-25:
If anyone interested a project manage bl602 now : https://www.elektroda.pl/rtvforum/topic3889041.html#19992780

openshwprojects, 2022-04-26:
Thank you for linking, but you should rather link to English version of the article and not to Polish one. 
Here it is:
https://www.elektroda.com/rtvforum/viewtopic.php?p=19992780#19992780
I am personally developing my firmware which supports, at the moment, four platforms, BK7231T, BK7231N, XR809 and BL602 with a fair chance of supporting also RTL (Realtek, RTL8710 etc) chips and at least 50% chance of supporting W600 modules.

Is there anyone with BL602 module like that led strip and would like to help me with futher testing? I am testing all my devices 24 hours per day in room, but still, so far I have only one BL602 thing.

normanr, 2022-06-10:
Did you get OTA working with the BL602 yet? I tried to get OTA working for NuttX, but without any success.

Minims, 2022-06-10:
OTA is not implemented for now on BL602. I have asked for it too : 
https://www.elektroda.com/rtvforum/viewtopic.php?p=20028737#20028737

normanr, 2022-06-12:
I figured out to enable logging for the bl602 bootloader (using hexedit on the blob compiled into blflash), so 
I managed to debug why the xz decompression was failing (not enough memory, shrinking the dictionary to 16kb fixed that). I currently have a hacky shell script that downloads the fw image using tftp (and just replaces the contents of the partition tables to force trigger the update logic in the bootloader), but (finally!) it does work.

Minims, 2022-06-13:
> I figured out to enable logging for the bl602 bootloader (using hexedit on the blob compiled into blflash), so
> I managed to debug why the xz decompression was failing (not enough memory, shrinking [...]

Ok so you can upload the latest FW via wifi. Is so I'm interested in your scripts. 
Can you share them, with a small how to use ? Thx !

normanr, 2022-06-13:
If you're doing this manually like I am for my NuttX build:
- take the built firmware, and prepend it with a bootheader,
- compress it with `xz -v --check=crc32 --lzma2=dict=16KiB` (otherwise the boot2 bootloader can't decompress it),
- look in the partition_cfg .toml file to see what the `FW` address1 is
- copy the compressed image to that location (using tftp, or http GET, or upload to httpd)

Open_BL602 uses a lot more of the bl_iot_sdk, so it does a lot of this for you:
- starting with `build_out/bl602_sharedApp.bin` as the built firmware,
- `build_out/ota/dts40M_pt2M_boot2release_ef4015/FW_OTA.bin` has `build_out/bootheader.bin` prepended
- `build_out/ota/dts40M_pt2M_boot2release_ef4015/FW_OTA.bin.xz` is the compressed file
- `build_out/ota/dts40M_pt2M_boot2release_ef4015/FW_OTA.bin.xz.ota` has an OTA header that
- `customer_app/system/ota/demo_ota` can consume and write to the flash at the right place based on the partition table

Minims, 2022-06-15:
Thanks for this, but it's a bit too complicated for me I think.
I'm just running a make in the src project and I have not the `ota` folder in `build_out` for example

normanr, 2022-06-16:
Yes, I think the OpenBeken bl602 code needs changing to be able to accept the ota file - there's not any support for it at the moment. I've gotten the ota process working with a different sdk (i.e. NuttX instead of the Bouffalolab bl_iot_sdk)

openshwprojects, 2023-01-22:
@Minims  @normanr   OpenBL602 now supports OTA and the OpenBL602 development speed will now seriously increase due to the higher number of BL602 devices in the market.
[image]
More info: https://www.elektroda.com/rtvforum/topic3889041.html

Minims, 2023-01-22:
@openshwprojects Many thanks, will test it on the next releases.